avrdudes/avrdude #954: Windows shared library build problem under both VS2019 and MSYS2

mcuee, 2022-05-09:
I am not able to build the shared library under either VS2019 or MSYS2. Static build is okay.
Ref: https://github.com/avrdudes/avrdude/issues/952#issuecomment-1120810691
Ref: https://github.com/avrdudes/avrdude/issues/951#issuecomment-1120991143

mcuee, 2022-05-09:
Config log for VS2019.

```
1> CMake generation started for configuration: 'x64'.
1> Command line: "C:\WINDOWS\system32\cmd.exe" /c "%SYSTEMROOT%\System32\chcp.com 65001 >NUL && "C:\PROGRAM FILES\MICROSOFT VISUAL STUDIO\2022\COMMUNITY\COMMON7\IDE\COMMONEXTENSIONS\MICROSOFT\CMAKE\CMake\bin\cmake.exe"  -G "Ninja"  -DCMAKE_BUILD_TYPE:STRING="Release" -DCMAKE_INSTALL_PREFIX:PATH="C:\work\avr\avrdude\out\install\x64" -DUSE_EXTERNAL:BOOL="True" -DCMAKE_BUILD_TYPE:STRING="Release" -DBUILD_SHARED_LIBS:BOOL="True"  -DCMAKE_MAKE_PROGRAM="C:\PROGRAM FILES\MICROSOFT VISUAL STUDIO\2022\COMMUNITY\COMMON7\IDE\COMMONEXTENSIONS\MICROSOFT\CMAKE\Ninja\ninja.exe" "C:\work\avr\avrdude" 2>&1"
1> Working directory: C:\work\avr\avrdude\out\build\x64
1> [CMake] -- Fetching external libraries, please wait...
1> [CMake] -- Using external library 'libelf'
1> [CMake] -- Using external library 'libusb'
1> [CMake] -- Using external library 'libhidapi'
1> [CMake] -- Using external library 'libftdi'
1> [CMake] -- Configuration summary:
1> [CMake] -- ----------------------
1> [CMake] -- DO HAVE    libelf
1> [CMake] -- DO HAVE    libusb
1> [CMake] -- DON'T HAVE libusb_1_0
1> [CMake] -- DO HAVE    libhidapi
1> [CMake] -- DO HAVE    libftdi
1> [CMake] -- DON'T HAVE libftdi1
1> [CMake] -- DISABLED   doc
1> [CMake] -- DISABLED   parport
1> [CMake] -- DISABLED   linuxgpio
1> [CMake] -- DISABLED   linuxspi
1> [CMake] -- ----------------------
1> [CMake] -- Configuring done
1> [CMake] -- Generating done
1> [CMake] -- Build files have been written to: C:/work/avr/avrdude/out/build/x64
1> Extracted CMake variables.
1> Extracted source files and headers.
1> Extracted code model.
1> Extracted toolchain configurations.
1> Extracted includes paths.
1> CMake generation finished.
```

Failed build log.

<details>
<summary> Click for full build log </summary>

```
>------ Build All started: Project: avrdude, Configuration: x64 ------
  [1/56] Building C object src\CMakeFiles\libavrdude.dir\dfu.c.obj
  [2/56] Building C object src\CMakeFiles\libavrdude.dir\avr910.c.obj
  [3/56] Building C object src\CMakeFiles\libavrdude.dir\arduino.c.obj
  [4/56] Building C object src\CMakeFiles\libavrdude.dir\fileio.c.obj
  [5/56] Building C object src\CMakeFiles\libavrdude.dir\avr.c.obj
  [6/56] Building C object src\CMakeFiles\libavrdude.dir\avrftdi.c.obj
  [7/56] Building C object src\CMakeFiles\libavrdude.dir\avrftdi_tpi.c.obj
  [8/56] Building C object src\CMakeFiles\libavrdude.dir\avrpart.c.obj
  [9/56] Building C object src\CMakeFiles\libavrdude.dir\buspirate.c.obj
  [10/56] Building C object src\CMakeFiles\libavrdude.dir\crc16.c.obj
  [11/56] Building C object src\CMakeFiles\libavrdude.dir\config.c.obj
  [12/56] Building C object src\CMakeFiles\libavrdude.dir\bitbang.c.obj
  [13/56] Building C object src\CMakeFiles\libavrdude.dir\butterfly.c.obj
  [14/56] Building C object src\CMakeFiles\libavrdude.dir\flip2.c.obj
  [15/56] Building C object src\CMakeFiles\libavrdude.dir\jtagmkI.c.obj
  [16/56] Building C object src\CMakeFiles\libavrdude.dir\ppi.c.obj
  [17/56] Building C object src\CMakeFiles\libavrdude.dir\serbb_posix.c.obj
  [18/56] Building C object src\CMakeFiles\libavrdude.dir\jtag3.c.obj
  [19/56] Building C object src\CMakeFiles\libavrdude.dir\jtagmkII.c.obj
  [20/56] Building C object src\CMakeFiles\libavrdude.dir\ft245r.c.obj
  [21/56] Building C object src\CMakeFiles\libavrdude.dir\flip1.c.obj
  [22/56] Building C object src\CMakeFiles\libavrdude.dir\linuxgpio.c.obj
  [23/56] Building C object src\CMakeFiles\libavrdude.dir\ppiwin.c.obj
  [24/56] Building C object src\CMakeFiles\libavrdude.dir\linuxspi.c.obj
  [25/56] Building C object src\CMakeFiles\libavrdude.dir\ser_avrdoper.c.obj
  [26/56] Building C object src\CMakeFiles\libavrdude.dir\lists.c.obj
  [27/56] Building C object src\CMakeFiles\libavrdude.dir\pgm.c.obj
  [28/56] Building C object src\CMakeFiles\libavrdude.dir\pgm_type.c.obj
[... 163 lines cut ...]
C:\work\avr\avrdude\out\build\x64\usb_libusb.c.obj : error LNK2001: unresolved external symbol verbose
C:\work\avr\avrdude\out\build\x64\usbtiny.c.obj : error LNK2001: unresolved external symbol verbose
C:\work\avr\avrdude\out\build\x64\jtagmkII.c.obj : error LNK2001: unresolved external symbol verbose
C:\work\avr\avrdude\out\build\x64\jtag3.c.obj : error LNK2001: unresolved external symbol verbose
C:\work\avr\avrdude\out\build\x64\ser_avrdoper.c.obj : error LNK2001: unresolved external symbol verbose
C:\work\avr\avrdude\out\build\x64\ser_win32.c.obj : error LNK2001: unresolved external symbol verbose
C:\work\avr\avrdude\out\build\x64\flip1.c.obj : error LNK2001: unresolved external symbol verbose
C:\work\avr\avrdude\out\build\x64\flip2.c.obj : error LNK2001: unresolved external symbol verbose
C:\work\avr\avrdude\out\build\x64\ft245r.c.obj : error LNK2001: unresolved external symbol verbose
C:\work\avr\avrdude\out\build\x64\jtagmkI.c.obj : error LNK2001: unresolved external symbol verbose
C:\work\avr\avrdude\out\build\x64\avrftdi.c.obj : error LNK2001: unresolved external symbol verbose
C:\work\avr\avrdude\out\build\x64\bitbang.c.obj : error LNK2001: unresolved external symbol verbose
C:\work\avr\avrdude\out\build\x64\butterfly.c.obj : error LNK2001: unresolved external symbol verbose
C:\work\avr\avrdude\out\build\x64\dfu.c.obj : error LNK2001: unresolved external symbol verbose
C:\work\avr\avrdude\out\build\x64\fileio.c.obj : error LNK2019: unresolved external symbol quell_progress referenced in function fileio
C:\work\avr\avrdude\out\build\x64\stk500v2.c.obj : error LNK2001: unresolved external symbol quell_progress
C:\work\avr\avrdude\out\build\x64\usbtiny.c.obj : error LNK2001: unresolved external symbol quell_progress
C:\work\avr\avrdude\out\build\x64\update.c.obj : error LNK2001: unresolved external symbol quell_progress
C:\work\avr\avrdude\out\build\x64\src\libavrdude.dll : fatal error LNK1120: 8 unresolved externals
  ninja: build stopped: subcommand failed.

Build All failed.
```

</details>

mariusgreuel, 2022-05-09:
The avrdude code is not designed to build a Windows DLL. Is there a specific reason why you are trying to build a DLL?

dl8dtl, 2022-05-09:
> The avrdude code is not designed to build a Windows DLL.

I still think this must somehow be a CMake issue.

MacOS shows the same error messages when trying to build a shared library using CMake as the Windows build shows (missing global symbols). However, autoconf has no issues building libavrdude.1.dylib in the same environment. There must be something they're handling differently … so regardless of whether there is anyone using the shared libs right now, I think we'd better analyze the reason. After all, building a shared lib is not supposed to cause any additional effort at all (and it has never been additional effort with autoconf/automake).

dl8dtl, 2022-05-09:
> I don't think you can make this work on Windows, as DLLs cannot import symbols.

Seems a little strange to me that a DLL cannot access the global namespace of the target application … anyway, would it help to turn "progname" into a library-internal (not really entirely) global symbol? We could then have something like `set_progname()` that gets called once when the application starts.

mariusgreuel, 2022-05-09:
> would it help to turn "progname" into a library-internal (not really entirely) global symbol?

Sure. IMHO, that is the way to go. A library should never expose global data, let alone data structures, and the dependencies should only point one way.

But, again I ask, is it worth the trouble if nobody is using it? If there is ever a need, one could make the changes then.

dl8dtl, 2022-05-09:
The biggest problem would be `avrdude_message()`. This has been purposely moved out of the library, so the target application can implement the way it wants to handle messages. Thus, this is always supposed to be implemented outside the DLL.
OK, could then also be worked around by providing a registration function that "installs" the message handling function as a kind of callback.
Still strange to me why the MacOS dylib build shows the very same error messages as Windows DLL when being built with CMake, wheres the automake build completes without errors.

For reference, here's the dylib linker command lines.
automake:
```
libtool: link: gcc -dynamiclib -Wl,-undefined -Wl,dynamic_lookup -o .libs/libavrdude.1.dylib  .libs/libavrdude_la-config_gram.o .libs/libavrdude_la-lexer.o .libs/libavrdude_la-arduino.o .libs/libavrdude_la-avr.o .libs/libavrdude_la-avr910.o .libs/libavrdude_la-avrftdi.o .libs/libavrdude_la-avrftdi_tpi.o .libs/libavrdude_la-avrpart.o .libs/libavrdude_la-bitbang.o .libs/libavrdude_la-buspirate.o .libs/libavrdude_la-butterfly.o .libs/libavrdude_la-config.o .libs/libavrdude_la-confwin.o .libs/libavrdude_la-crc16.o .libs/libavrdude_la-dfu.o .libs/libavrdude_la-fileio.o .libs/libavrdude_la-flip1.o .libs/libavrdude_la-flip2.o .libs/libavrdude_la-ft245r.o .libs/libavrdude_la-jtagmkI.o .libs/libavrdude_la-jtagmkII.o .libs/libavrdude_la-jtag3.o .libs/libavrdude_la-linuxgpio.o .libs/libavrdude_la-linuxspi.o .libs/libavrdude_la-lists.o .libs/libavrdude_la-micronucleus.o .libs/libavrdude_la-par.o .libs/libavrdude_la-pgm.o .libs/libavrdude_la-pgm_type.o .libs/libavrdude_la-pickit2.o .libs/libavrdude_la-pindefs.o .libs/libavrdude_la-ppi.o .libs/libavrdude_la-ppiwin.o .libs/libavrdude_la-serbb_posix.o .libs/libavrdude_la-serbb_win32.o .libs/libavrdude_la-ser_avrdoper.o .libs/libavrdude_la-ser_posix.o .libs/libavrdude_la-ser_win32.o .libs/libavrdude_la-stk500.o .libs/libavrdude_la-stk500v2.o .libs/libavrdude_la-stk500generic.o .libs/libavrdude_la-teensy.o .libs/libavrdude_la-usbasp.o .libs/libavrdude_la-serialupdi.o .libs/libavrdude_la-updi_link.o .libs/libavrdude_la-updi_state.o .libs/libavrdude_la-updi_readwrite.o .libs/libavrdude_la-updi_nvm.o .libs/libavrdude_la-usb_hidapi.o .libs/libavrdude_la-usb_libusb.o .libs/libavrdude_la-usbtiny.o .libs/libavrdude_la-update.o .libs/libavrdude_la-wiring.o .libs/libavrdude_la-xbee.o   -L/opt/local/lib -lhidapi -lreadline -lncurses -ltermcap    -install_name  /usr/local/lib/libavrdude.1.dylib -compatibility_version 2 -current_version 2.0 -Wl,-single_module
```
CMake:
```
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc -I/opt/local/include -O2 -g -DNDEBUG -arch arm64 -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX12.3.sdk -dynamiclib -Wl,-headerpad_max_install_names -compatibility_version 1.0.0 -current_version 1.0.0 -o libavrdude.1.0.0.dylib -install_name @rpath/libavrdude.1.dylib CMakeFiles/libavrdude.dir/arduino.c.o CMakeFiles/libavrdude.dir/avr.c.o CMakeFiles/libavrdude.dir/avr910.c.o CMakeFiles/libavrdude.dir/avrftdi.c.o CMakeFiles/libavrdude.dir/avrftdi_tpi.c.o CMakeFiles/libavrdude.dir/avrpart.c.o CMakeFiles/libavrdude.dir/bitbang.c.o CMakeFiles/libavrdude.dir/buspirate.c.o CMakeFiles/libavrdude.dir/butterfly.c.o CMakeFiles/libavrdude.dir/config.c.o CMakeFiles/libavrdude.dir/confwin.c.o CMakeFiles/libavrdude.dir/crc16.c.o CMakeFiles/libavrdude.dir/dfu.c.o CMakeFiles/libavrdude.dir/fileio.c.o CMakeFiles/libavrdude.dir/flip1.c.o CMakeFiles/libavrdude.dir/flip2.c.o CMakeFiles/libavrdude.dir/ft245r.c.o CMakeFiles/libavrdude.dir/jtagmkI.c.o CMakeFiles/libavrdude.dir/jtagmkII.c.o CMakeFiles/libavrdude.dir/jtag3.c.o CMakeFiles/libavrdude.dir/linuxgpio.c.o CMakeFiles/libavrdude.dir/linuxspi.c.o CMakeFiles/libavrdude.dir/lists.c.o CMakeFiles/libavrdude.dir/micronucleus.c.o CMakeFiles/libavrdude.dir/par.c.o CMakeFiles/libavrdude.dir/pgm.c.o CMakeFiles/libavrdude.dir/pgm_type.c.o CMakeFiles/libavrdude.dir/pickit2.c.o CMakeFiles/libavrdude.dir/pindefs.c.o CMakeFiles/libavrdude.dir/ppi.c.o CMakeFiles/libavrdude.dir/ppiwin.c.o CMakeFiles/libavrdude.dir/serbb_posix.c.o CMakeFiles/libavrdude.dir/serbb_win32.c.o CMakeFiles/libavrdude.dir/ser_avrdoper.c.o CMakeFiles/libavrdude.dir/ser_posix.c.o CMakeFiles/libavrdude.dir/ser_win32.c.o CMakeFiles/libavrdude.dir/serialupdi.c.o CMakeFiles/libavrdude.dir/stk500.c.o CMakeFiles/libavrdude.dir/stk500v2.c.o CMakeFiles/libavrdude.dir/stk500generic.c.o CMakeFiles/libavrdude.dir/teensy.c.o CMakeFiles/libavrdude.dir/updi_link.c.o CMakeFiles/libavrdude.dir/updi_nvm.c.o CMakeFiles/libavrdude.dir/updi_readwrite.c.o CMakeFiles/libavrdude.dir/updi_state.c.o CMakeFiles/libavrdude.dir/usbasp.c.o CMakeFiles/libavrdude.dir/usb_hidapi.c.o CMakeFiles/libavrdude.dir/usb_libusb.c.o CMakeFiles/libavrdude.dir/usbtiny.c.o CMakeFiles/libavrdude.dir/update.c.o CMakeFiles/libavrdude.dir/wiring.c.o CMakeFiles/libavrdude.dir/xbee.c.o CMakeFiles/libavrdude.dir/__/lexer.c.o CMakeFiles/libavrdude.dir/__/config_gram.c.o  -lm /opt/local/lib/libelf.a /opt/local/lib/libusb.dylib /opt/local/lib/libusb-1.0.dylib /opt/local/lib/libhidapi.dylib /opt/local/lib/libftdi.dylib /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX12.3.sdk/usr/lib/libreadline.tbd
```

dl8dtl, 2022-05-09:
> Sure. IMHO, that is the way to go. A library should never expose global data, let alone data structures, and the dependencies should only point one way.

It's not about *exposing* global data, it's about *using* global data (from the application).

> But, again I ask, is it worth the trouble if nobody is using it?

Because the entire idea of separating the library backend is fairly useless if it is not usable in the end. We once started to separate the backend code, so I think we should walk that way to the end (and it's my feeling that we already managed 95 % of the way, so it's not that we were just at the beginning).

subnut, 2022-05-10:
> OK, could then also be worked around by providing a registration function that "installs" the message handling function as a kind of callback.

Hmm... would a function pointer work?

I mean, the application defines the function `foo()` and then registers it like `avrdude_message_handler(&foo)`.

dl8dtl, 2022-05-10:
> Hmm... would a function pointer work?

Yes, that's the idea if directly accessing global namespace (of the application) doesn't work.

mcuee, 2022-05-14:
VS2019 configure log.

```
1> CMake generation started for configuration: 'x64-Debug'.
1> Found and using vcpkg toolchain file (C:/src/vcpkg/vcpkg/scripts/buildsystems/vcpkg.cmake).
1> Command line: "C:\WINDOWS\system32\cmd.exe" /c "%SYSTEMROOT%\System32\chcp.com 65001 >NUL && "C:\PROGRAM FILES (X86)\MICROSOFT VISUAL STUDIO\2019\COMMUNITY\COMMON7\IDE\COMMONEXTENSIONS\MICROSOFT\CMAKE\CMake\bin\cmake.exe"  -G "Ninja"  -DCMAKE_BUILD_TYPE:STRING="Debug" -DCMAKE_INSTALL_PREFIX:PATH="C:\work\avr\avrdude_test\avrdude-fork\out\install\x64-Debug" -DUSE_EXTERNAL:BOOL="True" -DUSE_STATIC_LIBS:BOOL="True" -DHAVE_LIBFTDI:FILEPATH="" -DHAVE_LIBFTDI1:FILEPATH="" -DCMAKE_VERBOSE_MAKEFILE:BOOL="True" -DDEBUG_CMAKE:BOOL="True" -DCMAKE_C_COMPILER:FILEPATH="C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x64/cl.exe" -DCMAKE_CXX_COMPILER:FILEPATH="C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x64/cl.exe"  -DCMAKE_MAKE_PROGRAM="C:\PROGRAM FILES (X86)\MICROSOFT VISUAL STUDIO\2019\COMMUNITY\COMMON7\IDE\COMMONEXTENSIONS\MICROSOFT\CMAKE\Ninja\ninja.exe" -DCMAKE_TOOLCHAIN_FILE="C:/src/vcpkg/vcpkg/scripts/buildsystems/vcpkg.cmake" "C:\work\avr\avrdude_test\avrdude-fork" 2>&1"
1> Working directory: C:\work\avr\avrdude_test\avrdude-fork\out\build\x64-Debug
1> [CMake] -- Fetching external libraries, please wait...
1> [CMake] -- Using external library 'libelf'
1> [CMake] -- Using external library 'libusb'
1> [CMake] -- Using external library 'libhidapi'
1> [CMake] -- Using external library 'libftdi'
1> [CMake] -- Configuration summary:
1> [CMake] -- ----------------------
1> [CMake] -- CMAKE_HOST_SYSTEM: Windows-10.0.22000
1> [CMake] -- CMAKE_SYSTEM: Windows-10.0.22000
1> [CMake] -- CMAKE_FIND_ROOT_PATH: C:/src/vcpkg/vcpkg/installed/x64-windows/debug;C:/src/vcpkg/vcpkg/installed/x64-windows
1> [CMake] -- CMAKE_C_COMPILER: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x64/cl.exe
1> [CMake] -- CONFIG_DIR: C:/work/avr/avrdude_test/avrdude-fork/out/install/x64-Debug/etc
1> [CMake] -- AVRDUDE_FULL_VERSION: 7.0-20220511 (65bb41f)
1> [CMake] -- USE_EXTERNAL: True
1> [CMake] -- USE_LIBUSBWIN32: OFF
1> [CMake] -- HAVE_LIBELF: 1
1> [CMake] -- HAVE_LIBUSB: 1
1> [CMake] -- HAVE_LIBUSB_1_0: HAVE_LIBUSB_1_0-NOTFOUND
1> [CMake] -- HAVE_LIBUSB_WIN32: HAVE_LIBUSB_WIN32-NOTFOUND
1> [CMake] -- HAVE_LIBHID: C:/Program Files (x86)/Windows Kits/10/Lib/10.0.22000.0/um/x64/hid.lib
1> [CMake] -- HAVE_LIBHIDAPI: 1
1> [CMake] -- HAVE_LIBFTDI: 1
1> [CMake] -- HAVE_LIBFTDI1: 
1> [CMake] -- HAVE_LIBREADLINE: HAVE_LIBREADLINE-NOTFOUND
1> [CMake] -- HAVE_LIBELF_H: 1
1> [CMake] -- HAVE_LIBELF_LIBELF_H: 
1> [CMake] -- HAVE_USB_H: 
1> [CMake] -- HAVE_LUSB0_USB_H: 1
1> [CMake] -- HAVE_LIBUSB_H: 
1> [CMake] -- HAVE_LIBUSB_1_0_LIBUSB_H: 
1> [CMake] -- HAVE_HIDAPI_HIDAPI_H: 1
1> [CMake] -- LIBUSB_COMPAT_DIR: 
1> [CMake] -- ----------------------
1> [CMake] -- DO HAVE    libelf
1> [CMake] -- DO HAVE    libusb
1> [CMake] -- DON'T HAVE libusb_1_0
1> [CMake] -- DO HAVE    libhidapi
1> [CMake] -- DO HAVE    libftdi
1> [CMake] -- DON'T HAVE libftdi1
1> [CMake] -- DISABLED   doc
1> [CMake] -- DISABLED   parport
1> [CMake] -- DISABLED   linuxgpio
1> [CMake] -- DISABLED   linuxspi
1> [CMake] -- ----------------------
1> [CMake] -- Configuring done
1> [CMake] -- Generating done
1> [CMake] -- Build files have been written to: C:/work/avr/avrdude_test/avrdude-fork/out/build/x64-Debug
1> Extracted CMake variables.
1> Extracted source files and headers.
1> Extracted code model.
1> Extracted toolchain configurations.
1> Extracted includes paths.
1> CMake generation finished.

```

closed 2022-06-08 by mariusgreuel

mcuee, 2022-06-10:
I agree to close this issue as it can be tracked in #952 or #819
* #952
* #819

#952 is still relevant as it is not possible to build the shared library under Windows even with #962.

Fail log under Windows using MSYS2 mingw64.
```
[ 91%] Linking C shared library libavrdude.dll
cd /d C:\work\avr\avrdude\build_mingw64\src && "C:\Program Files\CMake\bin\cmake.exe" -E cmake_link_script CMakeFiles\libavrdude.dir\link.txt --verbose=1
C:\msys64\mingw64\bin\gcc.exe -shared  -o libavrdude.dll CMakeFiles/libavrdude.dir/arduino.c.obj CMakeFiles/libavrdude.dir/avr.c.obj CMakeFiles/libavrdude.dir/avr910.c.obj CMakeFiles/libavrdude.dir/avrftdi.c.obj CMakeFiles/libavrdude.dir/avrftdi_tpi.c.obj CMakeFiles/libavrdude.dir/avrpart.c.obj CMakeFiles/libavrdude.dir/bitbang.c.obj CMakeFiles/libavrdude.dir/buspirate.c.obj CMakeFiles/libavrdude.dir/butterfly.c.obj CMakeFiles/libavrdude.dir/config.c.obj CMakeFiles/libavrdude.dir/confwin.c.obj CMakeFiles/libavrdude.dir/crc16.c.obj CMakeFiles/libavrdude.dir/dfu.c.obj CMakeFiles/libavrdude.dir/fileio.c.obj CMakeFiles/libavrdude.dir/flip1.c.obj CMakeFiles/libavrdude.dir/flip2.c.obj CMakeFiles/libavrdude.dir/ft245r.c.obj CMakeFiles/libavrdude.dir/jtagmkI.c.obj CMakeFiles/libavrdude.dir/jtagmkII.c.obj CMakeFiles/libavrdude.dir/jtag3.c.obj CMakeFiles/libavrdude.dir/linuxgpio.c.obj CMakeFiles/libavrdude.dir/linuxspi.c.obj CMakeFiles/libavrdude.dir/lists.c.obj CMakeFiles/libavrdude.dir/micronucleus.c.obj CMakeFiles/libavrdude.dir/par.c.obj CMakeFiles/libavrdude.dir/pgm.c.obj CMakeFiles/libavrdude.dir/pgm_type.c.obj CMakeFiles/libavrdude.dir/pickit2.c.obj CMakeFiles/libavrdude.dir/pindefs.c.obj CMakeFiles/libavrdude.dir/ppi.c.obj CMakeFiles/libavrdude.dir/ppiwin.c.obj CMakeFiles/libavrdude.dir/serbb_posix.c.obj CMakeFiles/libavrdude.dir/serbb_win32.c.obj CMakeFiles/libavrdude.dir/ser_avrdoper.c.obj CMakeFiles/libavrdude.dir/ser_posix.c.obj CMakeFiles/libavrdude.dir/ser_win32.c.obj CMakeFiles/libavrdude.dir/serialupdi.c.obj CMakeFiles/libavrdude.dir/stk500.c.obj CMakeFiles/libavrdude.dir/stk500v2.c.obj CMakeFiles/libavrdude.dir/stk500generic.c.obj CMakeFiles/libavrdude.dir/teensy.c.obj CMakeFiles/libavrdude.dir/updi_link.c.obj CMakeFiles/libavrdude.dir/updi_nvm.c.obj CMakeFiles/libavrdude.dir/updi_readwrite.c.obj CMakeFiles/libavrdude.dir/updi_state.c.obj CMakeFiles/libavrdude.dir/usbasp.c.obj CMakeFiles/libavrdude.dir/usb_hidapi.c.obj CMakeFiles/libavrdude.dir/usb_libusb.c.obj CMakeFiles/libavrdude.dir/usbtiny.c.obj CMakeFiles/libavrdude.dir/update.c.obj CMakeFiles/libavrdude.dir/wiring.c.obj CMakeFiles/libavrdude.dir/xbee.c.obj CMakeFiles/libavrdude.dir/__/lexer.c.obj CMakeFiles/libavrdude.dir/__/config_gram.c.obj  -lm C:\msys64\mingw64\lib\libelf.a C:\msys64\mingw64\lib\libusb.a C:\msys64\mingw64\lib\libusb-1.0.a C:\msys64\mingw64\lib\libhid.a C:\msys64\mingw64\lib\libhidapi.a C:\msys64\mingw64\lib\libftdi1.a C:\msys64\mingw64\lib\libreadline.dll.a -lsetupapi -lws2_32 -lkernel32 -luser32 -lgdi32 -lwinspool -lshell32 -lole32 -loleaut32 -luuid -lcomdlg32 -ladvapi32
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/11.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: CMakeFiles/libavrdude.dir/arduino.c.obj:arduino.c:(.text+0x3d): undefined reference to `avrdude_message'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/11.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: CMakeFiles/libavrdude.dir/arduino.c.obj:arduino.c:(.text+0xd8): undefined reference to `avrdude_message'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/11.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: CMakeFiles/libavrdude.dir/arduino.c.obj:arduino.c:(.text+0x11c): undefined reference to `avrdude_message'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/11.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: CMakeFiles/libavrdude.dir/arduino.c.obj:arduino.c:(.text+0x15d): undefined reference to `avrdude_message'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/11.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: CMakeFiles/libavrdude.dir/arduino.c.obj:arduino.c:(.rdata$.refptr.progname[.refptr.progname]+0x0): undefined reference to `progname'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/11.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: CMakeFiles/libavrdude.dir/avr.c.obj:avr.c:(.text+0xc2): undefined reference to `avrdude_message'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/11.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: CMakeFiles/libavrdude.dir/avr.c.obj:avr.c:(.text+0x18d): undefined reference to `avrdude_message'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/11.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: CMakeFiles/libavrdude.dir/avr.c.obj:avr.c:(.text+0x265): undefined reference to `avrdude_message'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/11.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: CMakeFiles/libavrdude.dir/avr.c.obj:avr.c:(.text+0x324): undefined reference to `avrdude_message'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/11.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: CMakeFiles/libavrdude.dir/avr.c.obj:avr.c:(.text+0x338): undefined reference to `avrdude_message'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/11.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: CMakeFiles/libavrdude.dir/avr.c.obj:avr.c:(.text+0x35a): more undefined references to `avrdude_message' follow
[... cut ...]
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/11.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: CMakeFiles/libavrdude.dir/flip1.c.obj:flip1.c:(.text+0x390): undefined reference to `avrdude_message'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/11.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: CMakeFiles/libavrdude.dir/flip1.c.obj:flip1.c:(.text+0x3cb): more undefined references to `avrdude_message' follow
collect2.exe: error: ld returned 1 exit status
mingw32-make[2]: [src\CMakeFiles\libavrdude.dir\build.make:911: src/libavrdude.dll] Error 1 (ignored)
mingw32-make[2]: Leaving directory 'C:/work/avr/avrdude/build_mingw64'

```

PR #962 (2022-06-08): CMake Enhancements
This serves as a bucket for various CMake issues and wishes that came up recently.

I will merge myself.

- Fixes #959
- Partially fixes #954