avrdudes/avrdude #955: Dynamic link to dependency libraries

mcuee, 2022-05-10:
My main interest here is for Windows. How can I shared a version of avrdude binary which is dynamically linked to libusb-1.0.dll, libftdi1.dll and libhidapi.dll under Windows? I do not really need to build the shared library as it may take some time as per #954.
* #954

mcuee, 2022-05-10:
One possibility is to let the user to select the header file name and library file name for these dependencies libraries. So the user can easily select the static library (eg: libusb-1.0.a) or the dynamic library (eg: libusb-1.0.dll.a).

mcuee, 2022-05-10:
Interestingly the "classic" release tarball (generated by auto-tools) will build the avrdude binary I want -- which means that it has dynamic link to the dependancy libraries.

```
 /bin/sh ./libtool  --tag=CC   --mode=link gcc -Wall -Wno-pointer-sign -g -O2 -DWIN32NATIVE  -static -o avrdude.exe avrdude-main.o avrdude-whereami.o avrdude-term.o ./libavrdude.a -lusb-1.0  -lusb -lftdi1  -lhid -lsetupapi -lelf -lpthread -lm -lhidapi -lreadline -lncurses -ltermcap  -lftdi1 -lws2_32
libtool: link: gcc -Wall -Wno-pointer-sign -g -O2 -DWIN32NATIVE -o avrdude.exe avrdude-main.o avrdude-whereami.o avrdude-term.o  ./libavrdude.a -lusb-1.0 -lusb -lhid -lsetupapi -lelf -lpthread -lhidapi -lreadline -lncurses -ltermcap -lftdi1 -lws2_32
```

However, CMake is using static libraries.
```
cd /C/work/avr/avrdude-7.0_cmake/avrdude-7.0/build/src && /C/msys64/mingw64/bin/gcc.exe -Wl,--whole-archive CMakeFiles/avrdude.dir/objects.a -Wl,--no-whole-archive -o avrdude.exe -Wl,--out-implib,libavrdude.dll.a -Wl,--major-image-version,0,--minor-image-version,0  libavrdude.a /C/msys64/mingw64/lib/libelf.a /C/msys64/mingw64/lib/libusb.a /C/msys64/mingw64/lib/libusb-1.0.a /C/msys64/mingw64/lib/libhid.a /C/msys64/mingw64/lib/libhidapi.a /C/msys64/mingw64/lib/libftdi1.a /C/msys64/mingw64/lib/libreadline.dll.a -lsetupapi -lws2_32 -lkernel32 -luser32 -lgdi32 -lwinspool -lshell32 -lole32 -loleaut32 -luuid -lcomdlg32 -ladvapi32
```

mcuee, 2022-05-10:
So the hacky way for CMake is to change the build command to use dynamic dependancy libs.

```
 cd /C/work/avr/avrdude-7.0_cmake/avrdude-7.0/build/src && /C/msys64/mingw64/bin/gcc.exe -Wl,--whole-archive CMakeFiles/avrdude.dir/objects.a -Wl,--no-whole-archive -o avrdude.exe -Wl,--out-implib,libavrdude.dll.a -Wl,--major-image-version,0,--minor-image-version,0  libavrdude.a -lelf -lusb -lusb-1.0 -lhid -lhidapi -lftdi1 /C/msys64/mingw64/lib/libreadline.a -lsetupapi -lws2_32 -lkernel32 -luser32 -lgdi32 -lwinspool -lshell32 -lole32 -loleaut32 -luuid -lcomdlg32 -ladvapi32
```

mcuee, 2022-05-10:
Here are three versions of the avrdude 7.0 release 64bit Windows binary using MSYS2 mingw64 (CMake static, CMake dynmaic and auto-tools dynamic). The dependancies dll are also included.
[avrdude-7.0_bin64.zip](https://github.com/avrdudes/avrdude/files/8661981/avrdude-7.0_bin64.zip)

mcuee, 2022-05-11:
The resultant shared avrdude is working fine.

```
PS C:\work\avr\avrdude-7.0_cmake\avrdude-7.0_bin64> .\avrdude_shared.exe -p m328p -P usb -c dragon_isp -v

avrdude_shared.exe: Version 7.0
                    Copyright (c) Brian Dean, http://www.bdmicro.com/
                    Copyright (c) Joerg Wunsch

                    System wide configuration file is "C:/work/avr/avrdude-7.0_cmake/avrdude-7.0_bin64/avrdude.conf"

                    Using Port                    : usb
                    Using Programmer              : dragon_isp
avrdude_shared.exe: usbdev_open(): Found AVRDRAGON, serno: 00xxxxxxxxxx
JTAG ICE mkII sign-on message:
Communications protocol version: 1
M_MCU:
  boot-loader FW version:        255
  firmware version:              7.14
  hardware version:              1
S_MCU:
  boot-loader FW version:        255
  firmware version:              7.14
  hardware version:              2
Serial number:                   00:xx:xx:xx:xx:xx
Device ID:                       AVRDRAGON
                    AVR Part                      : ATmega328P
                    Chip Erase delay              : 9000 us
                    PAGEL                         : PD7
                    BS2                           : PC2
                    RESET disposition             : dedicated
                    RETRY pulse                   : SCK
                    Serial program mode           : yes
                    Parallel program mode         : yes
                    Timeout                       : 200
                    StabDelay                     : 100
                    CmdexeDelay                   : 25
                    SyncLoops                     : 32
                    PollIndex                     : 3
                    PollValue                     : 0x53
                    Memory Detail                 :

                                                      Block Poll               Page                       Polled
                      Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
                      ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
                      eeprom                 65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
                      flash                  65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
                      lfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
                      hfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
                      efuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
                      lock                    0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
                      calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00
                      signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00

                    Programmer Type : DRAGON_ISP
                    Description     : Atmel AVR Dragon in ISP mode
                    Vtarget         : 4.7 V
                    SCK period      : 1.00 us

avrdude_shared.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.17s

avrdude_shared.exe: Device signature = 0x1e950f (probably m328p)

avrdude_shared.exe done.  Thank you.

```

mcuee, 2022-05-14:
#962 is working for MSYS2 mingw32/mingw64 build.

Still it will be great that users can define the option to link to static/shared lib individually (per library basis). For example, only use dynamic link for libusb-1.0, hidapi and libftdi1; and use static link to libelf and readline.

mcuee, 2022-06-11:
#962 has been merged and the option to use static lib or dynamic lib was added. But it has not been implemented yet.
* https://github.com/avrdudes/avrdude/pull/962#issuecomment-1150075472

@mariusgreuel wrote in the pull request comments.
>Note that the CMake files now include the option USE_STATIC_LIBS, which people can use to select either static or dynamic linking. Note, however, that I leave the implementation for somebody else in a separate PR. The current implementation will not work for non-defaults, i.e. statically linking for Linux or dynamically on Windows.

mcuee, 2022-08-13:
For example, the default under Linux is dynamic linking.
```
mcuee@UbuntuSwift3:~/build/avr/avrdude/build_linux/src$ ldd avrdude
	linux-vdso.so.1 (0x00007fffd11d1000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f8844b7b000)
	libelf.so.1 => /lib/x86_64-linux-gnu/libelf.so.1 (0x00007f8844b5f000)
	libusb-0.1.so.4 => /usr/local/lib/libusb-0.1.so.4 (0x00007f8844b57000)
	libusb-1.0.so.0 => /usr/local/lib/libusb-1.0.so.0 (0x00007f8844b37000)
	libhidapi-libusb.so.0 => /usr/local/lib/libhidapi-libusb.so.0 (0x00007f8844b2c000)
	libftdi.so.1 => /lib/x86_64-linux-gnu/libftdi.so.1 (0x00007f8844b20000)
	libftdi1.so.2 => /usr/local/lib/libftdi1.so.2 (0x00007f8844b0f000)
	libreadline.so.8 => /lib/x86_64-linux-gnu/libreadline.so.8 (0x00007f8844abf000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f88448cd000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f8844d7f000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f88448b1000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f884488e000)
	libudev.so.1 => /lib/x86_64-linux-gnu/libudev.so.1 (0x00007f884485f000)
	libtinfo.so.6 => /lib/x86_64-linux-gnu/libtinfo.so.6 (0x00007f884482f000)

```

mcuee, 2022-08-13:
Same for macOS.
```
mcuee@mcuees-Mac-mini avr % otool -L ./avrdude/build_darwin/src/avrdude
./avrdude/build_darwin/src/avrdude:
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1311.100.3)
	/opt/homebrew/opt/libusb-compat/lib/libusb-0.1.4.dylib (compatibility version 9.0.0, current version 9.4.0)
	/opt/homebrew/opt/libusb/lib/libusb-1.0.0.dylib (compatibility version 4.0.0, current version 4.0.0)
	/opt/homebrew/opt/libhid/lib/libhid.0.dylib (compatibility version 1.0.0, current version 1.0.0)
	/opt/homebrew/opt/hidapi/lib/libhidapi.0.dylib (compatibility version 0.0.0, current version 0.12.0)
	/opt/homebrew/opt/libftdi0/lib/libftdi.1.dylib (compatibility version 22.0.0, current version 22.0.0)
	/opt/homebrew/opt/libftdi/lib/libftdi1.2.dylib (compatibility version 2.0.0, current version 2.5.0)
	/usr/lib/libedit.3.dylib (compatibility version 2.0.0, current version 3.0.0)
```

On the othe hand, the build from Arduino alreay uses static linking for macOS.
https://github.com/arduino/avrdude-packing/releases/tag/7.0-arduino.3
```
mcuee@mcuees-Mac-mini avr % otool -L ./avrdude_test/avrdude_macOS_64bit/bin/avrdude 
./avrdude_test/avrdude_macOS_64bit/bin/avrdude:
	/System/Library/Frameworks/Foundation.framework/Versions/C/Foundation (compatibility version 300.0.0, current version 1673.126.0)
	/System/Library/Frameworks/IOKit.framework/Versions/A/IOKit (compatibility version 1.0.0, current version 275.0.0)
	/System/Library/Frameworks/Cocoa.framework/Versions/A/Cocoa (compatibility version 1.0.0, current version 23.0.0)
	/System/Library/Frameworks/Security.framework/Versions/A/Security (compatibility version 1.0.0, current version 59306.41.2)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1281.0.0)
	/System/Library/Frameworks/AppKit.framework/Versions/C/AppKit (compatibility version 45.0.0, current version 1894.10.126)
	/System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation (compatibility version 150.0.0, current version 1673.126.0)
```

mcuee, 2022-08-13:
Take note the Windows MSYS2 mingw32/mingw64 build will have a dynamic link to libusb0.dll if you link to libusb-win32. 

Ref:
* https://github.com/avrdudes/avrdude/issues/987#issuecomment-1204256639

The solution for  is to use libusb-compat-0.1 to replace libusb-win32. 

You can of course use the MSVC build for Windows as well which as of now only works with static linking.

mcuee, 2022-08-13:
Arduino is using the following patch to use static linking for Linux/macOS and mingw32.
https://github.com/arduino/avrdude-packing/blob/main/patches/0005-Cmake-always-prefer-static-libraries.patch

Maybe this can be adapted to the latest git so that user can select either static linking or dynamic linking for Linux/,macOS and mingw32/64 (not for MSVC).

```
From 3f5e03457330d3f75887d7e098d875b4d553124b Mon Sep 17 00:00:00 2001
From: Umberto Baldi <u.baldi@arduino.cc>
Date: Tue, 3 May 2022 16:59:10 +0200
Subject: [PATCH] CMake always prefer static libraries

---
 CMakeLists.txt | 22 ++++++----------------
 1 file changed, 6 insertions(+), 16 deletions(-)

diff --git a/CMakeLists.txt b/CMakeLists.txt
index 95c89d2..f4546b5 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -110,22 +110,12 @@ endif()
 # Detect installed libraries
 # =====================================
 
-# Prefer static libraries over DLLs on Windows
-if(WIN32)
-    set(PREFERRED_LIBELF libelf.a elf)
-    set(PREFERRED_LIBUSB libusb.a usb)
-    set(PREFERRED_LIBUSB_1_0 libusb-1.0.a usb-1.0)
-    set(PREFERRED_LIBHIDAPI libhidapi.a libhidapi-libusb.a libhidapi-hidraw.a hidapi hidapi-libusb hidapi-hidraw)
-    set(PREFERRED_LIBFTDI libftdi.a ftdi)
-    set(PREFERRED_LIBFTDI1 libftdi1.a ftdi1)
-else()
-    set(PREFERRED_LIBELF elf)
-    set(PREFERRED_LIBUSB usb)
-    set(PREFERRED_LIBUSB_1_0 usb-1.0)
-    set(PREFERRED_LIBHIDAPI hidapi hidapi-libusb hidapi-hidraw)
-    set(PREFERRED_LIBFTDI ftdi)
-    set(PREFERRED_LIBFTDI1 ftdi1)
-endif()
+set(PREFERRED_LIBELF libelf.a elf)
+set(PREFERRED_LIBUSB libusb.a usb)
+set(PREFERRED_LIBUSB_1_0 libusb-1.0.a usb-1.0)
+set(PREFERRED_LIBHIDAPI libhidapi.a libhidapi-libusb.a libhidapi-hidraw.a hidapi hidapi-libusb hidapi-hidraw)
+set(PREFERRED_LIBFTDI libftdi.a ftdi)
+set(PREFERRED_LIBFTDI1 libftdi1.a ftdi1)
 
 # -------------------------------------
 # Find libelf
-- 
2.17.1

```

mcuee, 2022-08-13:
> Arduino is using the following patch on top of avrdude 7.0 release to use static linking for Linux/macOS and mingw32. https://github.com/arduino/avrdude-packing/blob/main/patches/0005-Cmake-always-prefer-static-libraries.patch
> 
> Maybe this can be adapted to the latest git so that user can select either static linking or dynamic linking for Linux/,macOS and mingw32/64 (not for MSVC). Take note #962 has been merged, so the changes need to be on top of that.

I think the patch may not work as it is already included in the main git tree.
Ref:
https://github.com/avrdudes/avrdude/blob/main/CMakeLists.txt#L116-L136

mcuee, 2022-10-24:
Interestingly the following dirty patch fixes the issue for mingw.
```
$ git diff
diff --git a/CMakeLists.txt b/CMakeLists.txt
index 400da29..5931f42 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -37,7 +37,7 @@ option(BUILD_SHARED_LIBS "Build shared libraries" OFF)

 if(WIN32)
     # Prefer static libraries over DLLs on Windows
-    option(USE_STATIC_LIBS "Use static libraries" ON)
+    option(USE_STATIC_LIBS "Use static libraries" OFF)
 else()
     option(USE_STATIC_LIBS "Use static libraries" OFF)
 endif()

$ cmake -G"MSYS Makefiles" -D CMAKE_BUILD_TYPE=RelWithDebInfo -D CMAKE_VERBOSE_MAKEFILE=ON -B build_mingw64_shared

$ cmake --build build_mingw64_shared
...
[100%] Linking C executable avrdude.exe
cd /C/work/avr/avrdude_test/avrdude_main/build_mingw64_shared/src && /C/msys64/mingw64/bin/cmake.exe -E rm -f CMakeFiles/avrdude.dir/objects.a
cd /C/work/avr/avrdude_test/avrdude_main/build_mingw64_shared/src && /C/msys64/mingw64/bin/ar.exe qc CMakeFiles/avrdude.dir/objects.a "CMakeFiles/avrdude.dir/main.c.obj" "CMakeFiles/avrdude.dir/term.c.obj" "CMakeFiles/avrdude.dir/avrintel.c.obj" "CMakeFiles/avrdude.dir/developer_opts.c.obj" "CMakeFiles/avrdude.dir/whereami.c.obj" "CMakeFiles/avrdude.dir/windows.rc.obj"
cd /C/work/avr/avrdude_test/avrdude_main/build_mingw64_shared/src && /C/msys64/mingw64/bin/cc.exe -O2 -g -DNDEBUG -Wl,--whole-archive CMakeFiles/avrdude.dir/objects.a -Wl,--no-whole-archive -o avrdude.exe -Wl,--out-implib,libavrdude.dll.a -Wl,--major-image-version,0,--minor-image-version,0  libavrdude.a -lm /C/msys64/mingw64/lib/libelf.a /C/msys64/mingw64/lib/libusb.dll.a /C/msys64/mingw64/lib/libusb-1.0.dll.a /C/msys64/mingw64/lib/libhid.a /C/msys64/mingw64/lib/libhidapi.dll.a /C/msys64/mingw64/lib/libftdi1.dll.a /C/msys64/mingw64/lib/libreadline.dll.a -lsetupapi -lws2_32 -lkernel32 -luser32 -lgdi32 -lwinspool -lshell32 -lole32 -loleaut32 -luuid -lcomdlg32 -ladvapi32
make[2]: Leaving directory '/c/work/avr/avrdude_test/avrdude_main/build_mingw64_shared'
[100%] Built target avrdude
make[1]: Leaving directory '/c/work/avr/avrdude_test/avrdude_main/build_mingw64_shared'
/C/msys64/mingw64/bin/cmake.exe -E cmake_progress_start /C/work/avr/avrdude_test/avrdude_main/build_mingw64_shared/CMakeFiles 0

$ ldd ./build_mingw64_shared/src/avrdude.exe
        ntdll.dll => /c/WINDOWS/SYSTEM32/ntdll.dll (0x7ffbb8c40000)
        KERNEL32.DLL => /c/WINDOWS/System32/KERNEL32.DLL (0x7ffbb7990000)
        KERNELBASE.dll => /c/WINDOWS/System32/KERNELBASE.dll (0x7ffbb6410000)
        msvcrt.dll => /c/WINDOWS/System32/msvcrt.dll (0x7ffbb7150000)
        SETUPAPI.dll => /c/WINDOWS/System32/SETUPAPI.dll (0x7ffbb7200000)
        WS2_32.dll => /c/WINDOWS/System32/WS2_32.dll (0x7ffbb6ac0000)
        RPCRT4.dll => /c/WINDOWS/System32/RPCRT4.dll (0x7ffbb8ae0000)
        HID.DLL => /c/WINDOWS/SYSTEM32/HID.DLL (0x7ffbb4a30000)
        libhidapi-0.dll => /mingw64/bin/libhidapi-0.dll (0x7ffb968b0000)
        libftdi1.dll => /mingw64/bin/libftdi1.dll (0x7ffb77450000)
        libusb-1.0.dll => /mingw64/bin/libusb-1.0.dll (0x7ffb773e0000)
        libreadline8.dll => /mingw64/bin/libreadline8.dll (0x7ffb73490000)
        USER32.dll => /c/WINDOWS/System32/USER32.dll (0x7ffbb7b80000)
        win32u.dll => /c/WINDOWS/System32/win32u.dll (0x7ffbb6a20000)
        GDI32.dll => /c/WINDOWS/System32/GDI32.dll (0x7ffbb70c0000)
        gdi32full.dll => /c/WINDOWS/System32/gdi32full.dll (0x7ffbb6790000)
        msvcp_win.dll => /c/WINDOWS/System32/msvcp_win.dll (0x7ffbb6370000)
        ucrtbase.dll => /c/WINDOWS/System32/ucrtbase.dll (0x7ffbb6110000)
        libusb-0-1-4.dll => /mingw64/bin/libusb-0-1-4.dll (0x7ffb96860000)
        libtermcap-0.dll => /mingw64/bin/libtermcap-0.dll (0x6ac40000)

```

But the following commands failed to build dynamic link version. This is very strange.
```
$  cmake -G"MSYS Makefiles" -D CMAKE_BUILD_TYPE=RelWithDebInfo -D USE_STATIC_LIBS=ON -B build_mingw64_shared

$ cmake --build build_mingw64_shared
```

mcuee, 2022-10-25:
I will close this one as dynamic link works for MinGW. I will create another issue for static link for Linux and macOS.

closed 2022-10-25 by mcuee

umbynos, 2022-10-26:
> Indeed git main static link build for Linux and macOS failed. This may pose some challenges for Arduino people. Maybe @umbynos wants to take a look. I think the fix may not be too complicated but it does involve changing CMake files.

I have a PR ready for when the new version of avrdude comes out to [drop the patch for the cmake](https://github.com/arduino/avrdude-packing/blob/main/patches/0005-Cmake-always-prefer-static-libraries.patch) and to use https://github.com/avrdudes/avrdude/commit/65bb41f8e9bd500d2afe2462a7ff7f93b3c277b7

umbynos, 2022-10-26:
https://github.com/arduino/avrdude-packing/pull/22