avrdudes/avrdude #956: Release tarball with configure script -- build issues under macOS M1 Homebrew

mcuee, 2022-05-10:
Ref: https://github.com/avrdudes/avrdude/releases/tag/v7.0
> File avrdude-7.0.tar.gz contains the source tarball with a configure script, as it used to be the case in all releases before v7.0.

This one does not seem to work for macOS M1 Homebrew environment.

> File v7.0.tar.gz (labelled "Source code") contains the automatically generated tarball from Github, meant to be built using CMake.
This one works perfectly.

mcuee, 2022-05-10:
Issue -- can not find the dependancy libraries. Not so sure if the build script is not looking at /opt/homebrew.

```
mcuee@mcuees-Mac-mini avrdude-7.0 % ./configure 
checking build system type... arm-apple-darwin21.4.0
checking host system type... arm-apple-darwin21.4.0
checking target system type... arm-apple-darwin21.4.0
checking for a BSD-compatible install... /opt/homebrew/bin/ginstall -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /opt/homebrew/bin/gmkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking how to print strings... printf
checking whether make supports the include directive... yes (GNU style)
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking dependency style of gcc... gcc3
checking for a sed that does not truncate output... /opt/homebrew/bin/gsed
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for fgrep... /usr/bin/grep -F
checking for ld used by gcc... /Library/Developer/CommandLineTools/usr/bin/ld
checking if the linker (/Library/Developer/CommandLineTools/usr/bin/ld) is GNU ld... no
checking for BSD- or MS-compatible name lister (nm)... /usr/bin/nm -B
checking the name lister (/usr/bin/nm -B) interface... BSD nm
checking whether ln -s works... yes
checking the maximum length of command line arguments... 786432
checking how to convert arm-apple-darwin21.4.0 file names to arm-apple-darwin21.4.0 format... func_convert_file_noop
checking how to convert arm-apple-darwin21.4.0 file names to toolchain format... func_convert_file_noop
checking for /Library/Developer/CommandLineTools/usr/bin/ld option to reload object files... -r
checking for objdump... objdump
checking how to recognize dependent libraries... pass_all
checking for dlltool... no
checking how to associate runtime and link libraries... printf %s\n
checking for ar... ar
checking for archiver @FILE support... no
checking for strip... strip
checking for ranlib... ranlib
checking command to parse /usr/bin/nm -B output from gcc object... ok
checking for sysroot... no
checking for a working dd... /bin/dd
checking how to truncate binary pipes... /bin/dd bs=4096 count=1
checking for mt... no
checking if : is a manifest tool... no
checking for dsymutil... dsymutil
checking for nmedit... nmedit
checking for lipo... lipo
checking for otool... otool
checking for otool64... no
checking for -single_module linker flag... yes
checking for -exported_symbols_list linker flag... yes
checking for -force_load linker flag... yes
checking how to run the C preprocessor... gcc -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
[... 88 lines cut ...]
checking that generated files are newer than configure... done
configure: creating ./config.status
config.status: creating doc/Makefile
config.status: creating avrdude.spec
config.status: creating Makefile
config.status: creating avrdude.conf.tmp
config.status: creating ac_cfg.h
config.status: executing depfiles commands
config.status: executing libtool commands

Configuration summary:
----------------------
DON'T HAVE libelf
DON'T HAVE libusb
DON'T HAVE libusb_1_0
DON'T HAVE libftdi1
DON'T HAVE libftdi
DON'T HAVE libhid
DON'T HAVE libhidapi
DO HAVE    pthread
DISABLED   doc
DISABLED   parport
DISABLED   linuxgpio
DISABLED   linuxspi
```

mcuee, 2022-05-10:
The CMake tarball has no issues.
```

mcuee@mcuees-Mac-mini build % cmake ..
-- The C compiler identification is AppleClang 13.1.6.13160021
-- The CXX compiler identification is AppleClang 13.1.6.13160021
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Library/Developer/CommandLineTools/usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Git: /opt/homebrew/bin/git (found version "2.36.0") 
-- Found FLEX: /usr/bin/flex (found version "2.6.4") 
-- Found BISON: /usr/bin/bison (found version "2.3")
-- Looking for libelf.h
-- Looking for libelf.h - not found
-- Looking for libelf/libelf.h
-- Looking for libelf/libelf.h - not found
-- Looking for usb.h
-- Looking for usb.h - not found
-- Looking for lusb0_usb.h
-- Looking for lusb0_usb.h - not found
-- Looking for libusb.h
-- Looking for libusb.h - not found
-- Looking for libusb-1.0/libusb.h
-- Looking for libusb-1.0/libusb.h - not found
-- Looking for hidapi/hidapi.h
-- Looking for hidapi/hidapi.h - not found
-- Looking for ftdi_tcioflush
-- Looking for ftdi_tcioflush - not found
-- Configuration summary:
-- ----------------------
-- DO HAVE    libelf
-- DO HAVE    libusb
-- DO HAVE    libusb_1_0
-- DO HAVE    libhidapi
-- DO HAVE    libftdi (but prefer to use libftdi1)
-- DO HAVE    libftdi1
-- DISABLED   doc
-- DISABLED   parport
-- DISABLED   linuxgpio
-- DISABLED   linuxspi
-- ----------------------
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/mcuee/build/avr/cmake7build/avrdude-7.0/build
```

mcuee, 2022-05-10:
Ah, I see, I need to use the following configure script.
```

mcuee@mcuees-Mac-mini avrdude-7.0 % CFLAGS="-I/opt/homebrew/include" LDFLAGS="-L/opt/homebrew/lib" ./configure
checking build system type... arm-apple-darwin21.4.0
checking host system type... arm-apple-darwin21.4.0
checking target system type... arm-apple-darwin21.4.0
checking for a BSD-compatible install... /opt/homebrew/bin/ginstall -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /opt/homebrew/bin/gmkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking how to print strings... printf
checking whether make supports the include directive... yes (GNU style)
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking dependency style of gcc... gcc3
checking for a sed that does not truncate output... /opt/homebrew/bin/gsed
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for fgrep... /usr/bin/grep -F
checking for ld used by gcc... /Library/Developer/CommandLineTools/usr/bin/ld
checking if the linker (/Library/Developer/CommandLineTools/usr/bin/ld) is GNU ld... no
checking for BSD- or MS-compatible name lister (nm)... /usr/bin/nm -B
checking the name lister (/usr/bin/nm -B) interface... BSD nm
checking whether ln -s works... yes
checking the maximum length of command line arguments... 786432
checking how to convert arm-apple-darwin21.4.0 file names to arm-apple-darwin21.4.0 format... func_convert_file_noop
checking how to convert arm-apple-darwin21.4.0 file names to toolchain format... func_convert_file_noop
checking for /Library/Developer/CommandLineTools/usr/bin/ld option to reload object files... -r
checking for objdump... objdump
checking how to recognize dependent libraries... pass_all
checking for dlltool... no
checking how to associate runtime and link libraries... printf %s\n
checking for ar... ar
checking for archiver @FILE support... no
checking for strip... strip
checking for ranlib... ranlib
checking command to parse /usr/bin/nm -B output from gcc object... ok
checking for sysroot... no
checking for a working dd... /bin/dd
checking how to truncate binary pipes... /bin/dd bs=4096 count=1
checking for mt... no
checking if : is a manifest tool... no
checking for dsymutil... dsymutil
checking for nmedit... nmedit
checking for lipo... lipo
checking for otool... otool
checking for otool64... no
checking for -single_module linker flag... yes
checking for -exported_symbols_list linker flag... yes
checking for -force_load linker flag... yes
checking how to run the C preprocessor... gcc -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
[... 242 lines cut ...]
  CC       libavrdude_la-ppi.lo
  CC       libavrdude_la-ppiwin.lo
  CC       libavrdude_la-serbb_posix.lo
  CC       libavrdude_la-serbb_win32.lo
  CC       libavrdude_la-ser_avrdoper.lo
  CC       libavrdude_la-ser_posix.lo
  CC       libavrdude_la-ser_win32.lo
  CC       libavrdude_la-stk500.lo
  CC       libavrdude_la-stk500v2.lo
  CC       libavrdude_la-stk500generic.lo
  CC       libavrdude_la-teensy.lo
  CC       libavrdude_la-usbasp.lo
  CC       libavrdude_la-serialupdi.lo
  CC       libavrdude_la-updi_link.lo
  CC       libavrdude_la-updi_state.lo
  CC       libavrdude_la-updi_readwrite.lo
  CC       libavrdude_la-updi_nvm.lo
  CC       libavrdude_la-usb_hidapi.lo
  CC       libavrdude_la-usb_libusb.lo
  CC       libavrdude_la-usbtiny.lo
  CC       libavrdude_la-update.lo
  CC       libavrdude_la-wiring.lo
  CC       libavrdude_la-xbee.lo
  CCLD     libavrdude.la
```

dl8dtl, 2022-05-10:
If the CMake-generated tarball works for you, this is pretty fine; this is going to be the preferred build system anyway.
I've only rolled the "classic" one for those people who might want to run `./configure` as they used to do before.

mcuee, 2022-05-10:
It is good to keep the "classic" tarball if possible. There are people who like to be able to run "./configure" without extra tool like CMake.

closed 2022-06-20 by mcuee