avrdudes/avrdude #958: Adding PICkit 3 scripting FW support to avrdude

mcuee, 2022-05-11:
Ref: https://www.eevblog.com/forum/projects/programming-atmega328p-using-pickit3/
Ref: https://github.com/martonmiklos/pk2cmd
Ref: http://kair.us/projects/pickitminus/index.html

PICkit 3 Scripting FW (http://ww1.microchip.com/downloads/en/DeviceDoc/PICkit3%20Programmer%20Application%20v3.10.zip) basically emulate PICkit 2. So it is probably good to add PICkit 3 support to the existing PICkit 2 code base.

fancsali, 2022-06-07:
Just out of curiosity: isn't it possible to drive the PICkit3 "natively", similarly how PICkit2 is used by `avrdude`.

If not, why not? What are the technical difficulties? Can we overcome them?

mcuee, 2022-06-07:
@fancsali Because PICkit 3 normal FW is not open. If that can be done (though reverse engineering) that will be the best so that people do not need to switch FW.

fancsali, 2022-06-07:
@mcuee ... and let me have a wild guess, it doesn't even have a bit-banging mode, at least...

mcuee, 2022-06-07:
> @mcuee ... and let me have a wild guess, it doesn't even have a bit-banging mode, at least...

I tend to not calling PICKit 2 and PICKit 3 Scripting FW "bit bang" mode but I understand different people may have different definition of "bit bang".

Ref: https://github.com/avrdudes/avrdude/blob/main/src/pickit2.c#L25
```
/*
 * avrdude interface for PicKit2 programmer
 *
 * The PicKit2 programmer is a cheap device capable
 * of 2 (bidirectional data line), 3, 4 wire SPI comms
 *
 * The PICkit2 software license doesn't allow the source to be
 * modified to program other devices - nor can we distribute
 * their source code. This program is not derived from nor does it
 * contain any of the pickit2 source and should be exempt from any
 * licensing issues.
 *
 * ISP Pinout (AVR - PICKit2 (pin)):
 * RST  - VPP/MCLR (1)
 * VDD  - VDD Target (2) -- possibly optional if AVR self powered
 * GND  - GND (3)
 * MISO - PGD (4)
 * SCLK - PDC (5)
 * MOSI - AUX (6)
 */
```

fancsali, 2022-06-07:
To be honest, I am not familiar with the ins and outs of how the PICkit2 talks to AVRs (or anything else, for that matter), as a result I didn't intend to imply it is or isn't bit banging.

What I was alluding to was, somewhat of a wishful thinking: "If at least the PICkit3 would have some means, even S/W bit-bang to avoid any 'magic' to be needed to make this work..."

mcuee, 2022-06-07:
> To be honest, I am not familiar with the ins and outs of how the PICkit2 talks to AVRs (or anything else, for that matter), as a result I didn't intend to imply it is or isn't bit banging.
>
> What I [...]

You will have to carry out reverse engineering to check out how the Microchip PICKit 3 normal FW works. I think it is not that simple --- it needs to download different FW (or debug executives) with different PIC parts under MPLAB or MPLAB X. 

Microchip parts speaks ICSP, which I think is very different from AVRs. So I am not positive that your "wishful thinking" is possible or not.

fancsali, 2022-06-07:
Exactly. I might buy one PICkit2 and one PICkit3 (well clones), and call it a new hobby of collecting them. Then I have covered all my bases PIC/AVR-wise at least... ;)

mcuee, 2022-10-23:
I will close this one as the scripting FW is not really supported by Microchip.

closed 2022-10-23 by MCUdude