avrdudes/avrdude #959: CMake MSYS2 mingw32/mingw64 build is using libreadline.dll.a and not libreadline.a

mcuee, 2022-05-11:
There is a bit of inconsistency in the CMake build. It is using the static libraries for the dependancies like libusb-1.0, libfti1 and hidapi. But somehow it is using the dynamic link library for readline.

```
cd /C/work/avr/avrdude-7.0_cmake/avrdude-7.0/build/src && /C/msys64/mingw64/bin/gcc.exe -Wl,--whole-archive CMakeFiles/avrdude.dir/objects.a -Wl,--no-whole-archive -o avrdude.exe -Wl,--out-implib,libavrdude.dll.a -Wl,--major-image-version,0,--minor-image-version,0  libavrdude.a /C/msys64/mingw64/lib/libelf.a /C/msys64/mingw64/lib/libusb.a /C/msys64/mingw64/lib/libusb-1.0.a /C/msys64/mingw64/lib/libhid.a /C/msys64/mingw64/lib/libhidapi.a /C/msys64/mingw64/lib/libftdi1.a /C/msys64/mingw64/lib/libreadline.dll.a -lsetupapi -lws2_32 -lkernel32 -luser32 -lgdi32 -lwinspool -lshell32 -lole32 -loleaut32 -luuid -lcomdlg32 -ladvapi32
```

mariusgreuel, 2022-05-11:
You are really trying to poke holes in my CMake files, aren't you? :-)

But you are right, when libreadline is installed on MSYS2, it is picking up the wrong version. I will prepare a CMake PR that fixes a few issues that were raised by you and others.

dl8dtl, 2022-05-11:
The CMake stuff is your domain anyway, so feel free to integrate that yourself. I don't think there's anything gained if you want me to approve that before :-), my CMake knowledge is close to zero.

mariusgreuel, 2022-05-11:
Thanks, I will submit a PR just for the CI checks, and then commit myself.

mcuee, 2022-05-12:
> You are really trying to poke holes in my CMake files, aren't you? :-)

Haha, I play testing/supporting role for the open source projects I am involved (eg: libusb, hidapi, libusb-win32, libusbk, libusbdotnet, pyusb and, libftdi). Other than these projects, I like to use projects like OpenOCD and avrdude to carry out the tests as well as these projects depend on libusb, hidapi and libftdi.

mcuee, 2022-05-14:
I can confirm that #962 fixed this issue.
```
 cmake -G "MSYS Makefiles" ..  -DCMAKE_VERBOSE_MAKEFILE="ON"
...
[100%] Linking C executable avrdude.exe
cd /C/work/avr/avrdude_test/avrdude-fork/build/src && /C/msys64/mingw64/bin/cmake.exe -E rm -f CMakeFiles/avrdude.dir/objects.a
cd /C/work/avr/avrdude_test/avrdude-fork/build/src && /C/msys64/mingw64/bin/ar.exe qc CMakeFiles/avrdude.dir/objects.a "CMakeFiles/avrdude.dir/main.c.obj" "CMakeFiles/avrdude.dir/term.c.obj" "CMakeFiles/avrdude.dir/whereami.c.obj" "CMakeFiles/avrdude.dir/windows.rc.obj"
cd /C/work/avr/avrdude_test/avrdude-fork/build/src && /C/msys64/mingw64/bin/gcc.exe -Wl,--whole-archive CMakeFiles/avrdude.dir/objects.a -Wl,--no-whole-archive -o avrdude.exe -Wl,--out-implib,libavrdude.dll.a -Wl,--major-image-version,0,--minor-image-version,0  libavrdude.a -lm /C/msys64/mingw64/lib/libelf.a /C/msys64/mingw64/lib/libusb.a /C/msys64/mingw64/lib/libusb-1.0.a /C/msys64/mingw64/lib/libhid.a /C/msys64/mingw64/lib/libhidapi.a /C/msys64/mingw64/lib/libftdi1.a /C/msys64/mingw64/lib/libreadline.a -lsetupapi -lws2_32 -lkernel32 -luser32 -lgdi32 -lwinspool -lshell32 -lole32 -loleaut32 -luuid -lcomdlg32 -ladvapi32
make[2]: Leaving directory '/c/work/avr/avrdude_test/avrdude-fork/build'
[100%] Built target avrdude

```

mcuee, 2022-05-14:
I will close this one as I expected #962 to be merged.

PR #962 (2022-06-08): CMake Enhancements
This serves as a bucket for various CMake issues and wishes that came up recently.

I will merge myself.

- Fixes #959
- Partially fixes #954
closed 2022-05-14 by mcuee