avrdudes/avrdude #961: CMake build with vcpkg and Visual Studio

mcuee, 2022-05-13:
Other than libelf, the other dependancies have vcpkg ports.

I kind of can almost build avrdude with manually install libelf but there is a problem related to windows header file problem (winsock.h). Not so sure if this can be fixed.

CMake configure is good.
```
c:\work\avr\avrdude-7.0_cmake\avrdude-7.0\build>cmake -DCMAKE_TOOLCHAIN_FILE=C:/src/vcpkg/vcpkg/scripts/buildsystems/vcpkg.cmake ..
-- Looking for libelf.h
-- Looking for libelf.h - found
-- Looking for libelf/libelf.h
-- Looking for libelf/libelf.h - not found
-- Configuration summary:
-- ----------------------
-- DO HAVE    libelf
-- DO HAVE    libusb
-- DO HAVE    libusb_1_0
-- DO HAVE    libhidapi
-- DO HAVE    libftdi (but prefer to use libftdi1)
-- DO HAVE    libftdi1
-- DISABLED   doc
-- DISABLED   parport
-- DISABLED   linuxgpio
-- DISABLED   linuxspi
-- ----------------------
-- Configuring done
-- Generating done
-- Build files have been written to: C:/work/avr/avrdude-7.0_cmake/avrdude-7.0/build

```

Build failed.
```
Build started...
1>------ Build started: Project: ZERO_CHECK, Configuration: Debug x64 ------
2>------ Build started: Project: libavrdude, Configuration: Debug x64 ------
2>arduino.c
2>avr.c
2>avr910.c
2>avrftdi.c
2>C:\Program Files (x86)\Windows Kits\10\Include\10.0.22000.0\um\winsock.h(108,16): error C2011: 'timeval': 'struct' type redefinition
2>C:\work\avr\avrdude-7.0_cmake\avrdude-7.0\src\msvc\sys/time.h(27): message : see declaration of 'timeval'
2>WARNING: You need to implement deprecated #define for this compiler
2>avrftdi_tpi.c
2>WARNING: You need to implement deprecated #define for this compiler
2>avrpart.c
2>bitbang.c
2>buspirate.c
2>butterfly.c
2>config.c
2>confwin.c
2>dfu.c
2>fileio.c
2>flip1.c
2>flip2.c
2>ft245r.c
2>C:\Program Files (x86)\Windows Kits\10\Include\10.0.22000.0\um\winsock.h(108,16): error C2011: 'timeval': 'struct' type redefinition
2>C:\work\avr\avrdude-7.0_cmake\avrdude-7.0\src\msvc\sys/time.h(27): message : see declaration of 'timeval'
2>WARNING: You need to implement deprecated #define for this compiler
2>jtagmkI.c
2>jtagmkII.c
2>jtag3.c
2>linuxgpio.c
2>Generating Code...
2>Compiling...
2>linuxspi.c
2>lists.c
2>micronucleus.c
2>par.c
2>pgm.c
2>pgm_type.c
2>pickit2.c
2>pindefs.c
2>ppiwin.c
2>serbb_win32.c
2>ser_avrdoper.c
2>ser_win32.c
2>serialupdi.c
2>stk500.c
2>stk500v2.c
2>stk500generic.c
2>teensy.c
2>updi_link.c
2>updi_nvm.c
2>updi_readwrite.c
2>Generating Code...
2>Compiling...
2>updi_state.c
2>usbasp.c
2>C:\Program Files (x86)\Windows Kits\10\Include\10.0.22000.0\um\winsock.h(108,16): error C2011: 'timeval': 'struct' type redefinition
2>C:\work\avr\avrdude-7.0_cmake\avrdude-7.0\src\msvc\sys/time.h(27): message : see declaration of 'timeval'
2>usb_hidapi.c
2>usb_libusb.c
2>C:\Program Files (x86)\Windows Kits\10\Include\10.0.22000.0\um\winsock.h(108,16): error C2011: 'timeval': 'struct' type redefinition
2>C:\work\avr\avrdude-7.0_cmake\avrdude-7.0\src\msvc\sys/time.h(27): message : see declaration of 'timeval'
2>usbtiny.c
2>C:\Program Files (x86)\Windows Kits\10\Include\10.0.22000.0\um\winsock.h(108,16): error C2011: 'timeval': 'struct' type redefinition
2>C:\work\avr\avrdude-7.0_cmake\avrdude-7.0\src\msvc\sys/time.h(27): message : see declaration of 'timeval'
2>update.c
2>wiring.c
2>xbee.c
2>lexer.c
2>config_gram.c
2>Generating Code...
2>Done building project "libavrdude.vcxproj" -- FAILED.
3>------ Build started: Project: avrdude, Configuration: Debug x64 ------
3>main.c
3>term.c
3>Generating Code...
3>LINK : fatal error LNK1104: cannot open file 'Debug\libavrdude.lib'
3>Done building project "avrdude.vcxproj" -- FAILED.
4>------ Build started: Project: ALL_BUILD, Configuration: Debug x64 ------
5>------ Skipped Build: Project: INSTALL, Configuration: Debug x64 ------
5>Project not selected to build for this solution configuration 
========== Build: 2 succeeded, 2 failed, 0 up-to-date, 1 skipped ==========

```

mariusgreuel, 2022-05-13:
> Other than libelf, the other dependancies have vcpkg ports

I actually did check vcpkg awhile back, but they did not serve my needs. The only library did not require patching was libhidapi, which at that time was not available in vcpkg.

I had to customize/rewrite the libraries because:

- libusb: For the lack of libusb0.sys support
- libftdi: For the lack of FTDI driver support
- libelf: CMake changes, smaller fixes
- libhidapi: CMake changes

My primary goal with the MSVC build is to have a functionally working Windows version of avrdude, otherwise I would have gone the easy route with MSYS2. If you are interested, most of my reasoning is still documented in my old repo at <https://github.com/mariusgreuel/avrdude>.

mcuee, 2022-05-13:
> I had to customize/rewrite the libraries because:
>
> * libusb: For the lack of libusb0.sys support
> * libftdi: For the lack of FTDI driver support
> * libelf: CMake changes, smaller fixes
> * libhidapi: [...]

Yes I am interested in the MSVC build (libusb and hidapi do support VS build out of the box and hidapi has moved to CMake; and libftdi does support MSVC build for the library but not the examples; libusb-win32 and libusbk are basically Windows only with VS support and not really supporting GCC). 

Your reasoning about custom libusb (which should support WinUSB, libusb0.sys and libusbk.sys) and custom libftdi (which will support FTDI vendor driver) make perfect sense since you want to have a good out of the box user experiences for the Windows users without switch drivers using Zadig.
1) WinUSB support does not exist in libusb-win32
2) FTDI vendor driver does not work with libftdi (0.1 or 1.0).

So I am fine with the current approach.

mcuee, 2022-05-13:
That being said, I think it is still good to give the users the options of using their preferred library (eg: using the same libusb-1.0 and libftdi-1.0 library as in Linux and macOS) and they may use programmers with WinUSB driver already.

Another thing, some people may take issues with your approach of patching the libftdi to load D2XX DLLs to support FTDI vendor driver out of the box because of licensing concerns due to the fact libftdi being LGPL and avrdude being GPL.

Many years ago there was a heated discussion in OpenOCD (GPL license) which led to the decision to remove D2XX support and only using libftdi or libusb (without libftdi) for FTDI devices. Users will have to use Zadig to switch driver as a result.

mcuee, 2022-05-14:
Ticket created here for libusb-win32 project.
* https://github.com/mcuee/libusb-win32/issues/33

mcuee, 2022-05-14:
In any case, I will not persue vcpkg anyway after playing it for a while. I just do not like it and I have removed, I will close this issue,

mariusgreuel, 2022-05-15:
> there was a heated discussion in OpenOCD (GPL license) which led to the decision to remove D2XX support

That is very unfortunate. Not sure what the reasoning was back then, but loading the FTDI D2XX drivers is clearly covered by the GPL system library exception.

BTW, my libftdi version is a complete rewrite (it only includes the ftdi.h header and the readme) and does not link against ftdi.lib at all, but uses run-time dynamic linking because I do not want a static dependency to `ftd2xx.dll`.

mcuee, 2022-05-15:
> > there was a heated discussion in OpenOCD (GPL license) which led to the decision to remove D2XX support
> 
> That is very unfortunate. Not sure what the reasoning was back then, but loading the FTDI D2XX drivers is clearly covered by the GPL system library exception.

That was the main focus of the GPL discussion. I support your interpretation as the driver (once installed) becomes part of the Windows system. Unfortunately there were strong opposition in that discussion which led to the exit of some OpenOCD developers, including the original author.

> BTW, my libftdi version is a complete rewrite (it only includes the ftdi.h header and the readme) and does not link against ftdi.lib at all, but uses run-time dynamic linking because I do not want a static dependency to `ftd2xx.dll`.

I understand that. 

I think there is potential way out in this case to avoid the licensing discussion. You can probably change the license for your implementation to BSD license even though you use the ftdi.h header. FreeBSD project took a similar approach with their libusb implementation (BSD licensed) by adapting libusb-0.1 and libusb-1.0 header file only. But I am not so sure if this is really worth doing.

Ref: FreeBSD libusb
https://svnweb.freebsd.org/base/head/lib/libusb/
https://svnweb.freebsd.org/base/head/lib/libusb/usb.h?view=markup
https://svnweb.freebsd.org/base/head/lib/libusb/libusb.h?view=markup

mcuee, 2022-05-15:
OpenOCD mailing list archive has lots of discussion back in 2009 (around June time).

This one came from the original developer about the license.
https://sourceforge.net/p/openocd/mailman/openocd-devel/thread/1245813300.8477.545.camel%40localhost/
https://sourceforge.net/p/openocd/mailman/openocd-devel/thread/4A429A93.1050609%40duaneellis.com/

I contributed to this one.
https://sourceforge.net/p/openocd/mailman/openocd-devel/thread/JLEAKDMELBINENLADICFKEGJDDAA.fischermi@t-online.de/

This discussion was part of the motivation for me and Travis Robinson to take over the libusb-win32 project which was not working well back then (broken in Vista). Then Travis also created libusbk project later in 2011 because of the limitation of libusb0.sys and the lack of isoc transfer support of WinUSB back then. I managed to convince Travis to use dual license for libusbk project (BSD and LGPL3/GPL3)
https://sourceforge.net/p/libusb-win32/mailman/message/25069980/
https://sourceforge.net/p/libusb-win32/mailman/message/27321984/

Then I joined the libusbx fork in 2012.
https://sourceforge.net/p/libusbx/mailman/message/29016417/

mcuee, 2022-05-16:
@mariusgreuel @dl8dtl 
Just want to share the history of the D2XX discussion in OpenOCD. It has some impact on the distribution of the Windows binary of avrdude.

closed 2022-06-20 by mcuee

mcuee, 2022-12-08:
Work-around and hacks to get vcpkg working, with libusb-1.0 and GNU Readlie
https://github.com/avrdudes/avrdude/issues/1211#issuecomment-1341820882