avrdudes/avrdude #964: Compiler warnings in different OS and compiler

mcuee, 2022-05-14:
Not so sure if we need to report such issues or not.

For example, under macOS (Mac Mini M1) with Apple clang.

```
[ 34%] Building C object src/CMakeFiles/libavrdude.dir/jtagmkII.c.o
cd /Users/mcuee/build/avr/avrdude/build/src && /Library/Developer/CommandLineTools/usr/bin/cc -DCONFIG_DIR=\"/usr/local/etc\" -I/Users/mcuee/build/avr/avrdude/build/src -I/Users/mcuee/build/avr/avrdude -I/Users/mcuee/build/avr/avrdude/build -I/Users/mcuee/build/avr/avrdude/src -I/opt/homebrew/include -O2 -g -DNDEBUG -arch arm64 -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX12.3.sdk -Wall -std=gnu11 -MD -MT src/CMakeFiles/libavrdude.dir/jtagmkII.c.o -MF CMakeFiles/libavrdude.dir/jtagmkII.c.o.d -o CMakeFiles/libavrdude.dir/jtagmkII.c.o -c /Users/mcuee/build/avr/avrdude/src/jtagmkII.c
/Users/mcuee/build/avr/avrdude/src/jtagmkII.c:491:18: warning: variable 'checksum' set but not used [-Wunused-but-set-variable]
  unsigned short checksum = 0;
                 ^
1 warning generated.
```

mcuee, 2022-05-15:
Compiler warnings under FreeBSD.

```
[  6%] Building C object src/CMakeFiles/libavrdude.dir/avrftdi.c.o
/home/mcuee/build/avrdude/src/avrftdi.c:727:4: warning: 'ftdi_usb_purge_buffers' is deprecated [-Wdeprecated-declarations]
        E(ftdi_usb_purge_buffers(pdata->ftdic), pdata->ftdic);
          ^
/usr/local/include/libftdi1/ftdi.h:566:9: note: 'ftdi_usb_purge_buffers' has been explicitly marked deprecated here
    int DEPRECATED(ftdi_usb_purge_buffers(struct ftdi_context *ftdi));
        ^
/usr/local/include/libftdi1/ftdi.h:247:42: note: expanded from macro 'DEPRECATED'
#define DEPRECATED(func) __attribute__ ((deprecated)) func
                                         ^
1 warning generated.
```

mcuee, 2022-05-16:
> @mcuee could you post your CMake output? Looks like it checks `ftdi_tcioflush` for the old ftdi version but compiles against the new one.

Please refer to the log here.
https://github.com/avrdudes/avrdude/pull/950#issuecomment-1127608470

mcuee, 2022-05-16:
Here is the case with both libftdi and libftdi1 installed.

```
mcuee@mcuees-Mac-mini avrdude % cmake -D CMAKE_C_FLAGS=-I/opt/homebrew/include -D CMAKE_EXE_LINKER_FLAGS=-L/opt/homebrew/lib -D CMAKE_BUILD_TYPE=RelWithDebInfo -D CMAKE_INSTALL_PREFIX:PATH=/opt/homebrew/Cellar/avrdude/7.0 -D CMAKE_VERBOSE_MAKEFILE=ON -B build_macos2
-- The C compiler identification is AppleClang 13.1.6.13160021
-- The CXX compiler identification is AppleClang 13.1.6.13160021
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Library/Developer/CommandLineTools/usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Git: /opt/homebrew/bin/git (found version "2.36.1") 
-- Found FLEX: /usr/bin/flex (found version "2.6.4") 
-- Found BISON: /usr/bin/bison (found version "2.3")
-- Looking for libelf.h
-- Looking for libelf.h - not found
-- Looking for libelf/libelf.h
-- Looking for libelf/libelf.h - found
-- Looking for usb.h
-- Looking for usb.h - found
-- Looking for lusb0_usb.h
-- Looking for lusb0_usb.h - not found
-- Looking for libusb.h
-- Looking for libusb.h - not found
-- Looking for libusb-1.0/libusb.h
-- Looking for libusb-1.0/libusb.h - found
-- Looking for hidapi/hidapi.h
-- Looking for hidapi/hidapi.h - found
-- Looking for ftdi_tcioflush
-- Looking for ftdi_tcioflush - found
-- Configuration summary:
-- ----------------------
-- DO HAVE    libelf
-- DO HAVE    libusb
-- DO HAVE    libusb_1_0
-- DO HAVE    libhidapi
-- DO HAVE    libftdi (but prefer to use libftdi1)
-- DO HAVE    libftdi1
-- DISABLED   doc
-- DISABLED   parport
-- DISABLED   linuxgpio
-- DISABLED   linuxspi
-- ----------------------
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/mcuee/build/avr/avrdude/build_macos2

mcuee@mcuees-Mac-mini avrdude % cmake --build build_macos2 
...
[ 34%] Building C object src/CMakeFiles/libavrdude.dir/jtagmkII.c.o
cd /Users/mcuee/build/avr/avrdude/build_macos2/src && /Library/Developer/CommandLineTools/usr/bin/cc -DCONFIG_DIR=\"/etc/opt/homebrew/Cellar/avrdude/7.0\" -I/Users/mcuee/build/avr/avrdude/build_macos2/src -I/Users/mcuee/build/avr/avrdude -I/Users/mcuee/build/avr/avrdude/build_macos2 -I/Users/mcuee/build/avr/avrdude/src -I/opt/homebrew/include -O2 -g -DNDEBUG -arch arm64 -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX12.3.sdk -Wall -std=gnu11 -MD -MT src/CMakeFiles/libavrdude.dir/jtagmkII.c.o -MF CMakeFiles/libavrdude.dir/jtagmkII.c.o.d -o CMakeFiles/libavrdude.dir/jtagmkII.c.o -c /Users/mcuee/build/avr/avrdude/src/jtagmkII.c
/Users/mcuee/build/avr/avrdude/src/jtagmkII.c:491:18: warning: variable 'checksum' set but not used [-Wunused-but-set-variable]
  unsigned short checksum = 0;
                 ^
1 warning generated.
...
```

mcuee, 2022-05-16:
Here is the log without libftdi0 installed.

```
mcuee@mcuees-Mac-mini avrdude % brew uninstall libftdi0
Uninstalling /opt/homebrew/Cellar/libftdi0/0.20... (23 files, 582.3KB)

mcuee@mcuees-Mac-mini avrdude % cmake -D CMAKE_C_FLAGS=-I/opt/homebrew/include -D CMAKE_EXE_LINKER_FLAGS=-L/opt/homebrew/lib -D CMAKE_BUILD_TYPE=RelWithDebInfo -D CMAKE_INSTALL_PREFIX:PATH=/opt/homebrew/Cellar/avrdude/7.0 -D CMAKE_VERBOSE_MAKEFILE=ON -B build_macos3

-- The C compiler identification is AppleClang 13.1.6.13160021
-- The CXX compiler identification is AppleClang 13.1.6.13160021
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Library/Developer/CommandLineTools/usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Git: /opt/homebrew/bin/git (found version "2.36.1") 
-- Found FLEX: /usr/bin/flex (found version "2.6.4") 
-- Found BISON: /usr/bin/bison (found version "2.3")
-- Looking for libelf.h
-- Looking for libelf.h - not found
-- Looking for libelf/libelf.h
-- Looking for libelf/libelf.h - found
-- Looking for usb.h
-- Looking for usb.h - found
-- Looking for lusb0_usb.h
-- Looking for lusb0_usb.h - not found
-- Looking for libusb.h
-- Looking for libusb.h - not found
-- Looking for libusb-1.0/libusb.h
-- Looking for libusb-1.0/libusb.h - found
-- Looking for hidapi/hidapi.h
-- Looking for hidapi/hidapi.h - found
-- Looking for ftdi_tcioflush
-- Looking for ftdi_tcioflush - found
-- Configuration summary:
-- ----------------------
-- DO HAVE    libelf
-- DO HAVE    libusb
-- DO HAVE    libusb_1_0
-- DO HAVE    libhidapi
-- DON'T HAVE libftdi
-- DO HAVE    libftdi1
-- DISABLED   doc
-- DISABLED   parport
-- DISABLED   linuxgpio
-- DISABLED   linuxspi
-- ----------------------
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/mcuee/build/avr/avrdude/build_macos3

mcuee@mcuees-Mac-mini avrdude % cmake --build build_macos3 
...

[ 34%] Building C object src/CMakeFiles/libavrdude.dir/jtagmkII.c.o
cd /Users/mcuee/build/avr/avrdude/build_macos3/src && /Library/Developer/CommandLineTools/usr/bin/cc -DCONFIG_DIR=\"/etc/opt/homebrew/Cellar/avrdude/7.0\" -I/Users/mcuee/build/avr/avrdude/build_macos3/src -I/Users/mcuee/build/avr/avrdude -I/Users/mcuee/build/avr/avrdude/build_macos3 -I/Users/mcuee/build/avr/avrdude/src -I/opt/homebrew/include -O2 -g -DNDEBUG -arch arm64 -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX12.3.sdk -Wall -std=gnu11 -MD -MT src/CMakeFiles/libavrdude.dir/jtagmkII.c.o -MF CMakeFiles/libavrdude.dir/jtagmkII.c.o.d -o CMakeFiles/libavrdude.dir/jtagmkII.c.o -c /Users/mcuee/build/avr/avrdude/src/jtagmkII.c
/Users/mcuee/build/avr/avrdude/src/jtagmkII.c:491:18: warning: variable 'checksum' set but not used [-Wunused-but-set-variable]
  unsigned short checksum = 0;
                 ^
1 warning generated.
...
```

yegorich, 2022-05-17:
@mcuee thanks. So, for now, you get only `-Wunused-but-set-variable` warning. No libftdi related warnings.

@dl8dtl can the `checksum` variable be removed? It was unsued since its introduction in 2005.

mcuee, 2022-05-17:
@yegorich I was talking about macOS. I will check out FreeBSD later.

mcuee, 2022-05-17:
FreeBSD build log with both libftdi (0.x API_ and libftdi1 (1.0 API) installed. No warnings in this case.

```
mcuee@freebsdx64vm:~/build/avrdude $ cmake -D CMAKE_C_FLAGS=-I/usr/local/include -D CMAKE_EXE_LINKER_FLAGS=-L/usr/local/lib -D CMAKE_BUILD_TYPE=RelWithDebInfo -D CMAKE_VERBOSE_MAKEFILE=ON -B build_freebsd
-- The C compiler identification is Clang 11.0.1
-- The CXX compiler identification is Clang 11.0.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Git: /usr/local/bin/git (found version "2.35.2") 
-- Found FLEX: /usr/bin/flex (found version "2.6.4") 
-- Found YACC: /usr/bin/yacc
-- Looking for libelf.h
-- Looking for libelf.h - found
-- Looking for libelf/libelf.h
-- Looking for libelf/libelf.h - not found
-- Looking for usb.h
-- Looking for usb.h - found
-- Looking for lusb0_usb.h
-- Looking for lusb0_usb.h - not found
-- Looking for libusb.h
-- Looking for libusb.h - found
-- Looking for libusb-1.0/libusb.h
-- Looking for libusb-1.0/libusb.h - not found
-- Looking for hidapi/hidapi.h
-- Looking for hidapi/hidapi.h - found
-- Looking for ftdi_tcioflush
-- Looking for ftdi_tcioflush - found
-- Configuration summary:
-- ----------------------
-- DO HAVE    libelf
-- DO HAVE    libusb
-- DO HAVE    libusb_1_0
-- DO HAVE    libhidapi
-- DO HAVE    libftdi (but prefer to use libftdi1)
-- DO HAVE    libftdi1
-- DISABLED   doc
-- DISABLED   parport
-- DISABLED   linuxgpio
-- DISABLED   linuxspi
-- ----------------------
-- Configuring done
-- Generating done
-- Build files have been written to: /home/mcuee/build/avrdude/build_freebsd
```

```

mcuee, 2022-05-17:
FreeBSD build log without libftdi and only with libftdi1.
I do not see warning as well this time. It is quite strange. I got the previous warning messages from the build log here.
https://github.com/avrdudes/avrdude/pull/925#issuecomment-1126922982

```
mcuee@freebsdx64vm:~/build/avrdude $ cmake -D CMAKE_C_FLAGS=-I/usr/local/include -D CMAKE_EXE_LINKER_FLAGS=-L/usr/local/lib -D CMAKE_BUILD_TYPE=RelWithDebInfo -D CMAKE_VERBOSE_MAKEFILE=ON -B build_freebsd2
-- The C compiler identification is Clang 11.0.1
-- The CXX compiler identification is Clang 11.0.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Git: /usr/local/bin/git (found version "2.35.2") 
-- Found FLEX: /usr/bin/flex (found version "2.6.4") 
-- Found YACC: /usr/bin/yacc
-- Looking for libelf.h
-- Looking for libelf.h - found
-- Looking for libelf/libelf.h
-- Looking for libelf/libelf.h - not found
-- Looking for usb.h
-- Looking for usb.h - found
-- Looking for lusb0_usb.h
-- Looking for lusb0_usb.h - not found
-- Looking for libusb.h
-- Looking for libusb.h - found
-- Looking for libusb-1.0/libusb.h
-- Looking for libusb-1.0/libusb.h - not found
-- Looking for hidapi/hidapi.h
-- Looking for hidapi/hidapi.h - found
-- Looking for ftdi_tcioflush
-- Looking for ftdi_tcioflush - found
-- Configuration summary:
-- ----------------------
-- DO HAVE    libelf
-- DO HAVE    libusb
-- DO HAVE    libusb_1_0
-- DO HAVE    libhidapi
-- DON'T HAVE libftdi
-- DO HAVE    libftdi1
-- DISABLED   doc
-- DISABLED   parport
-- DISABLED   linuxgpio
-- DISABLED   linuxspi
-- ----------------------
-- Configuring done
-- Generating done
-- Build files have been written to: /home/mcuee/build/avrdude/build_freebsd2

```
```

mcuee, 2022-05-17:
> @mcuee thanks. So, for now, you get only -Wunused-but-set-variable warning. No libftdi related warnings.

@yegorich Yes looks like maybe the libftdi warning message is a false alarm on FreeBSD. So yes there is only the `-Wunused-but-set-variable` warning under macOS.

mcuee, 2022-06-10:
More compiler warnings with MSYS2 mingw64.

```
[ 58%] Building C object src/CMakeFiles/libavrdude.dir/ser_win32.c.obj
cd /d C:\work\avr\avrdude\build_mingw64\src && C:\msys64\mingw64\bin\gcc.exe -DCONFIG_DIR="\"C:/Program Files (x86)/avrdude/etc\"" @CMakeFiles/libavrdude.dir/includes_C.rsp -Wall -std=gnu11 -MD -MT src/CMakeFiles/libavrdude.dir/ser_win32.c.obj -MF CMakeFiles\libavrdude.dir\ser_win32.c.obj.d -o CMakeFiles\libavrdude.dir\ser_win32.c.obj -c C:\work\avr\avrdude\src\ser_win32.c
C:\work\avr\avrdude\src\ser_win32.c: In function 'net_send':
C:\work\avr\avrdude\src\ser_win32.c:396:36: warning: pointer targets in passing argument 2 of 'send' differ in signedness [-Wpointer-sign]
  396 |                 rc = send(fd->ifd, p, (len > 1024) ? 1024 : len, 0);
      |                                    ^
      |                                    |
      |                                    const unsigned char *
In file included from C:\work\avr\avrdude\src\ser_win32.c:30:
C:/msys64/mingw64/include/winsock2.h:1033:60: note: expected 'const char *' but argument is of type 'const unsigned char *'
 1033 |   WINSOCK_API_LINKAGE int WSAAPI send(SOCKET s,const char *buf,int len,int flags);
      |                                                ~~~~~~~~~~~~^~~
C:\work\avr\avrdude\src\ser_win32.c: In function 'net_recv':
C:\work\avr\avrdude\src\ser_win32.c:530:36: warning: pointer targets in passing argument 2 of 'recv' differ in signedness [-Wpointer-sign]
  530 |                 rc = recv(fd->ifd, p, (buflen - len > 1024) ? 1024 : buflen - len, 0);
      |                                    ^
      |                                    |
      |                                    unsigned char *
In file included from C:\work\avr\avrdude\src\ser_win32.c:30:
C:/msys64/mingw64/include/winsock2.h:1028:54: note: expected 'char *' but argument is of type 'unsigned char *'
 1028 |   WINSOCK_API_LINKAGE int WSAAPI recv(SOCKET s,char *buf,int len,int flags);
      |                                                ~~~~~~^~~
C:\work\avr\avrdude\src\ser_win32.c: In function 'net_drain':
C:\work\avr\avrdude\src\ser_win32.c:696:36: warning: pointer targets in passing argument 2 of 'recv' differ in signedness [-Wpointer-sign]
  696 |                 rc = recv(fd->ifd, &buf, 1, 0);
      |                                    ^~~~
      |                                    |
      |                                    unsigned char *
In file included from C:\work\avr\avrdude\src\ser_win32.c:30:
C:/msys64/mingw64/include/winsock2.h:1028:54: note: expected 'char *' but argument is of type 'unsigned char *'
 1028 |   WINSOCK_API_LINKAGE int WSAAPI recv(SOCKET s,char *buf,int len,int flags);
      |                                                ~~~~~~^~~

C:\work\avr\avrdude\build_mingw64>gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/11.2.0/lto-wrapper.exe
Target: x86_64-w64-mingw32
Configured with: ../gcc-11.2.0/configure --prefix=/mingw64 --with-local-prefix=/mingw64/local --build=x86_64-w64-mingw32 --host=x86_64-w64-mingw32 --target=x86_64-w64-mingw32 --with-native-system-header-dir=/mingw64/include --libexecdir=/mingw64/lib --enable-bootstrap --enable-checking=release --with-arch=x86-64 --with-tune=generic --enable-languages=c,lto,c++,fortran,ada,objc,obj-c++,jit --enable-shared --enable-static --enable-libatomic --enable-threads=posix --enable-graphite --enable-fully-dynamic-string --enable-libstdcxx-filesystem-ts --enable-libstdcxx-time --disable-libstdcxx-pch --disable-libstdcxx-debug --enable-lto --enable-libgomp --disable-multilib --disable-rpath --disable-win32-registry --disable-nls --disable-werror --disable-symvers --with-libiconv --with-system-zlib --with-gmp=/mingw64 --with-mpfr=/mingw64 --with-mpc=/mingw64 --with-isl=/mingw64 --with-pkgversion='Rev10, Built by MSYS2 project' --with-bugurl=https://github.com/msys2/MINGW-packages/issues --with-gnu-as --with-gnu-ld --with-boot-ldflags='-pipe -Wl,--disable-dynamicbase -static-libstdc++ -static-libgcc' LDFLAGS_FOR_TARGET=-pipe --enable-linker-plugin-flags='LDFLAGS=-static-libstdc++\ -static-libgcc\ -pipe\ -Wl,--stack,12582912'
Thread model: posix
Supported LTO compression algorithms: zlib zstd
gcc version 11.2.0 (Rev10, Built by MSYS2 project)
```

mcuee, 2022-06-19:
More compiler warnings with MSYS2 mingw gcc 11.3.0.

```
[ 61%] Building C object src/CMakeFiles/libavrdude.dir/ser_win32.c.obj
C:/work/avr/avrdude_test/avrdude_sr/src/ser_win32.c: In function 'net_send':
C:/work/avr/avrdude_test/avrdude_sr/src/ser_win32.c:396:36: warning: pointer targets in passing argument 2 of 'send' differ in signedness [-Wpointer-sign]
  396 |                 rc = send(fd->ifd, p, (len > 1024) ? 1024 : len, 0);
      |                                    ^
      |                                    |
      |                                    const unsigned char *
In file included from C:/work/avr/avrdude_test/avrdude_sr/src/ser_win32.c:30:
C:/msys64/mingw64/include/winsock2.h:1033:60: note: expected 'const char *' but argument is of type 'const unsigned char *'
 1033 |   WINSOCK_API_LINKAGE int WSAAPI send(SOCKET s,const char *buf,int len,int flags);
      |                                                ~~~~~~~~~~~~^~~
C:/work/avr/avrdude_test/avrdude_sr/src/ser_win32.c: In function 'net_recv':
C:/work/avr/avrdude_test/avrdude_sr/src/ser_win32.c:530:36: warning: pointer targets in passing argument 2 of 'recv' differ in signedness [-Wpointer-sign]
  530 |                 rc = recv(fd->ifd, p, (buflen - len > 1024) ? 1024 : buflen - len, 0);
      |                                    ^
      |                                    |
      |                                    unsigned char *
In file included from C:/work/avr/avrdude_test/avrdude_sr/src/ser_win32.c:30:
C:/msys64/mingw64/include/winsock2.h:1028:54: note: expected 'char *' but argument is of type 'unsigned char *'
 1028 |   WINSOCK_API_LINKAGE int WSAAPI recv(SOCKET s,char *buf,int len,int flags);
      |                                                ~~~~~~^~~
C:/work/avr/avrdude_test/avrdude_sr/src/ser_win32.c: In function 'net_drain':
C:/work/avr/avrdude_test/avrdude_sr/src/ser_win32.c:696:36: warning: pointer targets in passing argument 2 of 'recv' differ in signedness [-Wpointer-sign]
  696 |                 rc = recv(fd->ifd, &buf, 1, 0);
      |                                    ^~~~
      |                                    |
      |                                    unsigned char *
In file included from C:/work/avr/avrdude_test/avrdude_sr/src/ser_win32.c:30:
C:/msys64/mingw64/include/winsock2.h:1028:54: note: expected 'char *' but argument is of type 'unsigned char *'
 1028 |   WINSOCK_API_LINKAGE int WSAAPI recv(SOCKET s,char *buf,int len,int flags);
      |                                                ~~~~~~^~~
[ 62%] Building C object src/CMakeFiles/libavrdude.dir/serialupdi.c.obj
[ 64%] Building C object src/CMakeFiles/libavrdude.dir/stk500.c.obj
[ 66%] Building C object src/CMakeFiles/libavrdude.dir/stk500v2.c.obj
C:/work/avr/avrdude_test/avrdude_sr/src/stk500v2.c: In function 'stk500v2_setparm':
C:/work/avr/avrdude_test/avrdude_sr/src/stk500v2.c:3098:6: warning: 'current_value' may be used uninitialized in this function [-Wmaybe-uninitialized]
 3098 |   if (value == current_value) {
      |      ^
C:/work/avr/avrdude_test/avrdude_sr/src/stk500v2.c: In function 'stk500v2_print_parms1':
C:/work/avr/avrdude_test/avrdude_sr/src/stk500v2.c:3295:24: warning: 'osc_cmatch' may be used uninitialized in this function [-Wmaybe-uninitialized]
 3295 |       f /= (osc_cmatch + 1);
      |            ~~~~~~~~~~~~^~~~
C:/work/avr/avrdude_test/avrdude_sr/src/stk500v2.c:3242:35: warning: 'osc_pscale' may be used uninitialized in this function [-Wmaybe-uninitialized]
 3242 |   unsigned char vtarget, vadjust, osc_pscale, osc_cmatch, sck_duration =0; //XXX 0 is not correct, check caller
      |                                   ^~~~~~~~~~
C:/work/avr/avrdude_test/avrdude_sr/src/stk500v2.c:3278:5: warning: 'vadjust' may be used uninitialized in this function [-Wmaybe-uninitialized]
 3278 |     avrdude_message(MSG_INFO, "%sVaref           : %.1f V\n", p, vadjust / 10.0);
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
C:/work/avr/avrdude_test/avrdude_sr/src/stk500v2.c:3267:5: warning: 'vtarget' may be used uninitialized in this function [-Wmaybe-uninitialized]
 3267 |     avrdude_message(MSG_INFO, "%sVtarget         : %.1f V\n", p, vtarget / 10.0);
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[ 67%] Building C object src/CMakeFiles/libavrdude.dir/stk500generic.c.obj

$ gcc -v
Using built-in specs.
COLLECT_GCC=C:\msys64\mingw64\bin\gcc.exe
COLLECT_LTO_WRAPPER=C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/11.3.0/lto-wrapper.exe
Target: x86_64-w64-mingw32
Configured with: ../gcc-11.3.0/configure --prefix=/mingw64 --with-local-prefix=/mingw64/local --build=x86_64-w64-mingw32 --host=x86_64-w64-mingw32 --target=x86_64-w64-mingw32 --with-native-system-header-dir=/mingw64/include --libexecdir=/mingw64/lib --enable-bootstrap --enable-checking=release --with-arch=x86-64 --with-tune=generic --enable-languages=c,lto,c++,fortran,ada,objc,obj-c++,jit --enable-shared --enable-static --enable-libatomic --enable-threads=posix --enable-graphite --enable-fully-dynamic-string --enable-libstdcxx-filesystem-ts --enable-libstdcxx-time --disable-libstdcxx-pch --disable-libstdcxx-debug --enable-lto --enable-libgomp --disable-multilib --disable-rpath --disable-win32-registry --disable-nls --disable-werror --disable-symvers --with-libiconv --with-system-zlib --with-gmp=/mingw64 --with-mpfr=/mingw64 --with-mpc=/mingw64 --with-isl=/mingw64 --with-pkgversion='Rev1, Built by MSYS2 project' --with-bugurl=https://github.com/msys2/MINGW-packages/issues --with-gnu-as --with-gnu-ld
Thread model: posix
Supported LTO compression algorithms: zlib zstd
gcc version 11.3.0 (Rev1, Built by MSYS2 project)

```

mcuee, 2022-08-12:
This has been addressed in PR #1059 and can be closed once #1059 is merged.
* #1059

stefanrueger, 2022-08-14:
Fixed by PR #1059 (as far as poss)

closed 2022-08-14 by stefanrueger