avrdudes/avrdude #965: avrdude seems to no longer be able to read fuse bytes

K4LCIFER, 2022-05-16:
When I call avrdude to have it read from a chip, I get the following response:
```
% sudo avrdude -c atmelice_isp -p m328p

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e950f (probably m328p)

avrdude done.  Thank you.
```
Despite specifying `-p m328p`, it doesn't seem to see that that is what I have since it just returns the device signiture, and it also does not return the fuse bytes (although perhaps this was a change that was made since v7.0, since this issue all of a sudden appeared after I updated to that version.)

dl8dtl, 2022-05-16:
From thge NEWS file in AVRDUDE 7.0:

```
    - The "safemode" feature has been removed. The major class of
      programmers it has been designed for (lowlevel bitbang
      programmers on parallel or serial ports) virtually doesn't exist
      anymore, and the fuse combination that was covered by it do not
      match the fuses of modern AVR devices anyway.
```
Reading and displaying these fuse values has been part of "safemode".

Basically, you specified a command line that is not supposed to do anything beyond attaching to the ISP, and leaving it again. AVRDUDE does just this.

On the pro side: there is no error message, so the command tells you that your ISP connection is working fine.

If you want to do more, try "terminal mode" (`-t`). You can browse around interactively through your device, including the fuse values. Non-interactively, it would need `-U` optoins, like:

`avrdude … -U lfuse:r:-:h -U hfuse:r:-:h -U efuse:r:-:h`

The fuse names may be abbreviated to any unique string, so "lfuse" to "lf" etc.

mcuee, 2022-05-29:
Since this is deemed invalid, I think this can be closed.

closed 2022-05-29 by MCUdude