avrdudes/avrdude #966: avrdude  option -c micronucleus does not work

gfernval, 2022-05-16:
Doing "avrdude -p t85 -c micronucleus -x wait -V -U flash:w:blink.hex:i " gives:
...
avrdude: No device found, waiting for device to be plugged in...
avrdude: Press CTRL-C to terminate

If I then connect a small board with USB and an ATTINY85 chip avrdude does not progress, shows the above waiting message,
but in Control Panel the board connected appears as a: libusb-win32 devices -> Digispark Bootloader with VID 16D0 and
PID 0753. Micronucleus.exe utility works (doing "micronucleus --run blink.hex" and then plugging the small usb board programs ATTINY85 with blink.hex) All the tests I have done in Windows XP

mariusgreuel, 2022-05-16:
> All the tests I have done in Windows XP

avrdude v7.0 does not run on Windows XP, so you appear not to be running the v7.0 release from Github, but some build that does not support the libusb0.sys driver.

gfernval, 2022-05-16:
The tests I have done have been compiling the latest source code of avrdude 7.0, the resultant binary works well in Windows XP with an usbasp programmer and serial programmer, the problem is with micronucleus

gfernval, 2022-05-16:
There is something wrong in micronucleus code, because compiling latest micronucleus.exe source code from avrdude 7.0 works well as explained in my first post

mariusgreuel, 2022-05-17:
> The tests I have done have been compiling the latest source code of avrdude 7.0

That is what I am saying. You are not using our executable but built your own. I knew that because our build uses libwinusb, which does not run on XP. Hence the problem.

Which libusb do you use for your build? Which driver do you use for usbasp?

gfernval, 2022-05-17:
I have used msys2 for windows XP and the following libraries:

mingw-w64-i686-hidapi-0.10.0-1any
mingw-w64-i686-libelf-0.8.13-4-any
mingw-w64-i686-libftdi-1.5-1-any
mingw-w64-i686-libusb-1.0.23-1-any
mingw-w64-i686-libusb-compat-git-r76.b5db9d0-1-any
mingw-w64-i686-libusb-win32-1.2.6.0-1-any

gfernval, 2022-05-17:
The usbasp driver I use is version 3.0.7.0 (provider libusbK)

mariusgreuel, 2022-05-17:
> mingw-w64-i686-libusb-1.0.23-1-any

This is your problem. The driver libusb0.sys, which Digistump uses, does not work with libusb-1.0.

> The usbasp driver I use is version 3.0.7.0 (provider libusbK)

libusbK.sys does work with libusb-1.0. If you use this driver (or WinUSB.sys) for the Micronucleus bootloader it should work.

Also, see <https://github.com/avrdudes/avrdude/wiki/Troubleshooting-USB-devices-in-Windows>

gfernval, 2022-05-17:
Don´t know well the different drivers, what have to do? Use a more recent library instead than mingw-w64-i686-libusb-1.0.23-1? do not compile with such library the avrdude.exe in windows xp? or install a more recent usbasp driver than version 3.0.7.0?

mariusgreuel, 2022-05-17:
> Don´t know well the different drivers, what have to do?

Dump Windows XP and use our build.

You can't have the cake and eat it too. For a compatibility matrix, see
<https://github.com/avrdudes/avrdude/wiki/The-story-on-libusb-for-Windows>

mcuee, 2022-05-18:
@gfernval 
I agree with Marius that your best option is to dump XP and use a newer OS and the official avrdude Windows binary.

If you still want to try under Windows XP, take note of the following.
1) libusb-1.0 last version to support Windows XP is 1.0.23 (released on 2019-08-29). Windows XP support is dropped in 1.0.24.
https://github.com/libusb/libusb/commit/96e0af79b383fa805dd4bff1e4f8bebe3c6fdb46

2) libusb-1.0's support of libusb0.sys was not good prior to 1.0.25 so you should not really use libusb0.sys. You can however use either WinUSB or libusbk.sys.

3) Zadig also no longer supports Windows XP (as well as Vista). The last version to support Windows XP is the 2.2 version released back in 2016-02-04. You can use it to switch the driver to WinUSB or libusbk to see if that helps your situation.
https://github.com/pbatard/libwdi/releases/download/v1.2.5/zadig_xp-2.2.exe

You can use the libusbk GUI inf-wizard as well under XP to switch between different drivers. libusbk 3.0.7.0 should work under Windows XP.

4) libusb-win32 1.2.6.0 works under Windows XP. You may try to build your avrdude binary without libusb-1.0.23 and libftdi-1.5 support to see if that helps.
mingw-w64-i686-hidapi-0.10.0-1any
mingw-w64-i686-libelf-0.8.13-4-any
mingw-w64-i686-libusb-win32-1.2.6.0-1-any

mcuee, 2022-05-25:
BTW, avrdude official Windows binary should work with Windows 7/8/8.1/10/11.

```
PS C:\work\avr\avrdude_test\avrdude-v7.0-windows-x64> .\avrdude.exe -p t88 -c micronucleus -vv

avrdude.exe: Version 7.0
             Copyright (c) Brian Dean, http://www.bdmicro.com/
             Copyright (c) Joerg Wunsch

             System wide configuration file is "C:/work/avr/avrdude_test/avrdude-v7.0-windows-x64/avrdude.conf"

             Using Port                    : usb
             Using Programmer              : micronucleus
avrdude.exe: Found device with Micronucleus V2.2, bus:device: bus-0:\\?\usb#vid_16d0&pid_0753#5&586b51a&0&1#{a5dcbf10-6530-11d2-901f-00c04fb951ed}--libusb0
             AVR Part                      : ATtiny88
             Chip Erase delay              : 9000 us
             PAGEL                         : PD7
             BS2                           : PC2
             RESET disposition             : dedicated
             RETRY pulse                   : SCK
             Serial program mode           : yes
             Parallel program mode         : yes
             Timeout                       : 200
             StabDelay                     : 100
             CmdexeDelay                   : 25
             SyncLoops                     : 32
             PollIndex                     : 3
             PollValue                     : 0x53
             Memory Detail                 :

                                               Block Poll               Page                       Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               eeprom                 65    20     4    0 no         64    4      0  3600  3600 0xff 0xff
               flash                  65     6    64    0 yes      8192   64    128  4500  4500 0xff 0xff
               lfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
               hfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
               efuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
               lock                    0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
               calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00
               signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00

             Programmer Type : Micronucleus V2.0
             Description     : Micronucleus Bootloader

avrdude.exe: Bootloader version: 2.2
avrdude.exe: Available flash size: 6650
avrdude.exe: Page size: 32
avrdude.exe: Bootloader start: 0x1A00
avrdude.exe: Write sleep: 7ms
avrdude.exe: Erase sleep: 1456ms
avrdude.exe: Signature1: 0x93
avrdude.exe: Signature2: 0x11
avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude.exe: Device signature = 0x1e9311 (probably t88)

avrdude.exe done.  Thank you.

````

mcuee, 2022-05-26:
Since this issue is only for Windows XP and the official binary does not support Windows XP, probably this issue can be closed.

closed 2022-05-26 by gfernval