avrdudes/avrdude #969: A few niggling issues in `stk500.c`

stefanrueger, 2022-05-17:
The last error message in `stk500_loadaddr()` says

 https://github.com/avrdudes/avrdude/blob/4601bee4afa76fb4aa8392ffdd4a4dd56ede1bae/src/stk500.c#L785-L791 

It should say **stk500_loadaddr()** expected **Resp_STK_OK**, not that loadaddr() expected Resp_STK_INSYNC.

Similarly wrong error messages (again copy/paste relicts) exist in `stk500_paged_write()` and in `stk500_getparm()`.

---------

In `stk500_getsync()` the return value of `stk500_recv(pgm, resp, 1)` is not checked; this means that `resp[0]` might not have been set, so `stk500_getsync()` might act as if in sync when it's not:

 https://github.com/avrdudes/avrdude/blob/4601bee4afa76fb4aa8392ffdd4a4dd56ede1bae/src/stk500.c#L123-L126

---------

Issue #942 mentions other occasions in `stk500.c` when variables are not (fully) initialised before use.

---------

Not doing a PR yet as there is a bit of a queue already (including PR #943 that would solve Issue #942) and I might see more small issues wanting to be mended.

mcuee, 2022-06-07:
I think both findings (error message and `stk500_getsync()`) are valid.

MCUdude, 2022-07-17:
@stefanrueger now that PR #942 has been merged, this issue might be ready to be worked on.

Apart from the issues mentioned above, the most serious one appears to be the one mentioned in #942

> (@stefanrueger) Rationale: There have been quite a few errors in .conf by users making mistakes in the SPI command descriptions. This intervention makes x bits behave like 0 and removes a potential pitfall. In case the part cares about the value of x bits (and, hence, the command description in .conf was wrong) the suggested change makes avrdude behave predictably and therefore eases debugging. Some of the 48 calls of avr_set_bits() set bits on uninitialised memory leaving x bits in a truly unknown state (mostly in stk500.c and stk500v2.c), thereby giving utility to clear the x bits of SPI commands in avr_set_bits().

mcuee, 2022-07-31:
I believe #1046 will be merged soon to fix this issue.

stefanrueger, 2022-08-04:
Closed by PR #1046

closed 2022-08-04 by stefanrueger