avrdudes/avrdude #971: Arduino as ISP problem under Windows (no problem under Linux)

mcuee, 2022-05-21:
I accidentally erased the bootloader of my clone Arduino Uno (with CH340). And my AVRdragon is not with me right now.

So I am trying to use my other original Arduino Leonardo as avrisp programmer to program the Uno. Somehow it does not work under Windows 11 but works fine under Linux (Ubuntu 20.04) on the same laptop (dual boot).

```
mcuee@UbuntuSwift3:~/build/avr$ avrdude -c avrisp -p m328p -P /dev/ttyACM0 -b 19200 -U blinky_uno.ino.with_bootloader.standard.hex 

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e950f (probably m328p)
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file "blinky_uno.ino.with_bootloader.standard.hex"
avrdude: input file blinky_uno.ino.with_bootloader.standard.hex auto detected as Intel Hex
avrdude: writing flash (32768 bytes):

Writing | ################################################## | 100% 0.54s

avrdude: 32768 bytes of flash written
avrdude: verifying flash memory against blinky_uno.ino.with_bootloader.standard.hex:
avrdude: input file blinky_uno.ino.with_bootloader.standard.hex auto detected as Intel Hex

Reading | ################################################## | 100% 0.30s

avrdude: 32768 bytes of flash verified

avrdude done.  Thank you.

```

mcuee, 2022-05-21:
More verbose log under Windows.
```
PS C:\work\avr\avrdude_test\avrdude-v7.0-windows-x64> .\avrdude.exe -p m328p -c avrisp -P COM5 -b 19200 -vvvv
executable_abspath = C:/work/avr/avrdude_test/avrdude-v7.0-windows-x64/avrdude.exe
executable_abspath_len = 61
executable_dirpath = C:/work/avr/avrdude_test/avrdude-v7.0-windows-x64
executable_dirpath_len = 49
sys_config = C:/work/avr/avrdude_test/avrdude-v7.0-windows-x64/avrdude.conf
sys_config_found = true

avrdude.exe: Version 7.0
             Copyright (c) Brian Dean, http://www.bdmicro.com/
             Copyright (c) Joerg Wunsch

             System wide configuration file is "C:/work/avr/avrdude_test/avrdude-v7.0-windows-x64/avrdude.conf"

             Using Port                    : COM5
             Using Programmer              : avrisp
             Overriding Baud Rate          : 19200
avrdude.exe: Send: 0 [30]   [20]
avrdude.exe: Send: 0 [30]   [20]
avrdude.exe: Send: 0 [30]   [20]
avrdude.exe: ser_recv(): programmer is not responding
avrdude.exe: stk500_recv(): programmer is not responding
avrdude.exe: stk500_getsync() attempt 1 of 10: not in sync: resp=0xf0
avrdude.exe: Send: 0 [30]   [20]
avrdude.exe: ser_recv(): programmer is not responding
avrdude.exe: stk500_recv(): programmer is not responding
avrdude.exe: stk500_getsync() attempt 2 of 10: not in sync: resp=0xf0
avrdude.exe: Send: 0 [30]   [20]
avrdude.exe: ser_recv(): programmer is not responding
avrdude.exe: stk500_recv(): programmer is not responding
avrdude.exe: stk500_getsync() attempt 3 of 10: not in sync: resp=0xf0
avrdude.exe: Send: 0 [30]   [20]
avrdude.exe: ser_recv(): programmer is not responding
avrdude.exe: stk500_recv(): programmer is not responding
avrdude.exe: stk500_getsync() attempt 4 of 10: not in sync: resp=0xf0
avrdude.exe: Send: 0 [30]   [20]
avrdude.exe: ser_recv(): programmer is not responding
avrdude.exe: stk500_recv(): programmer is not responding
avrdude.exe: stk500_getsync() attempt 5 of 10: not in sync: resp=0xf0
avrdude.exe: Send: 0 [30]   [20]
avrdude.exe: ser_recv(): programmer is not responding
avrdude.exe: stk500_recv(): programmer is not responding
avrdude.exe: stk500_getsync() attempt 6 of 10: not in sync: resp=0xf0
avrdude.exe: Send: 0 [30]   [20]
avrdude.exe: ser_recv(): programmer is not responding
avrdude.exe: stk500_recv(): programmer is not responding
avrdude.exe: stk500_getsync() attempt 7 of 10: not in sync: resp=0xf0
avrdude.exe: Send: 0 [30]   [20]
avrdude.exe: ser_recv(): programmer is not responding
avrdude.exe: stk500_recv(): programmer is not responding
avrdude.exe: stk500_getsync() attempt 8 of 10: not in sync: resp=0xf0
avrdude.exe: Send: 0 [30]   [20]
avrdude.exe: ser_recv(): programmer is not responding
avrdude.exe: stk500_recv(): programmer is not responding
avrdude.exe: stk500_getsync() attempt 9 of 10: not in sync: resp=0xf0
avrdude.exe: Send: 0 [30]   [20]
avrdude.exe: ser_recv(): programmer is not responding
avrdude.exe: stk500_recv(): programmer is not responding
avrdude.exe: stk500_getsync() attempt 10 of 10: not in sync: resp=0xf0
avrdude.exe: opening programmer "avrisp" on port "COM5" failed

avrdude.exe done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude-v7.0-windows-x64> .\avrdude.exe -p m328p -c avrisp -P COM5 -b 19200 -vv

avrdude.exe: Version 7.0
             Copyright (c) Brian Dean, http://www.bdmicro.com/
             Copyright (c) Joerg Wunsch

             System wide configuration file is "C:/work/avr/avrdude_test/avrdude-v7.0-windows-x64/avrdude.conf"

             Using Port                    : COM5
             Using Programmer              : avrisp
             Overriding Baud Rate          : 19200
avrdude.exe: ser_recv(): programmer is not responding
avrdude.exe: stk500_recv(): programmer is not responding
avrdude.exe: stk500_getsync() attempt 1 of 10: not in sync: resp=0xd0
avrdude.exe: ser_recv(): programmer is not responding
avrdude.exe: stk500_recv(): programmer is not responding
avrdude.exe: stk500_getsync() attempt 2 of 10: not in sync: resp=0xd0
avrdude.exe: ser_recv(): programmer is not responding
avrdude.exe: stk500_recv(): programmer is not responding
avrdude.exe: stk500_getsync() attempt 3 of 10: not in sync: resp=0xd0
avrdude.exe: ser_recv(): programmer is not responding
avrdude.exe: stk500_recv(): programmer is not responding
avrdude.exe: stk500_getsync() attempt 4 of 10: not in sync: resp=0xd0
avrdude.exe: ser_recv(): programmer is not responding
avrdude.exe: stk500_recv(): programmer is not responding
avrdude.exe: stk500_getsync() attempt 5 of 10: not in sync: resp=0xd0
avrdude.exe: ser_recv(): programmer is not responding
avrdude.exe: stk500_recv(): programmer is not responding
avrdude.exe: stk500_getsync() attempt 6 of 10: not in sync: resp=0xd0
avrdude.exe: ser_recv(): programmer is not responding
avrdude.exe: stk500_recv(): programmer is not responding
avrdude.exe: stk500_getsync() attempt 7 of 10: not in sync: resp=0xd0
avrdude.exe: ser_recv(): programmer is not responding
avrdude.exe: stk500_recv(): programmer is not responding
avrdude.exe: stk500_getsync() attempt 8 of 10: not in sync: resp=0xd0
avrdude.exe: ser_recv(): programmer is not responding
avrdude.exe: stk500_recv(): programmer is not responding
avrdude.exe: stk500_getsync() attempt 9 of 10: not in sync: resp=0xd0
avrdude.exe: ser_recv(): programmer is not responding
avrdude.exe: stk500_recv(): programmer is not responding
avrdude.exe: stk500_getsync() attempt 10 of 10: not in sync: resp=0xd0
avrdude.exe: opening programmer "avrisp" on port "COM5" failed

avrdude.exe done.  Thank you.

```

per1234, 2022-05-21:
Hi @mcuee. I think the information here might provide the solution: https://github.com/arduino/ArduinoCore-avr/pull/56

mcuee, 2022-05-21:
> Hi @mcuee. I think the information here might provide the solution: [arduino/ArduinoCore-avr#56](https://github.com/arduino/ArduinoCore-avr/pull/56)

I actually tried to use Arduino and choose "Arduino as ISP (ATmega32U4)" as the programmer. But it failed no matter I want to upload the blinky sketch or try to burn the bootloader of the Uno's m328p.

Error message is the same as the above.

mcuee, 2022-05-21:
@per1234 But it seems to work if I change the programmer type to "arduino".

```
PS C:\work\avr\avrdude_test\avrdude-v7.0-windows-x64> .\avrdude.exe -p m328p -c arduino -P COM5 -vv

avrdude.exe: Version 7.0
             Copyright (c) Brian Dean, http://www.bdmicro.com/
             Copyright (c) Joerg Wunsch

             System wide configuration file is "C:/work/avr/avrdude_test/avrdude-v7.0-windows-x64/avrdude.conf"

             Using Port                    : COM5
             Using Programmer              : arduino
             AVR Part                      : ATmega328P
             Chip Erase delay              : 9000 us
             PAGEL                         : PD7
             BS2                           : PC2
             RESET disposition             : dedicated
             RETRY pulse                   : SCK
             Serial program mode           : yes
             Parallel program mode         : yes
             Timeout                       : 200
             StabDelay                     : 100
             CmdexeDelay                   : 25
             SyncLoops                     : 32
             PollIndex                     : 3
             PollValue                     : 0x53
             Memory Detail                 :

                                               Block Poll               Page                       Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               eeprom                 65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
               flash                  65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
               lfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
               hfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
               efuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
               lock                    0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
               calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00
               signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00

             Programmer Type : Arduino
             Description     : Arduino
             Hardware Version: 2
             Firmware Version: 1.18
             Topcard         : Unknown

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude.exe: Device signature = 0x1e950f (probably m328p)

avrdude.exe done.  Thank you.
```

mcuee, 2022-05-21:
I am not so sure if this is still revelant or not. But I was trying different older versions of Windows binary and they all seem to have the same error message.
1) http://download.savannah.gnu.org/releases/avrdude/
2) https://github.com/dioannidis/usbasp/tree/master/bin/avrdude-winusb

The interesting output is from here
https://github.com/mariusgreuel/avrdude/releases/tag/v6.3.1.1-windows

```
PS C:\work\avr\avrdude_test\avrdude-v6.3.1.1-windows> .\avrdude.exe -p m328p -c avrisp -P COM5 -vvvv

avrdude.exe: Version 6.3.1.1-windows
             Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
             Copyright (c) 2007-2014 Joerg Wunsch

             System wide configuration file is "C:\work\avr\avrdude_test\avrdude-v6.3.1.1-windows\avrdude.conf"

             Using Port                    : COM5
             Using Programmer              : avrisp
avrdude.exe: Found device at port 'COM5'
avrdude.exe: - InstanceId: USB\VID_2341&PID_8036&MI_00\7&1553EAE4&0&0000
avrdude.exe: - Port: COM5
avrdude.exe: - Location: Port_#0002.Hub_#0004
avrdude.exe: - ProductName: Arduino Leonardo
avrdude.exe: - IsPresent: true
avrdude.exe: - IsCompositeDevice: true
avrdude.exe: Entering bootloader mode...
(hang)
```

@mariusgreuel I guess this is not relevant now. Please confirm.

mcuee, 2022-05-21:
>  But it seems to work if I change the programmer type to "arduino".

So the question is whether "arduino" is the right programmer type or "avrisp" for "Arduino as an ISP (ATmega32U4)".

mcuee, 2022-05-21:
I changed the title, as there is no issue under Windows, if I use the Uno as the ISP now that I have recovered it.

Windows run log of using Arduino as ISP to program the ATmega32U4 on the Arduino Leonardo.

```
PS C:\work\avr\avrdude_test\avrdude-v7.0-windows-x64> .\avrdude.exe -p m32u4 -c avrisp -P COM12 -b 19200 -vv

avrdude.exe: Version 7.0
             Copyright (c) Brian Dean, http://www.bdmicro.com/
             Copyright (c) Joerg Wunsch

             System wide configuration file is "C:/work/avr/avrdude_test/avrdude-v7.0-windows-x64/avrdude.conf"

             Using Port                    : COM12
             Using Programmer              : avrisp
             Overriding Baud Rate          : 19200
             AVR Part                      : ATmega32U4
             Chip Erase delay              : 9000 us
             PAGEL                         : PD7
             BS2                           : PA0
             RESET disposition             : dedicated
             RETRY pulse                   : SCK
             Serial program mode           : yes
             Parallel program mode         : yes
             Timeout                       : 200
             StabDelay                     : 100
             CmdexeDelay                   : 25
             SyncLoops                     : 32
             PollIndex                     : 3
             PollValue                     : 0x53
             Memory Detail                 :

                                               Block Poll               Page                       Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               eeprom                 65    20     4    0 no       1024    4      0  9000  9000 0x00 0x00
               flash                  65     6   128    0 yes     32768  128    256  4500  4500 0x00 0x00
               lfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               hfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               efuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               lock                    0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00
               signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00

             Programmer Type : STK500
             Description     : Atmel AVR ISP
             Hardware Version: 2
             Firmware Version: 1.18
             Topcard         : Unknown
             Vtarget         : 0.0 V
             Varef           : 0.0 V
             Oscillator      : Off
             SCK period      : 0.1 us

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.02s

avrdude.exe: Device signature = 0x1e9587 (probably m32u4)

avrdude.exe done.  Thank you.
```

mariusgreuel, 2022-05-21:
> I guess this is not relevant now. Please confirm.

Hm, not relevant to you, but to me. You broke my Leonardo auto-reset code!

The heuristic looks for an Arduino, and when it is a composite devices, I assumed it needs resetting because it is running in sketch mode. That is true for your device as well, but you do not want to reset it but continue ISP programming. I guess I will need a custom switch for sketch resetting. I though I could get around that with a clever heuristic and not changing too much code.

mcuee, 2022-05-21:
> Hm, not relevant to you, but to me. You broke my Leonardo auto-reset code!

@mariusgreuel 
Haha, not intentional, just a side effect of my testing of the Arduino Leonardo as ISP.

> The heuristic looks for an Arduino, and when it is a composite devices, I assumed it needs resetting because it is running in sketch mode. That is true for your device as well, but you do not want to reset it but continue ISP programming. I guess I will need a custom switch for sketch resetting. I thought I could get around that with a clever heuristic and not changing too much code.

Interesting. This is in your repo here (https://github.com/mariusgreuel/avrdude) and not in the main line avrdude, right?

Going back to the reported issue (Arduino Leonardo as AVRISP does not work under Windows with `-c avrisp`), do you think it is a real bug for avrdude? 

I do not understand why `-c arduino` is working in this case. Please help to explain as well. Thanks.

mcuee, 2022-06-11:
> So the question is whether "arduino" is the right programmer type or "avrisp" for "Arduino as an ISP (ATmega32U4)".

I think I will close this for now, as there is at least a work-around to use "arduino" as the programmer type.

closed 2022-06-11 by mcuee