avrdudes/avrdude #976: Wrong timings for ATmega328PB in avrdude.conf

stefanrueger, 2022-05-22:
While designing a PCB with the [ATmeg328PB](https://ww1.microchip.com/downloads/en/DeviceDoc/40001906C.pdf) I noticed that some timings are different from [ATmega328P](https://ww1.microchip.com/downloads/en/DeviceDoc/ATmega48A-PA-88A-PA-168A-PA-328-P-DS-DS40002061B.pdf)'s, eg, chip erase delay is 10.5 ms, not 9 ms. `avrdude.conf` does not reflect that. Also noticed that chip erase delay misses for the `ATmega406` (along possibly other timings).

Not sure these things are still relevant today for some programmers (as they could as well poll to establish the operation is finished).

In my experience, errors are more likely to happen in sections of a project that already have seen a few errors, so if these timings are relevant in `avrdude.conf` it's probably a good idea to review other parameters and parts, too.

----

[image]

----

[image]

----

mcuee, 2022-05-23:
I am a Hardware engineer by trade.
Take note the datasheet for the ATmega328PB says "Typical Wait Delay" and the datasheet for ATmega328P says "Minimum Wait Delay". So they are not exactly comparable.

MCUdude, 2022-05-23:
@stefanrueger feel free to create a PR for small issues like this. They can easily be merged right away.
I've seen cases where people are having trouble flashing their 324PB/328PB, where flash verification fails for mysterious reasons.

stefanrueger, 2022-05-23:
> I've seen cases where people are having trouble flashing their 324PB/328PB, where flash verification fails for mysterious reasons.

Good to know. So, the timings in `.conf` might still be relevant for today's programmers (I had imagined they poll if possible). 

> feel free to create a PR

Thanks, but no: I have stopped suggesting PRs on a number of small-ish issues for fear of putting too much pressure on the current PR queue and, hence, diluting the limited resources the project has. This issue can serve as a useful reminder, though, for me or anyone else later on.

> small issues like this

Whilst the issue is small, the solution is not as easy as it may look: some of the timings are specific for memories, which `avrdude.conf`'s grammar file zaps (not extends), so they need to be copied in their entirety before making changes.

A good case study illustrating some pitfalls is the history of `ATmega64M1`'s entry in `avrdude.conf`. Recognising the wrong flash size triggered that change. Good. However, the change effected that the SPI programming ops and all other memory parameters were zapped. While a subsequent correction copied the memory descriptions over from the parent part, that in turn overlooked that the op address bits only catered for the parent device's smaller flash requiring a further intervention. Speaking of the **`ATmega64M1`, its entry might still be wrong:** it is inherited from `ATmega328`, so if even the `ATmegega328PB` differs in timings etc from it, how can we be confident that the `ATmega64M1` shares its parameters (other than size-related)? Answering this question for this part, or any other part for that matter, including the mentioned `ATmega324PB`, is well above my expertise (and pay grade :).

stefanrueger, 2022-05-23:
> the datasheet for the ATmega328PB says "Typical Wait Delay"

You are not wrong but the table caption probably *is*: I fully expect t<sub>WD_XYZ</sub> to mean "unless polling the *user must wait at least* that time after XYZ", and that is how AVRDUDE uses t<sub>WD_ERASE</sub>. (Search where WD_ is used in Microchip's MCU datasheets.)

mcuee, 2022-06-01:
I just got two official Microchip boards (AVR128DB48 Curiosity Nano) and ATmega328PB-xmini.

avrdude seems to work fine with ATmega328PB. But I will try the updated timing as mentioned by @stefanrueger 

```
C:\work\avr\binary\avrdude-v7.0-windows-x64> .\avrdude.exe -c xplainedmini -p m328pb -vv

avrdude.exe: Version 7.0
             Copyright (c) Brian Dean, http://www.bdmicro.com/
             Copyright (c) Joerg Wunsch

             System wide configuration file is "C:/work/avr/binary/avrdude-v7.0-windows-x64/avrdude.conf"

             Using Port                    : usb
             Using Programmer              : xplainedmini
avrdude.exe: stk500v2_jtag3_open()
avrdude.exe: Found CMSIS-DAP compliant device, using EDBG protocol
avrdude.exe: Sending sign-on command: 0x80 (3 bytes msg)
             AVR Part                      : ATmega328PB
             Chip Erase delay              : 9000 us
             PAGEL                         : PD7
             BS2                           : PC2
             RESET disposition             : dedicated
             RETRY pulse                   : SCK
             Serial program mode           : yes
             Parallel program mode         : yes
             Timeout                       : 200
             StabDelay                     : 100
             CmdexeDelay                   : 25
             SyncLoops                     : 32
             PollIndex                     : 3
             PollValue                     : 0x53
             Memory Detail                 :

                                               Block Poll               Page                       Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               eeprom                 65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
               flash                  65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
               lfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
               hfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
               lock                    0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
               calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00
               signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
               efuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00

             Programmer Type : JTAG3_ISP
             Description     : Atmel AVR XplainedMini in ISP mode
avrdude.exe: jtag3_getparm()
avrdude.exe: Sending get parameter (scope 0x01, section 1, parm 0) command: 0x84 (6 bytes msg)
             Vtarget         : 5.0 V
             SCK period      : 8.00 us

avrdude.exe: jtag3_setparm()
avrdude.exe: Sending set parameter (scope 0x12, section 0, parm 0) command: 0x80 (3 bytes msg)
avrdude.exe: jtag3_setparm()
avrdude.exe: Sending set parameter (scope 0x12, section 0, parm 1) command: 0x80 (3 bytes msg)
avrdude.exe: jtag3_setparm()
avrdude.exe: Sending set parameter (scope 0x12, section 1, parm 0) command: 0x80 (3 bytes msg)
avrdude.exe: AVR device initialized and ready to accept instructions

Reading |                                                    | 0% 0.00savrdude.exe: stk500isp_read_byte(.., signature, 0x0, ...)
avrdude.exe: stk500isp_read_byte(): Sending read memory command: avrdude.exe: stk500isp_read_byte(.., signature, 0x1, ...)
Reading | #################                                  | 33% 0.03savrdude.exe: stk500isp_read_byte(.., signature, 0x2, ...)
Reading | ################################################## | 100% 0.05s

avrdude.exe: Device signature = 0x1e9516 (probably m328pb)
avrdude.exe: stk500v2_jtag3_close()
avrdude.exe: jtag3_close()
avrdude.exe: Sending AVR sign-off command: 0x80 (3 bytes msg)
avrdude.exe: Sending sign-off command: 0x80 (3 bytes msg)

avrdude.exe done.  Thank you.
```

MCUdude, 2022-06-19:
@stefanrueger I had a look at the ATmega406 datasheet, and its ATDF file (attached below) that I've pulled from the most recent Microchip ATmega pack. To my surprise, I couldn't find any delays as I could in other ATDF files. Perhaps the avrdude.conf file is correct after all?

[ATmega406.atdf.zip](https://github.com/avrdudes/avrdude/files/8936064/ATmega406.atdf.zip)

dl8dtl, 2022-06-19:
> > the datasheet for the ATmega328PB says "Typical Wait Delay"
> 
> You are not wrong but the table caption probably _is_: I fully expect tWD_XYZ to mean "unless polling the _user must wait at least_ that time after XYZ", and that is how AVRDUDE uses tWD_ERASE. (Search where WD_ is used in Microchip's MCU datasheets.)

Ǐ'm not so sure about yoir interpretation.

I read it as one table describing the delays to *expect* when using the polled algorithm, while the other one describes the minimal time you have to wait when not polling.

stefanrueger, 2022-06-20:
> couldn't find any delays [ATmega406] ... Perhaps the avrdude.conf file is correct after all?

You are not wrong, but do we want the programmers to sleep 0 µs for that chip where other chips require O(10 ms)? I would argue it makes sense to put some typical values in just to be on the safe side

> one table describing the delays to _expect_ 

If it was *expect* to wait rather than *min* wait then `avrdude.conf` should hold an *even higher* value to be on the safe side, as AVRDUDE interprets that value as min wait needed for the programming to work.

I still think that the table captions are wrong: The page before the table says trice *must wait at least*. Technically t<sub>WD_ERASE</sub> is not mentioned in the text once, so in programming parlance a variable that is set but not used :), but other Atmel MCU data sheets that I read have WD_xxx as *min* delay. (I had verified in a couple of them with a text search for WD_ in the pdf.). Another fun fact: *Both* tables have captions saying typical delay. Only one table has a *column heading* with *minimum delay*. So, shoddy writing on part of Atmel/Microchip.

[image]

mcuee, 2022-06-20:
I will label this as `bug` first, for the `unset wait delays` thingy and not necessarily for the ATmega328PB itself.

stefanrueger, 2022-06-25:
@MCUdude You are right re the ATmega406: according to the data sheet it has JTAG programming, but no SPI programming, and therefore no t<sub>WD_...</sub> delays; looks like neither `wd_chip_erase` nor any other `wd_...` delays are needed. 

I am guessing AVRDUDE does not need to know delays for JTAG programming as delegated to the (physical) programmer.

stefanrueger, 2022-07-04:
Fixed in PR #1001 (thanks @MCUdude!)

closed 2022-07-04 by stefanrueger

mcuee, 2022-07-17:
Just  tried the partdesc branch from @stefanrueger under Linux (Ubuntu 20.04) since it does not work for Windows.

```
mcuee@UbuntuSwift3:~/build/avr/avrdude_bin$ ./avrdude_partdesc -p\* 2>&1 | grep ^.wd_".* "0.000.ms
.wd_eeprom 0.000 ms ATtiny11
.wd_flash 0.000 ms ATtiny11
.wd_lock 0.000 ms ATtiny11
.wd_fuse 0.000 ms ATtiny11
.wd_fuse 0.000 ms AT90S1200
.wd_fuse 0.000 ms AT90S4414
.wd_fuse 0.000 ms AT90S2313
.wd_fuse 0.000 ms AT90S8515
.wd_chip_erase 0.000 ms ATtiny28
.wd_flash 0.000 ms ATtiny28
.wd_lock 0.000 ms ATtiny28
.wd_fuse 0.000 ms ATtiny28
.wd_chip_erase 0.000 ms ATmega406
.wd_eeprom 0.000 ms ATmega406
.wd_flash 0.000 ms ATmega406
.wd_hfuse 0.000 ms ATmega406
.wd_lfuse 0.000 ms ATmega406
.wd_lockbits 0.000 ms ATmega406

mcuee@UbuntuSwift3:~/build/avr/avrdude_bin$ ./avrdude_partdesc -p\* 2>&1 | grep ^.desc | awk '{ x=and(strtonum($14), 48); if(x == 16) print $2; }'
'ATtiny12'
'ATtiny15'
'AT90S1200'
'AT90S4414'
'AT90S2313'
'AT90S2333'
'AT90S2343'
'AT90S4433'
'AT90S4434'
'AT90S8515'
'AT90S8535'
'ATmega103'
'ATmega64'
'ATmega64A'
'ATmega128'
'ATmega128A'
'ATmega163'
'ATmega161'
'ATmega8'
'ATmega8A'
'ATmega8515'
'ATmega8535'
'ATtiny26'

mcuee@UbuntuSwift3:~/build/avr/avrdude_bin$ ./avrdude_partdesc -p\* 2>&1 | grep ^.desc | gawk '{ if(strtonum($12) == 1) print $2; }' | tr "'\n" '` ' 
`ATtiny11` `ATtiny12` `ATtiny15` `AT90S1200` `AT90S4414` `AT90S2313` `AT90S2333` `AT90S2343` `AT90S4433` `AT90S4434` `AT90S8515` `AT90S8535` `ATmega103` `ATmega163` `ATmega161` `ATmega8515` `ATmega8535` `ATtiny26` `AVR32DA28` `AVR32DA32` `AVR32DA48` `AVR64DA28` `AVR64DA32` `AVR64DA48` `AVR64DA64` `AVR128DA28` `AVR128DA32` `AVR128DA48` `AVR128DA64` `AVR32DB28` `AVR32DB32` `AVR32DB48` `AVR64DB28` `AVR64DB32` `AVR64DB48` `AVR64DB64` `AVR128DB28` `AVR128DB32` `AVR128DB48` `AVR128DB64` `AVR16DD14` `AVR16DD20` `AVR16DD28` `AVR16DD32` `AVR32DD14` `AVR32DD20` `AVR32DD28` `AVR32DD32` `AVR64DD14` `AVR64DD20` `AVR64DD28` `AVR64DD32` 

```

mcuee, 2022-07-17:
@stefanrueger and @MCUdude 

The following may still be wrong, right?
```
.wd_chip_erase 0.000 ms ATmega406
.wd_eeprom 0.000 ms ATmega406
.wd_flash 0.000 ms ATmega406
.wd_hfuse 0.000 ms ATmega406
.wd_lfuse 0.000 ms ATmega406
.wd_lockbits 0.000 ms ATmega406
```

MCUdude, 2022-07-17:
> The following may still be wrong, right?

Nope, it's correct. See @stefanrueger answer below. The ATmega406 doesn't have an ISP interface, only JTAG.

> @MCUdude You are right re the ATmega406: according to the datasheet it has JTAG programming, but no SPI programming, and therefore no tWD_... delays; looks like neither wd_chip_erase nor any other wd_... delays are needed.

I am guessing AVRDUDE does not need to know delays for JTAG programming as delegated to the (physical) programmer.

mcuee, 2022-07-17:
> Nope, it's correct. See @stefanrueger answer below. The ATmega406 doesn't have an ISP interface, only JTAG.
>
> I am guessing AVRDUDE does not need to know delays for JTAG programming as delegated to the (physical) programmer.

I see. Thanks for the explanation.