avrdudes/avrdude #978: Leveraging static code analysis tools

mcuee, 2022-05-29:
Ref: https://github.com/avrdudes/avrdude/issues/247#issuecomment-1140424421

Not so sure if the project wants to adopt static code analysis tools like Coverity. It is free for Open Source projects.

libusb project uses Coverity and it does catch some errors.
https://scan.coverity.com/projects/libusb-libusb

mcuee, 2022-06-06:
github seems to have the function as well as part of github action.
https://docs.github.com/en/code-security/code-scanning/automatically-scanning-your-code-for-vulnerabilities-and-errors/about-code-scanning

dl8dtl, 2022-06-15:
I'm not against that, but it certainly needs volunteers.

mcuee, 2022-06-21:
Done here.

<a href="https://scan.coverity.com/projects/avrdude">
  <img alt="Coverity Scan Build Status"
       src="https://scan.coverity.com/projects/25149/badge.svg"/>
</a>

mcuee, 2022-06-21:
@dl8dtl 
I have sent you the invitation as the owner.