avrdudes/avrdude #986: avrdude EEPROM read/write problem with optiboot and stk500v2 bootloader

mcuee, 2022-06-05:
There are known problems with EEPROM and some bootloader/parts combination.
* https://github.com/avrdudes/avrdude/issues/970
* https://github.com/avrdudes/avrdude/issues/967

There is also a known stk500v2 bootloader (wiring bootloader) for the Arduino Mega2560 with regard to EEPROM reading. But it is said that EEPROM writing should work as per @MCUdude.
* https://github.com/avrdudes/avrdude/issues/515
* https://github.com/arduino/ArduinoCore-avr/pull/24

However, I can not even get the basic EEPROM writing working with either Arduino Uno (optiboot, subset of stk500v1 protocol) and Arduino Mega2560 (subset of stk500v2 protocol).

So I am wondering if this is just my issue or is a more generic issue.

mcuee, 2022-06-05:
I am not so sure if this EEPROM write problem is specific to my Arduino Mega2560 Clone or a generic problem.
https://github.com/avrdudes/avrdude/issues/231#issuecomment-1146714563

////////////////////////////////////////////////////////////////////////////////////////////////
Erasing the chip (including EEPROM and the reflash the booloader, here is the result.
BTW, I use a slightly updated stk500v2 bootloader this time.
https://haarer.github.io/arduino,bootloader/2018/03/25/building-an-arduino-mega-2560-bootloader.html

Now the read and write seem to match. But using AVR Dragon shows that the write is not that correct.

```
PS C:\work\avr\avrdude_test\avrdude-v7.0-windows-x64> echo "read eeprom 0 0x40" | ./avrdude -c wiring -P COM6 -p atmega2560 -t

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude.exe: Device signature = 0x1e9801 (probably m2560)
avrdude> >>> read eeprom 0 0x40

Reading | ################################################## | 100% 0.08s

0000  01 02 03 04 05 06 07 08  09 0a 0b 0c ff ff ff ff  |........    ....|
0010  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0020  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0030  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|

avrdude>
avrdude.exe done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude-v7.0-windows-x64> echo "read eeprom 0 0x40" | ./avrdude -c dragon_isp -p atmega2560 -t

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.15s

avrdude.exe: Device signature = 0x1e9801 (probably m2560)
avrdude> >>> read eeprom 0 0x40

Reading | ################################################## | 100% 0.82s

0000  01 02 03 04 05 06 07 08  ff ff ff ff ff ff ff ff  |................|
0010  09 0a 0b 0c ff ff ff ff  ff ff ff ff ff ff ff ff  |    ............|
0020  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0030  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|

avrdude>
avrdude.exe done.  Thank you.

```

mcuee, 2022-06-05:
Here is for Arduino Uno. I tried the original bootloader and another optiboot bootloader from @MCUdude 

* https://github.com/avrdudes/avrdude/issues/967#issuecomment-1146600561
/////////////////////////////////

I tried the hex file here from @MCUdude which is supposed to work with EEPROM and it does not seem to work either. I have tried two Arduino Uno boards (one clone with ATmega16U2 and one clone with CH340).

https://github.com/MCUdude/MiniCore/blob/master/avr/bootloaders/optiboot_flash/bootloaders/atmega328p/16000000L/optiboot_flash_atmega328p_UART0_115200_16000000L_B5.hex

```
PS C:\work\avr\avrdude_test\avrdude-v7.0-windows-x64> avrdude -p atmega328p -c arduino -b 115200 -P COM9 -t

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude.exe: Device signature = 0x1e950f
avrdude> d ee
>>> d ee
avrdude.exe: stk500_cmd(): programmer is out of sync

PS C:\work\avr\avrdude_test\avrdude-v7.0-windows-x64> avrdude -p atmega328p -c arduino -b 115200 -P COM9 -U eeprom:w:entest.eep:i

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude.exe: Device signature = 0x1e950f
avrdude.exe: reading input file "entest.eep"
avrdude.exe: writing eeprom (512 bytes):

Writing |                                                    | 0% 0.00s
avrdude.exe: stk500_paged_write(): (a) protocol error, expect=0x14, resp=0x14
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
Writing | #                                                  | 1% 10.44s ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
...
 ***failed;
 ***failed;
 ***failed;
 ***failed;
Writing | ################################################## | 100% 45.61s

avrdude.exe: 512 bytes of eeprom written
avrdude.exe: verifying eeprom memory against entest.eep:
avrdude.exe: load data eeprom data from input file entest.eep:
avrdude.exe: input file entest.eep contains 512 bytes
avrdude.exe: reading on-chip eeprom data:

Reading | ################################################## | 100% 1.16s

avrdude.exe: verifying ...
avrdude.exe: verification error, first mismatch at byte 0x0000
             0x54 != 0xff
avrdude.exe: verification error; content mismatch

avrdude.exe: safemode: Fuses OK

avrdude.exe done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude-v7.0-windows-x64> avrdude -p atmega328p -c arduino -b 115200 -P COM6 -t

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude.exe: Device signature = 0x1e950f
avrdude> d ee
>>> d ee
0000  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
0010  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
0020  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
0030  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|

avrdude> w ee 0 1 2 3 4 5 6 7
>>> w ee 0 1 2 3 4 5 6 7
avrdude.exe: stk500_cmd(): programmer is out of sync

```

mcuee, 2022-06-05:
As per the command here, bootloader may not work well with terminal mode.
* https://github.com/avrdudes/avrdude/discussions/940#discussioncomment-2795383

But even if I do not use the terminal mode on the Uno, it still does not work either.

```

PS C:\work\avr\avrdude_test\avrdude-v7.0-windows-x64> avrdude -p atmega328p -c arduino -b 115200 -P COM9 -U eeprom:w:entest.eep:i

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude.exe: Device signature = 0x1e950f
avrdude.exe: reading input file "entest.eep"
avrdude.exe: writing eeprom (512 bytes):

Writing |                                                    | 0% 0.00s
avrdude.exe: stk500_paged_write(): (a) protocol error, expect=0x14, resp=0x14
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
Writing | #                                                  | 1% 10.44s ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
...
 ***failed;
 ***failed;
 ***failed;
 ***failed;
Writing | ################################################## | 100% 45.61s

avrdude.exe: 512 bytes of eeprom written
avrdude.exe: verifying eeprom memory against entest.eep:
avrdude.exe: load data eeprom data from input file entest.eep:
avrdude.exe: input file entest.eep contains 512 bytes
avrdude.exe: reading on-chip eeprom data:

Reading | ################################################## | 100% 1.16s

avrdude.exe: verifying ...
avrdude.exe: verification error, first mismatch at byte 0x0000
             0x54 != 0xff
avrdude.exe: verification error; content mismatch

avrdude.exe: safemode: Fuses OK

avrdude.exe done.  Thank you.

```

mcuee, 2022-06-05:
For the Uno, read and verify seem to work fine.
```
PS C:\work\avr\avrdude_test\avrdude-v7.0-windows-x64> .\avrdude -p atmega328p -c arduino -b 115200 -P COM23 -U eeprom:r:enread.eep:i -vv

avrdude.exe: Version 7.0
             Copyright (c) Brian Dean, http://www.bdmicro.com/
             Copyright (c) Joerg Wunsch

             System wide configuration file is "C:/work/avr/avrdude_test/avrdude-v7.0-windows-x64/avrdude.conf"

             Using Port                    : COM23
             Using Programmer              : arduino
             Overriding Baud Rate          : 115200
             AVR Part                      : ATmega328P
             Chip Erase delay              : 9000 us
             PAGEL                         : PD7
             BS2                           : PC2
             RESET disposition             : dedicated
             RETRY pulse                   : SCK
             Serial program mode           : yes
             Parallel program mode         : yes
             Timeout                       : 200
             StabDelay                     : 100
             CmdexeDelay                   : 25
             SyncLoops                     : 32
             PollIndex                     : 3
             PollValue                     : 0x53
             Memory Detail                 :

                                               Block Poll               Page                       Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               eeprom                 65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
               flash                  65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
               lfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
               hfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
               efuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
               lock                    0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
               calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00
               signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00

             Programmer Type : Arduino
             Description     : Arduino
             Hardware Version: 3
             Firmware Version: 8.0

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude.exe: Device signature = 0x1e950f (probably m328p)
avrdude.exe: reading eeprom memory:

Reading | ################################################## | 100% 2.10s

avrdude.exe: writing output file "enread.eep"

avrdude.exe done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude-v7.0-windows-x64> .\avrdude -p atmega328p -c arduino -b 115200 -P COM23 -U eeprom:v:enread.eep:i -vv

avrdude.exe: Version 7.0
             Copyright (c) Brian Dean, http://www.bdmicro.com/
             Copyright (c) Joerg Wunsch

             System wide configuration file is "C:/work/avr/avrdude_test/avrdude-v7.0-windows-x64/avrdude.conf"

             Using Port                    : COM23
             Using Programmer              : arduino
             Overriding Baud Rate          : 115200
[... 61 lines cut ...]
:200140003091040123E0230F2D3768F126E8230F0296A11DB11D209304018093050190938F
:200160000601A0930701B09308018091000190910101A0910201B09103010196A11DB11D20
:200180008093000190930101A0930201B0930301BF91AF919F918F913F912F910F900FBECD
:2001A0000F901F9018950196A11DB11DD4CF3FB7F8948091000190910101A0910201B09152
:2001C000030126B5A89B05C02F3F19F00196A11DB11D3FBFBA2FA92F982F8827BC01CD01DE
:2001E000620F711D811D911D42E0660F771F881F991F4A95D1F708958F929F92AF92BF9200
:20020000CF92DF92EF92FF926B017C010E94D7004B015C01C114D104E104F104E9F00E94F0
:2002200000020E94D700681979098A099B09683E73408105910570F321E0C21AD108E10892
:20024000F10888EE880E83E0981EA11CB11CC114D104E104F10429F7FF90EF90DF90CF9076
:20026000BF90AF909F908F900895789484B5836084BD85B5836085BDEEE6F0E080818160B7
:200280008083E1E8F0E082E08083808181608083E0E8F0E0808181608083E1EBF0E080815E
:2002A00084608083E0EBF0E080818160808387E880937A001092C1000895843081F028F49A
:2002C0008230E9F08330F9F00895883071F0893091F08530B9F4809180008F7D03C0809134
:2002E00080008F778093800008958091B0008F778093B00008958091B0008F7DF9CF84B553
:200300008F7784BD089584B58F7DFBCFCF93DF9390E0FC01E158FF4F24918A569F4FFC01A2
:2003200084918823C9F090E0880F991FFC01E954FF4FA591B491FC01E355FF4FC591D49144
:2003400061110DC09FB7F8948C91209582238C938881282328839FBFDF91CF91089562308A
:2003600051F49FB7F8943C91822F809583238C93E8812E2BEFCF8FB7F894EC912E2B2C9317
:200380008FBFEACF1F93CF93DF93282F30E0F901E859FF4F8491F901E158FF4FD491F901EB
:2003A000EA56FF4FC491CC23A9F0162F81110E945D01EC2FF0E0EE0FFF1FE355FF4FA59139
:2003C000B4918FB7F894EC91111108C0D095DE23DC938FBFDF91CF911F910895DE2BF8CF8F
:2003E00008950E9435010E94F0010E947500C0E0D0E00E9479002097E1F30E940000F9CF7E
:00000001FF

```

mcuee, 2022-06-05:
Okay, with the help here, it seems to work with the Uno using the bootloader from @westfw.
https://forum.arduino.cc/t/optiboot-w-eeprom-support-through-avrdude/966939/5

```
> .\avrdude -v -patmega328p -c dragon_isp -e -Ulock:w:0x3F:m -Uefuse:w:0xFD:m -Uhfuse:w:0xDC:m -Ulfuse:w
:0xFF:m -Uflash:w:bigboot_328.hex

avrdude.exe: Version 7.0
             Copyright (c) Brian Dean, http://www.bdmicro.com/
             Copyright (c) Joerg Wunsch

             System wide configuration file is "C:/work/avr/avrdude_test/avrdude-v7.0-windows-x64/avrdude.conf"

             Using Port                    : usb
             Using Programmer              : dragon_isp
avrdude.exe: usbdev_open(): Found AVRDRAGON, serno: 00A200001487
JTAG ICE mkII sign-on message:
Communications protocol version: 1
M_MCU:
  boot-loader FW version:        255
  firmware version:              7.39
  hardware version:              1
S_MCU:
  boot-loader FW version:        255
  firmware version:              7.39
  hardware version:              2
Serial number:                   00:a2:00:00:14:87
Device ID:                       AVRDRAGON
             AVR Part                      : ATmega328P
             Chip Erase delay              : 9000 us
             PAGEL                         : PD7
             BS2                           : PC2
             RESET disposition             : dedicated
             RETRY pulse                   : SCK
             Serial program mode           : yes
             Parallel program mode         : yes
             Timeout                       : 200
             StabDelay                     : 100
             CmdexeDelay                   : 25
             SyncLoops                     : 32
             PollIndex                     : 3
             PollValue                     : 0x53
             Memory Detail                 :

                                               Block Poll               Page                       Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               eeprom                 65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
               flash                  65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
               lfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
               hfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
               efuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
               lock                    0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
               calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00
               signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00

             Programmer Type : DRAGON_ISP
             Description     : Atmel AVR Dragon in ISP mode
             Vtarget         : 4.8 V
             SCK period      : 1.00 us

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.15s

avrdude.exe: Device signature = 0x1e950f (probably m328p)
avrdude.exe: erasing chip
avrdude.exe: reading input file "0x3F"
avrdude.exe: writing lock (1 bytes):

[... 128 lines cut ...]
:20014000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFBF
:20016000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF9F
:20018000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF7F
:2001A000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF5F
:2001C000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF3F
:2001E000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF1F
:20020000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFE
:20022000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFDE
:20024000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFBE
:20026000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF9E
:20028000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF7E
:2002A000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF5E
:2002C000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF3E
:2002E000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF1E
:20030000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFD
:20032000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFDD
:20034000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFBD
:20036000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF9D
:20038000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF7D
:2003A000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF5D
:2003C000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF3D
:2003E000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF1D
:00000001FF

```

mcuee, 2022-06-05:
So it seems there is no issue with avrdude EEPROM function with the correct bootloader. So the next thing is to find a correct bootloader for the Arduini Mega2560 clone.

mcuee, 2022-06-05:
Going back to Arduino Mega2560. Even if I do not use terminal mode, it does not seem to work properly with write.

```
PS C:\work\avr\avrdude_test\avrdude-v7.0-windows-x64> .\avrdude -p m2560 -c wiring -b 115200 -P COM6 -U eeprom:w:entest.eep:i -vv

avrdude.exe: Version 7.0
             Copyright (c) Brian Dean, http://www.bdmicro.com/
             Copyright (c) Joerg Wunsch

             System wide configuration file is "C:/work/avr/avrdude_test/avrdude-v7.0-windows-x64/avrdude.conf"

             Using Port                    : COM6
             Using Programmer              : wiring
             Overriding Baud Rate          : 115200
avrdude.exe: wiring_open(): releasing DTR/RTS
avrdude.exe: wiring_open(): asserting DTR/RTS
             AVR Part                      : ATmega2560
             Chip Erase delay              : 9000 us
             PAGEL                         : PD7
             BS2                           : PA0
             RESET disposition             : dedicated
             RETRY pulse                   : SCK
             Serial program mode           : yes
             Parallel program mode         : yes
             Timeout                       : 200
             StabDelay                     : 100
             CmdexeDelay                   : 25
             SyncLoops                     : 32
             PollIndex                     : 3
             PollValue                     : 0x53
             Memory Detail                 :

                                               Block Poll               Page                       Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               eeprom                 65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
               flash                  65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
               lfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               hfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               efuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               lock                    0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00
               signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00

             Programmer Type : Wiring
             Description     : Wiring
             Programmer Model: AVRISP
             Hardware Version: 15
             Firmware Version Master : 2.10
             Vtarget         : 0.0 V
             SCK period      : 0.1 us

avrdude.exe: AVR device initialized and ready to accept instructions

Reading |                                                    | 0% 0.00savrdude.exe: stk500isp_read_byte(.., signature, 0x0, ...)
avrdude.exe: stk500isp_read_byte(): Sending read memory command: avrdude.exe: stk500isp_read_byte(.., signature, 0x1, ...)
Reading | #################                                  | 33% 0.01savrdude.exe: stk500isp_read_byte(.., signature, 0x2, ...)
Reading | ################################################## | 100% 0.01s

avrdude.exe: Device signature = 0x1e9801 (probably m2560)
avrdude.exe: reading input file "entest.eep"
avrdude.exe: writing eeprom (512 bytes):

Writing | ################################################## | 100% 0.54s

avrdude.exe: 512 bytes of eeprom written
avrdude.exe: verifying eeprom memory against entest.eep:
avrdude.exe: load data eeprom data from input file entest.eep:
avrdude.exe: input file entest.eep contains 512 bytes
avrdude.exe: reading on-chip eeprom data:

Reading | ################################################## | 100% 0.53s

avrdude.exe: verifying ...
avrdude.exe: verification error, first mismatch at byte 0x0000
             0xff != 0x54
avrdude.exe: verification error; content mismatch

avrdude.exe done.  Thank you.

```

mcuee, 2022-06-05:
It is said that the production hex file is buggy and I should use the other one.
https://github.com/arduino/ArduinoCore-avr/blob/master/bootloaders/stk500v2/stk500boot_v2_mega2560.hex

I still got the failed result for EEPROM write.
```
PS C:\work\avr\avrdude_test\avrdude-v7.0-windows-x64> .\avrdude -p m2560 -c wiring -b 115200 -P COM6 -U eeprom:w:entest.eep:i -vv

avrdude.exe: Version 7.0
             Copyright (c) Brian Dean, http://www.bdmicro.com/
             Copyright (c) Joerg Wunsch

             System wide configuration file is "C:/work/avr/avrdude_test/avrdude-v7.0-windows-x64/avrdude.conf"

             Using Port                    : COM6
             Using Programmer              : wiring
             Overriding Baud Rate          : 115200
avrdude.exe: wiring_open(): releasing DTR/RTS
avrdude.exe: wiring_open(): asserting DTR/RTS
             AVR Part                      : ATmega2560
             Chip Erase delay              : 9000 us
             PAGEL                         : PD7
             BS2                           : PA0
             RESET disposition             : dedicated
             RETRY pulse                   : SCK
             Serial program mode           : yes
             Parallel program mode         : yes
             Timeout                       : 200
             StabDelay                     : 100
             CmdexeDelay                   : 25
             SyncLoops                     : 32
             PollIndex                     : 3
             PollValue                     : 0x53
             Memory Detail                 :

                                               Block Poll               Page                       Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               eeprom                 65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
               flash                  65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
               lfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               hfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               efuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               lock                    0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00
               signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00

             Programmer Type : Wiring
             Description     : Wiring
             Programmer Model: AVRISP
             Hardware Version: 15
             Firmware Version Master : 2.10
             Vtarget         : 0.0 V
             SCK period      : 0.1 us

avrdude.exe: AVR device initialized and ready to accept instructions

Reading |                                                    | 0% 0.00savrdude.exe: stk500isp_read_byte(.., signature, 0x0, ...)
avrdude.exe: stk500isp_read_byte(): Sending read memory command: avrdude.exe: stk500isp_read_byte(.., signature, 0x1, ...)
Reading | #################                                  | 33% 0.01savrdude.exe: stk500isp_read_byte(.., signature, 0x2, ...)
Reading | ################################################## | 100% 0.01s

avrdude.exe: Device signature = 0x1e9801 (probably m2560)
avrdude.exe: reading input file "entest.eep"
avrdude.exe: writing eeprom (512 bytes):

Writing | ################################################## | 100% 2.08s

avrdude.exe: 512 bytes of eeprom written
avrdude.exe: verifying eeprom memory against entest.eep:
avrdude.exe: load data eeprom data from input file entest.eep:
avrdude.exe: input file entest.eep contains 512 bytes
avrdude.exe: reading on-chip eeprom data:

Reading | ################################################## | 100% 0.52s

avrdude.exe: verifying ...
avrdude.exe: verification error, first mismatch at byte 0x0008
             0x66 != 0x6b
avrdude.exe: verification error; content mismatch

avrdude.exe done.  Thank you.
```

mcuee, 2022-06-05:
> It is said that the production hex file is buggy and I should use the other one. https://github.com/arduino/ArduinoCore-avr/blob/master/bootloaders/stk500v2/stk500boot_v2_mega2560.hex
> 
> I still got the content missmatch failure for EEPROM write.

Okay, that is caused by the known issue that EEPROM read will fail. In reality the EEPROM read is good.
```
PS C:\work\avr\avrdude_test\avrdude-v7.0-windows-x64>
                                                      .\avrdude -p m2560 -c dragon_isp -t

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.15s

avrdude.exe: Device signature = 0x1e9801 (probably m2560)
avrdude> d ee
>>> d ee

Reading | ################################################## | 100% 3.26s

0000  54 68 65 20 71 75 69 63  6b 20 62 72 6f 77 6e 20  |The quick brown |
0010  66 6f 78 20 6a 75 6d 70  73 20 6f 76 65 72 20 74  |fox jumps over t|
0020  68 65 20 6c 61 7a 79 20  64 6f 67 0a 54 68 65 20  |he lazy dog The |
0030  71 75 69 63 6b 20 62 72  6f 77 6e 20 66 6f 78 20  |quick brown fox |
0040  6a 75 6d 70 73 20 6f 76  65 72 20 74 68 65 20 6c  |jumps over the l|
0050  61 7a 79 20 64 6f 67 0a  54 68 65 20 71 75 69 63  |azy dog The quic|
0060  6b 20 62 72 6f 77 6e 20  66 6f 78 20 6a 75 6d 70  |k brown fox jump|
0070  73 20 6f 76 65 72 20 74  68 65 20 6c 61 7a 79 20  |s over the lazy |
0080  64 6f 67 0a 54 68 65 20  71 75 69 63 6b 20 62 72  |dog The quick br|
0090  6f 77 6e 20 66 6f 78 20  6a 75 6d 70 73 20 6f 76  |own fox jumps ov|
00a0  65 72 20 74 68 65 20 6c  61 7a 79 20 64 6f 67 0a  |er the lazy dog |
00b0  54 68 65 20 71 75 69 63  6b 20 62 72 6f 77 6e 20  |The quick brown |
00c0  66 6f 78 20 6a 75 6d 70  73 20 6f 76 65 72 20 74  |fox jumps over t|
00d0  68 65 20 6c 61 7a 79 20  64 6f 67 0a 54 68 65 20  |he lazy dog The |
00e0  71 75 69 63 6b 20 62 72  6f 77 6e 20 66 6f 78 20  |quick brown fox |
00f0  6a 75 6d 70 73 20 6f 76  65 72 20 74 68 65 20 6c  |jumps over the l|

avrdude> q
>>> q

avrdude.exe done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude-v7.0-windows-x64> .\avrdude -p m2560 -c dragon_isp -U eeprom:v:entest.eep:i

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.15s

avrdude.exe: Device signature = 0x1e9801 (probably m2560)
avrdude.exe: verifying eeprom memory against entest.eep:

Reading | ################################################## | 100% 6.49s

avrdude.exe: 512 bytes of eeprom verified

avrdude.exe done.  Thank you.

```

mcuee, 2022-06-05:
I think I can close this issue as the EEPROM read issue for the original Arduino Mega2560 is well known.
https://github.com/arduino/ArduinoCore-avr/pull/24

I will see if I can recompile the bootloader and check again.

closed 2022-06-05 by mcuee

mcuee, 2022-06-05:
> I think I can close this issue as the EEPROM read issue for the original Arduino Mega2560 is well known. [arduino/ArduinoCore-avr#24](https://github.com/arduino/ArduinoCore-avr/pull/24)
> 
> I will see if I can recompile the bootloader and check again.

Yes I can rebuild the hex file with the pull-request with WinAVR-20100110 version and it does help.

```
PS C:\work\avr\avrdude_test\avrdude-v7.0-windows-x64> .\avrdude -p m2560 -c wiring -b 115200 -P COM6 -U eeprom:w:entest.eep:i -vv

avrdude.exe: Version 7.0
             Copyright (c) Brian Dean, http://www.bdmicro.com/
             Copyright (c) Joerg Wunsch

             System wide configuration file is "C:/work/avr/avrdude_test/avrdude-v7.0-windows-x64/avrdude.conf"

             Using Port                    : COM6
             Using Programmer              : wiring
             Overriding Baud Rate          : 115200
avrdude.exe: wiring_open(): releasing DTR/RTS
avrdude.exe: wiring_open(): asserting DTR/RTS
             AVR Part                      : ATmega2560
             Chip Erase delay              : 9000 us
             PAGEL                         : PD7
             BS2                           : PA0
             RESET disposition             : dedicated
             RETRY pulse                   : SCK
             Serial program mode           : yes
             Parallel program mode         : yes
             Timeout                       : 200
             StabDelay                     : 100
             CmdexeDelay                   : 25
             SyncLoops                     : 32
             PollIndex                     : 3
             PollValue                     : 0x53
             Memory Detail                 :

                                               Block Poll               Page                       Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               eeprom                 65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
               flash                  65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
               lfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               hfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               efuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               lock                    0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00
               signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00

             Programmer Type : Wiring
             Description     : Wiring
             Programmer Model: AVRISP
             Hardware Version: 15
             Firmware Version Master : 2.10
             Vtarget         : 0.0 V
             SCK period      : 0.1 us

avrdude.exe: AVR device initialized and ready to accept instructions

Reading |                                                    | 0% 0.00savrdude.exe: stk500isp_read_byte(.., signature, 0x0, ...)
avrdude.exe: stk500isp_read_byte(): Sending read memory command: avrdude.exe: stk500isp_read_byte(.., signature, 0x1, ...)
Reading | #################                                  | 33% 0.01savrdude.exe: stk500isp_read_byte(.., signature, 0x2, ...)
Reading | ################################################## | 100% 0.01s

avrdude.exe: Device signature = 0x1e9801 (probably m2560)
avrdude.exe: reading input file "entest.eep"
avrdude.exe: writing eeprom (512 bytes):

Writing | ################################################## | 100% 2.08s

avrdude.exe: 512 bytes of eeprom written
[... 85 lines cut ...]

avrdude.exe: AVR device initialized and ready to accept instructions

Reading |                                                    | 0% 0.00savrdude.exe: stk500isp_read_byte(.., signature, 0x0, ...)
avrdude.exe: stk500isp_read_byte(): Sending read memory command: avrdude.exe: stk500isp_read_byte(.., signature, 0x1, ...)
Reading | #################                                  | 33% 0.10savrdude.exe: stk500isp_read_byte(.., signature, 0x2, ...)
Reading | ################################################## | 100% 0.15s

avrdude.exe: Device signature = 0x1e9801 (probably m2560)
avrdude.exe: verifying eeprom memory against entest.eep:
avrdude.exe: load data eeprom data from input file entest.eep:
avrdude.exe: input file entest.eep contains 512 bytes
avrdude.exe: reading on-chip eeprom data:

Reading | ################################################## | 100% 6.49s

avrdude.exe: verifying ...
avrdude.exe: 512 bytes of eeprom verified
avrdude.exe: stk500v2_jtagmkII_close()
avrdude.exe: jtagmkII_close()
avrdude.exe: jtagmkII_close(): Sending sign-off command: 0x80 (1 bytes msg)

avrdude.exe done.  Thank you.

```

mcuee, 2022-06-11:
* https://github.com/avrdudes/avrdude/discussions/940#discussioncomment-2928411

The hex file provided by @stefanrueger works fine as well.
https://github.com/avrdudes/avrdude/files/8882669/atmega2560_v3.txt

stefanrueger, 2022-06-11:
I have not read carefully the full history of this issue, but I note that `-c wiring` uses the STK500**v2** protocol while `-c arduino` uses the STK500**v1** protocol with one extension for ext addr load for parts with more than 128k flash (ie, the ATmega2560).

Optiboot implements a skeleton of STK500**v1**, so looks to work with `-c arduino` and not with `-c wiring`.

Glad to know my bootloader works with EEPROM r/w under `-c arduino` programmer. It should also erase flash when requested by AVRDUDE. For LED blinking you need to patch the bootloader as described in Discussion #940. I plan to release the source of my bootloaders over the summer once I have finished writing the new programmer for AVRDUDE that allows the bootloader to be still smaller: currently I am at 256 bytes for all supported parts (including the ATmega2560) albeit without EEPROM r/w. With the latter you need to spend typically one more memory page (ie, 384 bytes for ATmega328P and 512 bytes for the ATmega2560).

WestfW, 2022-06-11:
> need to spend typically one more memory page

Unless you're skipping the HW support for bootloaders (ala "virtual boot partition" in Optiboot), the boot partition sizes in classic AVRs grow in powers of two, rather than "pagesize increments."  So the next bootloader size up from 256 bytes is 512 bytes, and the _minimum_ bootloader size on ATmega1280/2560 is 1k.

(Not that having "extra space" in the bootloader section is all bad.  this is one of the reasons that I stuck so much "useless cruft" (like the build parameters) in the Optiboot "BIGBOOT" (1k) versions - once you go even a little bit past the 512 byte mark, you have a whole additional 512 bytes to use...)

The "new" avrs let you change the boot partition size in multiples of 256 (mega0, tiny0/1/2) or 512 (avr-dX) bytes.  Still not "pages", but significantly more flexible than the older AVRs.

mcuee, 2022-06-12:
Initially I had problems with the optiboot Mega2560 bootloader from MCUdude's Megacore, later the issue is sorted out. It also works with EEPROM read/write as it has BIGBOOT enabled.
* https://github.com/MCUdude/MegaCore/issues/188

From @MCUdude's comments
* https://github.com/avrdudes/avrdude/discussions/940#discussioncomment-2927458
> for reference, [I host a whole bunch of pre-compiled bootloaders](https://github.com/MCUdude/optiboot_flash) for most AVR ATmega targets.
> In each target folder, there's also a [build info file](https://github.com/MCUdude/optiboot_flash/blob/master/bootloaders/atmega328p/atmega328p_build_info.txt) that contains information about each build. To keep the bootloader size below 512 bytes for 32kB targets and smaller, I've left out EEPROM support. EEPROM support is present on all targets with 64kB flash or more.

mcuee, 2022-06-12:
Just for completeness, the hex file by stefanrueger for the ATmega328P also works fine.
https://github.com/avrdudes/avrdude/files/8882667/atmega328p.txt

```
PS C:\work\avr\avrdude_test\avrdude-v7.0-windows-x64> .\avrdude.exe -p m328p -c usbasp -vv -U flash:w:.\atmega328p_sr940.hex:i

avrdude.exe: Version 7.0
             Copyright (c) Brian Dean, http://www.bdmicro.com/
             Copyright (c) Joerg Wunsch

             System wide configuration file is "C:/work/avr/avrdude_test/avrdude-v7.0-windows-x64/avrdude.conf"

             Using Port                    : usb
             Using Programmer              : usbasp
avrdude.exe: seen device from vendor ->www.fischl.de<-
avrdude.exe: seen product ->USBasp<-
             AVR Part                      : ATmega328P
             Chip Erase delay              : 9000 us
             PAGEL                         : PD7
             BS2                           : PC2
             RESET disposition             : dedicated
             RETRY pulse                   : SCK
             Serial program mode           : yes
             Parallel program mode         : yes
             Timeout                       : 200
             StabDelay                     : 100
             CmdexeDelay                   : 25
             SyncLoops                     : 32
             PollIndex                     : 3
             PollValue                     : 0x53
             Memory Detail                 :

                                               Block Poll               Page                       Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               eeprom                 65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
               flash                  65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
               lfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
               hfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
               efuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
               lock                    0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
               calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00
               signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00

             Programmer Type : usbasp
             Description     : USBasp, http://www.fischl.de/usbasp/

avrdude.exe: auto set sck period (because given equals null)
avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude.exe: Device signature = 0x1e950f (probably m328p)
avrdude.exe: NOTE: "flash" memory has been specified, an erase cycle will be performed
             To disable this feature, specify the -D option.
avrdude.exe: erasing chip
avrdude.exe: auto set sck period (because given equals null)
avrdude.exe: reading input file ".\atmega328p_sr940.hex"
avrdude.exe: writing flash (32768 bytes):

Writing | ################################################## | 100% 0.00s

avrdude.exe: 32768 bytes of flash written
avrdude.exe: verifying flash memory against .\atmega328p_sr940.hex:
avrdude.exe: load data flash data from input file .\atmega328p_sr940.hex:
avrdude.exe: input file .\atmega328p_sr940.hex contains 32768 bytes
avrdude.exe: reading on-chip flash data:

Reading | ################################################## | 100% 0.00s

[... 48 lines cut ...]

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude.exe: Device signature = 0x1e950f (probably m328p)
avrdude.exe: reading input file "entest.eep"
avrdude.exe: writing eeprom (512 bytes):

Writing | ################################################## | 100% 2.34s

avrdude.exe: 512 bytes of eeprom written
avrdude.exe: verifying eeprom memory against entest.eep:
avrdude.exe: load data eeprom data from input file entest.eep:
avrdude.exe: input file entest.eep contains 512 bytes
avrdude.exe: reading on-chip eeprom data:

Reading | ################################################## | 100% 1.07s

avrdude.exe: verifying ...
avrdude.exe: 512 bytes of eeprom verified

avrdude.exe done.  Thank you.

```

stefanrueger, 2022-06-12:
> Unless you're skipping the HW support for bootloaders (ala "virtual boot partition" in Optiboot), the boot partition sizes in classic AVRs grow in powers of two, rather than "pagesize increments." 

@WestfW Correct, and that's what I do. 

I call these bootloaders *vector bootloaders*, but to all extents and purposes the idea originates from your virtual boot partition. 

Feel free to head over to Discussion #940, where I propose a new programmer for AVRDUDE that patches vector bootloaders so that mechanism costs *zero* bytes in the actual bootloader (which jumps to the agreed vector number rather than reset - both are constant). I have a draft implementation for that AVRDUDE SPM programmer (which I call `urclock`). With that my bootloaders (called `urboot`) can be quite small. I only ever use HW support for urboot bootloaders when, by coincidence, their size matches a HW support size. I actually don't know why I bother even in these cases, because, as you say, if they get over only a little they then need double size. 

Would be *very* interested in your comments in Discussion #940!

I plan to publish the whole lot (SPM programmer for AVRDUDE, urboot bootloader and an AVR uploader sketch that deals with fuses, watchdog timeout, LED position)  in the next few months once I figured out a way of integrating that with AVRDUDE so that as little as possible needs changing in the AVRDUDE code base. I predict the new SPM programmer might be useful for the optiboot project, too. 

My work on the  `urboot` bootloader was inspired by your optiboot. So, first of all *many thanks* for your work and inspiration. I managed to produce urboot bootloaders that are considerably smaller. Many of my tricks are not for the faint-hearted such as asking gcc-avr to allocate a *global* variable uint16_t zaddress in r30/r31 where the address needs to be eventually for SMP/LPM later on anyway. My early urboot bootloaders work with `-c arduino`. For example, [atmega328p.hex](https://github.com/avrdudes/avrdude/files/8882667/atmega328p.txt) only has 454 bytes code plus a 6 byte table at the top of flash, *erases flash* (except itself) when AVRDUDE issues a chip erase, carries out EEPROM read and write, sports an API for calling writepage() from the application and puts the reset cause in r2 (at zero bytes cost because I asked gcc-avr to  allocate the local variable there); it can even twiggle LEDs but since it's not clear which LEDs a board has I use different nops as placeholders for the sbi/cbi port/ddr opcodes by the AVR uploader sketch.

Anyway, with the proposed `urclock` SPM programmer for AVRDUDE these little `urboot` bootloaders can become smaller still... and, because I put some effort in to be backwards compatible with `-c arduino` other bootloader writers should benefit, too.

stefanrueger, 2022-06-12:
> The "new" avrs let you change the boot partition size in multiples of 256 (mega0, tiny0/1/2) or 512 (avr-dX) bytes

I had looked at the data sheet, and I think vector bootloaders should still work (but I haven't tried yet).