avrdudes/avrdude #988: Two HORRIBLE error messages that have wasted hundreds or thousands of man hours

SpenceKonde, 2022-06-11:
More than any other error, this error leads new users astray. When using ISP programming on classic AVRs, and the signature doesn't match. 
  a. If the signature reads as 0xFFFFFF we should report that the likely cause is a wiring error (MISO shorted to Vcc, GND not connected to target, etc
  b. If the signature reads as 0x000000, the most common case, we should report the likely cause either wiring, or that the user previously set the chip to a non-internal clock source which isn't present, and that supplying an 4-8 MHz squarewave to the XTAL:1 pin should allow it to be reprogrammed. 
  c. Only if neither of the above is, we should print what chip we were expected, and what we saw, and mention -F

The reference to -F when it's reading as 0xFFFFFF or 0x000000 is misleading, because it implies that -F might fix it,. when it absolutely never would, because the problem is either MISO shorted high or chip set to use crystal or ext clock that isn;t present. We should mention workaround only when they might be helpful. Many relative newbies (based on what I read on forums) seem to get sidetracked by that note, and spend hours trying to figure out how to get that -F into the upload command, instead of finding the modified arduinoasISP that outputs an 8 MHz clock which can be used to revive 0x000000 chips that wound up that way due to incorrect clock setting.

dl8dtl, 2022-06-11:
> because it implies that -F might fix it,. when it absolutely never would

You are probably too young then. ;-)

In the early days of the AVRs (and that's where all that including the message originate from), with those sometimes quite flakey self-built parallel-port ISP adapters, it could occasionally happen that you triggered an erasure of the signature fuses. This case is even mentioned in one of the old AVR appnotes. Likewise, I've got one AT90S1200 where I somehow managed it to reprogram the signature fuses to all 0's – even though that controller still functions normally.

That's why we invented `-F` as an override, and gave that hint in the error message.

Covering all the cases there which you mention is certainly way too much for a simple error message. In that case, it makes more sense to rewrite the error message into some kind of "please read the documentation", and then mention all possible cases in the FAQ.

You're welcome to make suggestions for that, preferrably as pull requests.

MCUdude, 2022-06-11:
Well, I do agree that showing a more informative error message when the device ID is read as 0x000000 or 0xFFFFFF is a good idea, but information about the -F flag is still relevant.

How about something like:

```
avrdude: Device signature = 0x000000
avrdude: Yikes!  Invalid device signature (should have been 0x1e9801). 
         This may be caused by improper wiring or a missing clock signal on the XTAL1 pin.
         Double-check connections, check the XTAL1 pin and try again.
         -F may be used to skip device signature checking.
```

```
avrdude: Device signature = 0xFFFFFF
avrdude: Yikes!  Invalid device signature (should have been 0x1e9801).
         This may be caused by improper wiring.
         Double-check connections and try again.
         -F may be used to skip device signature checking.
```

I can provide a PR if we can figure out some error messages that are descriptive and not too verbose.
BTW @SpenceKonde which programmer can I use to reproduce the 0x000000 and 0xFFFFFF signature?
When using my USBasp, I'm only getting this error:

```
         Programmer Type : usbasp
         Description     : USBasp, http://www.fischl.de/usbasp/

avrdude: auto set sck period (because given equals null)
avrdude: error: program enable: target doesn't answer. 1 
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

avrdude done.  Thank you.
```

dl8dtl, 2022-06-11:
PR welcome, but nevertheless, I think a respective entry in the FAQ would be worth the while. It could mention the wiring issues, but could also hint there might be chances that a device signature is damaged.

As faulty wiring or clock setup causes a complete communication failure, while a wrong signature requires a prior successful ISP initialzation, it should even be possible to clearly distinguish both situations.

SpenceKonde, 2022-06-11:
You are probably too old then :-) 
Parallel programmers have gone the way of the dinosaurs. Nobody uses them these days!

It sounds like the possible messageswe want to show are:
```
avrdude: Yikes!  Invalid device signature (should have been 0x1e9801 saw 0xFFFFFF).
         This is likely caused by improper wiring. 
         Double-check connections and try again.
```
```
avrdude: Yikes!  Invalid device signature (should have been 0x1e9801 saw 0x000000).
        This may be caused by improper wiring, or the chip may be set to use a clock source
         that is not preset. (an 8 MHz square wave applied to pin XTAL1 will permit programming)
         Double-check connections and try again.
```
```
avrdude: Yikes!  Invalid device signature (should have been 0x1e9801 saw 0xsomeotherchipid) .
         Make sure you have the correct chip specfied. 
         Double-check connections and try again.
         [If parallel, show -F message]
````
Rare indeed is the wrong-but-plausible-signature. (I've bricked my share of AVRs, but only one ever ended up with the wrong signature.

dl8dtl, 2022-06-11:
> Parallel programmers have gone the way of the dinosaurs. Nobody uses them these days!

Well, the remark was meant in response to your "this could never ever fix it" comment. Btw., the fact that the signature might be damaged isn't restricted to parallel programmers at all, but it's certainly much less likely to happen with more modern AVRs.

The more I think about it though, this message is never supposed to be triggered by faulty MISO/MOSI wiring, or SCK issues. Faulty MISO/MOSI prevents the sign-in from ever completing, so the signature is never read at all.

SCK issues will normally also prevent the sign-in from completing. In corner cases (SCK just slightly over the limit), the sign-in might complete, but then, the signature is usually read out as garbage rather than 0x000000 or 0xFFFFFF. The above message though only triggers for 0x000000 and 0xFFFFFF – all other cases cause a "signature mismatch" message.

MCUdude, 2022-06-11:
> The more I think about it though, this message is never supposed to be triggered by faulty MISO/MOSI wiring, or SCK issues. Faulty MISO/MOSI prevents the sign-in from ever completing, so the signature is never read at all.

It looks like this only happens with the [Arduino as ISP](https://docs.arduino.cc/built-in-examples/arduino-isp/ArduinoISP/) programmer. When searching online for "Avrdude 0x000000 error", all programmers in use are Arduino as ISPs.

dl8dtl, 2022-06-11:
Sounds like a  bug in the Arduino ISP implementation (either in the AVRDUDE side, or in the firmware).

That bug needs to be fixed, rather than fiddling with that message.

mcuee, 2022-06-14:
> It looks like this only happens with the [Arduino as ISP](https://docs.arduino.cc/built-in-examples/arduino-isp/ArduinoISP/) programmer. When searching online for "Avrdude 0x000000 error", all [...]

Sometimes it can be other values. For example, here is what it will happen if I miss-wire the 6-pin ISP connector on the target (Arduino Uno) -- 3 pins inserted only, using "Arduino as ISP" sketch.
```
C:\work\avr\binary\avrdude-v7.0-windows-x64> .\avrdude.exe -p m328p -c stk500v1 -P COM29 -b 19200 -vv

avrdude.exe: Version 7.0
             Copyright (c) Brian Dean, http://www.bdmicro.com/
             Copyright (c) Joerg Wunsch

             System wide configuration file is "C:/work/avr/binary/avrdude-v7.0-windows-x64/avrdude.conf"

             Using Port                    : COM29
             Using Programmer              : stk500v1
             Overriding Baud Rate          : 19200
             AVR Part                      : ATmega328P
             Chip Erase delay              : 9000 us
             PAGEL                         : PD7
             BS2                           : PC2
             RESET disposition             : dedicated
             RETRY pulse                   : SCK
             Serial program mode           : yes
             Parallel program mode         : yes
             Timeout                       : 200
             StabDelay                     : 100
             CmdexeDelay                   : 25
             SyncLoops                     : 32
             PollIndex                     : 3
             PollValue                     : 0x53
             Memory Detail                 :

                                               Block Poll               Page                       Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               eeprom                 65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
               flash                  65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
               lfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
               hfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
               efuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
               lock                    0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
               calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00
               signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00

             Programmer Type : STK500
             Description     : Atmel STK500 Version 1.x firmware
             Hardware Version: 2
             Firmware Version: 1.18
             Topcard         : Unknown
             Vtarget         : 0.0 V
             Varef           : 0.0 V
             Oscillator      : Off
             SCK period      : 0.1 us

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.02s

avrdude.exe: Device signature = 0x00ffff
avrdude.exe: Expected signature for ATmega328P is 1E 95 0F
             Double check chip, or use -F to override this check.

avrdude.exe done.  Thank you.

```

dl8dtl, 2022-06-14:
Something is pretty broken then: it should not even start trying to read a signature then.
But now that I see it's using STK500v1 protocol, I'm not that surprised at all: nobody cared about making the STK500v1 protocol implementation in AVRDUDE robust once v2 was the way to go on the real STK500.

So if anyone is going to fix anything here, don't start with the error message, please start with the true error: if the ISP setup is not working, the sign-on must never succeed, and AVRDUDE is not supposed to even start reading a signature.

Someone needs to analyze then whether it's actually AVRDUDE that proceeds even though the sign-on failed, or whether it's a bug in that firmware that pretends a successful sign-on even though the ISP setup is broken.

Just "fixing" the symptoms (by changing the message, as the subject of the ticket suggests) is certainly the wrong thing to do.

mcuee, 2022-06-14:
I tend to think it is an issue with the "Arduino as ISP sketch". The code is here.
https://github.com/arduino/arduino-examples/blob/main/examples/11.ArduinoISP/ArduinoISP/ArduinoISP.ino

mcuee, 2022-06-14:
Ref: 
https://github.com/arduino/arduino-examples/blob/main/examples/11.ArduinoISP/ArduinoISP/ArduinoISP.ino#L646
```
void avrisp() {
  uint8_t ch = getch();
  switch (ch) {
    case '0': // signon
      ISPError = 0;
      empty_reply();
      break;
```
```

mcuee, 2022-06-14:
This is what I get if I connect nothing to the main Uno as the ISP programmer, with an imaginary ATmega8 chip. :-)

```
C:\work\avr\binary\avrdude-v7.0-windows-x64> .\avrdude.exe -p m8 -c stk500v1 -P COM29  -b 19200 -vv

avrdude.exe: Version 7.0
             Copyright (c) Brian Dean, http://www.bdmicro.com/
             Copyright (c) Joerg Wunsch

             System wide configuration file is "C:/work/avr/binary/avrdude-v7.0-windows-x64/avrdude.conf"

             Using Port                    : COM29
             Using Programmer              : stk500v1
             Overriding Baud Rate          : 19200
             AVR Part                      : ATmega8
             Chip Erase delay              : 10000 us
             PAGEL                         : PD7
             BS2                           : PC2
             RESET disposition             : dedicated
             RETRY pulse                   : SCK
             Serial program mode           : yes
             Parallel program mode         : yes
             Timeout                       : 200
             StabDelay                     : 100
             CmdexeDelay                   : 25
             SyncLoops                     : 32
             PollIndex                     : 3
             PollValue                     : 0x53
             Memory Detail                 :

                                               Block Poll               Page                       Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               eeprom                  4    20   128    0 no        512    4      0  9000  9000 0xff 0xff
               flash                  33    10    64    0 yes      8192   64    128  4500  4500 0xff 0x00
               lfuse                   0     0     0    0 no          1    1      0  2000  2000 0x00 0x00
               hfuse                   0     0     0    0 no          1    1      0  2000  2000 0x00 0x00
               lock                    0     0     0    0 no          1    1      0  2000  2000 0x00 0x00
               calibration             0     0     0    0 no          4    1      0     0     0 0x00 0x00
               signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00

             Programmer Type : STK500
             Description     : Atmel STK500 Version 1.x firmware
             Hardware Version: 2
             Firmware Version: 1.18
             Topcard         : Unknown
             Vtarget         : 0.0 V
             Varef           : 0.0 V
             Oscillator      : Off
             SCK period      : 0.1 us

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.02s

avrdude.exe: Device signature = 0x00ffff
avrdude.exe: Expected signature for ATmega8 is 1E 93 07
             Double check chip, or use -F to override this check.

avrdude.exe done.  Thank you.
```

dl8dtl, 2022-06-14:
> case '0': // signon
>       ISPError = 0;
>       empty_reply();

:-(((

@SpenceKonde: please blame the author of that firmware rather than AVRDUDE.

closed 2022-06-14 by dl8dtl

dl8dtl, 2022-06-14:
If that site had a ticket system, I'd open a ticket there now … alas, I don't see something like that.

per1234, 2022-06-14:
>I don't see something like that

Right here:

https://github.com/arduino/arduino-examples/issues

mcuee, 2022-06-14:
Just FYI, I like ScratchMonkey (stk500v2 and more) better than "Arduino as ISP".
https://github.com/microtherion/ScratchMonkey/tree/master/ScratchMonkey

This is what I get when I did not connect anything, at least avrdude says "Double check connections and try again" before "or use -F to override  this check".

```
C:\work\avr\binary\avrdude-v7.0-windows-x64> .\avrdude.exe -p m328p -c stk500v2 -P COM29  -vv

avrdude.exe: Version 7.0
             Copyright (c) Brian Dean, http://www.bdmicro.com/
             Copyright (c) Joerg Wunsch

             System wide configuration file is "C:/work/avr/binary/avrdude-v7.0-windows-x64/avrdude.conf"

             Using Port                    : COM29
             Using Programmer              : stk500v2
avrdude.exe: stk500v2_getsync(): got response from unknown programmer SCRATCHMONKEY, assuming STK500
             AVR Part                      : ATmega328P
             Chip Erase delay              : 9000 us
             PAGEL                         : PD7
             BS2                           : PC2
             RESET disposition             : dedicated
             RETRY pulse                   : SCK
             Serial program mode           : yes
             Parallel program mode         : yes
             Timeout                       : 200
             StabDelay                     : 100
             CmdexeDelay                   : 25
             SyncLoops                     : 32
             PollIndex                     : 3
             PollValue                     : 0x53
             Memory Detail                 :

                                               Block Poll               Page                       Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               eeprom                 65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
               flash                  65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
               lfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
               hfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
               efuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
               lock                    0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
               calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00
               signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00

             Programmer Type : STK500V2
             Description     : Atmel STK500 Version 2.x firmware
             Programmer Model: STK500
             Hardware Version: 0
             Firmware Version Master : 2.00
             Topcard         : Unknown
             Vtarget         : 5.0 V
             SCK period      : 8.7 us
             Varef           : 5.0 V
             Oscillator      : Off

avrdude.exe: stk500v2_command(): command failed
avrdude.exe: initialization failed, rc=-1
             Double check connections and try again, or use -F to override
             this check.

avrdude.exe done.  Thank you.

```