avrdudes/avrdude #989: To integrate the optional patches for ScratchMonkey

mcuee, 2022-06-12:
ScratchMonkey seems to be interesting as it creates an stk5002 prorammer. It also has some unique features like TPI support and LED status.

ScratchMonkey documentation:
https://microtherion.github.io/ScratchMonkey/GettingStarted.html

ScratchMonkey optional patch
https://github.com/microtherion/ScratchMonkey/blob/master/avrdude.patch

mcuee, 2022-06-12:
It is probably not as popular as the Arduino as ISP sketch (https://docs.arduino.cc/built-in-examples/arduino-isp/ArduinoISP) but still pretty popular.

The pin out is more or less compatible with Arduino as ISP sketch.
https://microtherion.github.io/ScratchMonkey/ISP.html (ScratchMonkey pin ISP connection)
https://docs.arduino.cc/built-in-examples/arduino-isp/ArduinoISP (Arduino as ISP connection)

I also supports high voltage programming, high voltage serial programming and high voltage parallel programming with some additional circuits.

Another example using Pro Mini.
https://prominimicros.com/how-to-make-an-avr-programmer-out-of-a-pro-mini-part-1-isp-icsp/

mcuee, 2022-06-12:
It seems to work fine with stk500v2 protocol under avrdude.

```
PS C:\work\avr\avrdude_test\avrdude-v7.0-windows-x64> .\avrdude.exe -p m328p -c stk500v2 -P COM3 -D -U flash:w:.\Blink.ino.standard.hex:i -v

avrdude.exe: Version 7.0
             Copyright (c) Brian Dean, http://www.bdmicro.com/
             Copyright (c) Joerg Wunsch

             System wide configuration file is "C:/work/avr/avrdude_test/avrdude-v7.0-windows-x64/avrdude.conf"

             Using Port                    : COM3
             Using Programmer              : stk500v2
avrdude.exe: stk500v2_ReceiveMessage(): timeout
avrdude.exe: stk500v2_getsync(): got response from unknown programmer SCRATCHMONKEY, assuming STK500
             AVR Part                      : ATmega328P
             Chip Erase delay              : 9000 us
             PAGEL                         : PD7
             BS2                           : PC2
             RESET disposition             : dedicated
             RETRY pulse                   : SCK
             Serial program mode           : yes
             Parallel program mode         : yes
             Timeout                       : 200
             StabDelay                     : 100
             CmdexeDelay                   : 25
             SyncLoops                     : 32
             PollIndex                     : 3
             PollValue                     : 0x53
             Memory Detail                 :

                                               Block Poll               Page                       Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               eeprom                 65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
               flash                  65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
               lfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
               hfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
               efuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
               lock                    0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
               calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00
               signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00

             Programmer Type : STK500V2
             Description     : Atmel STK500 Version 2.x firmware
             Programmer Model: STK500
             Hardware Version: 0
             Firmware Version Master : 2.00
             Topcard         : Unknown
             Vtarget         : 5.0 V
             SCK period      : 8.7 us
             Varef           : 5.0 V
             Oscillator      : Off

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude.exe: Device signature = 0x1e950f (probably m328p)
avrdude.exe: reading input file ".\Blink.ino.standard.hex"
avrdude.exe: writing flash (924 bytes):

Writing | ################################################## | 100% 1.32s

avrdude.exe: 924 bytes of flash written
avrdude.exe: verifying flash memory against .\Blink.ino.standard.hex:

Reading | ################################################## | 100% 1.29s

avrdude.exe: 924 bytes of flash verified

avrdude.exe done.  Thank you.

```

mcuee, 2022-06-13:
BTW, based on my testing, it is also a bit more stable and faster than "Arduino as ISP".

```
PS C:\work\avr\avrdude_test\avrdude-v7.0-windows-x64> .\avrdude.exe -p m328p -c stk500v2 -P COM3 -U flash:w:.\Blink.ino.with_bootloader.standard.hex:i -v

avrdude.exe: Version 7.0
             Copyright (c) Brian Dean, http://www.bdmicro.com/
             Copyright (c) Joerg Wunsch

             System wide configuration file is "C:/work/avr/avrdude_test/avrdude-v7.0-windows-x64/avrdude.conf"

             Using Port                    : COM3
             Using Programmer              : stk500v2
avrdude.exe: stk500v2_ReceiveMessage(): timeout
avrdude.exe: stk500v2_getsync(): got response from unknown programmer SCRATCHMONKEY, assuming STK500
             AVR Part                      : ATmega328P
             Chip Erase delay              : 9000 us
             PAGEL                         : PD7
             BS2                           : PC2
             RESET disposition             : dedicated
             RETRY pulse                   : SCK
             Serial program mode           : yes
             Parallel program mode         : yes
             Timeout                       : 200
             StabDelay                     : 100
             CmdexeDelay                   : 25
             SyncLoops                     : 32
             PollIndex                     : 3
             PollValue                     : 0x53
             Memory Detail                 :

                                               Block Poll               Page                       Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               eeprom                 65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
               flash                  65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
               lfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
               hfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
               efuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
               lock                    0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
               calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00
               signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00

             Programmer Type : STK500V2
             Description     : Atmel STK500 Version 2.x firmware
             Programmer Model: STK500
             Hardware Version: 0
             Firmware Version Master : 2.00
             Topcard         : Unknown
             Vtarget         : 5.0 V
             SCK period      : 8.7 us
             Varef           : 5.0 V
             Oscillator      : Off

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude.exe: Device signature = 0x1e950f (probably m328p)
avrdude.exe: NOTE: "flash" memory has been specified, an erase cycle will be performed
             To disable this feature, specify the -D option.
avrdude.exe: erasing chip
avrdude.exe: reading input file ".\Blink.ino.with_bootloader.standard.hex"
avrdude.exe: writing flash (32768 bytes):

Writing | ################################################## | 100% 0.44s

avrdude.exe: 32768 bytes of flash written
avrdude.exe: verifying flash memory against .\Blink.ino.with_bootloader.standard.hex:

[... 53 lines cut ...]
             SCK period      : 0.1 us

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.02s

avrdude.exe: Device signature = 0x1e950f (probably m328p)
avrdude.exe: NOTE: "flash" memory has been specified, an erase cycle will be performed
             To disable this feature, specify the -D option.
avrdude.exe: erasing chip
avrdude.exe: reading input file ".\Blink.ino.with_bootloader.standard.hex"
avrdude.exe: writing flash (32768 bytes):

Writing | ################################################## | 100% 1.13s

avrdude.exe: 32768 bytes of flash written
avrdude.exe: verifying flash memory against .\Blink.ino.with_bootloader.standard.hex:

Reading | ################################################## | 100% 0.62s

avrdude.exe: 32768 bytes of flash verified

avrdude.exe done.  Thank you.

```

MCUdude, 2022-06-13:
From the Scratchmonkey docs (https://microtherion.github.io/ScratchMonkey/GettingStarted.html):

> ScratchMonkey communicates using the STK500v2 protocol, which is supported by default in avrdude, the communication program that the Arduino IDE uses to connect to its targets. However, the standard protocol cannot support some ScratchMonkey features, such as TPI programming and status LEDs.
For most uses, these features are no great loss, and you can simply use the (STK500 Mode) programmer entries to use the standard protocol. However, if you're interested in these features, and comfortable with avrdude development, you can apply the file avrdude.patch to your copy of the avrdude source code, recompile avrdude, and reinstall it, along with the updated configuration file.
> 
> While I will attempt to get the patch incorporated into the standard distribution of avrdude, the Arduino IDE tends to rarely pick up new version of avrdude, so it might take years for this to show up in an Arduino IDE by default.

MCUdude, 2022-06-13:
I don't see how the Scratchmonkey Avrdude patch can enable TPI programming? To me, it mostly looks like a patch to get the LEDs working. Please correct me if I'm wrong!

mcuee, 2022-06-14:
BTW, there is a emulation of STK500v2 using usbtinyisp here.
https://cdn-learn.adafruit.com/downloads/pdf/usbtinyisp.pdf
https://learn.adafruit.com/usbtinyisp/download

After installation of the signed com0com driver (https://sourceforge.net/p/com0com/bugs/32/) and the USBtiny500 bridge, it kinds of working half half -- detection working with avrdude but reading/writing do not really work. 

```
PS C:\work\avr\avrdude_test\avrdude-v7.0-windows-x64> .\avrdude.exe -p m328p -c stk500v2 -P COM4 -D -vv

avrdude.exe: Version 7.0
             Copyright (c) Brian Dean, http://www.bdmicro.com/
             Copyright (c) Joerg Wunsch

             System wide configuration file is "C:/work/avr/avrdude_test/avrdude-v7.0-windows-x64/avrdude.conf"

             Using Port                    : COM4
             Using Programmer              : stk500v2
             AVR Part                      : ATmega328P
             Chip Erase delay              : 9000 us
             PAGEL                         : PD7
             BS2                           : PC2
             RESET disposition             : dedicated
             RETRY pulse                   : SCK
             Serial program mode           : yes
             Parallel program mode         : yes
             Timeout                       : 200
             StabDelay                     : 100
             CmdexeDelay                   : 25
             SyncLoops                     : 32
             PollIndex                     : 3
             PollValue                     : 0x53
             Memory Detail                 :

                                               Block Poll               Page                       Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               eeprom                 65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
               flash                  65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
               lfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
               hfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
               efuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
               lock                    0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
               calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00
               signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00

             Programmer Type : STK500V2
             Description     : Atmel STK500 Version 2.x firmware
             Programmer Model: AVRISP
             Hardware Version: 1
             Firmware Version Master : 2.00
             Vtarget         : 0.0 V
             SCK period      : 0.1 us

avrdude.exe: AVR device initialized and ready to accept instructions

Reading |                                                    | 0% 0.00savrdude.exe: stk500isp_read_byte(.., signature, 0x0, ...)
avrdude.exe: stk500isp_read_byte(): Sending read memory command: avrdude.exe: stk500isp_read_byte(.., signature, 0x1, ...)
Reading | #################                                  | 33% 0.13savrdude.exe: stk500isp_read_byte(.., signature, 0x2, ...)
Reading | ################################################## | 100% 0.24s

avrdude.exe: Device signature = 0x1e950f (probably m328p)

avrdude.exe done.  Thank you.

```

mcuee, 2022-07-17:
Updates: I have got one AVRISP mkii clone and three STK500 v2 programmer clones in the end (two of them are supposed to support HVSP/HVPP but I have not tested that). So I do not really need ScratchMonkey now.

stefanrueger, 2022-07-18:
I had a quick(!) look at @microtherion's patch:  It is solid, and expect that it works and will support ScratchMonkey correctly, *but* it 
 - Uses the name of the programmer itself in the code (not a bug per se, but ideally AVRDUDE's functional behaviour should be invariant under renaming of programmers)
 - Introduces a new programmer type `PGMTYPE_SCRATCHMONKEY` as equal sibling to the likes of `PGMTYPE_JTAGICE3` and `PGMTYPE_STK600` without actually representing a different programming protocol/paradigm

Looking at the meat of the patch code my impression is that ScratchMonkey is rather a derivative of other programmers, undoubtedly with neat features, but it should therefore be integrated with a clearer separation of code and programmer description in avrdude.conf. This is one of the features that makes AVRDUDE an outstanding programmer and relatively manageable to maintain, so is important for the AVRDUDE project.

My gut feeling is that the patch is nearly there. Let's come back to that at another point once the issue queue has dropped down a bit. And ... with a bit of luck ... a slightly modified patch for ScratchMonkey comes along that addresses both concerns above, perhaps even by @microtherion?

mcuee, 2022-08-26:
Performance of ScratchMonkey is actullay pretty decent, especially flash writing.

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c stk500v2 -P COM4 -p m328p -U .\m328p_readback.hex -v

avrdude.exe: Version 7.0-20220824 (3e49f07)
             Copyright (c) Brian Dean, http://www.bdmicro.com/
             Copyright (c) Joerg Wunsch

             System wide configuration file is "C:/work/avr/avrdude_test/avrdude_bin/avrdude.conf"

avrdude.exe: input file .\m328p_readback.hex auto detected as Intel Hex
             Using Port                    : COM4
             Using Programmer              : stk500v2
avrdude.exe: stk500v2_getsync(): got response from unknown programmer SCRATCHMONKEY, assuming STK500
             AVR Part                      : ATmega328P
             Chip Erase delay              : 9000 us
             PAGEL                         : PD7
             BS2                           : PC2
             RESET disposition             : possible i/o
             RETRY pulse                   : SCK
             Serial program mode           : yes
             Parallel program mode         : yes
             Timeout                       : 200
             StabDelay                     : 100
             CmdexeDelay                   : 25
             SyncLoops                     : 32
             PollIndex                     : 3
             PollValue                     : 0x53
             Memory Detail                 :

                                               Block Poll               Page                       Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               eeprom                 65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
               flash                  65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
               lfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
               hfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
               efuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
               lock                    0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
               signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
               calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00

             Programmer Type : STK500V2
             Description     : Atmel STK500 Version 2.x firmware
             Programmer Model: STK500
             Hardware Version: 0
             Firmware Version Master : 2.00
             Topcard         : Unknown
             Vtarget         : 5.0 V
             SCK period      : 8.7 us
             Varef           : 5.0 V
             Oscillator      : Off

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.02s

avrdude.exe: Device signature = 0x1e950f (probably m328p)
avrdude.exe: NOTE: "flash" memory has been specified, an erase cycle will be performed
             To disable this feature, specify the -D option.
avrdude.exe: erasing chip
avrdude.exe: reading input file .\m328p_readback.hex for flash
             with 32768 bytes in 1 section within [0, 0x7fff]
             using 256 pages and 0 pad bytes
avrdude.exe: writing 32768 bytes flash ...

Writing | ################################################## | 100% 14.59s

avrdude.exe: 32768 bytes of flash written
avrdude.exe: verifying flash memory against .\m328p_readback.hex

Reading | ################################################## | 100% 14.34s

avrdude.exe: 32768 bytes of flash verified

avrdude.exe done.  Thank you.

```

mcuee, 2023-06-20:
> The patch is relatively simple, so it's fairly easy to rewrite it in order to drive the Scratchmonkey programmers from the existing stk500v2 isp/tpi, hvsp, and hvpp programmers.
> 
> However, Avrdude already supports stk500v2 TPI, so the _only_ thing the patch actually adds is status LEDs? And is that a good enough reason to add three new programmers to avrdude.conf?

It seems to be not that popular even though I think it is better than `Arduino As ISP` (not considerig the HVPP and HVSP part as they require extra HW). Maybe we can close this one as well.

In the future if @microtherion or others are interested and submit a PR, then we can of course review the PR.

mcuee, 2023-06-20:
@stefanrueger 

Any objection to close this issue as not planned?

stefanrueger, 2023-06-20:
>  the _only_ thing the patch actually adds is status LEDs?

So could Scratchmonkey be programmed using an existing `-c` programmer?  If so, we should document this somewhere as a comment above the `programmer` entry that works for scratchmonkey. IIRC, the `id` entry of a programmer is a list, so could we just add `, "scratchmonkey"` to the `id` of a  programmer that works?

mcuee, 2023-06-21:
> So could Scratchmonkey be programmed using an existing `-c` programmer? If so, we should document this somewhere as a comment above the `programmer` entry that works for scratchmonkey. IIRC, the [...]

Yes it should work with existing `stk500v2`.

I will test later to see if the warning goes away by adding the extra `id` to `stk500v2`.

If that works, then we can add the following `id` entries to the existing programmers
1) `scratchmonkey` ro `stk500v2`
2) `scratchmonkey_pp` ro `stk500pp`
2) `scratchmonkey_hvsp` ro `stk500hvsp`

stefanrueger, 2023-06-23:
> I will test later to see if the warning goes away by adding the extra id to stk500v2.

The curiosity has gotten the better of me: Has it? Would be an easy way to close this issue?

> avrdude.exe: NOTE: "flash" memory has been specified, an erase cycle will be performed

Is this a notice from git mainline? These days I would have expected the `.exe` to be gone in the messaging functions.

mcuee, 2023-06-23:
> > avrdude.exe: NOTE: "flash" memory has been specified, an erase cycle will be performed
> 
> Is this a notice from git mainline? These days I would have expected the `.exe` to be gone in the messaging functions.

It is not from git main, but rather copy/paste from the previous test results.

WIth git main, then the message is like the following,
```
avrdude: stk500v2_getsync(): got response from unknown programmer SCRATCHMONKEY, assuming STK500
```

mcuee, 2023-06-23:
> > I will test later to see if the warning goes away by adding the extra id to stk500v2.
> 
> The curiosity has gotten the better of me: Has it? Would be an easy way to close this issue?

I just test this and the warning is still there. I think we need a minor change in `stk500v2.c`.

```
PS C:\work\avr\avrdude_test\avrdude_bin> cat .\avrdude.rc
allow_subshells = yes;
programmer parent "dryrun"
    id                     = "dryrun1", "dryrun2";
;

#------------------------------------------------------------
# stk500v2
#------------------------------------------------------------

programmer
    id                     = "stk500v2", "SCRATCHMONKEY";
    desc                   = "Atmel STK500 version 2.x firmware";
    type                   = "stk500v2";
    prog_modes             = PM_TPI | PM_ISP;
    extra_features         = HAS_VTARG_ADJ | HAS_VTARG_READ | HAS_FOSC_ADJ | HAS_VAREF_ADJ;
    connection_type        = serial;
;
PS C:\work\avr\avrdude_test\avrdude_bin> cat .\avrdude.rc
allow_subshells = yes;
programmer parent "dryrun"
    id                     = "dryrun1", "dryrun2";
;

#------------------------------------------------------------
# stk500v2
#------------------------------------------------------------

programmer
    id                     = "stk500v2", "scratchmonkey";
    desc                   = "Atmel STK500 version 2.x firmware";
    type                   = "stk500v2";
    prog_modes             = PM_TPI | PM_ISP;
    extra_features         = HAS_VTARG_ADJ | HAS_VTARG_READ | HAS_FOSC_ADJ | HAS_VAREF_ADJ;
    connection_type        = serial;
;
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c scratchmonkey -P COM50 -p m32u4  -v

avrdude: Version 7.1-20230621 (9269102b)
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is C:\work\avr\avrdude_test\avrdude_bin\avrdude.conf
         User configuration file is C:\work\avr\avrdude_test\avrdude_bin\avrdude.rc
avrdude yywarning() warning: programmer stk500v2 overwrites previous definition C:\work\avr\avrdude_test\avrdude_bin\avrdude.conf:1073. [C:\work\avr\avrdude_test\avrdude_bin\avrdude.rc:11]

         Using Port                    : COM50
         Using Programmer              : scratchmonkey
avrdude: stk500v2_getsync(): got response from unknown programmer SCRATCHMONKEY, assuming STK500
         AVR Part                      : ATmega32U4
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : possible i/o
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom                 65    20     4    0 no       1024    4      0  9000  9000 0x00 0x00
           flash                  65     6   128    0 yes     32768  128    256  4500  4500 0x00 0x00
           lfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
           hfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
           efuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
           lock                    0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
           signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
           calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00

         Programmer Type : STK500V2
         Description     : Atmel STK500 version 2.x firmware
         Programmer Model: STK500
         Hardware Version: 0
         Firmware Version Controller : 2.00
         Topcard         : Unknown
         Vtarget         : 5.0 V
         SCK period      : 8.7 us
         Varef           : 5.0 V
         Oscillator      : Off
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9587 (probably m32u4)

avrdude done.  Thank you.
```

mcuee, 2023-06-23:
@stefanrueger 

BTW, the following warning message is also not so good.
```
avrdude yywarning() warning: programmer stk500v2 overwrites previous definition C:\work\avr\avrdude_test\avrdude_bin\avrdude.conf:1073. [C:\work\avr\avrdude_test\avrdude_bin\avrdude.rc:11]
```

I think it should start with `avrdude warning:`.

mcuee, 2023-06-23:
> I just test this and the warning is still there. I think we need a minor change in stk500v2.c.

The following quick fix seems to work.
```
$ git diff
diff --git a/src/stk500v2.c b/src/stk500v2.c
index a95af29c..aa302514 100644
--- a/src/stk500v2.c
+++ b/src/stk500v2.c
@@ -755,6 +755,9 @@ retry:
       if (siglen >= strlen("STK500_2") &&
          memcmp(resp + 3, "STK500_2", strlen("STK500_2")) == 0) {
        PDATA(pgm)->pgmtype = PGMTYPE_STK500;
+         } else if (siglen >= strlen("SCRATCHMONKEY") &&
+         memcmp(resp + 3, "SCRATCHMONKEY", strlen("SCRATCHMONKEY")) == 0) {
+       PDATA(pgm)->pgmtype = PGMTYPE_STK500;
       } else if (siglen >= strlen("AVRISP_2") &&
                 memcmp(resp + 3, "AVRISP_2", strlen("AVRISP_2")) == 0) {
        PDATA(pgm)->pgmtype = PGMTYPE_AVRISP;

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_issue989 -c scratchmonkey -P COM50 -p m32u4  -v

avrdude_issue989: Version 7.1-20230621 (9269102b)
                  Copyright the AVRDUDE authors;
                  see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

                  System wide configuration file is C:\work\avr\avrdude_test\avrdude_bin\avrdude.conf
                  User configuration file is C:\work\avr\avrdude_test\avrdude_bin\avrdude.rc
avrdude_issue989 yywarning() warning: programmer stk500v2 overwrites previous definition C:\work\avr\avrdude_test\avrdude_bin\avrdude.conf:1073. [C:\work\avr\avrdude_test\avrdude_bin\avrdude.rc:11]

                  Using Port                    : COM50
                  Using Programmer              : scratchmonkey
                  AVR Part                      : ATmega32U4
                  Chip Erase delay              : 9000 us
                  PAGEL                         : PD7
                  BS2                           : PA0
                  RESET disposition             : possible i/o
                  RETRY pulse                   : SCK
                  Serial program mode           : yes
                  Parallel program mode         : yes
                  Timeout                       : 200
                  StabDelay                     : 100
                  CmdexeDelay                   : 25
                  SyncLoops                     : 32
                  PollIndex                     : 3
                  PollValue                     : 0x53
                  Memory Detail                 :

                                                    Block Poll               Page                       Polled
                    Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
                    ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
                    eeprom                 65    20     4    0 no       1024    4      0  9000  9000 0x00 0x00
                    flash                  65     6   128    0 yes     32768  128    256  4500  4500 0x00 0x00
                    lfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
                    hfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
                    efuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
                    lock                    0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
                    signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
                    calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00

                  Programmer Type : STK500V2
                  Description     : Atmel STK500 version 2.x firmware
                  Programmer Model: STK500
                  Hardware Version: 0
                  Firmware Version Controller : 2.00
                  Topcard         : Unknown
                  Vtarget         : 5.0 V
                  SCK period      : 8.7 us
                  Varef           : 5.0 V
                  Oscillator      : Off
avrdude_issue989: AVR device initialized and ready to accept instructions
avrdude_issue989: device signature = 0x1e9587 (probably m32u4)

avrdude_issue989 done.  Thank you.
```

stefanrueger, 2023-06-23:
> should start with avrdude warning: 

No, that's actually OK. These are messages from the the grammar part.

mcuee, 2023-06-23:
> > The following quick fix seems to work.
> 
> Yes, that's what I would have done. Do you want to do a PR, if not I can do that.

Please help to create the PR. Thanks.

mcuee, 2023-06-23:
> > should start with avrdude warning:
> 
> No, that's actually OK. These are messages from the the grammar part.

I see. Good to know that.

PR #1429 (2023-06-25): Add scratchmonkey programmers
Fixes #989.
closed 2023-06-25 by stefanrueger