avrdudes/avrdude #990: Bad EEPROM dump from ATmega328p using ArduinoISP programmer

x0urc3, 2022-06-12:
1. Using the following as a simple test case, avrdude produce a bad EEPROM dump. 
```
#include <avr/io.h>
#include <avr/eeprom.h>

int main(void) {
    uint16_t data=0;

    for (uint16_t i=0; i<1024; i++) { 
        eeprom_write_word((uint16_t *) i, 0xff);
    }

    for (uint16_t i=0; i<512; i+=0x20) {
        eeprom_write_word((uint16_t *) i, data);
        data += 0x11;
    }

    for (uint16_t i=0; i<512; i+=0x20) {
        data = eeprom_read_word((uint16_t *) i);
    }

    return(0);
}
``` 
2. EEPROM address 0x100 suppose to contain 0x88 and this is verified using serial print and avrdude terminal (-t). 
3. Using memory operation (-U) produce a bad EEPROM dump (
[bad-dump.hex.gz](https://github.com/avrdudes/avrdude/files/8885866/bad-dump.hex.gz). The data seems to repeat itself at the 256 boundary i.e. 0x100, 0x200, 0x300. Command used are as follows:
```
 avrdude -c stk500v1 -p m328p -P /dev/ttyUSB0 -b 19200 -U eeprom:r:dump.hex:i
```

x0urc3, 2022-06-12:
I inspect  ATmega328 EEPROM section in avrdude.conf [here](https://github.com/avrdudes/avrdude/blob/cb114233ef68625332df5fff876467f57648cd9d/src/avrdude.conf.in#L9678-L9681) and notice that paged=no but page_size=4. Changing page_size=0 or remove the line fix the bad EEPROM dump above.

x0urc3, 2022-06-12:
I could also confirm that the configuration for ATmega48/88/168 also have the same offending line but I don't have the physical uC to verify.

stefanrueger, 2022-06-12:
Maybe try `avrdude -c arduinoisp ...` not `avrdude -c stk500v1 ...` if it's an ArduinoISP programmer? AVRDUDE relies on being told  the programmer that the PC connects to for working correctly.

If you want to erase EEPROM consider sth like
```
for(uint16_t i=0; i<1024; i+=2)
        eeprom_write_word((uint16_t *) i, 0xffff);
```
Your first for loop looks wrong (though it erases EEPROM OK).

x0urc3, 2022-06-13:
I disabled EESAVE fuse in my project so EEPROM erase is a precaution. My erase loop looks funky because later I'll be saving 2 data byte in a word since eeprom_read_byte() only use byte-based addressing.

Short update. Switching the programmer to  ```arduinoisp``` cause the following error
```
avrdude: Error: Invalid -P value: '/dev/ttyUSB0'
         Use -P usb:bus:device
```
Since my simple modification [above](https://github.com/avrdudes/avrdude/issues/990#issuecomment-1153169075) sort out my specific need, I will come back to this issue later.

x0urc3, 2022-06-13:
My bad, I did not realize there is a difference between ArduinoISP and 'Arduino as ISP'. I like to clarify that I'm using 'Arduino as ISP' programmer soo ```avrdude -c stk500v1 ...``` is correct

mcuee, 2022-06-13:
> My bad, I did not realize there is a difference between ArduinoISP and 'Arduino as ISP'. I like to clarify that I'm using 'Arduino as ISP' programmer soo `avrdude -c stk500v1 ...` is correct

You may want to try `avrdude -c avrisp` instead if you use "Arduino as ISP".

mcuee, 2022-06-13:
BTW, it seems to me ScratchMonkey is better than "Arduino as ISP".

This is the testing using ScratchMonkey (stk500v2) and it seems to work fine. So I tend to think avrdude.conf is okay.

```
PS C:\work\avr\avrdude_test\avrdude-v7.0-windows-x64> cat .\entest.eep
:020000040000FA
:2000000054686520717569636B2062726F776E20666F78206A756D7073206F76657220740E
:200020006865206C617A7920646F670A54686520717569636B2062726F776E20666F78207C
:200040006A756D7073206F76657220746865206C617A7920646F670A5468652071756963FD
:200060006B2062726F776E20666F78206A756D7073206F76657220746865206C617A7920D4
:20008000646F670A54686520717569636B2062726F776E20666F78206A756D7073206F76B5
:2000A000657220746865206C617A7920646F670A54686520717569636B2062726F776E20FE
:2000C000666F78206A756D7073206F76657220746865206C617A7920646F670A54686520C2
:2000E000717569636B2062726F776E20666F78206A756D7073206F76657220746865206C16
:20010000617A7920646F670A54686520717569636B2062726F776E20666F78206A756D7038
:2001200073206F76657220746865206C617A7920646F670A54686520717569636B20627279
:200140006F776E20666F78206A756D7073206F76657220746865206C617A7920646F670A0E
:2001600054686520717569636B2062726F776E20666F78206A756D7073206F7665722074AD
:200180006865206C617A7920646F670A54686520717569636B2062726F776E20666F78201B
:2001A0006A756D7073206F76657220746865206C617A7920646F670A54686520717569639C
:2001C0006B2062726F776E20666F78206A756D7073206F76657220746865206C617A792073
:2001E000646F670A54686520717569636B2062726F776E20666F78206A756D7073206F7654
:00000001FF

PS C:\work\avr\avrdude_test\avrdude-v7.0-windows-x64> ./avrdude -c stk500v2 -P COM3 -p atmega328p -U eeprom:w:entest.eep:i -v

avrdude.exe: Version 7.0
             Copyright (c) Brian Dean, http://www.bdmicro.com/
             Copyright (c) Joerg Wunsch

             System wide configuration file is "C:/work/avr/avrdude_test/avrdude-v7.0-windows-x64/avrdude.conf"

             Using Port                    : COM3
             Using Programmer              : stk500v2
avrdude.exe: stk500v2_getsync(): got response from unknown programmer SCRATCHMONKEY, assuming STK500
             AVR Part                      : ATmega328P
             Chip Erase delay              : 9000 us
             PAGEL                         : PD7
             BS2                           : PC2
             RESET disposition             : dedicated
             RETRY pulse                   : SCK
             Serial program mode           : yes
             Parallel program mode         : yes
             Timeout                       : 200
             StabDelay                     : 100
             CmdexeDelay                   : 25
             SyncLoops                     : 32
             PollIndex                     : 3
             PollValue                     : 0x53
             Memory Detail                 :

                                               Block Poll               Page                       Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               eeprom                 65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
               flash                  65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
               lfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
               hfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
               efuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
               lock                    0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
               calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00
               signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00

             Programmer Type : STK500V2
             Description     : Atmel STK500 Version 2.x firmware
             Programmer Model: STK500
             Hardware Version: 0
             Firmware Version Master : 2.00
             Topcard         : Unknown
[... 63 lines cut ...]
01c0  6b 20 62 72 6f 77 6e 20  66 6f 78 20 6a 75 6d 70  |k brown fox jump|
01d0  73 20 6f 76 65 72 20 74  68 65 20 6c 61 7a 79 20  |s over the lazy |
01e0  64 6f 67 0a 54 68 65 20  71 75 69 63 6b 20 62 72  |dog The quick br|
01f0  6f 77 6e 20 66 6f 78 20  6a 75 6d 70 73 20 6f 76  |own fox jumps ov|
0200  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0210  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0220  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0230  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0240  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0250  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0260  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0270  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0280  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0290  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
02a0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
02b0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
02c0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
02d0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
02e0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
02f0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|

avrdude>
avrdude.exe done.  Thank you.

```

mcuee, 2022-06-13:
WIth the same setup (one Arduino Uno clone with m16u2, one Arduino Uno clone with CH340) , somehow I can not get "Arduino as ISP", strange.

```
PS C:\work\avr\avrdude_test\avrdude-v7.0-windows-x64> .\avrdude.exe -c stk500v1 -p m328p -P COM23 -vv

avrdude.exe: Version 7.0
             Copyright (c) Brian Dean, http://www.bdmicro.com/
             Copyright (c) Joerg Wunsch

             System wide configuration file is "C:/work/avr/avrdude_test/avrdude-v7.0-windows-x64/avrdude.conf"

             Using Port                    : COM23
             Using Programmer              : stk500v1
avrdude.exe: ser_recv(): programmer is not responding
avrdude.exe: stk500_recv(): programmer is not responding
avrdude.exe: stk500_getsync() attempt 1 of 10: not in sync: resp=0xa0
avrdude.exe: stk500_getsync() attempt 2 of 10: not in sync: resp=0xe0
avrdude.exe: stk500_getsync() attempt 3 of 10: not in sync: resp=0xe0
avrdude.exe: stk500_getsync() attempt 4 of 10: not in sync: resp=0xe0
avrdude.exe: stk500_getsync() attempt 5 of 10: not in sync: resp=0x00
avrdude.exe: stk500_getsync() attempt 6 of 10: not in sync: resp=0xe0
avrdude.exe: stk500_getsync() attempt 7 of 10: not in sync: resp=0xe0
avrdude.exe: stk500_getsync() attempt 8 of 10: not in sync: resp=0xe0
avrdude.exe: stk500_getsync() attempt 9 of 10: not in sync: resp=0x00
avrdude.exe: stk500_getsync() attempt 10 of 10: not in sync: resp=0xe0
avrdude.exe: opening programmer "stk500v1" on port "COM23" failed

avrdude.exe done.  Thank you.
```

mcuee, 2022-06-13:
Okay I have to reduce the baud rate to 19200 for my setup to work. I can not reproduce the issue.

```
PS C:\work\avr\avrdude_test\avrdude-v7.0-windows-x64> .\avrdude.exe -p m328p -c avrisp -P COM6 -b 19200 -U eeprom:w:entest.eep:i -v

avrdude.exe: Version 7.0
             Copyright (c) Brian Dean, http://www.bdmicro.com/
             Copyright (c) Joerg Wunsch

             System wide configuration file is "C:/work/avr/avrdude_test/avrdude-v7.0-windows-x64/avrdude.conf"

             Using Port                    : COM6
             Using Programmer              : avrisp
             Overriding Baud Rate          : 19200
             AVR Part                      : ATmega328P
             Chip Erase delay              : 9000 us
             PAGEL                         : PD7
             BS2                           : PC2
             RESET disposition             : dedicated
             RETRY pulse                   : SCK
             Serial program mode           : yes
             Parallel program mode         : yes
             Timeout                       : 200
             StabDelay                     : 100
             CmdexeDelay                   : 25
             SyncLoops                     : 32
             PollIndex                     : 3
             PollValue                     : 0x53
             Memory Detail                 :

                                               Block Poll               Page                       Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               eeprom                 65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
               flash                  65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
               lfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
               hfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
               efuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
               lock                    0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
               calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00
               signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00

             Programmer Type : STK500
             Description     : Atmel AVR ISP
             Hardware Version: 2
             Firmware Version: 1.18
             Topcard         : Unknown
             Vtarget         : 0.0 V
             Varef           : 0.0 V
             Oscillator      : Off
             SCK period      : 0.1 us

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.02s

avrdude.exe: Device signature = 0x1e950f (probably m328p)
avrdude.exe: reading input file "entest.eep"
avrdude.exe: writing eeprom (512 bytes):

Writing | ################################################## | 100% 24.89s

avrdude.exe: 512 bytes of eeprom written
avrdude.exe: verifying eeprom memory against entest.eep:

Reading | ################################################## | 100% 1.66s

avrdude.exe: 512 bytes of eeprom verified

avrdude.exe done.  Thank you.
[... 40 lines cut ...]
01c0  6b 20 62 72 6f 77 6e 20  66 6f 78 20 6a 75 6d 70  |k brown fox jump|
01d0  73 20 6f 76 65 72 20 74  68 65 20 6c 61 7a 79 20  |s over the lazy |
01e0  64 6f 67 0a 54 68 65 20  71 75 69 63 6b 20 62 72  |dog The quick br|
01f0  6f 77 6e 20 66 6f 78 20  6a 75 6d 70 73 20 6f 76  |own fox jumps ov|
0200  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0210  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0220  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0230  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0240  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0250  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0260  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0270  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0280  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0290  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
02a0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
02b0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
02c0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
02d0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
02e0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
02f0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|

avrdude>
avrdude.exe done.  Thank you.

```

mcuee, 2022-06-14:
> > My bad, I did not realize there is a difference between ArduinoISP and 'Arduino as ISP'. I like to clarify that I'm using 'Arduino as ISP' programmer soo `avrdude -c stk500v1 ...` is correct
> 
> You may want to try `avrdude -c avrisp` instead if you use "Arduino as ISP".

@x0urc3 
Even with `stk500v1` it works for me as well.

```
PS C:\work\avr\avrdude_test\avrdude-v7.0-windows-x64> echo "read eeprom 0 0x300" | ./avrdude -c stk500v1 -P COM3 -b 19200 -p atmega328p -t

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.02s

avrdude.exe: Device signature = 0x1e950f (probably m328p)
avrdude> >>> read eeprom 0 0x300

Reading | ################################################## | 100% 5.10s

0000  54 68 65 20 71 75 69 63  6b 20 62 72 6f 77 6e 20  |The quick brown |
0010  66 6f 78 20 6a 75 6d 70  73 20 6f 76 65 72 20 74  |fox jumps over t|
0020  68 65 20 6c 61 7a 79 20  64 6f 67 0a 54 68 65 20  |he lazy dog The |
0030  71 75 69 63 6b 20 62 72  6f 77 6e 20 66 6f 78 20  |quick brown fox |
0040  6a 75 6d 70 73 20 6f 76  65 72 20 74 68 65 20 6c  |jumps over the l|
0050  61 7a 79 20 64 6f 67 0a  54 68 65 20 71 75 69 63  |azy dog The quic|
0060  6b 20 62 72 6f 77 6e 20  66 6f 78 20 6a 75 6d 70  |k brown fox jump|
0070  73 20 6f 76 65 72 20 74  68 65 20 6c 61 7a 79 20  |s over the lazy |
0080  64 6f 67 0a 54 68 65 20  71 75 69 63 6b 20 62 72  |dog The quick br|
0090  6f 77 6e 20 66 6f 78 20  6a 75 6d 70 73 20 6f 76  |own fox jumps ov|
00a0  65 72 20 74 68 65 20 6c  61 7a 79 20 64 6f 67 0a  |er the lazy dog |
00b0  54 68 65 20 71 75 69 63  6b 20 62 72 6f 77 6e 20  |The quick brown |
00c0  66 6f 78 20 6a 75 6d 70  73 20 6f 76 65 72 20 74  |fox jumps over t|
00d0  68 65 20 6c 61 7a 79 20  64 6f 67 0a 54 68 65 20  |he lazy dog The |
00e0  71 75 69 63 6b 20 62 72  6f 77 6e 20 66 6f 78 20  |quick brown fox |
00f0  6a 75 6d 70 73 20 6f 76  65 72 20 74 68 65 20 6c  |jumps over the l|
0100  61 7a 79 20 64 6f 67 0a  54 68 65 20 71 75 69 63  |azy dog The quic|
0110  6b 20 62 72 6f 77 6e 20  66 6f 78 20 6a 75 6d 70  |k brown fox jump|
0120  73 20 6f 76 65 72 20 74  68 65 20 6c 61 7a 79 20  |s over the lazy |
0130  64 6f 67 0a 54 68 65 20  71 75 69 63 6b 20 62 72  |dog The quick br|
0140  6f 77 6e 20 66 6f 78 20  6a 75 6d 70 73 20 6f 76  |own fox jumps ov|
0150  65 72 20 74 68 65 20 6c  61 7a 79 20 64 6f 67 0a  |er the lazy dog |
0160  54 68 65 20 71 75 69 63  6b 20 62 72 6f 77 6e 20  |The quick brown |
0170  66 6f 78 20 6a 75 6d 70  73 20 6f 76 65 72 20 74  |fox jumps over t|
0180  68 65 20 6c 61 7a 79 20  64 6f 67 0a 54 68 65 20  |he lazy dog The |
0190  71 75 69 63 6b 20 62 72  6f 77 6e 20 66 6f 78 20  |quick brown fox |
01a0  6a 75 6d 70 73 20 6f 76  65 72 20 74 68 65 20 6c  |jumps over the l|
01b0  61 7a 79 20 64 6f 67 0a  54 68 65 20 71 75 69 63  |azy dog The quic|
01c0  6b 20 62 72 6f 77 6e 20  66 6f 78 20 6a 75 6d 70  |k brown fox jump|
01d0  73 20 6f 76 65 72 20 74  68 65 20 6c 61 7a 79 20  |s over the lazy |
01e0  64 6f 67 0a 54 68 65 20  71 75 69 63 6b 20 62 72  |dog The quick br|
01f0  6f 77 6e 20 66 6f 78 20  6a 75 6d 70 73 20 6f 76  |own fox jumps ov|
0200  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0210  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0220  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0230  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0240  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0250  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0260  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0270  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0280  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0290  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
02a0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
02b0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
02c0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
02d0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
02e0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
02f0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|

avrdude>
avrdude.exe done.  Thank you.
```

mcuee, 2022-06-14:
@dl8dtl 
Just wondering what is the difference between stk500v1 and `avrisp`? Both seem to work with the "Arduino as ISP". I know that AVRISP lacks some features of the STK500.

BTW, Atmel document https://github.com/avrdudes/avrdude/blob/main/atmel-docs/STK500v2-AVR068.pdf seems to say that AVRISP is the same as STK500v2 in most cases except some unique features of the STK500v2. But then avrdude's `avrisp` apparently talks STK500v1 protocol.

Is it that the original AVRISP FW is compatible with STK500v1 and later FW is compatible with STK500v2?

x0urc3, 2022-06-14:
@mcuee as I mentioned in my first post, terminal mode (-t) works fine. I am having trouble when using ```-U eeprom:r:dump.hex:i```. Did not work for ```stk500v1``` or ```avrisp```

mcuee, 2022-06-14:
> @mcuee as I mentioned in my first post, terminal mode (-t) works fine. I am having trouble when using `-U eeprom:r:dump.hex:i`. Did not work for `stk500v1` or `avrisp`

No issue here as well.

C:\work\avr\binary\avrdude-v7.0-windows-x64> .\avrdude.exe -p m328p -c stk500v1 -P COM29 -b 19200 -U eeprom:w:entest.eep:i -v

avrdude.exe: Version 7.0
             Copyright (c) Brian Dean, http://www.bdmicro.com/
             Copyright (c) Joerg Wunsch

             System wide configuration file is "C:/work/avr/binary/avrdude-v7.0-windows-x64/avrdude.conf"

             Using Port                    : COM29
             Using Programmer              : stk500v1
             Overriding Baud Rate          : 19200
             AVR Part                      : ATmega328P
             Chip Erase delay              : 9000 us
             PAGEL                         : PD7
             BS2                           : PC2
             RESET disposition             : dedicated
             RETRY pulse                   : SCK
             Serial program mode           : yes
             Parallel program mode         : yes
             Timeout                       : 200
             StabDelay                     : 100
             CmdexeDelay                   : 25
             SyncLoops                     : 32
             PollIndex                     : 3
             PollValue                     : 0x53
             Memory Detail                 :

                                               Block Poll               Page                       Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               eeprom                 65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
               flash                  65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
               lfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
               hfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
               efuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
               lock                    0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
               calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00
               signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00

             Programmer Type : STK500
             Description     : Atmel STK500 Version 1.x firmware
             Hardware Version: 2
             Firmware Version: 1.18
             Topcard         : Unknown
             Vtarget         : 0.0 V
             Varef           : 0.0 V
             Oscillator      : Off
             SCK period      : 0.1 us

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.02s

avrdude.exe: Device signature = 0x1e950f (probably m328p)
avrdude.exe: reading input file "entest.eep"
avrdude.exe: writing eeprom (512 bytes):

Writing | ################################################## | 100% 24.82s

avrdude.exe: 512 bytes of eeprom written
avrdude.exe: verifying eeprom memory against entest.eep:

Reading | ################################################## | 100% 1.61s

avrdude.exe: 512 bytes of eeprom verified

[... 321 lines cut ...]
02b0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
02c0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
02d0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
02e0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
02f0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0300  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0310  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0320  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0330  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0340  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0350  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0360  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0370  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0380  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0390  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
03a0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
03b0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
03c0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
03d0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
03e0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
03f0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|

avrdude>
avrdude.exe done.  Thank you.
```

x0urc3, 2022-06-14:
I want to replicate @mcuee test case by writing and reading using -U argument, but my setup is not cooperating. 
```
./avrdude -c avrisp -p m328p -P /dev/ttyUSB0 -b 19200 -U eeprom:w:entest.epp:i -v

avrdude: Version 7.0-20220610 (cb11423)
...

Reading | ################################################## | 100% 2.76s

avrdude: verification error, first mismatch at byte 0x0000
         0x61 != 0x54
avrdude: verification error; content mismatch

avrdude done.  Thank you.
```

x0urc3, 2022-06-14:
BTW based on avrdude.conf both ```stk500v1``` and ```avrisp``` programmer are ```type = "stk500"``` soo I guess they are the same.

mcuee, 2022-06-14:
> I want to replicate @mcuee test case by writing and reading using -U argument, but my setup is not cooperating.

Just FYI for my setup.

Programmer: Arduino Uno with "Arduino as ISP" sketch.
Device: Arduino Uno

@x0urc3 Please give the "ScratchMonkey" sketch a try, it is emulating stk500v2 protocol and works better for me.

Ref: you can see my test results here, no need to patch avrdude.
* https://github.com/avrdudes/avrdude/issues/989

x0urc3, 2022-06-14:
1. I have tracked the cause of the bad EEPROM dump. I have used a verstion of the 'Arduino as ISP' from Adafruit [here](https://github.com/adafruit/Adafruit_Learning_System_Guides/tree/main/ArduinoISP) instead of the official Arduino [here](https://github.com/arduino/arduino-examples/tree/main/examples/11.ArduinoISP/ArduinoISP)
2. The Adafruit 'Arduino as ISP' does NOT properly handle word-based address when reading/writing eeprom causing it to repeat at 256 boundary.
3. This resolves this issue. 

Thanks @mcuee for the feedback and suggestion

closed 2022-06-15 by MCUdude