avrdudes/avrdude #991: avrdude with Arduino Nano Every: conf file with Arduino vs conf file from avrdude

CiaranW5, 2022-06-13:
I am programming (successfully) an Atmega4809 in Arduino Nano Every using Microchip Studio external tools (Windows 10 or 11). 
I copied avrdude and the conf file from an Arduno IDE download, and the conf file contains a programmer entry 'jtag2updi'. This is reported  as Version 6.3.
If I download Ver 6.3 (also Ver 6.4) from the AVRDUDE github then there is no entry for jtag2updi. Why is this?
Also should I use jtagmkii_pdi in the github downloaded versions in place of 'jtag2updi'?
Thanks,
Ciaran

mcuee, 2022-06-13:
Just use the official avrdude 7.0 release here and it will be okay. You need to use a trick to simulate what Arduino is doing.

Please refer to my run log here:
https://github.com/avrdudes/avrdude/issues/980

mcuee, 2022-06-13:
jtag2updi is only added to the official avrdude repo on 29 Dec 2021. Therefore it will not be available for avrdude 6.3 and 6.4 release.
https://github.com/avrdudes/avrdude/commit/0a335e2f73ac24d14ab4df731712346aa0922883

Arduino probbaly has their own patches for there 6.3 version.

CiaranW5, 2022-06-13:
1. Yes, it looks like the avrdude.conf in the Arduino download has this patch.
2. I have just downloaded avrdude version 7.0 and I have programmed my Nano Every M4809 successfully using it (also using the 1200bps workaround).
Thanks for the prompt reply. I prefer to use this release version, Version 7.0. Looks good.

mcuee, 2022-06-13:
@CiaranW5 In this case, please close this issue. Thanks.

CiaranW5, 2022-06-13:
Thanks - issue resolved now.

closed 2022-06-13 by CiaranW5

mirh, 2023-05-14:
https://github.com/facchinm/avrdude/commit/46a9c3bfec913314a72865020c23aac2447f09a5 this was it