avrdudes/avrdude #992: USBtinyISP and 256KB Flash Parts: enhancement possible from avrdude side?

mcuee, 2022-06-15:
I just got an USBtinyISP and I found out from the official documentation that it can not program parts with 128KB/256KB parts.

https://learn.adafruit.com/usbtinyisp
> Works with any AVR ISP chip with 64K of flash (or less) - does not work with Atmega1281/1280/2561/2560

But @westfw seems to indicate that it is possible by using `raw SPI commands`. Is this true?
https://forums.adafruit.com/viewtopic.php?f=24&t=26091

He has put a reference here -- people use the trick to program 128KB flash parts.
https://forums.adafruit.com/viewtopic.php?f=20&t=24409

dl8dtl, 2022-06-15:
Well, the first question is whether there will be any firmware fix.

Since usbtiny.c has its own implementation, a hack could be for any device with more than 64 KiB of flash, to see whether all the data that have been read from above 64 KiB are identical to those from the lower addresses, and if so, implement a "soft hack" to perform what has been described by @WestfW . It's quite hackish though.

mcuee, 2022-06-16:
@dl8dtl I see quite some alternative FW for USBASP, but not really anything on USBtinyISP. So I believe there is unlikely any firmware fix any time soon. And it seems to me USBASP is much more popular. In fact I bought quite at least 4 USBASP to try out different FW.

mcuee, 2022-06-16:
@MCUdude Have you tried big hex file which cross the 64KB and 128KB size?

MCUdude, 2022-06-16:
@mcuee I can du that tomorrow!

mcuee, 2022-06-16:
This is what I get.

```
PS C:\work\avr\avrdude_test\avrdude-v7.0-windows-x64> .\avrdude.exe -c usbtiny -p m2560 -v -D -U flash:w:.\atmega2560_0x11000_offset.hex:i

avrdude.exe: Version 7.0
             Copyright (c) Brian Dean, http://www.bdmicro.com/
             Copyright (c) Joerg Wunsch

             System wide configuration file is "C:/work/avr/avrdude_test/avrdude-v7.0-windows-x64/avrdude.conf"

             Using Port                    : usb
             Using Programmer              : usbtiny
avrdude.exe: usbdev_open(): Found USBtinyISP, bus:device: bus-0:\\?\usb#vid_1781&pid_0c9f#6&8694159&0&3#{a5dcbf10-6530-11d2-901f-00c04fb951ed}--libusb0
             AVR Part                      : ATmega2560
             Chip Erase delay              : 9000 us
             PAGEL                         : PD7
             BS2                           : PA0
             RESET disposition             : dedicated
             RETRY pulse                   : SCK
             Serial program mode           : yes
             Parallel program mode         : yes
             Timeout                       : 200
             StabDelay                     : 100
             CmdexeDelay                   : 25
             SyncLoops                     : 32
             PollIndex                     : 3
             PollValue                     : 0x53
             Memory Detail                 :

                                               Block Poll               Page                       Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               eeprom                 65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
               flash                  65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
               lfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               hfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               efuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               lock                    0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00
               signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00

             Programmer Type : USBtiny
             Description     : USBtiny simple USB programmer, https://learn.adafruit.com/usbtinyisp
avrdude.exe: programmer operation not supported

avrdude.exe: Using SCK period of 10 usec
avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude.exe: Device signature = 0x1e9801 (probably m2560)
avrdude.exe: reading input file ".\atmega2560_0x11000_offset.hex"
avrdude.exe: writing flash (261414 bytes):

Writing | ################################################## | 100% 0.00s

avrdude.exe: 261414 bytes of flash written
avrdude.exe: verifying flash memory against .\atmega2560_0x11000_offset.hex:

Reading | ################################################## | 100% 0.00s

avrdude.exe: verification error, first mismatch at byte 0x11000
             0xff != 0x54
avrdude.exe: verification error; content mismatch

avrdude.exe done.  Thank you.

```

Then I verified with AVRDragon and it still shows error.
```
PS C:\work\avr\avrdude_test\avrdude-v7.0-windows-x64> .\avrdude.exe -c dragon_isp -p m2560 -U flash:v:.\atmega2560_0x11000_offset.hex:i

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.15s

avrdude.exe: Device signature = 0x1e9801 (probably m2560)
avrdude.exe: verifying flash memory against .\atmega2560_0x11000_offset.hex:

Reading | ################################################## | 100% 0.01s

avrdude.exe: verification error, first mismatch at byte 0x3fc00
             0x00 != 0x80
avrdude.exe: verification error; content mismatch

avrdude.exe done.  Thank you.
```

BTW, AVRDragon itself is okay.

```
PS C:\work\avr\avrdude_test\avrdude-v7.0-windows-x64>
                                                      .\avrdude.exe -c dragon_isp -p m2560 -U flash:w:.\atmega2560_0x11000_offset.hex:i

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.15s

avrdude.exe: Device signature = 0x1e9801 (probably m2560)
avrdude.exe: NOTE: "flash" memory has been specified, an erase cycle will be performed
             To disable this feature, specify the -D option.
avrdude.exe: erasing chip
avrdude.exe: reading input file ".\atmega2560_0x11000_offset.hex"
avrdude.exe: writing flash (261414 bytes):

Writing | ################################################## | 100% 0.01s

avrdude.exe: 261414 bytes of flash written
avrdude.exe: verifying flash memory against .\atmega2560_0x11000_offset.hex:

Reading | ################################################## | 100% 0.01s

avrdude.exe: 261414 bytes of flash verified

avrdude.exe done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude-v7.0-windows-x64> .\avrdude.exe -c dragon_isp -p m2560 -U flash:v:.\atmega2560_0x11000_offset.hex:i

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.15s

avrdude.exe: Device signature = 0x1e9801 (probably m2560)
avrdude.exe: verifying flash memory against .\atmega2560_0x11000_offset.hex:

Reading | ################################################## | 100% 0.01s

avrdude.exe: 261414 bytes of flash verified

avrdude.exe done.  Thank you.
```

mcuee, 2022-06-17:
The above hex file is not valid as it exceeded 256KB length.

mcuee, 2022-06-17:
I will close this issue as I do not think there is a need to the hackish solution. And the work-around is simple for people with 256KB flash devices -- that is to use usbasp or other programmer.

mcuee, 2022-06-17:
@MCUdude Interesting that your test hex file works with usbisptiny for write indeed and I can use an usbasp to verify,

```
PS C:\work\avr\avrdude_test\avrdude-v7.0-windows-x64> .\avrdude.exe -p m2560 -c usbtiny -v -U .\sketch_jun17a.ino.hex -B1

avrdude.exe: Version 7.0
             Copyright (c) Brian Dean, http://www.bdmicro.com/
             Copyright (c) Joerg Wunsch

             System wide configuration file is "C:/work/avr/avrdude_test/avrdude-v7.0-windows-x64/avrdude.conf"

             Using Port                    : usb
             Using Programmer              : usbtiny
             Setting bit clk period        : 1.0
avrdude.exe: usbdev_open(): Found USBtinyISP, bus:device: bus-0:\\?\usb#vid_1781&pid_0c9f#6&8694159&0&1#{a5dcbf10-6530-11d2-901f-00c04fb951ed}--libusb0
             AVR Part                      : ATmega2560
             Chip Erase delay              : 9000 us
             PAGEL                         : PD7
             BS2                           : PA0
             RESET disposition             : dedicated
             RETRY pulse                   : SCK
             Serial program mode           : yes
             Parallel program mode         : yes
             Timeout                       : 200
             StabDelay                     : 100
             CmdexeDelay                   : 25
             SyncLoops                     : 32
             PollIndex                     : 3
             PollValue                     : 0x53
             Memory Detail                 :

                                               Block Poll               Page                       Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               eeprom                 65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
               flash                  65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
               lfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               hfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               efuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               lock                    0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00
               signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00

             Programmer Type : USBtiny
             Description     : USBtiny simple USB programmer, https://learn.adafruit.com/usbtinyisp
avrdude.exe: programmer operation not supported

avrdude.exe: Setting SCK period to 1 usec
avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude.exe: Device signature = 0x1e9801 (probably m2560)
avrdude.exe: NOTE: "flash" memory has been specified, an erase cycle will be performed
             To disable this feature, specify the -D option.
avrdude.exe: erasing chip
avrdude.exe: Setting SCK period to 1 usec
avrdude.exe: reading input file ".\sketch_jun17a.ino.hex"
avrdude.exe: input file .\sketch_jun17a.ino.hex auto detected as Intel Hex
avrdude.exe: writing flash (176758 bytes):

Writing | ################################################## | 100% 50.18s

avrdude.exe: 176758 bytes of flash written
avrdude.exe: verifying flash memory against .\sketch_jun17a.ino.hex:
avrdude.exe: input file .\sketch_jun17a.ino.hex auto detected as Intel Hex

Reading | ################################################## | 100% 46.48s

avrdude.exe: verification error, first mismatch at byte 0x0000
[... 37 lines cut ...]
               hfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               efuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               lock                    0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00
               signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00

             Programmer Type : usbasp
             Description     : USBasp, http://www.fischl.de/usbasp/

avrdude.exe: auto set sck period (because given equals null)
avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude.exe: Device signature = 0x1e9801 (probably m2560)
avrdude.exe: verifying flash memory against .\sketch_jun17a.ino.hex:
avrdude.exe: input file .\sketch_jun17a.ino.hex auto detected as Intel Hex

Reading | ################################################## | 100% 27.45s

avrdude.exe: 176758 bytes of flash verified

avrdude.exe done.  Thank you.

```

mcuee, 2022-06-17:
And redo the test shows that it also works with my test hex file even though it can not verify properly. Again I am able to verify with the usbasp.

```
PS C:\work\avr\avrdude_test\avrdude-v7.0-windows-x64> .\avrdude.exe -p m2560 -c usbtiny -v -U .\eepromtest_intel.hex -v

avrdude.exe: Version 7.0
             Copyright (c) Brian Dean, http://www.bdmicro.com/
             Copyright (c) Joerg Wunsch

             System wide configuration file is "C:/work/avr/avrdude_test/avrdude-v7.0-windows-x64/avrdude.conf"

avrdude.exe: defaulting memtype in -U w:.\eepromtest_intel.hex option to "flash"
avrdude.exe: defaulting memtype in -U v:.\eepromtest_intel.hex option to "flash"
             Using Port                    : usb
             Using Programmer              : usbtiny
avrdude.exe: usbdev_open(): Found USBtinyISP, bus:device: bus-0:\\?\usb#vid_1781&pid_0c9f#6&8694159&0&1#{a5dcbf10-6530-11d2-901f-00c04fb951ed}--libusb0
             AVR Part                      : ATmega2560
             Chip Erase delay              : 9000 us
             PAGEL                         : PD7
             BS2                           : PA0
             RESET disposition             : dedicated
             RETRY pulse                   : SCK
             Serial program mode           : yes
             Parallel program mode         : yes
             Timeout                       : 200
             StabDelay                     : 100
             CmdexeDelay                   : 25
             SyncLoops                     : 32
             PollIndex                     : 3
             PollValue                     : 0x53
             Memory Detail                 :

                                               Block Poll               Page                       Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               eeprom                 65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
               flash                  65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
               lfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               hfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               efuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               lock                    0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00
               signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00

             Programmer Type : USBtiny
             Description     : USBtiny simple USB programmer, https://learn.adafruit.com/usbtinyisp
avrdude.exe: programmer operation not supported

avrdude.exe: Using SCK period of 10 usec
CMD: [ac 53 00 00] [2c 52 53 00]
avrdude.exe: AVR device initialized and ready to accept instructions

Reading |                                                    | 0% 0.00sCMD: [30 00 00 00] [00 30 00 1e]
CMD: [30 00 01 00] [00 30 00 98]
Reading | #################                                  | 33% 0.00sCMD: [30 00 02 00] [00 30 00 01]
Reading | ################################################## | 100% 0.01s

avrdude.exe: Device signature = 0x1e9801 (probably m2560)
avrdude.exe: NOTE: "flash" memory has been specified, an erase cycle will be performed
             To disable this feature, specify the -D option.
avrdude.exe: erasing chip
CMD: [ac 80 00 00] [00 ac 80 00]
avrdude.exe: Using SCK period of 10 usec
CMD: [ac 53 00 00] [00 ac 53 00]
avrdude.exe: reading input file ".\eepromtest_intel.hex"
avrdude.exe: input file .\eepromtest_intel.hex auto detected as Intel Hex
avrdude.exe: writing flash (70144 bytes):

Writing | ################################################## | 100% 0.00s

[... 51 lines cut ...]
               lfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               hfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               efuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               lock                    0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00
               signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00

             Programmer Type : usbasp
             Description     : USBasp, http://www.fischl.de/usbasp/

avrdude.exe: auto set sck period (because given equals null)
avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude.exe: Device signature = 0x1e9801 (probably m2560)
avrdude.exe: verifying flash memory against .\eepromtest_intel.hex:

Reading | ################################################## | 100% 0.00s

avrdude.exe: 70144 bytes of flash verified

avrdude.exe done.  Thank you.

```

mcuee, 2022-06-17:
Now it is interesting, why does the flash write works but not read?

MCUdude, 2022-06-17:
> Now it is interesting, why does the flash write work but not read?

I have no idea, but I'll have to admit that my [neat, ATtiny44 based USBtiny](https://www.ebay.com/itm/311566698886) programmer with a 2x3 female header soldered on the underside has been my go-to programmer for several years.

It's interesting to look at the USBtiny read. If you look at the data from address 0x0000 in the hex file, it's obvious that it didn't really start reading from address 0x0000, but instead from the 0x10000 or 0x20000 mark. Not sure if this can be dealt with by Avrdude or not.

mcuee, 2022-06-17:
@MCUdude 
Haha, I also bought the ATtiny44 based USBtinyISP programmer and it will reach me soon.

WestfW, 2022-06-19:
> why does the flash write work but not read?
One possibility:
the extension to support 256k parts in optiboot/stk500v1 essentially sends an "out of band" command to set the high address byte when programming high memory.
It's not clear that anything RESETS that address byte when the programmer switches back to read mode for the "verify" part.

mcuee, 2022-06-20:
> why does the flash write work but not read?

So you are saying that USBtinyISP is doing something like this "out of band" command to set the high address byte when programming high memory as well for write, and we need to see how to implement the same for read, right?

FT232R/FT245R seem to have the same problem as per #995.
* #995 

Bus Pirate may have issues with both write and read, but probably "out of band" command can be implemented as well.
* #459

mcuee, 2022-07-14:
Hacky patch for 128KB Flash read:
https://github.com/cwyse/avrdude-build-usbtiny/blob/master/avrdude-6.3-patches/0001-Support-slow-flash-128KB-read-via-USBTinyISP.patch

MCUdude, 2022-07-14:
> Hacky patch for 128KB Flash read: https://github.com/cwyse/avrdude-build-usbtiny/blob/master/avrdude-6.3-patches/0001-Support-slow-flash-128KB-read-via-USBTinyISP.patch

@mcuee thanks for digging this one up! I'll give it a try and report back later tonight.

MCUdude, 2022-07-14:
The patch _did_ actually work, but reading was painfully slow:

With patch:
```
$ ./avrdude -cusbtiny -patmega2560 -Uflash:w:/Users/hans/Downloads/avrdude/src/chaucer256k.hex:i -B1

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9801 (probably m2560)
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file "/Users/hans/Downloads/avrdude/src/chaucer256k.hex"
avrdude: writing flash (176758 bytes):

Writing | ################################################## | 100% 66.07s

avrdude: 176758 bytes of flash written
avrdude: verifying flash memory against /Users/hans/Downloads/avrdude/src/chaucer256k.hex:

Reading | ################################################## | 100% 304.57s

avrdude: 176758 bytes of flash verified

avrdude done.  Thank you.
```

Without patch:
```
$ ./avrdude -cusbtiny -patmega2560 -Uflash:w:/Users/hans/Downloads/avrdude/src/chaucer256k.hex:i -B1

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9801 (probably m2560)
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file "/Users/hans/Downloads/avrdude/src/chaucer256k.hex"
avrdude: writing flash (176758 bytes):

Writing | ################################################## | 100% 67.83s

avrdude: 176758 bytes of flash written
avrdude: verifying flash memory against /Users/hans/Downloads/avrdude/src/chaucer256k.hex:

Reading | ################################################## | 100% 46.61s

avrdude: verification error, first mismatch at byte 0x0000
         0x2c != 0x0d
avrdude: verification error; content mismatch

avrdude done.  Thank you.
```

mcuee, 2022-07-15:
> The patch did actually work, but reading was painfully slow:

Good to know it actually works. Haha, the patch is called `[Support-slow-flash-128KB-read-via-USBTinyISP.patch` for a reason. 

Hopefully someone can come out with a proper patch.

mcuee, 2022-07-15:
@MCUdude 
Indeed the patch works fine.

With the patch:

```
PS C:\work\avr\avrdude_test\avrdude-7.0_bin64> .\avrdude_git_issue443_issue992.exe -c usbtiny -qqp m2560 -U .\hex\blink-mega2560_lext-test.hex && echo OK
OK
```

Without the patch:
```
PS C:\work\avr\avrdude_test\avrdude-7.0_bin64> .\avrdude_pr1025v3.exe -c usbtiny -qqp m2560 -U .\hex\blink-mega2560_lext-test.hex && echo OK
avrdude_pr1025v3.exe: verification error, first mismatch at byte 0x0000
                      0xff != 0x76
avrdude_pr1025v3.exe: verification error; content mismatch

```

mcuee, 2022-07-15:
@MCUdude and @stefanrueger 
Is it possible to enable the patch only for 128KB/256KB flash parts and not the rest. Then it can still be a good enhancement.

MCUdude, 2022-07-15:
> Is it possible to enable the patch only for 128KB/256KB flash parts and not the rest. Then it can still be a good enhancement.

I'm not sure it's worth it, because readback is so slow it's practically useless for impatient people like me. It took a whopping 10 minutes to verify a 176kB program. IMO it's better to just accept the fact that the program _can_ be written correctly to flash, but not be read, at least without a proper fix.

> It is just pot luck the current mainline does not seem to want to play with the ATmega2560 on my usbtiny

I've been using the usbtiny programmer with the Avrdude 6.3 version bundled with Arduino IDE, and it behaves exactly like with Avrdude 7.0. I can write to _any_ AVR with an ISP interface, but I can only verify the flash memory on <= 64kB parts.

mcuee, 2022-07-16:
> @mcuee Can you confirm the current mainline avrdude (not the 6.3 version just tested) works with ATmega2560, and more importantly that commenting out the paged read in usbtiny.c is a quick fix, in principle.

Yes I can confirm. I am using the latest git plus the slow quick slow usbtinyisp patch plus fix for Issue 443 (for JTAG ICE 1).

```
PS C:\work\avr\avrdude_test\avrdude> git diff >mypatch.txt
PS C:\work\avr\avrdude_test\avrdude> cat .\mypatch.txt
diff --git a/src/jtagmkI.c b/src/jtagmkI.c
index 7a97217..19f874c 100644
--- a/src/jtagmkI.c
+++ b/src/jtagmkI.c
@@ -1058,7 +1058,7 @@ static int jtagmkI_write_byte(PROGRAMMER * pgm, AVRPART * p, AVRMEM * mem,
 {
   unsigned char cmd[6], datacmd[1 * 2 + 1];
   unsigned char resp[1], writedata;
-  int len, need_progmode = 1;
+  int len, need_progmode = 1, need_dummy_read = 0;

   avrdude_message(MSG_NOTICE2, "%s: jtagmkI_write_byte(.., %s, 0x%lx, ...)\n",
            progname, mem->desc, addr);
@@ -1075,17 +1075,22 @@ static int jtagmkI_write_byte(PROGRAMMER * pgm, AVRPART * p, AVRMEM * mem,
     PDATA(pgm)->eeprom_pageaddr = (unsigned long)-1L;
   } else if (strcmp(mem->desc, "lfuse") == 0) {
     cmd[1] = MTYPE_FUSE_BITS;
+       need_dummy_read = 1;
     addr = 0;
   } else if (strcmp(mem->desc, "hfuse") == 0) {
     cmd[1] = MTYPE_FUSE_BITS;
+       need_dummy_read = 1;
     addr = 1;
   } else if (strcmp(mem->desc, "efuse") == 0) {
     cmd[1] = MTYPE_FUSE_BITS;
+       need_dummy_read = 1;
     addr = 2;
   } else if (strcmp(mem->desc, "lock") == 0) {
     cmd[1] = MTYPE_LOCK_BITS;
+       need_dummy_read = 1;
   } else if (strcmp(mem->desc, "calibration") == 0) {
     cmd[1] = MTYPE_OSCCAL_BYTE;
+       need_dummy_read = 1;
   } else if (strcmp(mem->desc, "signature") == 0) {
     cmd[1] = MTYPE_SIGN_JTAG;
   }
@@ -1154,6 +1159,9 @@ static int jtagmkI_write_byte(PROGRAMMER * pgm, AVRPART * p, AVRMEM * mem,
       avrdude_message(MSG_NOTICE2, "OK\n");
   }

+  if(need_dummy_read)
+    jtagmkI_recv(pgm, resp, 1);
+
   return 0;
 }

diff --git a/src/usbtiny.c b/src/usbtiny.c
index 4bbaca7..d95fe0d 100644
--- a/src/usbtiny.c
+++ b/src/usbtiny.c
@@ -789,7 +789,7 @@ void usbtiny_initpgm ( PROGRAMMER* pgm )
   /* Optional Functions */
   pgm->powerup         = NULL;
   pgm->powerdown       = usbtiny_powerdown;
-  pgm->paged_load      = usbtiny_paged_load;
+  //pgm->paged_load    = usbtiny_paged_load;
   pgm->paged_write     = usbtiny_paged_write;
   pgm->set_sck_period  = usbtiny_set_sck_period;
   pgm->setup            = usbtiny_setup;
```

mcuee, 2022-07-16:
> Other than that, the avrdude v 6.3 patch seems to suggest, when commenting paged read it all works meaning byte read works, which sends the high byte of the 24-bit word address for _every_ byte read, so 256k (minus 2) unnecessary load extended address commands sent to the device for a full device readout. I still think the caching of the high address byte should be centralised and moved out of all 20 or so programmers and be shared with the generic avr_...() access functions.
> 
> So, in the short term it should be possible to implant the load extended address commands that avr_read_byte_default() carries out into usbtiny.c's paged read. I'll be sufficient to do that, without caching, once at the start of every page read. That's still 1024-2 too many load extended address calls, but there you go.

@stefanrueger 
That will be great. Other than usbasp, I am guessing that USBtinyISP is the second most popular programmer for avrdude users. So if you can fix the issue, it will be a great enhancement for avrdude.

mcuee, 2022-07-16:
> I'm not sure it's worth it, because readback is so slow it's practically useless for impatient people like me. It took a whopping 10 minutes to verify a 176kB program. IMO it's better to just accept [...]

Okay, I agree with you. Hopefully @stefanrueger can come out with a more proper fix.

mcuee, 2022-07-16:
> It is just pot luck the current mainline does not seem to want to play with the ATmega2560 on my usbtiny:
>
> I do get the same problem occasionally with an ATmega328P board, but with the ATmega2560 I get that all the time.
> 

Interesting. I think I occassionally have the same issue but unplug and plug back in the USBtinyISP will usually fix the issue. You may want to let the USBtinyISP power the target board to see if that helps.

My hypothesis for the issue is that the programmer is in a bad state in this case ( not so sure if it is related to the V-USB code base used).

As mentioned by @MCUdude, you may want to try to update the FW. I have not done that yet for my USBtinyISP units (I got two based on ATtiny2313 and one based on ATtiny44). 

@stefanrueger I think you may want to get a usbasp clone (make sure you buy the ATmega8/8A based version with jumpers for FW upgrade). It is probably the most popular programmer for avrdude users. Now I have 4 working usbasp units (three based on ATmeag8A and one based on ATmega88). Then I got two usbisp units based on ATmega88 which can only use the Chinese USB HID FW (progisp host application) but not avrdude..

In the case of usbasp, I have always upgraded my usbasp to a new version, usually the one here.
https://github.com/dioannidis/usbasp

stefanrueger, 2022-07-16:
If usbtiny.c cannot properly read/verify flash above **64 kB** (as @MCUdude found out) that points to an self-inflicted protocol error for usbtiny's approach to tell the physical programmer's firmware what to read (or a plain firmware error). The (good) idea of behind usbtiny's way of paged read means they send one command for a chunk read and just receive the result (rather than sending SPI commands for each byte), but that doesn't work above 64 kB. There is the independent problem for devices with flash above **128 kB** as they need the load extended address byte set. We know from the old patch that @mcuee found that byte read works. That uses one SPI command for each byte on small flash, and two SPI commands for flash > 128 kB.

I hope PR #1029 fixes both (independent) problems.

stefanrueger, 2022-07-16:
BTW, I tested/simulated the PR on the m328p, and am reasonably confident it works. On my usbtiny it takes paged read 1 s to access 1 kB below 64 kB, and 1.5 s to access 1 kB above 64 kB (simulated). The slow patch would be 3 s per kB everywhere for parts > 128 kB. 

A full ATmega2560 read should be just shy of 13 minutes on the slow old patch, a bit less than 6 minutes on the current PR and a bit over 4 minutes if usbtiny didn't have the protocol/firmware bug.

mcuee, 2022-07-16:
@stefanrueger 

Great. This works well and the time is roughly in line with your estimation for the full read in my case (455 seconds).

```
PS C:\work\avr\avrdude_test\avrdude-7.0_bin64> .\avrdude_pr1029.exe -c usbtiny -qqp m2560 
-U .\hex\blink-mega2560_lext-test.hex && echo OK
OK

PS C:\work\avr\avrdude_test\avrdude-7.0_bin64> .\avrdude_pr1029.exe -c usbtiny -p m2560 
-U .\hex\blink-mega2560_lext-test.hex

avrdude_pr1029.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.02s

avrdude_pr1029.exe: Device signature = 0x1e9801 (probably m2560)
avrdude_pr1029.exe: NOTE: "flash" memory has been specified, an erase cycle will be performed
                    To disable this feature, specify the -D option.
avrdude_pr1029.exe: erasing chip
avrdude_pr1029.exe: reading input file ".\hex\blink-mega2560_lext-test.hex"
avrdude_pr1029.exe: input file .\hex\blink-mega2560_lext-test.hex auto detected as Intel Hex
avrdude_pr1029.exe: writing flash (200814 bytes):

Writing | ################################################## | 100% 0.94s

avrdude_pr1029.exe: 200814 bytes of flash written
avrdude_pr1029.exe: verifying flash memory against .\hex\blink-mega2560_lext-test.hex:
avrdude_pr1029.exe: input file .\hex\blink-mega2560_lext-test.hex auto detected as Intel Hex

Reading | ################################################## | 100% 0.92s

avrdude_pr1029.exe: 200814 bytes of flash verified

avrdude_pr1029.exe done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude-7.0_bin64> .\avrdude_pr1029.exe -c usbtiny -p m2560 
-U flash:r:m2560_readback_pr1029.hex:i

avrdude_pr1029.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude_pr1029.exe: Device signature = 0x1e9801 (probably m2560)
avrdude_pr1029.exe: reading flash memory:

Reading | ################################################## | 100% 455.12s

avrdude_pr1029.exe: writing output file "m2560_readback_pr1029.hex"

avrdude_pr1029.exe done.  Thank you.

```

mcuee, 2022-07-16:
Under Ubuntu 20.04 Linux (dual boot, same Acer Swift 3 lapop as the Windows 11 OS) it is a bit faster  without `-B1` (355.65 seconds). With `-B1` it is much faster (186.5 seconds).

```
mcuee@UbuntuSwift3:~/build/avr/avrdude_sr/bin$ ./avrdude_pr1029 -c usbtiny -p m2560 -B1 -U flash:r:m2560_readback_pr1029.hex:i

avrdude_pr1029: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude_pr1029: Device signature = 0x1e9801 (probably m2560)
avrdude_pr1029: reading flash memory:

Reading | ################################################## | 100% 186.50s

avrdude_pr1029: writing output file "m2560_readback_pr1029.hex"

avrdude_pr1029 done.  Thank you.

mcuee@UbuntuSwift3:~/build/avr/avrdude_sr/bin$ ./avrdude_pr1029 -c usbtiny -p m2560 -U flash:r:m2560_readback_pr1029.hex_no_B1:i

avrdude_pr1029: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude_pr1029: Device signature = 0x1e9801 (probably m2560)
avrdude_pr1029: reading flash memory:

Reading | ################################################## | 100% 355.65s

avrdude_pr1029: writing output file "m2560_readback_pr1029.hex_no_B1"

avrdude_pr1029 done.  Thank you.

```

closed 2022-07-18 by stefanrueger

MCUdude, 2022-07-19:
> Then I got two usbisp units based on ATmega88 which can only use the Chinese USB HID FW (progisp host application) but not avrdude..

@mcuee I have a bunch of these as well, and can easily be re-flashed with USBasp firmware. They appear to be more rugged than "regular" USBasp's, since these have an aluminum shell. More info here: https://github.com/MCUdude/usbisp

mcuee, 2022-07-19:
> @mcuee I have a bunch of these as well, and can easily be re-flashed with USBasp firmware. They appear to be more rugged than "regular" USBasp's, since these have an aluminum shell. More info here: [...]

Thanks. I will give this a try. Maybe the fuse is the issue. Previously I did not change the fuse settings and use the following firmware.
https://github.com/dioannidis/usbasp/blob/master/firmware

mcuee, 2022-07-20:
@MCUdude 
Great, your hex file works for my usbisp clone. I have just converted one of them as usbasp. Thanks. I will still keep the remaining one as the Chinese HID FW (to work with ProgISP).

MCUdude, 2022-07-20:
@mcuee there's an even better firmware for the USBasp and its clones here (including the USBISP hardware):

https://github.com/nerdralph/usbasp

It has a maximum ISP clock speed of 1.5 MHz, so it's about twice as fast as the default USBasp firmware. It can also figure out the maximum clock speed the target can handle, so there's no need to use the `-B` anymore!

mcuee, 2022-07-20:
> @mcuee there's an even better firmware for the USBasp and its clones here (including the USBISP hardware):
>
> https://github.com/nerdralph/usbasp
>
> It has a maximum ISP clock speed of 1.5 MHz, so it's [...]

Yes that also works. Thanks.

mcuee, 2022-07-20:
@MCUdude 
BTW, there is a version 6 which uses ATmega88p.
* https://github.com/aleh/usbisp/issues/8

Version 5 is using ATmega88.
https://github.com/aleh/usbisp/issues/6

mcuee, 2022-07-22:
@MCUdude 
Just an update, the FW from @dioannidis actually works. I my have used the wrong image and one of my ATmega88 clone has HW issues.
Latest FW: https://github.com/dioannidis/usbasp/blob/master/firmware/m88.hex
Issue has been closed -- it is a false alarm.
* https://github.com/dioannidis/usbasp/issues/11

dioannidis, 2022-07-22:
@MCUdude
> @mcuee there's an even better firmware for the USBasp and its clones here (including the USBISP hardware):
> 
> https://github.com/nerdralph/usbasp

My firmware is actually the Ralph's firmware with the addition of ms os 2.0 descriptors and UART HID support ( 9600 max ). 

Ralph didn't want to add the HID UART feature because it needs more than 4KB SRAM and also [his opinion is that is a novelty](https://github.com/nerdralph/usbasp/issues/9#issuecomment-860280113).

I'll continue to track any Ralph's changes and merge them to my repo of course.

stefanrueger, 2022-08-31:
| Interesting. I think I occassionally have the same issue but unplug and plug back in the USBtinyISP will usually fix the issue. You may want to let the USBtinyISP power the target board to see if that helps.

Fun fact: you were absolutely spot on @mcuee! I closely checked the ISP header pins and they had a coat of oxides on them preventing good contact. I only used the ISP header pins with my ATmega2560 board (my other boards do not have a proper ISP header). After a thorough clean the USBtiny works like a charm...

PR #1029 (2022-07-18): Fix usbtiny read/verify for parts with more than 64 kB flash
Usbtiny has a protocol or firmware problem that prevents it from reading flash above 64 kB in page mode (used by `-U flash:r:...` and `-U flash:v:...`). This PR fixes that problem by falling back on byte access for flash paged reads above 64k. It also issues the correct load extended address command for parts with more than 128 kB flash thus extending support to ATmega2560 et al.
For details see issue #992

Resolves #992