avrdudes/avrdude #994: ATmega2560 wiring bootloader flash writing issue

mcuee, 2022-06-16:
Using the test hex file and it seems that the wiring bootloader has the same issue as the FT2232H based programmer.

```
PS C:\work\avr\avrdude_test\avrdude-v7.0-windows-x64> .\avrdude.exe -c wiring -P COM3 -p m2560 -D -U flash:w:.\atmega2560_0x11000_offset.hex:i -v

avrdude.exe: Version 7.0
             Copyright (c) Brian Dean, http://www.bdmicro.com/
             Copyright (c) Joerg Wunsch

             System wide configuration file is "C:/work/avr/avrdude_test/avrdude-v7.0-windows-x64/avrdude.conf"

             Using Port                    : COM3
             Using Programmer              : wiring
avrdude.exe: stk500v2_ReceiveMessage(): timeout
             AVR Part                      : ATmega2560
             Chip Erase delay              : 9000 us
             PAGEL                         : PD7
             BS2                           : PA0
             RESET disposition             : dedicated
             RETRY pulse                   : SCK
             Serial program mode           : yes
             Parallel program mode         : yes
             Timeout                       : 200
             StabDelay                     : 100
             CmdexeDelay                   : 25
             SyncLoops                     : 32
             PollIndex                     : 3
             PollValue                     : 0x53
             Memory Detail                 :

                                               Block Poll               Page                       Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               eeprom                 65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
               flash                  65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
               lfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               hfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               efuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               lock                    0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00
               signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00

             Programmer Type : Wiring
             Description     : Wiring
             Programmer Model: AVRISP
             Hardware Version: 15
             Firmware Version Master : 2.10
             Vtarget         : 0.0 V
             SCK period      : 0.1 us

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude.exe: Device signature = 0x1e9801 (probably m2560)
avrdude.exe: reading input file ".\atmega2560_0x11000_offset.hex"
avrdude.exe: writing flash (261414 bytes):

Writing | ################################################## | 100% 0.00s

avrdude.exe: 261414 bytes of flash written
avrdude.exe: verifying flash memory against .\atmega2560_0x11000_offset.hex:

Reading | ################################################## | 100% 0.01s

avrdude.exe: verification error, first mismatch at byte 0x11000
             0xff != 0x54
avrdude.exe: verification error; content mismatch

avrdude.exe done.  Thank you.

```

mcuee, 2022-06-16:
Verified with AVR Dragon as well.
```
PS C:\work\avr\avrdude_test\avrdude-v7.0-windows-x64> .\avrdude.exe -c dragon_isp -p m2560 -D -U flash:v:.\atmega2560_0x11000_offset.hex:i -v

avrdude.exe: Version 7.0
             Copyright (c) Brian Dean, http://www.bdmicro.com/
             Copyright (c) Joerg Wunsch

             System wide configuration file is "C:/work/avr/avrdude_test/avrdude-v7.0-windows-x64/avrdude.conf"

             Using Port                    : usb
             Using Programmer              : dragon_isp
avrdude.exe: usbdev_open(): Found AVRDRAGON, serno: 00A200001487
JTAG ICE mkII sign-on message:
Communications protocol version: 1
M_MCU:
  boot-loader FW version:        255
  firmware version:              7.39
  hardware version:              1
S_MCU:
  boot-loader FW version:        255
  firmware version:              7.39
  hardware version:              2
Serial number:                   00:a2:00:00:14:87
Device ID:                       AVRDRAGON
             AVR Part                      : ATmega2560
             Chip Erase delay              : 9000 us
             PAGEL                         : PD7
             BS2                           : PA0
             RESET disposition             : dedicated
             RETRY pulse                   : SCK
             Serial program mode           : yes
             Parallel program mode         : yes
             Timeout                       : 200
             StabDelay                     : 100
             CmdexeDelay                   : 25
             SyncLoops                     : 32
             PollIndex                     : 3
             PollValue                     : 0x53
             Memory Detail                 :

                                               Block Poll               Page                       Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               eeprom                 65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
               flash                  65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
               lfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               hfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               efuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               lock                    0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00
               signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00

             Programmer Type : DRAGON_ISP
             Description     : Atmel AVR Dragon in ISP mode
             Vtarget         : 4.8 V
             SCK period      : 2.00 us

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.15s

avrdude.exe: Device signature = 0x1e9801 (probably m2560)
avrdude.exe: verifying flash memory against .\atmega2560_0x11000_offset.hex:

Reading | ################################################## | 100% 0.01s

avrdude.exe: verification error, first mismatch at byte 0x11000
             0xff != 0x54
avrdude.exe: verification error; content mismatch

avrdude.exe done.  Thank you.

```

mcuee, 2022-06-16:
AVR Dragon seems to be okay.
```
PS C:\work\avr\avrdude_test\avrdude-v7.0-windows-x64> .\avrdude.exe -c dragon_isp -p m2560  -D -U flash:w:.\atmega2560_0x11000_offset.hex:i -v

avrdude.exe: Version 7.0
             Copyright (c) Brian Dean, http://www.bdmicro.com/
             Copyright (c) Joerg Wunsch

             System wide configuration file is "C:/work/avr/avrdude_test/avrdude-v7.0-windows-x64/avrdude.conf"

             Using Port                    : usb
             Using Programmer              : dragon_isp
avrdude.exe: usbdev_open(): Found AVRDRAGON, serno: 00A200001487
JTAG ICE mkII sign-on message:
Communications protocol version: 1
M_MCU:
  boot-loader FW version:        255
  firmware version:              7.39
  hardware version:              1
S_MCU:
  boot-loader FW version:        255
  firmware version:              7.39
  hardware version:              2
Serial number:                   00:a2:00:00:14:87
Device ID:                       AVRDRAGON
             AVR Part                      : ATmega2560
             Chip Erase delay              : 9000 us
             PAGEL                         : PD7
             BS2                           : PA0
             RESET disposition             : dedicated
             RETRY pulse                   : SCK
             Serial program mode           : yes
             Parallel program mode         : yes
             Timeout                       : 200
             StabDelay                     : 100
             CmdexeDelay                   : 25
             SyncLoops                     : 32
             PollIndex                     : 3
             PollValue                     : 0x53
             Memory Detail                 :

                                               Block Poll               Page                       Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               eeprom                 65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
               flash                  65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
               lfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               hfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               efuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               lock                    0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00
               signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00

             Programmer Type : DRAGON_ISP
             Description     : Atmel AVR Dragon in ISP mode
             Vtarget         : 4.8 V
             SCK period      : 2.00 us

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.15s

avrdude.exe: Device signature = 0x1e9801 (probably m2560)
avrdude.exe: reading input file ".\atmega2560_0x11000_offset.hex"
avrdude.exe: writing flash (261414 bytes):

Writing | ################################################## | 100% 0.01s

avrdude.exe: 261414 bytes of flash written
avrdude.exe: verifying flash memory against .\atmega2560_0x11000_offset.hex:
[... 44 lines cut ...]
               lock                    0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00
               signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00

             Programmer Type : Wiring
             Description     : Wiring
             Programmer Model: AVRISP
             Hardware Version: 15
             Firmware Version Master : 2.10
             Vtarget         : 0.0 V
             SCK period      : 0.1 us

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude.exe: Device signature = 0x1e9801 (probably m2560)
avrdude.exe: verifying flash memory against .\atmega2560_0x11000_offset.hex:

Reading | ################################################## | 100% 0.00s

avrdude.exe: 261414 bytes of flash verified

avrdude.exe done.  Thank you.
```

mcuee, 2022-06-16:
The test hex file is usng the ATmega2560 stk500v2 bootloader contents plus extra contents at 0x11000. The extra contents were generated by srec_cat.

```
$ srec_cat eepromtest.bin -offset 0x011000 -o eepromtest.srec
srec_cat: eepromtest.bin: 19: warning: no execution start address record

$ srec_cat eepromtest.srec  -o eepromtest_intel.hex -intel
srec_cat: eepromtest.srec: 19: warning: no execution start address record

$ cat eepromtest_intel.hex
:020000040001F9
:2010000054686520717569636B2062726F776E20666F78206A756D7073206F7665722074FE
:201020006865206C617A7920646F670A54686520717569636B2062726F776E20666F78206C
:201040006A756D7073206F76657220746865206C617A7920646F670A5468652071756963ED
:201060006B2062726F776E20666F78206A756D7073206F76657220746865206C617A7920C4
:20108000646F670A54686520717569636B2062726F776E20666F78206A756D7073206F76A5
:2010A000657220746865206C617A7920646F670A54686520717569636B2062726F776E20EE
:2010C000666F78206A756D7073206F76657220746865206C617A7920646F670A54686520B2
:2010E000717569636B2062726F776E20666F78206A756D7073206F76657220746865206C06
:20110000617A7920646F670A54686520717569636B2062726F776E20666F78206A756D7028
:2011200073206F76657220746865206C617A7920646F670A54686520717569636B20627269
:201140006F776E20666F78206A756D7073206F76657220746865206C617A7920646F670AFE
:2011600054686520717569636B2062726F776E20666F78206A756D7073206F76657220749D
:201180006865206C617A7920646F670A54686520717569636B2062726F776E20666F78200B
:2011A0006A756D7073206F76657220746865206C617A7920646F670A54686520717569638C
:2011C0006B2062726F776E20666F78206A756D7073206F76657220746865206C617A792063
:2011E000646F670A54686520717569636B2062726F776E20666F78206A756D7073206F7644
:00000001FF
```

mcuee, 2022-06-16:
Not only wiring bootloader, the one provided by @stefanrueger here ( https://github.com/avrdudes/avrdude/files/8882669/atmega2560_v3.txt ) in #940 discussion has a problem as well. But the problem may be a bit different.

```
PS C:\work\avr\avrdude_test\avrdude-v7.0-windows-x64> .\avrdude.exe -c arduino -P COM3 -p m2560 -D -U flash:w:.\atmega2560_0x11000_offset.hex:i -v

avrdude.exe: Version 7.0
             Copyright (c) Brian Dean, http://www.bdmicro.com/
             Copyright (c) Joerg Wunsch

             System wide configuration file is "C:/work/avr/avrdude_test/avrdude-v7.0-windows-x64/avrdude.conf"

             Using Port                    : COM3
             Using Programmer              : arduino
avrdude.exe: ser_open(): can't open device "\\.\COM3": The system cannot find the file specified.

avrdude.exe: opening programmer "arduino" on port "COM3" failed

avrdude.exe done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude-v7.0-windows-x64> .\avrdude.exe -c arduino -P COM6 -p m2560 -D -U flash:w:.\atmega2560_0x11000_offset.hex:i -v

avrdude.exe: Version 7.0
             Copyright (c) Brian Dean, http://www.bdmicro.com/
             Copyright (c) Joerg Wunsch

             System wide configuration file is "C:/work/avr/avrdude_test/avrdude-v7.0-windows-x64/avrdude.conf"

             Using Port                    : COM6
             Using Programmer              : arduino
             AVR Part                      : ATmega2560
             Chip Erase delay              : 9000 us
             PAGEL                         : PD7
             BS2                           : PA0
             RESET disposition             : dedicated
             RETRY pulse                   : SCK
             Serial program mode           : yes
             Parallel program mode         : yes
             Timeout                       : 200
             StabDelay                     : 100
             CmdexeDelay                   : 25
             SyncLoops                     : 32
             PollIndex                     : 3
             PollValue                     : 0x53
             Memory Detail                 :

                                               Block Poll               Page                       Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               eeprom                 65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
               flash                  65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
               lfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               hfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               efuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               lock                    0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00
               signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00

             Programmer Type : Arduino
             Description     : Arduino
             Hardware Version: 7
             Firmware Version: 7.6

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude.exe: Device signature = 0x1e9801 (probably m2560)
avrdude.exe: reading input file ".\atmega2560_0x11000_offset.hex"
avrdude.exe: writing flash (261414 bytes):

Writing | ################################################## | 100% 0.00s

avrdude.exe: 261414 bytes of flash written
avrdude.exe: verifying flash memory against .\atmega2560_0x11000_offset.hex:

Reading | ################################################## | 100% 0.00s

avrdude.exe: verification error, first mismatch at byte 0x3fd00
             0x11 != 0x99
avrdude.exe: verification error; content mismatch

avrdude.exe done.  Thank you.
```

stefanrueger, 2022-06-17:
> avrdude.exe: verification error, first mismatch at byte 0x3fd00

You are trying to overwrite the bootloader. Your file is 261414 bytes long, i.e. does not fit into the available flash of 256k minus 768 bytes for the bootloader.  My bootloader protects itself from being overwritten, but throws no error when writing (that costs too many code bytes :). You will notice during verification. The `- c arduino` programmer does not know about the size of the bootloader, so cannot do anything in this situation; my proposed `urclock` programmer knows about the available flash; its draft implementation complains with an appropriate error message and exits *before* attempts to write flash.

stefanrueger, 2022-06-17:
> avrdude.exe: verification error, first mismatch at byte 0x11000

That is an interesting and unusual address for the first verification error.  What I find intriguing is that apparently
```
$ .\avrdude.exe -c dragon_isp -p m2560 -D -U flash:v:.\atmega2560_0x11000_offset.hex:i -v
```
fails and that 
```
$ .\avrdude.exe -c dragon_isp -p m2560 -D -U flash:w:.\atmega2560_0x11000_offset.hex:i -v
```
works?!? The only difference is 'v' and 'w' in the `-U`, which in presence of `-v` won't make a difference at all.

Perhaps try test files that fit below the bootloader?

stefanrueger, 2022-06-17:
Your input file is to blame, I think. 
```
$ srec_cat atmega2560_0x11000_offset.hex -intel -o - -c-arra dopp -Output_Word | tail -9 | head -4
srec_cat: atmega2560_0x11000_offset.hex: 470: warning: data records not in
    strictly ascending order (expected >= 0x3FD26, got 0x11000)
const unsigned long dopp_termination = 0x00000000;
const unsigned long dopp_start       = 0x00011000;
const unsigned long dopp_finish      = 0x0003FD26;
const unsigned long dopp_length      = 0x0002ED26;
```

mcuee, 2022-06-17:
@stefanrueger 
Apparently the file generated is not correct. The intention is not to test above 256KB but just above 0x11000 (a bit above 64KB.

As for the bootloader, that is kind of intentional -- to see how the bootloader behaves with `-D`.

mcuee, 2022-06-17:
Redo the tests:

```
PS C:\work\avr\avrdude_test\avrdude-v7.0-windows-x64> .\avrdude.exe -p m2560 -c arduino -P COM3 -D -U .\eepromtest_intel.hex -vv

avrdude.exe: Version 7.0
             Copyright (c) Brian Dean, http://www.bdmicro.com/
             Copyright (c) Joerg Wunsch

             System wide configuration file is "C:/work/avr/avrdude_test/avrdude-v7.0-windows-x64/avrdude.conf"

avrdude.exe: defaulting memtype in -U w:.\eepromtest_intel.hex option to "flash"
avrdude.exe: defaulting memtype in -U v:.\eepromtest_intel.hex option to "flash"
             Using Port                    : COM3
             Using Programmer              : arduino
             AVR Part                      : ATmega2560
             Chip Erase delay              : 9000 us
             PAGEL                         : PD7
             BS2                           : PA0
             RESET disposition             : dedicated
             RETRY pulse                   : SCK
             Serial program mode           : yes
             Parallel program mode         : yes
             Timeout                       : 200
             StabDelay                     : 100
             CmdexeDelay                   : 25
             SyncLoops                     : 32
             PollIndex                     : 3
             PollValue                     : 0x53
             Memory Detail                 :

                                               Block Poll               Page                       Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               eeprom                 65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
               flash                  65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
               lfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               hfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               efuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               lock                    0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00
               signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00

             Programmer Type : Arduino
             Description     : Arduino
             Hardware Version: 7
             Firmware Version: 7.6

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude.exe: Device signature = 0x1e9801 (probably m2560)
avrdude.exe: reading input file ".\eepromtest_intel.hex"
avrdude.exe: input file .\eepromtest_intel.hex auto detected as Intel Hex
avrdude.exe: writing flash (70144 bytes):

Writing | ################################################## | 100% 0.00s

avrdude.exe: 70144 bytes of flash written
avrdude.exe: verifying flash memory against .\eepromtest_intel.hex:
avrdude.exe: load data flash data from input file .\eepromtest_intel.hex:
avrdude.exe: input file .\eepromtest_intel.hex auto detected as Intel Hex
avrdude.exe: input file .\eepromtest_intel.hex contains 70144 bytes
avrdude.exe: reading on-chip flash data:

Reading | ################################################## | 100% 0.00s

avrdude.exe: verifying ...
avrdude.exe: 70144 bytes of flash verified

avrdude.exe done.  Thank you.

```

mcuee, 2022-06-17:
But the stk500v2 wiring bootloader (with eeprom read fix) seems to still have a problem.
```
PS C:\work\avr\avrdude_test\avrdude-v7.0-windows-x64> .\avrdude.exe -p m2560 -c usbasp -U .\stk500boot_v2_mega2560_eeprom_fix.hex

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude.exe: Device signature = 0x1e9801 (probably m2560)
avrdude.exe: NOTE: "flash" memory has been specified, an erase cycle will be performed
             To disable this feature, specify the -D option.
avrdude.exe: erasing chip
avrdude.exe: reading input file ".\stk500boot_v2_mega2560_eeprom_fix.hex"
avrdude.exe: input file .\stk500boot_v2_mega2560_eeprom_fix.hex auto detected as Intel Hex
avrdude.exe: writing flash (261414 bytes):

Writing | ################################################## | 100% 0.01s

avrdude.exe: 261414 bytes of flash written
avrdude.exe: verifying flash memory against .\stk500boot_v2_mega2560_eeprom_fix.hex:
avrdude.exe: input file .\stk500boot_v2_mega2560_eeprom_fix.hex auto detected as Intel Hex

Reading | ################################################## | 100% 0.01s

avrdude.exe: 261414 bytes of flash verified

avrdude.exe done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude-v7.0-windows-x64> .\avrdude.exe -p m2560 -c wiring -P COM3 -D -U flash:v:.\eepromtest_intel.hex:i -vv

avrdude.exe: Version 7.0
             Copyright (c) Brian Dean, http://www.bdmicro.com/
             Copyright (c) Joerg Wunsch

             System wide configuration file is "C:/work/avr/avrdude_test/avrdude-v7.0-windows-x64/avrdude.conf"

             Using Port                    : COM3
             Using Programmer              : wiring
avrdude.exe: wiring_open(): releasing DTR/RTS
avrdude.exe: wiring_open(): asserting DTR/RTS
avrdude.exe: ser_recv(): programmer is not responding
avrdude.exe: stk500v2_ReceiveMessage(): timeout
             AVR Part                      : ATmega2560
             Chip Erase delay              : 9000 us
             PAGEL                         : PD7
             BS2                           : PA0
             RESET disposition             : dedicated
             RETRY pulse                   : SCK
             Serial program mode           : yes
             Parallel program mode         : yes
             Timeout                       : 200
             StabDelay                     : 100
             CmdexeDelay                   : 25
             SyncLoops                     : 32
             PollIndex                     : 3
             PollValue                     : 0x53
             Memory Detail                 :

                                               Block Poll               Page                       Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               eeprom                 65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
               flash                  65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
               lfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               hfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               efuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               lock                    0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00
               signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00

             Programmer Type : Wiring
             Description     : Wiring
             Programmer Model: AVRISP
             Hardware Version: 15
             Firmware Version Master : 2.10
             Vtarget         : 0.0 V
             SCK period      : 0.1 us

avrdude.exe: AVR device initialized and ready to accept instructions

Reading |                                                    | 0% 0.00savrdude.exe: stk500isp_read_byte(.., signature, 0x0, ...)
avrdude.exe: stk500isp_read_byte(): Sending read memory command: avrdude.exe: stk500isp_read_byte(.., signature, 0x1, ...)
Reading | #################                                  | 33% 0.01savrdude.exe: stk500isp_read_byte(.., signature, 0x2, ...)
Reading | ################################################## | 100% 0.01s

avrdude.exe: Device signature = 0x1e9801 (probably m2560)
avrdude.exe: verifying flash memory against .\eepromtest_intel.hex:
avrdude.exe: load data flash data from input file .\eepromtest_intel.hex:
avrdude.exe: input file .\eepromtest_intel.hex contains 70144 bytes
avrdude.exe: reading on-chip flash data:

Reading | ################################################## | 100% 0.00s

avrdude.exe: verifying ...
avrdude.exe: verification error, first mismatch at byte 0x11000
             0xff != 0x54
avrdude.exe: verification error; content mismatch

avrdude.exe done.  Thank you.
```

mcuee, 2022-06-17:
Using the bootloader from megacore from @MCUdude seems to be okay as wlel.

```
:\Users\xiaof\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino18/bin/avrdude -CC:\Users\xiaof\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\2.1.3/avrdude.conf -v -patmega2560 -cusbasp -Pusb -Uflash:w:C:\Users\xiaof\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\2.1.3/bootloaders/optiboot_flash/bootloaders/atmega2560/16000000L/optiboot_flash_atmega2560_UART0_115200_16000000L_B7_BIGBOOT.hex:i -Ulock:w:0x0f:m 

avrdude: Version 6.3-20201216
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\xiaof\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\2.1.3/avrdude.conf"

         Using Port                    : usb
         Using Programmer              : usbasp
         AVR Part                      : ATmega2560
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
Writing | ################################################## | 100% 0.00s

avrdude: 1 bytes of lock written
avrdude: verifying lock memory against 0x3f:
avrdude: load data lock data from input file 0x3f:
avrdude: input file 0x3f contains 1 bytes
avrdude: reading on-chip lock data:

Reading | ################################################## | 100% 0.00s

avrdude: verifying ...
avrdude: 1 bytes of lock verified
avrdude: reading input file "0xfd"
avrdude: writing efuse (1 bytes):

Writing | ################################################## | 100% 0.00s

avrdude: 1 bytes of efuse written
avrdude: verifying efuse memory against 0xfd:
avrdude: load data efuse data from input file 0xfd:
avrdude: input file 0xfd contains 1 bytes
avrdude: reading on-chip efuse data:

Reading | ################################################## | 100% 0.00s

avrdude: verifying ...
avrdude: 1 bytes of efuse verified
avrdude: reading input file "0b11010110"
avrdude: writing hfuse (1 bytes):

Writing | ################################################## | 100% 0.00s

avrdude: 1 bytes of hfuse written
avrdude: verifying hfuse memory against 0b11010110:
avrdude: load data hfuse data from input file 0b11010110:
avrdude: input file 0b11010110 contains 1 bytes
avrdude: reading on-chip hfuse data:

Reading | ################################################## | 100% 0.00s
[... 180 lines cut ...]
               signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00

             Programmer Type : Arduino
             Description     : Arduino
             Hardware Version: 3
             Firmware Version: 8.0

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude.exe: Device signature = 0x1e9801 (probably m2560)
avrdude.exe: verifying flash memory against .\eepromtest_intel.hex:
avrdude.exe: load data flash data from input file .\eepromtest_intel.hex:
avrdude.exe: input file .\eepromtest_intel.hex contains 70144 bytes
avrdude.exe: reading on-chip flash data:

Reading | ################################################## | 100% 0.00s

avrdude.exe: verifying ...
avrdude.exe: 70144 bytes of flash verified

avrdude.exe done.  Thank you.

```

mcuee, 2022-06-17:
BTW, programmer like usbasp and AVR Dragon have no issues.

stefanrueger, 2022-06-17:
Would it be fair to summarise that a particular bootloader does not seem to work (stk500v2 wiring) with `-c wiring`, but other bootloaders do? So this could well be the _bootloader_ that is unable to upload that file. 

Have you tried input files that start at 0? Some bootloaders autoincrement the address and ignore any set extended address commands. In this case your input file should have been folded by the defective bootloader into the [0, 0xffff] space. Try to find where your input has been placed (read out the flash and look at it). 

Anyway, AVRDUDE wouldn't be able to do anything about defective bootloaders...

mcuee, 2022-06-17:
@stefanrueger 
Yes I will close this issue as it seems to be the issue with the wiring bootloader and not avrdude, but it seems to be strange that it could have such an issue. It is used in Arduino Mega2560.

They do not bother to fix the issue with regard to EEPROM for so many years.
https://github.com/arduino/ArduinoCore-avr/pull/24

stefanrueger, 2022-06-17:
> Full debug log with -vvvv

BTW, putting longer logs into a file is a much better way to give additional information. The best issue descriptions are short and try to pinpoint the problem. I had a glance over the log file, and the protocol send/recv streams seem OK firming up my hypothesis that the bootloader is to blame. (You should prefer mine once I have published them :)

mcuee, 2022-06-17:
> Have you tried input files that start at 0? Some bootloaders autoincrement the address and ignore any set extended address commands. In this case your input file should have been folded by the [...]

It seems to write nothing. I can not find the test pattern from anywhere.
[readback.hex.txt](https://github.com/avrdudes/avrdude/files/8926989/readback.hex.txt)

stefanrueger, 2022-06-17:
Correct - no quick brown fox hiding anywhere. Only an 8k bootloader starting with a vector table at 0x3e000. That doesn't work.

mcuee, 2022-06-18:
I have some doubts about my own build of the bootlaoder, so I changed to the one used by @MCUdude's MegaCore.

```
/Users/mcuee/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino18/bin/avrdude -C/Users/mcuee/Library/Arduino15/packages/MegaCore/hardware/avr/2.2.0/avrdude.conf -v -patmega2560 -cusbasp -Pusb -e -Ulock:w:0xff:m -Uefuse:w:0xfd:m -Uhfuse:w:0b11011000:m -Ulfuse:w:0xf7:m 

avrdude: Version 6.3-20201216
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/Users/mcuee/Library/Arduino15/packages/MegaCore/hardware/avr/2.2.0/avrdude.conf"
         User configuration file is "/Users/mcuee/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : usb
         Using Programmer              : usbasp
         AVR Part                      : ATmega2560
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
           flash         65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : usbasp
         Description     : USBasp, http://www.fischl.de/usbasp/

avrdude: auto set sck period (because given equals null)
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9801 (probably m2560)
avrdude: erasing chip
avrdude: auto set sck period (because given equals null)
avrdude: reading input file "0xff"
avrdude: writing lock (1 bytes):

Writing |  ***failed;  
################################################## | 100% 0.06s

avrdude: 1 bytes of lock written
avrdude: verifying lock memory against 0xff:
avrdude: load data lock data from input file 0xff:
avrdude: input file 0xff contains 1 bytes
avrdude: reading on-chip lock data:

/Users/mcuee/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino18/bin/avrdude -C/Users/mcuee/Library/Arduino15/packages/MegaCore/hardware/avr/2.2.0/avrdude.conf -v -patmega2560 -cusbasp -Pusb -Uflash:w:/Users/mcuee/Library/Arduino15/packages/MegaCore/hardware/avr/2.2.0/bootloaders/stk500v2/bootloaders/atmega2560/stk500boot_v2_atmega2560_115200_16000000L.hex:i -Ulock:w:0xcf:m 

avrdude: Version 6.3-20201216
[... 119 lines cut ...]
avrdude: verifying ...
avrdude: 260138 bytes of flash verified
avrdude: reading input file "0xcf"
avrdude: writing lock (1 bytes):

Writing |  ***failed;  
################################################## | 100% 0.06s

avrdude: 1 bytes of lock written
avrdude: verifying lock memory against 0xcf:
avrdude: load data lock data from input file 0xcf:
avrdude: input file 0xcf contains 1 bytes
avrdude: reading on-chip lock data:

Reading | ################################################## | 100% 0.00s

avrdude: verifying ...
avrdude: WARNING: invalid value for unused bits in fuse "lock", should be set to 1 according to datasheet
This behaviour is deprecated and will result in an error in future version
You probably want to use 0x0f instead of 0xcf (double check with your datasheet first).
avrdude: 1 bytes of lock verified

avrdude done.  Thank you.

```

mcuee, 2022-06-18:
So I go back to the original stk500v2 bootloader hex file from Arduino (with the known EEPROM read bug).

```
/Users/mcuee/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17/bin/avrdude -C/Users/mcuee/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17/etc/avrdude.conf -v -patmega2560 -cusbasp -Pusb -e -Ulock:w:0x3F:m -Uefuse:w:0xFD:m -Uhfuse:w:0xD8:m -Ulfuse:w:0xFF:m 

avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/Users/mcuee/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17/etc/avrdude.conf"
         User configuration file is "/Users/mcuee/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : usb
         Using Programmer              : usbasp
         AVR Part                      : ATmega2560
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
           flash         65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : usbasp
         Description     : USBasp, http://www.fischl.de/usbasp/

avrdude: auto set sck period (because given equals null)
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9801 (probably m2560)
avrdude: erasing chip
avrdude: auto set sck period (because given equals null)
avrdude: reading input file "0x3F"
avrdude: writing lock (1 bytes):

/Users/mcuee/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17/bin/avrdude -C/Users/mcuee/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17/etc/avrdude.conf -v -patmega2560 -cusbasp -Pusb -Uflash:w:/Users/mcuee/Library/Arduino15/packages/arduino/hardware/avr/1.8.5/bootloaders/stk500v2/stk500boot_v2_mega2560.hex:i -Ulock:w:0x0F:m 

avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/Users/mcuee/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17/etc/avrdude.conf"
         User configuration file is "/Users/mcuee/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : usb
         Using Programmer              : usbasp
[... 110 lines cut ...]
avrdude: input file /Users/mcuee/Library/Arduino15/packages/arduino/hardware/avr/1.8.5/bootloaders/stk500v2/stk500boot_v2_mega2560.hex contains 261406 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 0.00s

avrdude: verifying ...
avrdude: 261406 bytes of flash verified
avrdude: reading input file "0x0F"
avrdude: writing lock (1 bytes):

Writing | ################################################## | 100% 0.01s

avrdude: 1 bytes of lock written
avrdude: verifying lock memory against 0x0F:
avrdude: load data lock data from input file 0x0F:
avrdude: input file 0x0F contains 1 bytes
avrdude: reading on-chip lock data:

Reading | ################################################## | 100% 0.00s

avrdude: verifying ...
avrdude: 1 bytes of lock verified

avrdude done.  Thank you.
```

mcuee, 2022-06-18:
Somehow the above process using Arduino under macOS created un-usable bootloader.

So I just use usbasp to program the original stk500v2 bootloader (with known EEPROM read bug) using usbasp and there seems to be no issues any more.
```
mcuee@mcuees-Mac-mini bin % ./avrdude -p m2560 -c usbasp -U ./stk500boot_v2_mega2560.hex -v

avrdude: Version 7.0
         Copyright (c) Brian Dean, http://www.bdmicro.com/
         Copyright (c) Joerg Wunsch

         System wide configuration file is "/Users/mcuee/build/avr/avrdude_test/avrdude_macOS_64bit/bin/../etc/avrdude.conf"
         User configuration file is "/Users/mcuee/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : usb
         Using Programmer              : usbasp
         AVR Part                      : ATmega2560
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom                 65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
           flash                  65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
           lfuse                   0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse                   0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           efuse                   0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lock                    0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           calibration             0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature               0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : usbasp
         Description     : USBasp, http://www.fischl.de/usbasp/

avrdude: auto set sck period (because given equals null)
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9801 (probably m2560)
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: auto set sck period (because given equals null)
avrdude: reading input file "./stk500boot_v2_mega2560.hex"
avrdude: input file ./stk500boot_v2_mega2560.hex auto detected as Intel Hex
avrdude: writing flash (261406 bytes):

Writing | ################################################## | 100% 0.00s

avrdude: 261406 bytes of flash written
avrdude: verifying flash memory against ./stk500boot_v2_mega2560.hex:
avrdude: input file ./stk500boot_v2_mega2560.hex auto detected as Intel Hex

Reading | ################################################## | 100% 0.00s

avrdude: 261406 bytes of flash verified
[... 195 lines cut ...]
         Firmware Version Master : 2.10
         Vtarget         : 0.0 V
         SCK period      : 0.1 us

avrdude: AVR device initialized and ready to accept instructions

Reading |                                                    | 0% 0.00savrdude: stk500isp_read_byte(.., signature, 0x0, ...)
avrdude: stk500isp_read_byte(): Sending read memory command: avrdude: stk500isp_read_byte(.., signature, 0x1, ...)
Reading | #################                                  | 33% 0.01savrdude: stk500isp_read_byte(.., signature, 0x2, ...)
Reading | ################################################## | 100% 0.01s

avrdude: Device signature = 0x1e9801 (probably m2560)
avrdude: verifying flash memory against ./entest.hex:
avrdude: load data flash data from input file ./entest.hex:
avrdude: input file ./entest.hex contains 512 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 0.06s

avrdude: verifying ...
avrdude: 512 bytes of flash verified

avrdude done.  Thank you.

```

mcuee, 2022-06-18:
@MCUdude The megacore m2560 stk500v2 bootloader burning process seems to have some issues. The following is under Windows.

```
C:\Users\xiaof\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino18/bin/avrdude -CC:\Users\xiaof\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\2.2.0/avrdude.conf -v -patmega2560 -cusbasp -Pusb -e -Ulock:w:0xff:m -Uefuse:w:0xfd:m -Uhfuse:w:0b11010000:m -Ulfuse:w:0xf7:m 

avrdude: Version 6.3-20201216
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\xiaof\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\2.2.0/avrdude.conf"

         Using Port                    : usb
         Using Programmer              : usbasp
         AVR Part                      : ATmega2560
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
           flash         65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : usbasp
         Description     : USBasp, http://www.fischl.de/usbasp/

avrdude: auto set sck period (because given equals null)
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9801 (probably m2560)
avrdude: erasing chip
avrdude: auto set sck period (because given equals null)
avrdude: reading input file "0xff"
avrdude: writing lock (1 bytes):

C:\Users\xiaof\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino18/bin/avrdude -CC:\Users\xiaof\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\2.2.0/avrdude.conf -v -patmega2560 -cusbasp -Pusb -Uflash:w:C:\Users\xiaof\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\2.2.0/bootloaders/stk500v2/bootloaders/atmega2560/stk500boot_v2_atmega2560_115200_16000000L.hex:i -Ulock:w:0xcf:m 
Writing |  ***failed;  
################################################## | 100% 0.06s

avrdude: 1 bytes of lock written
avrdude: verifying lock memory against 0xff:
avrdude: load data lock data from input file 0xff:
avrdude: input file 0xff contains 1 bytes
avrdude: reading on-chip lock data:

Reading | ################################################## | 100% 0.00s

avrdude: verifying ...
avrdude: WARNING: invalid value for unused bits in fuse "lock", should be set to 1 according to datasheet
[... 114 lines cut ...]
avrdude: verifying ...
avrdude: 260138 bytes of flash verified
avrdude: reading input file "0xcf"
avrdude: writing lock (1 bytes):

Writing |  ***failed;  
################################################## | 100% 0.06s

avrdude: 1 bytes of lock written
avrdude: verifying lock memory against 0xcf:
avrdude: load data lock data from input file 0xcf:
avrdude: input file 0xcf contains 1 bytes
avrdude: reading on-chip lock data:

Reading | ################################################## | 100% 0.00s

avrdude: verifying ...
avrdude: WARNING: invalid value for unused bits in fuse "lock", should be set to 1 according to datasheet
This behaviour is deprecated and will result in an error in future version
You probably want to use 0x0f instead of 0xcf (double check with your datasheet first).
avrdude: 1 bytes of lock verified

avrdude done.  Thank you.

```

mcuee, 2022-06-18:
Going back to the issue with the stk500v2 (wiring) bootloader, it seems to me the issue is with my build of the EEPROM fix. The hex file may exceeds the 256KB boundary.

```
PS C:\work\avr\avrdude_test\avrdude-v7.0-windows-x64> .\avrdude.exe -p m2560 -c usbasp -U .\stk500boot_v2_mega2560_eeprom_fix.hex

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude.exe: Device signature = 0x1e9801 (probably m2560)
avrdude.exe: NOTE: "flash" memory has been specified, an erase cycle will be performed
             To disable this feature, specify the -D option.
avrdude.exe: erasing chip
avrdude.exe: reading input file ".\stk500boot_v2_mega2560_eeprom_fix.hex"
avrdude.exe: input file .\stk500boot_v2_mega2560_eeprom_fix.hex auto detected as Intel Hex
avrdude.exe: writing flash (261414 bytes):

Writing | ################################################## | 100% 0.01s

avrdude.exe: 261414 bytes of flash written
avrdude.exe: verifying flash memory against .\stk500boot_v2_mega2560_eeprom_fix.hex:
avrdude.exe: input file .\stk500boot_v2_mega2560_eeprom_fix.hex auto detected as Intel Hex

Reading | ################################################## | 100% 0.01s

avrdude.exe: 261414 bytes of flash verified

avrdude.exe done.  Thank you.
```

MCUdude, 2022-06-18:
@mcuee I just changed the lock/unlock fuse bits so Avrdude wouldn't complain about setting unused bits. Seems like I'll have to revert that commit and go back to using 0x3f/0x0f.

Slightly on topic: The stk500v2 bootloader I've compiled and bundled with MegaCore; does this works as expected (flash read/write, EEPROM read/write) for you?

mcuee, 2022-06-18:
> @mcuee I just changed the lock/unlock fuse bits so Avrdude wouldn't complain about setting unused bits. Seems like I'll have to revert that commit and go back to using 0x3f/0x0f.

Looks like the old version of avrdude bundled with Arduino caused this issue. avrdude-v7.0 is okay with 0xff.

> Slightly on topic: The stk500v2 bootloader I've compiled and bundled with MegaCore; does this works as expected (flash read/write, EEPROM read/write) for you?

EEPROM read/write is fine. But flash write got problems -- sometimes it times out. I am not so sure why.

The official bootloader from Arduino has the EEPROM read problem, but it does not seem to have the flash write problem.

mcuee, 2022-06-18:
@MCUdude 

EEPROM is fine.

```
PS C:\work\avr\avrdude_test\avrdude-7.0_bin64> .\avrdude_sr474.exe -c jtagkey -v -patmega2560 -e -Ulock:w:0xff:m -Uefuse:w:0xfd:m -Uhfuse:w:0b11010000:m -Ulfuse:w:0xf7:m

avrdude_sr474.exe: Version 7.0-20220530 (feaa1c6)
                   Copyright (c) Brian Dean, http://www.bdmicro.com/
                   Copyright (c) Joerg Wunsch

                   System wide configuration file is "C:/work/avr/avrdude_test/avrdude-7.0_bin64/avrdude.conf"

                   Using Port                    : usb
                   Using Programmer              : jtagkey
                   AVR Part                      : ATmega2560
                   Chip Erase delay              : 9000 us
                   PAGEL                         : PD7
                   BS2                           : PA0
                   RESET disposition             : dedicated
                   RETRY pulse                   : SCK
                   Serial program mode           : yes
                   Parallel program mode         : yes
                   Timeout                       : 200
                   StabDelay                     : 100
                   CmdexeDelay                   : 25
                   SyncLoops                     : 32
                   PollIndex                     : 3
                   PollValue                     : 0x53
                   Memory Detail                 :

                                                     Block Poll               Page                       Polled
                     Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
                     ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
                     eeprom                 65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
                     flash                  65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
                     lfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
                     hfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
                     efuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
                     lock                    0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
                     calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00
                     signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00

                   Programmer Type : avrftdi
                   Description     : Amontec JTAGKey, JTAGKey-Tiny and JTAGKey2

avrdude_sr474.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude_sr474.exe: Device signature = 0x1e9801 (probably m2560)
avrdude_sr474.exe: erasing chip
avrdude_sr474.exe: reading input file "0xff"
avrdude_sr474.exe: writing lock (1 bytes):

Writing | ################################################## | 100% 0.00s

avrdude_sr474.exe: 1 bytes of lock written
avrdude_sr474.exe: verifying lock memory against 0xff:

Reading | ################################################## | 100% 0.00s

avrdude_sr474.exe: 1 bytes of lock verified
avrdude_sr474.exe: reading input file "0xfd"
avrdude_sr474.exe: writing efuse (1 bytes):

Writing | ################################################## | 100% 0.00s

avrdude_sr474.exe: 1 bytes of efuse written
avrdude_sr474.exe: verifying efuse memory against 0xfd:
[... 241 lines cut ...]
0090  6f 77 6e 20 66 6f 78 20  6a 75 6d 70 73 20 6f 76  |own fox jumps ov|
00a0  65 72 20 74 68 65 20 6c  61 7a 79 20 64 6f 67 0a  |er the lazy dog |
00b0  54 68 65 20 71 75 69 63  6b 20 62 72 6f 77 6e 20  |The quick brown |
00c0  66 6f 78 20 6a 75 6d 70  73 20 6f 76 65 72 20 74  |fox jumps over t|
00d0  68 65 20 6c 61 7a 79 20  64 6f 67 0a 54 68 65 20  |he lazy dog The |
00e0  71 75 69 63 6b 20 62 72  6f 77 6e 20 66 6f 78 20  |quick brown fox |
00f0  6a 75 6d 70 73 20 6f 76  65 72 20 74 68 65 20 6c  |jumps over the l|
0100  61 7a 79 20 64 6f 67 0a  54 68 65 20 71 75 69 63  |azy dog The quic|
0110  6b 20 62 72 6f 77 6e 20  66 6f 78 20 6a 75 6d 70  |k brown fox jump|
0120  73 20 6f 76 65 72 20 74  68 65 20 6c 61 7a 79 20  |s over the lazy |
0130  64 6f 67 0a 54 68 65 20  71 75 69 63 6b 20 62 72  |dog The quick br|
0140  6f 77 6e 20 66 6f 78 20  6a 75 6d 70 73 20 6f 76  |own fox jumps ov|
0150  65 72 20 74 68 65 20 6c  61 7a 79 20 64 6f 67 0a  |er the lazy dog |
0160  54 68 65 20 71 75 69 63  6b 20 62 72 6f 77 6e 20  |The quick brown |
0170  66 6f 78 20 6a 75 6d 70  73 20 6f 76 65 72 20 74  |fox jumps over t|
0180  68 65 20 6c 61 7a 79 20  64 6f 67 0a 54 68 65 20  |he lazy dog The |
0190  71 75 69 63 6b 20 62 72  6f 77 6e 20 66 6f 78 20  |quick brown fox |
01a0  6a 75 6d 70 73 20 6f 76  65 72 20 74 68 65 20 6c  |jumps over the l|
01b0  61 7a 79 20 64 6f 67 0a  54 68 65 20 71 75 69 63  |azy dog The quic|
01c0  6b 20 62 72 6f 77 6e 20  66 6f 78 20 6a 75 6d 70  |k brown fox jump|
01d0  73 20 6f 76 65 72 20 74  68 65 20 6c 61 7a 79 20  |s over the lazy |
01e0  64 6f 67 0a 54 68 65 20  71 75 69 63 6b 20 62 72  |dog The quick br|
01f0  6f 77 6e 20 66 6f 78 20  6a 75 6d 70 73 20 6f 76  |own fox jumps ov|

```

mcuee, 2022-06-18:
Flash write sometimes got problems.

Example: first time okay, second time not okay

```
PS C:\work\avr\avrdude_test\avrdude-7.0_bin64> .\avrdude_sr474.exe -p m2560 -c wiring -P COM12 -v -D -U flash:w:.\entest_offset_0x11000.hex:i

avrdude_sr474.exe: Version 7.0-20220530 (feaa1c6)
                   Copyright (c) Brian Dean, http://www.bdmicro.com/
                   Copyright (c) Joerg Wunsch

                   System wide configuration file is "C:/work/avr/avrdude_test/avrdude-7.0_bin64/avrdude.conf"

                   Using Port                    : COM12
                   Using Programmer              : wiring
                   AVR Part                      : ATmega2560
                   Chip Erase delay              : 9000 us
                   PAGEL                         : PD7
                   BS2                           : PA0
                   RESET disposition             : dedicated
                   RETRY pulse                   : SCK
                   Serial program mode           : yes
                   Parallel program mode         : yes
                   Timeout                       : 200
                   StabDelay                     : 100
                   CmdexeDelay                   : 25
                   SyncLoops                     : 32
                   PollIndex                     : 3
                   PollValue                     : 0x53
                   Memory Detail                 :

                                                     Block Poll               Page                       Polled
                     Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
                     ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
                     eeprom                 65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
                     flash                  65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
                     lfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
                     hfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
                     efuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
                     lock                    0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
                     calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00
                     signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00

                   Programmer Type : Wiring
                   Description     : Wiring
                   Programmer Model: AVRISP
                   Hardware Version: 15
                   Firmware Version Master : 2.10
                   Vtarget         : 0.0 V
                   SCK period      : 0.1 us

avrdude_sr474.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.02s

avrdude_sr474.exe: Device signature = 0x1e9801 (probably m2560)
avrdude_sr474.exe: reading input file ".\entest_offset_0x11000.hex"
avrdude_sr474.exe: writing flash (70144 bytes):

Writing | ################################################## | 100% 0.00s

avrdude_sr474.exe: 70144 bytes of flash written
avrdude_sr474.exe: verifying flash memory against .\entest_offset_0x11000.hex:

Reading | ################################################## | 100% 0.00s

avrdude_sr474.exe: 70144 bytes of flash verified

avrdude_sr474.exe done.  Thank you.

[... 35 lines cut ...]
                     calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00
                     signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00

                   Programmer Type : Wiring
                   Description     : Wiring
                   Programmer Model: AVRISP
                   Hardware Version: 15
                   Firmware Version Master : 2.10
                   Vtarget         : 0.0 V
                   SCK period      : 0.1 us

avrdude_sr474.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.02s

avrdude_sr474.exe: Device signature = 0x1e9801 (probably m2560)
avrdude_sr474.exe: reading input file ".\entest_offset_0x11000.hex"
avrdude_sr474.exe: writing flash (70144 bytes):

Writing | ################################################## | 100% 0.00s

avrdude_sr474.exe: stk500v2_command(): unknown status 0x45
avrdude_sr474.exe: stk500v2_paged_write: write command failed
avrdude_sr474.exe: stk500v2_ReceiveMessage(): timeout
```

mcuee, 2022-06-18:
@MCUdude I think your bootloader hex file is fine but my board may have some issue with 115200 baudrate. I tried the official Arduino stk500v2 bootloader file and I can reproduce the same issue when I tried more times.

mcuee, 2022-06-18:
@MCUdude 
Just FYI, the Chinese clone of Arduino Mega2560 tends to have the time out issue. There is a "workaround" with Arduino as mentioned in the following Youtube video -- press reset just before downloading the sketch.
Ref: https://youtu.be/tAzjO4v7oF4

I do not usually need to use this trick with avrdude-7.0. But I think I will give up on the stk500v2 bootloader and use your optiboot bootloader or the bootloader by @stefanrueger mentioned in #940. Both have no issues at all.

stefanrueger, 2022-06-18:
> Flash write sometimes got problems. Example: first time okay, second time not okay

Two programming sessions with optiboot in quick succession can result in a timeout failure the second time, indeed.

My hypothesis is that after the first programming session ended optiboot still waits a while for WDT to finally trigger, and any external reset during this period (caused by the second programming session) makes optiboot run the application instead of the bootloader. This hypothesis can be tested by stepping through the optiboot code that decides what to run based on MCUSR.

Workaround: Wait a bit longer than the actual optiboot WDT timeout between AVRDUDE runs with optiboot. There is nothing AVRDUDE usefully can do about this, this problem is a side effect of optiboot's desire to keep [MCUSR as faithful as possible](https://github.com/Optiboot/optiboot/issues/140).

I would exclude baud rate mismatch for the observed timeout problems (principally because it always works with my bootloader ;).

_However,_ if you see this type of timeout otherwise  then it's possible that something is wrong with the timings in AVRDUDE.

mcuee, 2022-06-18:
> Two programming sessions with optiboot in quick succession can result in a timeout failure the second time, indeed.

Interestingly I have not seen things like this with optiboot (including your bootloader with EEPROM support) on the ATmega328p.

stefanrueger, 2022-06-18:
> Interestingly I have not seen things like this with optiboot (including your bootloader with EEPROM support) on the ATmega328p 

I believe from optiboot v7.0 onwards the "what to start at reset" logic has changed. What is the version of your 328p optiboot? The last two bytes in flash show its version number (at address 0x7ffe for ATmega328P). My bootloaders are different from optiboot and do not exhibit this effect.

More crucially, try to see whether waiting between two runs with the stk500v2 bootloader @MCUdude  has compiled and bundled with MegaCore makes a difference!

mcuee, 2022-06-19:
> More crucially, try to see whether waiting between two runs with the stk500v2 bootloader from @MCUdude makes a difference!

With the 0x21000 offset hex file, the situation seems to be bad. I will reopen this issue.

```
PS C:\work\avr\avrdude_test\avrdude-v7.0-windows-x64> .\avrdude.exe -c usbasp -patmega2560 -e -Ulock:w:0xff:m -Uefuse:w:0xfd:m -Uhfuse:w:0b11010000:m -Ulfuse:w:0xf7:m

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude.exe: Device signature = 0x1e9801 (probably m2560)
avrdude.exe: erasing chip
avrdude.exe: reading input file "0xff"
avrdude.exe: writing lock (1 bytes):

Writing | ################################################## | 100% 0.00s

avrdude.exe: 1 bytes of lock written
avrdude.exe: verifying lock memory against 0xff:

Reading | ################################################## | 100% 0.00s

avrdude.exe: 1 bytes of lock verified
avrdude.exe: reading input file "0xfd"
avrdude.exe: writing efuse (1 bytes):

Writing | ################################################## | 100% 0.00s

avrdude.exe: 1 bytes of efuse written
avrdude.exe: verifying efuse memory against 0xfd:

Reading | ################################################## | 100% 0.00s

avrdude.exe: 1 bytes of efuse verified
avrdude.exe: reading input file "0b11010000"
avrdude.exe: writing hfuse (1 bytes):

Writing | ################################################## | 100% 0.00s

avrdude.exe: 1 bytes of hfuse written
avrdude.exe: verifying hfuse memory against 0b11010000:

Reading | ################################################## | 100% 0.00s

avrdude.exe: 1 bytes of hfuse verified
avrdude.exe: reading input file "0xf7"
avrdude.exe: writing lfuse (1 bytes):

Writing | ################################################## | 100% 0.00s

avrdude.exe: 1 bytes of lfuse written
avrdude.exe: verifying lfuse memory against 0xf7:

Reading | ################################################## | 100% 0.00s

avrdude.exe: 1 bytes of lfuse verified

avrdude.exe done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude-v7.0-windows-x64> .\avrdude.exe -c usbasp -patmega2560 -e -U .\hex\stk500boot_v2_atmega2560_115200_16000000L.hex

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude.exe: Device signature = 0x1e9801 (probably m2560)
avrdude.exe: erasing chip
avrdude.exe: reading input file ".\hex\stk500boot_v2_atmega2560_115200_16000000L.hex"
avrdude.exe: input file .\hex\stk500boot_v2_atmega2560_115200_16000000L.hex auto detected as Intel Hex
avrdude.exe: writing flash (260138 bytes):

Writing | ################################################## | 100% 0.01s

avrdude.exe: 260138 bytes of flash written
avrdude.exe: verifying flash memory against .\hex\stk500boot_v2_atmega2560_115200_16000000L.hex:
avrdude.exe: input file .\hex\stk500boot_v2_atmega2560_115200_16000000L.hex auto detected as Intel Hex

Reading | ################################################## | 100% 0.01s

avrdude.exe: 260138 bytes of flash verified

avrdude.exe done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude-7.0_bin64> .\avrdude -p m2560 -c wiring -P COM6 -D -U flash:w:.\entest_offset_0x21000.hex:i

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.02s

avrdude.exe: Device signature = 0x1e9801 (probably m2560)
avrdude.exe: reading input file ".\entest_offset_0x21000.hex"
avrdude.exe: writing flash (135680 bytes):

Writing | ################################################## | 100% 0.00s

avrdude.exe: stk500v2_command(): unknown status 0x45
avrdude.exe: stk500v2_paged_write: write command failed

```

mcuee, 2022-06-19:
Strange that official Arduino stk500v2 bootloader (with EEPROM read bug) indeed performs better. As long as I wait a bit longer (say 20s), usually it is okay. And even if it runs into problems like ` avrdude.exe: verification error, first mismatch at byte 0x21000              0xff != 0x54 avrdude.exe: verification error; content mismatch`, resetting the board will fix it.

mcuee, 2022-06-19:
The EEPROM bug fix is here. It does fix the EEPROM read issue. I do not think it has anything to do with the Flash write.
* https://github.com/arduino/ArduinoCore-avr/pull/24/files

But strangely even my build of the above fix (same fuse settings as the Arduino as I do not do anything about the fuse) still performs a bit worse than the Arduino official stk500v2 firmware: more prone to the `checksum error`.

But since there is another possibility that the CH340 clone itself is the problem, hopefully people with official Arduino Mega2560 can chime in and see if the MegaCore hex file is okay or not.

mcuee, 2022-06-19:
@stefanrueger 
Your bootloader file (compatible with `-c arduino`) seems to work much better than the Megacore stk500v2 bootloader file. 

Still sometimes it gets the mismatch error.

```
PS C:\work\avr\avrdude_test\avrdude-7.0_bin64> .\avrdude.exe -c usbasp -patmega2560 -U .\hex\atmega2560_sr940_v3.hex

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude.exe: Device signature = 0x1e9801 (probably m2560)
avrdude.exe: NOTE: "flash" memory has been specified, an erase cycle will be performed
             To disable this feature, specify the -D option.
avrdude.exe: erasing chip
avrdude.exe: reading input file ".\hex\atmega2560_sr940_v3.hex"
avrdude.exe: input file .\hex\atmega2560_sr940_v3.hex auto detected as Intel Hex
avrdude.exe: writing flash (262144 bytes):

Writing | ################################################## | 100% 0.01s

avrdude.exe: 262144 bytes of flash written
avrdude.exe: verifying flash memory against .\hex\atmega2560_sr940_v3.hex:
avrdude.exe: input file .\hex\atmega2560_sr940_v3.hex auto detected as Intel Hex

Reading | ################################################## | 100% 0.01s

avrdude.exe: 262144 bytes of flash verified

avrdude.exe done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude-7.0_bin64> .\avrdude -p m2560 -c arduino -P COM6 -D -U flash:w:.\entest_offset_0x21000.hex:i

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude.exe: Device signature = 0x1e9801 (probably m2560)
avrdude.exe: reading input file ".\entest_offset_0x21000.hex"
avrdude.exe: writing flash (135680 bytes):

Writing | ################################################## | 100% 0.00s

avrdude.exe: 135680 bytes of flash written
avrdude.exe: verifying flash memory against .\entest_offset_0x21000.hex:

Reading | ################################################## | 100% 0.00s

avrdude.exe: 135680 bytes of flash verified

avrdude.exe done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude-7.0_bin64> .\avrdude -p m2560 -c arduino -P COM6 -D -U flash:w:.\entest_offset_0x21000.hex:i

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude.exe: Device signature = 0x1e9801 (probably m2560)
avrdude.exe: reading input file ".\entest_offset_0x21000.hex"
avrdude.exe: writing flash (135680 bytes):

Writing | ################################################## | 100% 0.00s

avrdude.exe: 135680 bytes of flash written
avrdude.exe: verifying flash memory against .\entest_offset_0x21000.hex:

Reading | ################################################## | 100% 0.00s

avrdude.exe: verification error, first mismatch at byte 0x21000
             0xff != 0x54
avrdude.exe: verification error; content mismatch

avrdude.exe done.  Thank you.
```

mcuee, 2022-06-19:
@MCUdude 
The optibootloader in the MegaCore is also much better, similar to the performance of the bootloader by @stefanrueger. Occassionally it will have the mismatch error.

mcuee, 2022-06-19:
> But since there is another possibility that the CH340 clone itself is the problem, hopefully people with official Arduino Mega2560 can chime in and see if the MegaCore hex file is okay or not.

I happen to have another Arduino Mega2560 clone with CH340. And it has no Flash stability issues with different hex file (official stk500v2 bootloadder, MegaCore stk500v2 bootloader and the MagaCore optiboot bootloader or the bootloader by Stefan).

So in the end, I think this is due to the clone which may have some HW design issues with the reset.

I will close this issue again.

closed 2022-06-19 by mcuee

mcuee, 2022-06-19:
Side question @stefanrueger and @WestfW, the big boot optiboot hex file seems to have some warnings if I run the comands by Stefan. Are there any real issues?

```
$ srec_cat bigboot_328.hex -intel -o - -c-arra dopp -Output_Word | tail -9 | head -4
srec_cat: standard output: 32: The C-Array (16-bit) output format uses 16-bit
    data, but unaligned data is present. Use a "--fill 0xNN --within <input>
    --range-padding 2" filter to fix this problem.
0x10F1, 0xCFF8, 0xC016, 0x01F8, 0x9185, 0x018F, 0xD015, 0x94FA, 0x10F1,
0xCFF9, 0xC00E, 0x3785, 0xF439, 0xD027, 0xE18E, 0xD00C, 0xE985, 0xD00A,
0xE08F, 0xCF80, 0x3581, 0xF411, 0xE088, 0xD017, 0xD01C, 0xE180, 0xD001,
0xCF6D, 0x9190, 0x00C0, 0xFF95, 0xCFFC, 0x9380, 0x00C6, 0x9508, 0x9180,
```

mcuee, 2022-07-31:
Workaround of the reset HW issue is to reset the ATmega2560 using the 1200bps fix and then flash.

```
PS C:\work\avr\avrdude_test\avrdude_bin> mode COM5 baud=12 dtr=on > nul && mode COM5 baud=12 dtr=off > nul && .\avrdude_git.exe -c usbasp -p m2560 -D -U flash:v:.\hex\blink-mega2560_lext-test.hex:i

avrdude_git.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude_git.exe: Device signature = 0x1e9801 (probably m2560)
avrdude_git.exe: verifying flash memory against .\hex\blink-mega2560_lext-test.hex:

Reading | ################################################## | 100% 0.19s

avrdude_git.exe: 200814 bytes of flash verified

avrdude_git.exe done.  Thank you.

```