avrdudes/avrdude #995: ft232r based programmer reading flash problem for ATmega2560

mcuee, 2022-06-17:
Programming seems to be no issue.

```
PS C:\work\avr\avrdude_test\avrdude-v7.0-windows-x64> cat .\eepromtest_intel.hex
:020000040001F9
:2010000054686520717569636B2062726F776E20666F78206A756D7073206F7665722074FE
:201020006865206C617A7920646F670A54686520717569636B2062726F776E20666F78206C
:201040006A756D7073206F76657220746865206C617A7920646F670A5468652071756963ED
:201060006B2062726F776E20666F78206A756D7073206F76657220746865206C617A7920C4
:20108000646F670A54686520717569636B2062726F776E20666F78206A756D7073206F76A5
:2010A000657220746865206C617A7920646F670A54686520717569636B2062726F776E20EE
:2010C000666F78206A756D7073206F76657220746865206C617A7920646F670A54686520B2
:2010E000717569636B2062726F776E20666F78206A756D7073206F76657220746865206C06
:20110000617A7920646F670A54686520717569636B2062726F776E20666F78206A756D7028
:2011200073206F76657220746865206C617A7920646F670A54686520717569636B20627269
:201140006F776E20666F78206A756D7073206F76657220746865206C617A7920646F670AFE
:2011600054686520717569636B2062726F776E20666F78206A756D7073206F76657220749D
:201180006865206C617A7920646F670A54686520717569636B2062726F776E20666F78200B
:2011A0006A756D7073206F76657220746865206C617A7920646F670A54686520717569638C
:2011C0006B2062726F776E20666F78206A756D7073206F76657220746865206C617A792063
:2011E000646F670A54686520717569636B2062726F776E20666F78206A756D7073206F7644
:00000001FF

PS C:\work\avr\avrdude_test\avrdude-v7.0-windows-x64> .\avrdude.exe -p m2560 -c arduino-ft232r -v -U flash:w:.\eepromtest_intel.hex:i

avrdude.exe: Version 7.0
             Copyright (c) Brian Dean, http://www.bdmicro.com/
             Copyright (c) Joerg Wunsch

             System wide configuration file is "C:/work/avr/avrdude_test/avrdude-v7.0-windows-x64/avrdude.conf"

             Using Port                    : usb
             Using Programmer              : arduino-ft232r
avrdude.exe: ft245r_open(): no device identifier in portname, using default
             AVR Part                      : ATmega2560
             Chip Erase delay              : 9000 us
             PAGEL                         : PD7
             BS2                           : PA0
             RESET disposition             : dedicated
             RETRY pulse                   : SCK
             Serial program mode           : yes
             Parallel program mode         : yes
             Timeout                       : 200
             StabDelay                     : 100
             CmdexeDelay                   : 25
             SyncLoops                     : 32
             PollIndex                     : 3
             PollValue                     : 0x53
             Memory Detail                 :

                                               Block Poll               Page                       Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               eeprom                 65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
               flash                  65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
               lfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               hfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               efuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               lock                    0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00
               signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00

             Programmer Type : ftdi_syncbb
             Description     : Arduino: FT232R connected to ISP
             Pin assignment  : 0..7 = DBUS0..7
               VCC     =  (not used)
               BUFF    =  (not used)
               RESET   =  7
               SCK     =  5
               MOSI    =  6
[... 134 lines cut ...]
               BUFF    =  (not used)
               RESET   =  7
               SCK     =  5
               MOSI    =  6
               MISO    =  3
               ERR LED =  (not used)
               RDY LED =  (not used)
               PGM LED =  (not used)
               VFY LED =  (not used)

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude.exe: Device signature = 0x1e9801 (probably m2560)
avrdude.exe: verifying flash memory against .\eepromtest_intel.hex:

Reading | ################################################## | 100% 0.00s

avrdude.exe: verification error, first mismatch at byte 0x11000
             0xff != 0x54
avrdude.exe: verification error; content mismatch

avrdude.exe done.  Thank you.
```

mcuee, 2022-06-17:
Not so sure this is similat to #474.
* #474

mcuee, 2022-06-17:
Reading back using the FT232R as empty and it was really alow.

```
PS C:\work\avr\avrdude_test\avrdude-v7.0-windows-x64> .\avrdude.exe -p m2560 -c arduino-ft232r -v -U flash:r:.\eepromtest_intel_readback_ft232r.hex:i -b 38400

avrdude.exe: Version 7.0
             Copyright (c) Brian Dean, http://www.bdmicro.com/
             Copyright (c) Joerg Wunsch

             System wide configuration file is "C:/work/avr/avrdude_test/avrdude-v7.0-windows-x64/avrdude.conf"

             Using Port                    : usb
             Using Programmer              : arduino-ft232r
             Overriding Baud Rate          : 38400
avrdude.exe: ft245r_open(): no device identifier in portname, using default
             AVR Part                      : ATmega2560
             Chip Erase delay              : 9000 us
             PAGEL                         : PD7
             BS2                           : PA0
             RESET disposition             : dedicated
             RETRY pulse                   : SCK
             Serial program mode           : yes
             Parallel program mode         : yes
             Timeout                       : 200
             StabDelay                     : 100
             CmdexeDelay                   : 25
             SyncLoops                     : 32
             PollIndex                     : 3
             PollValue                     : 0x53
             Memory Detail                 :

                                               Block Poll               Page                       Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               eeprom                 65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
               flash                  65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
               lfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               hfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               efuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               lock                    0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00
               signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00

             Programmer Type : ftdi_syncbb
             Description     : Arduino: FT232R connected to ISP
             Pin assignment  : 0..7 = DBUS0..7
               VCC     =  (not used)
               BUFF    =  (not used)
               RESET   =  7
               SCK     =  5
               MOSI    =  6
               MISO    =  3
               ERR LED =  (not used)
               RDY LED =  (not used)
               PGM LED =  (not used)
               VFY LED =  (not used)

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.03s

avrdude.exe: Device signature = 0x1e9801 (probably m2560)
avrdude.exe: reading flash memory:

Reading | ################################################## | 100% 2632.17s

avrdude.exe: Flash is empty, resulting file has no contents.
avrdude.exe: writing output file ".\eepromtest_intel_readback_ft232r.hex"

avrdude.exe done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude-v7.0-windows-x64> cat .\eepromtest_intel_readback_ft232r.hex
:00000001FF

````

mcuee, 2022-06-17:
A quick glance into the code shows that flash write uses `AVR_OP_LOAD_EXT_ADDR` but not flash read. Not so sure if this is the root cause of the issue or not.

mcuee, 2022-06-18:
Same thing happens with a smaller hex file -- wirting is okay, reading is not okay.

```
mcuee@mcuees-Mac-mini bin % ./avrdude -p m2560 -c arduino-ft232r -v -D -U flash:w:./entest.hex:i

avrdude: Version 7.0
         Copyright (c) Brian Dean, http://www.bdmicro.com/
         Copyright (c) Joerg Wunsch

         System wide configuration file is "/Users/mcuee/build/avr/avrdude_test/avrdude_macOS_64bit/bin/../etc/avrdude.conf"
         User configuration file is "/Users/mcuee/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : usb
         Using Programmer              : arduino-ft232r
avrdude: ft245r_open(): no device identifier in portname, using default
         AVR Part                      : ATmega2560
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom                 65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
           flash                  65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
           lfuse                   0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse                   0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           efuse                   0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lock                    0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           calibration             0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature               0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : ftdi_syncbb
         Description     : Arduino: FT232R connected to ISP
         Pin assignment  : 0..7 = DBUS0..7
           VCC     =  (not used)
           BUFF    =  (not used)
           RESET   =  7
           SCK     =  5
           MOSI    =  6
           MISO    =  3
           ERR LED =  (not used)
           RDY LED =  (not used)
           PGM LED =  (not used)
           VFY LED =  (not used)

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9801 (probably m2560)
avrdude: reading input file "./entest.hex"
avrdude: writing flash (512 bytes):

Writing | ################################################## | 100% 0.32s

avrdude: 512 bytes of flash written
avrdude: verifying flash memory against ./entest.hex:
[... 129 lines cut ...]
           lfuse                   0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse                   0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           efuse                   0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lock                    0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           calibration             0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature               0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : Arduino
         Description     : Arduino
         Hardware Version: 7
         Firmware Version: 7.6

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9801 (probably m2560)
avrdude: verifying flash memory against ./entest.hex:

Reading | ################################################## | 100% 0.06s

avrdude: 512 bytes of flash verified

avrdude done.  Thank you.
```

mcuee, 2022-06-18:
Interestingly termail mode seems to work.

```

mcuee@mcuees-Mac-mini bin % ./avrdude -p m2560 -c arduino-ft232r -t

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9801 (probably m2560)
avrdude> d
>>> d 
Usage: d <memtype> [<start addr> <len>]
       d <memtype> [<start addr> <...>]
       d <memtype> <...>
       d <memtype>
avrdude> d flash 0 512
>>> d flash 0 512 

Reading | ################################################## | 100% 1.02s

0000  54 68 65 0a 54 68 65 0a  54 68 65 0a 54 68 65 0a  |The The The The |
0010  54 68 65 0a 54 68 65 0a  54 68 65 0a 54 68 65 0a  |The The The The |
0020  54 68 65 0a 54 68 65 0a  54 68 65 0a 54 68 65 0a  |The The The The |
0030  54 68 65 0a 54 68 65 0a  54 68 65 0a 54 68 65 0a  |The The The The |
0040  54 68 65 0a 54 68 65 0a  54 68 65 0a 54 68 65 0a  |The The The The |
0050  54 68 65 0a 54 68 65 0a  54 68 65 0a 54 68 65 0a  |The The The The |
0060  54 68 65 0a 54 68 65 0a  54 68 65 0a 54 68 65 0a  |The The The The |
0070  54 68 65 0a 54 68 65 0a  54 68 65 0a 54 68 65 0a  |The The The The |
0080  54 68 65 0a 54 68 65 0a  54 68 65 0a 54 68 65 0a  |The The The The |
0090  54 68 65 0a 54 68 65 0a  54 68 65 0a 54 68 65 0a  |The The The The |
00a0  54 68 65 0a 54 68 65 0a  54 68 65 0a 54 68 65 0a  |The The The The |
00b0  54 68 65 0a 54 68 65 0a  54 68 65 0a 54 68 65 0a  |The The The The |
00c0  54 68 65 0a 54 68 65 0a  54 68 65 0a 54 68 65 0a  |The The The The |
00d0  54 68 65 0a 54 68 65 0a  54 68 65 0a 54 68 65 0a  |The The The The |
00e0  54 68 65 0a 54 68 65 0a  54 68 65 0a 54 68 65 0a  |The The The The |
00f0  54 68 65 0a 54 68 65 0a  54 68 65 0a 54 68 65 0a  |The The The The |
0100  54 68 65 0a 54 68 65 0a  54 68 65 0a 54 68 65 0a  |The The The The |
0110  54 68 65 0a 54 68 65 0a  54 68 65 0a 54 68 65 0a  |The The The The |
0120  54 68 65 0a 54 68 65 0a  54 68 65 0a 54 68 65 0a  |The The The The |
0130  54 68 65 0a 54 68 65 0a  54 68 65 0a 54 68 65 0a  |The The The The |
0140  54 68 65 0a 54 68 65 0a  54 68 65 0a 54 68 65 0a  |The The The The |
0150  54 68 65 0a 54 68 65 0a  54 68 65 0a 54 68 65 0a  |The The The The |
0160  54 68 65 0a 54 68 65 0a  54 68 65 0a 54 68 65 0a  |The The The The |
0170  54 68 65 0a 54 68 65 0a  54 68 65 0a 54 68 65 0a  |The The The The |
0180  54 68 65 0a 54 68 65 0a  54 68 65 0a 54 68 65 0a  |The The The The |
0190  54 68 65 0a 54 68 65 0a  54 68 65 0a 54 68 65 0a  |The The The The |
01a0  54 68 65 0a 54 68 65 0a  54 68 65 0a 54 68 65 0a  |The The The The |
01b0  54 68 65 0a 54 68 65 0a  54 68 65 0a 54 68 65 0a  |The The The The |
01c0  54 68 65 0a 54 68 65 0a  54 68 65 0a 54 68 65 0a  |The The The The |
01d0  54 68 65 0a 54 68 65 0a  54 68 65 0a 54 68 65 0a  |The The The The |
01e0  54 68 65 0a 54 68 65 0a  54 68 65 0a 54 68 65 0a  |The The The The |
01f0  54 68 65 0a 54 68 65 0a  54 68 65 0a 54 68 65 0a  |The The The The |

avrdude> q
>>> q 

avrdude done.  Thank you.

```

mcuee, 2022-06-18:
> A quick glance into the code shows that flash write uses `AVR_OP_LOAD_EXT_ADDR` but not flash read. Not so sure if this is the root cause of the issue or not. However that can not explain why the verfication stage of the writing operation works fine.

@stefanrueger
I tend to think this may have something similar to #474, any recomended quick fix to prove that not using `AVR_OP_LOAD_EXT_ADDR` is the issue with flash read; or prove it has nothing to do with that? Sorry I am not good at reading codes.

mcuee, 2022-06-18:
> > A quick glance into the code shows that flash write uses `AVR_OP_LOAD_EXT_ADDR` but not flash read. Not so sure if this is the root cause of the issue or not. However that can not explain why the verfication stage of the writing operation works fine.

That is not correct as `USE_INLINE_WRITE_PAGE` is not defined.
https://github.com/avrdudes/avrdude/blob/main/src/ft245r.c#L110
```
#define FT245R_CYCLES	2
#define FT245R_FRAGMENT_SIZE  512
#define REQ_OUTSTANDINGS	10
//#define USE_INLINE_WRITE_PAGE
```

https://github.com/avrdudes/avrdude/blob/main/src/ft245r.c#L1108
```
#if defined(USE_INLINE_WRITE_PAGE)
        if (do_page_write) {
            int addr_wk = addr_save - (addr_save % m->page_size);
            /* If this device has a "load extended address" command, issue it. */
            if (m->op[AVR_OP_LOAD_EXT_ADDR]) {
                unsigned char cmd[4];
                OPCODE *lext = m->op[AVR_OP_LOAD_EXT_ADDR];

                memset(cmd, 0, 4);
                avr_set_bits(lext, cmd);
                avr_set_addr(lext, cmd, addr_wk/2);
                buf_pos += set_data(pgm, buf+buf_pos, cmd[0]);
                buf_pos += set_data(pgm, buf+buf_pos, cmd[1]);
                buf_pos += set_data(pgm, buf+buf_pos, cmd[2]);
                buf_pos += set_data(pgm, buf+buf_pos, cmd[3]);
            }
            buf_pos += set_data(pgm, buf+buf_pos, 0x4C); /* Issue Page Write */
            buf_pos += set_data(pgm, buf+buf_pos,(addr_wk >> 9) & 0xff);
            buf_pos += set_data(pgm, buf+buf_pos,(addr_wk >> 1) & 0xff);
            buf_pos += set_data(pgm, buf+buf_pos, 0);
        }
#endif
```

> @stefanrueger I tend to think this may have something similar to #474, any recomended quick fix to prove that not using `AVR_OP_LOAD_EXT_ADDR` is the issue with flash read; or prove it has nothing to do with that? Sorry I am not good at reading codes.

mcuee, 2022-06-18:
Now I am guessing that the FT245R/FT232R based programmer, as of now, do not support ATmega2560, even though write seems to work, probably similar to the USBtinyISP issue.
* https://github.com/avrdudes/avrdude/issues/992

mcuee, 2022-06-18:
It is quite strange that read always shows empty.

```
PS C:\work\avr\avrdude_test\avrdude-v7.0-windows-x64> cat .\empty.hex
:00000001FF
PS C:\work\avr\avrdude_test\avrdude-v7.0-windows-x64> ./avrdude -p m2560 -c arduino-ft232r -v -D -U flash:v:./empty.hex

avrdude.exe: Version 7.0
             Copyright (c) Brian Dean, http://www.bdmicro.com/
             Copyright (c) Joerg Wunsch

             System wide configuration file is "C:/work/avr/avrdude_test/avrdude-v7.0-windows-x64/avrdude.conf"

             Using Port                    : usb
             Using Programmer              : arduino-ft232r
avrdude.exe: ft245r_open(): no device identifier in portname, using default
             AVR Part                      : ATmega2560
             Chip Erase delay              : 9000 us
             PAGEL                         : PD7
             BS2                           : PA0
             RESET disposition             : dedicated
             RETRY pulse                   : SCK
             Serial program mode           : yes
             Parallel program mode         : yes
             Timeout                       : 200
             StabDelay                     : 100
             CmdexeDelay                   : 25
             SyncLoops                     : 32
             PollIndex                     : 3
             PollValue                     : 0x53
             Memory Detail                 :

                                               Block Poll               Page                       Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               eeprom                 65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
               flash                  65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
               lfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               hfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               efuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               lock                    0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00
               signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00

             Programmer Type : ftdi_syncbb
             Description     : Arduino: FT232R connected to ISP
             Pin assignment  : 0..7 = DBUS0..7
               VCC     =  (not used)
               BUFF    =  (not used)
               RESET   =  7
               SCK     =  5
               MOSI    =  6
               MISO    =  3
               ERR LED =  (not used)
               RDY LED =  (not used)
               PGM LED =  (not used)
               VFY LED =  (not used)

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude.exe: Device signature = 0x1e9801 (probably m2560)
avrdude.exe: verifying flash memory against ./empty.hex:
avrdude.exe: input file ./empty.hex auto detected as Intel Hex

Reading | ################################################## | 100% 0.00s

avrdude.exe: 0 bytes of flash verified

avrdude.exe done.  Thank you.
```

stefanrueger, 2022-06-18:
@mcuee I like your systematic tests of boards and programmers; it is useful to explore the boundaries of the gamut of AVRDUDE programmers. Have you thought about a systematic test protocol? For example, erase the ATmega2560 with a *reliable* programmer; upload a quick-brown-fox test file that starts at 0x**2**1000 with the *to be tested* programmer (0x21000 is better than 0x11000); verify with a *reliable* programmer. If it verifies incorrectly, check where the *test* programmer wrote the test file to before you report the problem.

May I suggest  reporting any problems you find with the *least* amount of info that is useful to reproduce/understand the problem. The verbose AVRDUDE outputs makes it hard to understand what has happened. Also useful, if you can break long command lines into two lines so that no horizontal scrolling is necessary.

I expect most programmers to be able to read/write/verify the first 128k of an ATmega256 correctly as the load extended address is only needed above 128k (0x20000). However, as soon as you start to write files that cross that 0x20000 boundary, any programmer that is unable to handle 24-bit *word* addresses will get into trouble one way or the other.

mcuee, 2022-06-19:
> @mcuee I like your systematic tests of boards and programmers; it is useful to explore the boundaries of the gamut of AVRDUDE programmers. Have you thought about a systematic test protocol? For [...]

Good point. Now that I am more familiar with avrdude's usage and I am getting more interested in trying to understand avrdude codes (usually I do not loop deep enough about the codes for projects like libusb and OpenOCD -- they are too complicated), I will think about the testing strategy a bit more.

> May I suggest reporting any problems you find with the _least_ amount of info that is useful to reproduce/understand the problem. The verbose AVRDUDE outputs makes it hard to understand what has [...]

Very good advice. Thanks.

> > even though write seems to work
> 
> I expect most programmers to be able to read/write/verify the first 128k of an ATmega256 correctly as the load extended address is only needed above 128k (0x20000). However, as soon as you start to write files that cross that 0x20000 boundary, any programmer that is unable to handle 24-bit _word_ addresses will get into trouble one way or the other.

Good point. I will try this -- I should be able to learn "just enough" of srecord to create the test hex files.

mcuee, 2022-06-21:
Using the hex file provided by Stefan (https://github.com/avrdudes/avrdude/files/8943338/blink-mega2560%2Blext-test.txt )and it is the same result.

```
C:\work\avr\binary\avrdude-7.0_bin64> .\avrdude_sr474 -qqp m2560 -c arduino-ft232r
 -U flash:v:blink-mega2560_lext-test.hex:i && echo OK
avrdude_sr474.exe: verification error, first mismatch at byte 0x0000
                   0xff != 0x76
avrdude_sr474.exe: verification error; content mismatch
```

mcuee, 2022-06-21:
Strange thing when writing a smaller hex file without crossing the 64KB bounday.

1) Writing is okay and the verification step within the writing operation is okay, verified by usbasp to be correct as well.
2) But reading still shows empty result.

```
C:\work\avr\binary\avrdude-7.0_bin64> .\avrdude -qqp m2560 -c arduino-ft232r -U flash:w:entest.hex:i && echo OK
OK
C:\work\avr\binary\avrdude-7.0_bin64> .\avrdude -qqp m2560 -c arduino-ft232r -U flash:v:entest.hex:i && echo OK
avrdude.exe: verification error, first mismatch at byte 0x0000
             0xff != 0x54
avrdude.exe: verification error; content mismatch
C:\work\avr\binary\avrdude-7.0_bin64> .\avrdude -qqp m2560 -c usbasp -U flash:v:entest.hex:i && echo OK
OK
```

mcuee, 2022-06-21:
Side note: no issues with ATmega328p.

```
C:\work\avr\binary\avrdude-7.0_bin64> .\avrdude -qqp m328p -c arduino-ft232r 
-U flash:w:.\entest.hex:i && echo OK
OK
C:\work\avr\binary\avrdude-7.0_bin64> .\avrdude -qqp m328p -c arduino-ft232r 
-U flash:v:.\entest.hex:i && echo OK
OK
C:\work\avr\binary\avrdude-7.0_bin64> .\avrdude -qqp m328p -c usbasp 
-U flash:v:.\entest.hex:i && echo OK
OK

```

mcuee, 2022-06-21:
I borrowed another FT232R based dongle and it has the same results.
```
C:\work\avr\binary\avrdude-7.0_bin64> .\avrdude -qqp m2560 -c ttl232r
 -U flash:w:.\entest.hex:i && echo OK
OK
C:\work\avr\binary\avrdude-7.0_bin64> .\avrdude -qqp m2560 -c ttl232r 
-U flash:v:.\entest.hex:i && echo OK
avrdude.exe: verification error, first mismatch at byte 0x0000
             0xff != 0x54
avrdude.exe: verification error; content mismatch
C:\work\avr\binary\avrdude-7.0_bin64> .\avrdude -qqp m2560 -c usbasp
 -U flash:v:.\entest.hex:i && echo OK
OK

```

stefanrueger, 2022-06-23:
> 1. Writing is okay and the verification step within the writing operation is okay, verified by usbasp to be correct as well.
> 2. But reading still shows empty result.
> 3. Terminal mode is correct.

@mcuee Great testing. Looks like a fiendishly difficult problem. Terminal mode deploys *byte-wise* access, so that seems to work. Some of the *paged* reads don't seem to work: verification (ie, reading) after writing does, but strangely verification alone does not, nor does read alone). 

I wonder what happens if you removed *paged* reads by commenting the following line
https://github.com/avrdudes/avrdude/blob/1aa59aaa98e6a79bb4a321be80c5113184dd2668/src/ft245r.c#L1281

stefanrueger, 2022-06-24:
> I wonder what happens if you removed paged reads

@mcuee The rationale for this test is that by removing the page read functionality, AVRDUDE should switch to byte-wise read, which might work (it does in the terminal, which is byte-read). If it does, we would know the problem is in `ft245r_paged_load()` (and we would have a quick fix while we look at this).

mcuee, 2022-06-25:
> @mcuee The rationale for this test is that by removing the page read functionality, AVRDUDE should switch to byte-wise read, which might work (it does in the terminal, which is byte-read). If it [...]

Yes now it works.

```
PS C:\work\avr\avrdude> git diff
diff --git a/src/ft245r.c b/src/ft245r.c
index d5af4ca..de0de1f 100644
--- a/src/ft245r.c
+++ b/src/ft245r.c
@@ -1278,7 +1278,7 @@ void ft245r_initpgm(PROGRAMMER * pgm) {
      * optional functions
      */
     pgm->paged_write = ft245r_paged_write;
-    pgm->paged_load = ft245r_paged_load;
+    //pgm->paged_load = ft245r_paged_load;

     pgm->rdy_led        = set_led_rdy;
     pgm->err_led        = set_led_err;

PS C:\work\avr\avrdude_test\avrdude-7.0_bin64> .\avrdude_issue995.exe -qqp m2560 -c arduino-ft232r 
-U flash:w:.\entest_offset_0x21000.hex:i && echo OK
OK

PS C:\work\avr\avrdude_test\avrdude-7.0_bin64> .\avrdude_issue995.exe -qqp m2560 -c arduino-ft232r 
-U flash:v:.\entest_offset_0x21000.hex:i && echo OK
OK

```

stefanrueger, 2022-06-27:
@mcuee Brilliant. Thank you. So, we got this working by falling back to byte-wise reads. And we know the culprit is the paged read routine. We could leave it at this was byte-wise access not a lot slower than paged as it has an overhead for each byte, *and* on an ATmega2560 there will be a load extended address command prepended for *each* byte.  

OK. I have an USB BUB II with an FT232R on it, so I'll try to see how I can use that to debug the paged read.
[image]
Reading the source didn't show an obvious culprit, but once I have my own FT232R set up I should be able to crack this conundrum. A few non-obvious lines already, just by reading the source:
https://github.com/avrdudes/avrdude/blob/362e6993acc863c282a16fd4d3642ebbeab4eae7/src/ft245r.c#L1096-L1099
Here the SPI write opcode is used directly 0x48 for hi byte and 0x40 for lo byte. The code really should consult mem->op[...]. 

And there is the curious line where 9 bit try to squeeze into an `unsigned char ft245r_ddr`:
https://github.com/avrdudes/avrdude/blob/362e6993acc863c282a16fd4d3642ebbeab4eae7/src/ft245r.c#L916-L925
This probably makes sense in some way or another (mapping 9 signals onto 8 data lines that the user somehow configures?), but made me do a double-take on first scanning.
 
I aim to have a closer look over next week.

stefanrueger, 2022-07-04:
See PR #1018.

Had second thoughts about how to fix this issue: I went with Option 2. While Option 3 above is a perfectly workable solution, it would have duplicated the functionality of `avr_read_byte_default()` and `avr_write_page()`. The *much better* approach is to

  4. optimise load extended address instructions for these *generic* functions in `avr.c` in addition to Option 2

Out of necessity this *requires* avr.c and the programmers share a *single* cached extended address byte, so that mixed use of *default* routines in avr.c and programmer functionality can occur. There really can only be *one* cached byte!

A natural place for the shared byte would be either
  1. in the AVRPART structure (where adding one byte will use up 312 bytes, the number of parts)
  2. or a single global byte 
  3. or a static hidden one in a to-be-written lext_cache() access function

For now, I will put this on the back burner, as implementing the single cache requires reqriting individual caching in programmers. Once done, though, terminal dumps of large-ish flash sections of ATmega2560 et al will run at roughly double speed.

closed 2022-07-12 by stefanrueger