avrdudes/avrdude #1005: Wrong number of bytes written/verified displayed when input file has holes

stefanrueger, 2022-06-21:
AVRDUDE wrongly shows the highest flash address+1 as the number of bytes written and verified, respectively, when programming a test file with holes, [blink-mega2560+lext-test.hex](https://github.com/avrdudes/avrdude/files/8943338/blink-mega2560%2Blext-test.txt) with a small code file and 3 tiny data:

```
$ avrdude -qp m2560 -c ... -U flash:w:blink-mega2560+lext-test.hex:i
avrdude: AVR device initialized and ready to accept instructions
avrdude: Device signature = 0x1e9801 (probably m2560)
avrdude: reading input file "blink-mega2560+lext-test.hex"
avrdude: writing flash (200814 bytes):
avrdude: 200814 bytes of flash written
avrdude: verifying flash memory against blink-mega2560+lext-test.hex:
avrdude: load data flash data from input file blink-mega2560+lext-test.hex:
avrdude: input file blink-mega2560+lext-test.hex contains 200814 bytes
avrdude: reading on-chip flash data:
avrdude: verifying ...
avrdude: 200814 bytes of flash verified

avrdude: safemode: Fuses OK (E:00, H:00, L:00)

avrdude done.  Thank you.
```

Actually, only 1346 bytes were programmed, and this number would be the one to display. Depending on the verbosity level, the output could be more  informative, eg,
```
avrdude: reading input file "blink-mega2560+lext-test.hex" with 1346 bytes in
         4 sections, the largest of which with has 936 bytes, spanning the
         interval [$0, $312A1] = [0, 200813] spread over 7 pages
```

Also, some programmers overwrite memory (think spm erase and write in bootloaders), rather than anding the input to the existing contents (which is what *low-level writes* to NOR flash memories do). So when using `-D` or when programming the second `-U flash:w:...` it might be neat to add `erasing 446 pad bytes`.

Thoughts?

mcuee, 2022-07-28:
This is indeed a common problem. The following hex file is the simple blinking FW for Arduino Uno (exported from the IDE, with bootloader).

```
C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_git.exe -c usbasp -p m328p -U .\blink_uno_with_bootloader.hex

avrdude_git.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude_git.exe: Device signature = 0x1e950f (probably m328p)
avrdude_git.exe: NOTE: "flash" memory has been specified, an erase cycle will be performed
                 To disable this feature, specify the -D option.
avrdude_git.exe: erasing chip
avrdude_git.exe: reading input file ".\blink_uno_with_bootloader.hex"
avrdude_git.exe: input file .\blink_uno_with_bootloader.hex auto detected as Intel Hex
avrdude_git.exe: writing flash (32768 bytes):

Writing | ################################################## | 100% 0.35s

avrdude_git.exe: 32768 bytes of flash written
avrdude_git.exe: verifying flash memory against .\blink_uno_with_bootloader.hex:
avrdude_git.exe: input file .\blink_uno_with_bootloader.hex auto detected as Intel Hex

Reading | ################################################## | 100% 0.21s

avrdude_git.exe: 32768 bytes of flash verified

avrdude_git.exe done.  Thank you.

```
[blink_uno_with_bootloader.hex.txt](https://github.com/avrdudes/avrdude/files/9205775/blink_uno_with_bootloader.hex.txt)

stefanrueger, 2022-08-04:
Fixed by PR #1053

closed 2022-08-04 by stefanrueger