avrdudes/avrdude #1007: github action MSVC build

mcuee, 2022-06-22:
It will be good to copy the avrdude.conf file in the same directory as the avrdude.exe file.

```
C:\work\avr\binary\build-msvc-x64\src\RelWithDebInfo> .\avrdude.exe -p m328p -c pickit2 -P usb -vvvv
executable_abspath = C:/work/avr/binary/build-msvc-x64/src/RelWithDebInfo/avrdude.exe
executable_abspath_len = 64
executable_dirpath = C:/work/avr/binary/build-msvc-x64/src/RelWithDebInfo
executable_dirpath_len = 52
sys_config =
sys_config_found = false

avrdude.exe: Version 7.0-20220619 (1aa59aa)
             Copyright (c) Brian Dean, http://www.bdmicro.com/
             Copyright (c) Joerg Wunsch

             System wide configuration file is ""
avrdude.exe: can't open config file "": Invalid argument
avrdude.exe: error reading system wide configuration file ""
```

mcuee, 2022-06-22:
Files copy from latest github action.
https://github.com/avrdudes/avrdude/actions/runs/2524832038

mariusgreuel, 2022-06-22:
The build artifacts (build-*) are just a complete dump of the CMake build folder. I would rather keep the build folder unmodified, as it is indented for build debugging.

If you want the executables, there are separate artifacts (avrdude-*), which in case of MSVC contain the exe, conf, and symbols.

mcuee, 2022-06-22:
No problem.  I will close this issue.

closed 2022-06-22 by mcuee