avrdudes/avrdude #1011: Problem downloading EEPROM from Arduino Mega 2650

ravenboy, 2022-06-24:
Mac Big Sur, avrdude 6.3

in a sketch running on a Mega2650: using EEPROM library I wrote "757343238393515142B0" starting at location Zero

char buffer[] = "757343238393515142B0";
for (int i=0; i<21; i++) {
     EEPROM.write(i,buffer[i]);
 }

reading it works fine

char buffer2[21] 
for (int i=0; i<21; i++) {
     buffer2[i] = EEPROM.read(i);
 }

buffer2[20]='\0';
Serial.printlin (buffer2);
prints "757343238393515142B0" to serial

when i download the EEPROM memory using AVRDude, i get no errors.
looking at the file with a hexdump utility, i see
0000-0010: 3a 32 30 30-30 30 30 30-30 33 37 33-35 33 37 33 :2000000 03735373
0000-0020: 33 33 34 33-33 33 32 33-33 33 34 33-32 34 32 33 33433323 33432423
0000-0030: 30 30 30 46-46 46 46 46-46 46 46 46-46 46 46 46 000FFFFF FFFFFFFF
0000-0040: 46 46 46 46-46 46 46 46-46 46 46 46-46 46 46 46 FFFFFFFF FFFFFFFF

which has only 12 of the bytes i wrote to EEPROM
hex 37 35 37 33 34 33 32 33 34 32 42 30
ascii 7   5   7   3   4   3   2    3   4   2   B   0

i wrote 20 bytes 757343238393515142B0
so the download doesn't have 8 bytes in the middle.
the bytes missing start at position 7 and are the 8 bytes 83935151
so the download has the first 7 bytes of what i wrote
plus the last 4, but lacks bytes 8 thru 15

any idea how avrdude's download would disagree with what the arduino sketch can READ??

i used
avrdude -p m2560 -c wiring -P [port] -U eeprom:r:file.hex:i

mcuee, 2022-06-25:
That is probably becasue the wiring bootloader has an issue with EEPROM read. You need to build the bootloader again with this patch.

Ref:
* https://github.com/arduino/ArduinoCore-avr/pull/24
* https://github.com/avrdudes/avrdude/issues/986

mcuee, 2022-06-25:
I will close this issue now. Please re-open if the above fix does not sort out your issue.

closed 2022-06-25 by mcuee

ravenboy, 2022-06-26:
manu thanks!

ravenboy, 2022-06-26:
Thanks so much for the speedy and excellent support.  Once i sussed out burning bootloaders, burning with stk500boot_v2_mega2560_eeprom_fix.hex is working!!!