avrdudes/avrdude #1012: Using brew to build on OSX Big Sur fails

ravenboy, 2022-06-24:
your directions say to use

brew install cmake hidapi libftdi1 libusb libelf

this command fails:

Warning: Treating cmake as a formula. For the cask, use homebrew/cask/cmake
Warning: No available formula with the name "libftdi1". Did you mean libftdi or libftdi0?
==> Searching for similarly named formulae...
These similarly named formulae were found:
libftdi                                  libftdi0 ✔
To install one of them, run (for example):
  brew install libftdi
==> Searching for a previously deleted formula (in the last month)...
Error: No previously deleted formula found.
==> Searching taps on GitHub...
Error: No formulae found in taps.
raventalking@Johns-MacBook-Pro ~ %

mcuee, 2022-06-25:
You are right, you should install libftdi instead.
https://github.com/avrdudes/avrdude/wiki/Building-AVRDUDE-for-macOS --> need to be updated for Homebrew

Ref:
https://formulae.brew.sh/formula/libftdi
https://formulae.brew.sh/formula/libftdi0 (this will work as well but not recommended).

mcuee, 2022-06-25:
@MCUdude or @mariusgreuel 
Please help to fix the Wiki. Thanks.

MCUdude, 2022-06-25:
@mcuee just change the brew command in the wiki to:

```
brew install cmake hidapi libftdi libusb libelf
```

Right?

mcuee, 2022-06-26:
Yes that is correct.

ravenboy, 2022-06-26:
thanks much

ravenboy, 2022-06-26:
thanks much

MCUdude, 2022-06-26:
Wiki updated!

closed 2022-06-26 by MCUdude