avrdudes/avrdude #1017: opening programmer "snap_isp" on port "usb" failed

lkoepsel, 2022-06-30:
I'm running Manjaro (Arch Linux) and am having problems using avrdude programming an Uno with a Microchip SNAP. I have been successful using the same hardware combination and command on macOS Monterey.
```
avrdude -c snap_isp -p atmega328p -P usb -vv -t 
```
On Manjaro, I get the following:
```
lkoepsel@manjaroMBP  » avrdude -c snap_isp -p atmega328p -P usb -vv -t                     1 ↵

avrdude: Version 7.0
         Copyright (c) Brian Dean, http://www.bdmicro.com/
         Copyright (c) Joerg Wunsch

         System wide configuration file is "/etc/avrdude.conf"
         User configuration file is "/home/lkoepsel/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : usb
         Using Programmer              : snap_isp
avrdude: stk500v2_jtag3_open()
avrdude: usbdev_open(): Found MPLAB Snap ICD CMSIS-DAP, serno: BUR190971368
avrdude: usbdev_open(): WARNING: failed to set configuration 1: Device or resource busy
avrdude: Found CMSIS-DAP compliant device, using EDBG protocol
avrdude: usbdev_send(): wrote -5 out of 912 bytes, err = Input/output error
avrdude: jtag3_edbg_prepare(): failed to send command to serial port
avrdude: failed to sync with the JTAGICE3 in ISP mode
avrdude: opening programmer "snap_isp" on port "usb" failed

avrdude done.  Thank you.
```

On the mac, I get the following:
```
avrdude -c snap_isp -p atmega328p -P usb -vv -t

avrdude: Version 7.0
         Copyright (c) Brian Dean, http://www.bdmicro.com/
         Copyright (c) Joerg Wunsch

         System wide configuration file is "/opt/homebrew/etc/avrdude.conf"
         User configuration file is "/Users/lkoepsel/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : usb
         Using Programmer              : snap_isp
avrdude: stk500v2_jtag3_open()
avrdude: usbhid_open(): No device found
avrdude: Found CMSIS-DAP compliant device, using EDBG protocol
avrdude: jtag3_edbg_prepare(): connection status 0x01
avrdude: Sending sign-on command: 0x80 (3 bytes msg)
         AVR Part                      : ATmega328P
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom                 65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
           flash                  65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
           lfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           hfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           efuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           lock                    0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00
           signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00

         Programmer Type : JTAG3_ISP
         Description     : MPLAB(R) SNAP in ISP mode
avrdude: jtag3_getparm()
avrdude: Sending get parameter (scope 0x01, section 1, parm 0) command: 0x84 (6 bytes msg)
         Vtarget         : 5.2 V
         SCK period      : 8.00 us

avrdude: jtag3_setparm()
avrdude: Sending set parameter (scope 0x12, section 0, parm 0) command: 0x80 (3 bytes msg)
avrdude: jtag3_setparm()
avrdude: Sending set parameter (scope 0x12, section 0, parm 1) command: 0x80 (3 bytes msg)
avrdude: jtag3_setparm()
avrdude: Sending set parameter (scope 0x12, section 1, parm 0) command: 0x80 (3 bytes msg)
avrdude: AVR device initialized and ready to accept instructions

Reading |                                                    | 0% 0.00savrdude: stk500isp_read_byte(.., signature, 0x0, ...)
avrdude: stk500isp_read_byte(): Sending read memory command: avrdude: stk500isp_read_byte(.., signature, 0x1, ...)
Reading | #################                                  | 33% 0.00savrdude: stk500isp_read_byte(.., signature, 0x2, ...)
Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e950f (probably m328p)
avrdude>
```

Any guidance as to how to solve?

Thank you,
Lief

mcuee, 2022-07-01:
> avrdude: usbdev_open(): WARNING: failed to set configuration 1: Device or resource busy

Looks like missing udev rules to have the proper access right.

lkoepsel, 2022-07-01:
Thank you, @dl8dtl, however, I'm not using a VM.

I did use the example from pyedbglib and was able to get information from the device:
```
pedgb.py
pyedbglib version 2.20.3.105
Debugger firmware is version 1.10.249
Target voltage is 5.10V
```
If I disconnect the target, the voltage goes to 0.3V as expected, so clearly pyedbglib is chatting with the board.

Bloom as in https://bloom.oscillate.io/ is also able to speak to the board. I thought I might have a contention/conflict, however when I uninstalled Bloom the problem remained.

I'll continue to debug...its a bit maddening.

lkoepsel, 2022-07-02:
Thanks to guidance from @navnavnav, I downloaded the latest artifact and it now works.

closed 2022-07-02 by lkoepsel