avrdudes/avrdude #1020: Write flash rarely works in terminal mode for classic AVRs

stefanrueger, 2022-07-07:
The reason for this is that terminal mode relies on the generic `avr_write_byte()` routine
https://github.com/avrdudes/avrdude/blob/64ee4858fd5e4a2ce65cf3c280274674c4320de5/src/term.c#L569-L570
 - Which in turn dispatches to the individual programmer's write byte routine
 - Which in turn more often than not (see below for affected programmers) falls back on `avr_write_byte_default()`
 - Which in turn only caters for low-level TPI (these cannot write a byte to flash!) and for low-level "classic" AVR ISP
 - Which in turn for flash only does a proper write for 11 parts with byte-write commands (since you asked: `ATtiny12`, `ATtiny15`, `AT90S1200`, `AT90S4414`, `AT90S2313`, `AT90S2333`, `AT90S2343`, `AT90S4433`, `AT90S4434`, `AT90S8515`, `AT90S8535`)
 - Which in turn means that for the other 153 classic AVR parts known to AVRDUDE (don't ask) only a page load is carried out
 - Which in turn means that when `term.c` reads back to check there will be a failure
https://github.com/avrdudes/avrdude/blob/64ee4858fd5e4a2ce65cf3c280274674c4320de5/src/term.c#L581-L586

Affected programmers: `avrftdi.c`, `buspirate.c`, `ft245r.c`, `linuxgpio.c`, `linuxspi.c`, `par.c`, `pickit2.c`, `serbb_posix.c`, `serbb_win32.c`, `stk500.c`, `usbasp.c`, `usbtiny.c`

For those who prefer to test, run `avrdude -t` with  your favourite AVR part and your favourite programmer:
```
$ echo  'write flash 0x3100 "hello, world\n"' | avrdude -qqp m328p -c usbtiny -t
avrdude> write flash 0x3100 "hello, world\n"
>>> write flash 0x100 "hello, world\n" 
avrdude (write): error writing 0x68 at 0x03100 cell=0xff
avrdude (write): error writing 0x65 at 0x03101 cell=0xff
avrdude (write): error writing 0x6c at 0x03102 cell=0xff
avrdude (write): error writing 0x6c at 0x03103 cell=0xff
avrdude (write): error writing 0x6f at 0x03104 cell=0xff
avrdude (write): error writing 0x2c at 0x03105 cell=0xff
avrdude (write): error writing 0x20 at 0x03106 cell=0xff
avrdude (write): error writing 0x77 at 0x03107 cell=0xff
avrdude (write): error writing 0x6f at 0x03108 cell=0xff
avrdude (write): error writing 0x72 at 0x03109 cell=0xff
avrdude (write): error writing 0x6c at 0x0310a cell=0xff
avrdude (write): error writing 0x64 at 0x0310b cell=0xff
avrdude (write): error writing 0x5c at 0x0310c cell=0xff
avrdude (write): error writing 0x6e at 0x0310d cell=0xff
avrdude (write): error writing 0x00 at 0x0310e cell=0xff
```
I have not looked at those programmers that specify their own write byte routine; these *might* work in terminal mode.

Of course, if any program should be able to modify a flash byte interactively, it should be AVRDUDE, not? Although AVRDUDE started out as a simple uploader/downloader 20 years ago, it's now considered *the* ultimate AVR programming tool these days.

stefanrueger, 2022-07-07:
My proposal is to provide in `avr.c` generic `avr_read_byte_cache()` and  `avr_write_byte_cache()` routines  that, for programmers that have paged read/write functions,
 - Maintain a page-sized *flash* read/write cache for all *parts with paged flash* memory (not only classic AVRs)
 - Maintain a 256-byte *EEPROM* read/write cache (or less if the EEPROM is smaller) for all parts with EEPROM memory
 - Flush the respective cache (if modified) on switching a page or trying to access a byte outside the valid address range
 - Fall back to `avr_read_byte_default()` and `avr_write_byte_default()`, respectively, otherwise

`term.c`  and SPM programmers (which serve bootloaders that normally *only* have paged r/w) can make use of the `avr_..._byte_cache()` functions instead of the current `avr_..._byte_default()` routines.

I have written a cache for the urclock SPM programmer draft that can be utilised here, see https://github.com/avrdudes/avrdude/discussions/940#discussioncomment-2977292

Thoughts?

stefanrueger, 2022-07-07:
> by reading a flash page, modifying the content and writing it back again

Exactly! With a r/w cache (for speed and lest wearing out memory unnecessarily) and using the *paged* r/w functions of the programmer (for generality)

stefanrueger, 2022-07-07:
>  support other data than bytes

Yes, I noted the addition of strings, etc. Vvv useful!

dl8dtl, 2022-07-07:
Interesting side note: when I tried to update my STK500 using an AVRISPmkII (as the built-in avr910 loader failed), that doesn't work either – it's an AT90S8535 (where the byte write ought to work) but somehow, stk500v2 broke that. Maybe due to decomissioning the SPI_MULTI command.

dl8dtl, 2022-07-07:
There are already some programmers implementing caches (I think JTAG programmers). Generalizing that might be a good move.

mcuee, 2022-07-08:
Just some quick test to confirm that there are problems.

```
PS C:\work\avr\avrdude_test\avrdude-7.0_bin64> echo  'write flash 0x3100 "hello, world\n"' | .\avrdude -qqp m2560 
-c usbasp -t
avrdude> >>> write flash 0x3100 "hello, world\n"
avrdude.exe (write): error writing 0x68 at 0x03100 cell=0xff
avrdude.exe (write): error writing 0x65 at 0x03101 cell=0xff
avrdude.exe (write): error writing 0x6c at 0x03102 cell=0xff
avrdude.exe (write): error writing 0x6c at 0x03103 cell=0xff
avrdude.exe (write): error writing 0x6f at 0x03104 cell=0xff
avrdude.exe (write): error writing 0x2c at 0x03105 cell=0xff
avrdude.exe (write): error writing 0x20 at 0x03106 cell=0xff
avrdude.exe (write): error writing 0x77 at 0x03107 cell=0xff
avrdude.exe (write): error writing 0x6f at 0x03108 cell=0xff
avrdude.exe (write): error writing 0x72 at 0x03109 cell=0xff
avrdude.exe (write): error writing 0x6c at 0x0310a cell=0xff
avrdude.exe (write): error writing 0x64 at 0x0310b cell=0xff
avrdude.exe (write): error writing 0x5c at 0x0310c cell=0xff
avrdude.exe (write): error writing 0x6e at 0x0310d cell=0xff
avrdude.exe (write): error writing 0x00 at 0x0310e cell=0xff
avrdude>

```

mcuee, 2022-07-08:
@stefanrueger Your prediction is pretty accurate except PICKit2.

stefanrueger, 2022-07-08:
@mcuee Many thanks for testing! And for developing a good test bench. I am convinced that this will improve the quality of AVRDUDE over time (actually, it already has!), and it gives us more confidence for merging PRs.

> pretty accurate except PICKit2

I still think terminal write won't work for PICKit2. Could you please try the test again after erasing flash. 

Explanation: before  testing `-c pickit2` the test `-c wiring` succeeded, which means that the test-string was in flash (for the first time in the test series). The test with `-c pickit2` tried writing to flash and will have failed, but since the string was actually there from the previous successful `-c wiring` test,  the `-c pickit2` avr_write_byte() routine did not notice the problem on reading back().

Maybe include into the test protocol the preparation of the MCU to defined state or alternating test strings between tests?

mcuee, 2022-07-08:
> I still think terminal write won't work for PICKit2. Could you please try the test again after erasing flash.
>
> Explanation: before testing `-c pickit2` the test `-c wiring` succeeded, which means [...]

@stefanrueger 
Ah, you are exactly right. PICKit 2 also does not work. 

```
PS C:\work\avr\avrdude_test\avrdude-7.0_bin64> .\avrdude_issue1004v3.exe -p m2560 -c pickit2 -e

avrdude_issue1004v3.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.02s

avrdude_issue1004v3.exe: Device signature = 0x1e9801 (probably m2560)
avrdude_issue1004v3.exe: erasing chip

avrdude_issue1004v3.exe done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude-7.0_bin64> echo  'write flash 0x3100 "hello, world\n"' | .\avrdude_issue1004v3.exe 
-qqp m2560 -c pickit2 -t
avrdude> >>> write flash 0x3100 "hello, world\n"
avrdude_issue1004v3.exe (write): error writing 0x68 at 0x03100 cell=0xff
avrdude_issue1004v3.exe (write): error writing 0x65 at 0x03101 cell=0xff
avrdude_issue1004v3.exe (write): error writing 0x6c at 0x03102 cell=0xff
avrdude_issue1004v3.exe (write): error writing 0x6c at 0x03103 cell=0xff
avrdude_issue1004v3.exe (write): error writing 0x6f at 0x03104 cell=0xff
avrdude_issue1004v3.exe (write): error writing 0x2c at 0x03105 cell=0xff
avrdude_issue1004v3.exe (write): error writing 0x20 at 0x03106 cell=0xff
avrdude_issue1004v3.exe (write): error writing 0x77 at 0x03107 cell=0xff
avrdude_issue1004v3.exe (write): error writing 0x6f at 0x03108 cell=0xff
avrdude_issue1004v3.exe (write): error writing 0x72 at 0x03109 cell=0xff
avrdude_issue1004v3.exe (write): error writing 0x6c at 0x0310a cell=0xff
avrdude_issue1004v3.exe (write): error writing 0x64 at 0x0310b cell=0xff
avrdude_issue1004v3.exe (write): error writing 0x5c at 0x0310c cell=0xff
avrdude_issue1004v3.exe (write): error writing 0x6e at 0x0310d cell=0xff
avrdude_issue1004v3.exe (write): error writing 0x00 at 0x0310e cell=0xff
avrdude>

```

mcuee, 2022-07-08:
And I verfied again that wiring bootloader works fine.

```
PS C:\work\avr\avrdude_test\avrdude-7.0_bin64> .\avrdude -qqp m2560 -c wiring -P COM6 -t
avrdude> write flash 0x3100 "hello, world\n"
>>> write flash 0x3100 "hello, world\n"
avrdude> dump flash 0x3100 0x10
>>> dump flash 0x3100 0x10
3100  68 65 6c 6c 6f 2c 20 77  6f 72 6c 64 5c 6e 00 ff  |hello, world\n..|

avrdude> q
>>> q

```

mcuee, 2022-07-08:
ATmega328PB should count as classic AVR. xplainedmini programer (built-in programmer for the Microchip ATmega328PB-XMINI) seems to work fine.

```
PS C:\work\avr\avrdude_test\avrdude-7.0_bin64> .\avrdude.exe -qqp m328pb -c xplainedmini -t
avrdude>  write flash 0x3100 "hello, world\n"
>>> write flash 0x3100 "hello, world\n"
avrdude> dump flash 0x3100 0x10
>>> dump flash 0x3100 0x10
3100  68 65 6c 6c 6f 2c 20 77  6f 72 6c 64 5c 6e 00 ff  |hello, world\n..|
```

Then I tested two new UPDI AVRs and they work fine as well.

1) Arduino Nano Every official board with ATmega4809 and built-in jtag2updi programmer
```
PS C:\work\avr\avrdude_test\avrdude-7.0_bin64> mode COM7 baud=12 dtr=on > nul
PS C:\work\avr\avrdude_test\avrdude-7.0_bin64> mode COM7 baud=12 dtr=off > nul
PS C:\work\avr\avrdude_test\avrdude-7.0_bin64> .\avrdude.exe -qqp m4809 -c jtag2updi -P COM7 -t
avrdude> write flash 0x3100 "hello, world\n"
>>> write flash 0x3100 "hello, world\n"
avrdude> dump flash 0x3100 0x10
>>> dump flash 0x3100 0x10
3100  68 65 6c 6c 6f 2c 20 77  6f 72 6c 64 5c 6e 00 ff  |hello, world\n..|
```

2) Microchip ATmega128DB48 Curiosity Nano with built-in pkobn_updi programmer.
```
PS C:\work\avr\avrdude_test\avrdude-7.0_bin64> .\avrdude.exe -c pkobn_updi -qqp avr128db48 -t
             Vtarget                      : 3.31 V
             PDI/UPDI clock Xmega/megaAVR : 100 kHz

avrdude> write flash 0x3100 "hello, world\n"
>>> write flash 0x3100 "hello, world\n"
avrdude> dump flash 0x3100 0x10
>>> dump flash 0x3100 0x10
3100  68 65 6c 6c 6f 2c 20 77  6f 72 6c 64 5c 6e 00 ff  |hello, world\n..|
```

mcuee, 2022-07-08:
AVR109 bootloader for Arduino Leonardo does not seem to support flash write at all in terminal mode.

```
PS C:\work\avr\avrdude_test\avrdude-7.0_bin64> .\leonardo_terminal_test.bat
COM5

Status for device COM5:
-----------------------
    Baud:            1200
    Parity:          None
    Data Bits:       8
    Stop Bits:       1
    Timeout:         OFF
    XON/XOFF:        OFF
    CTS handshaking: OFF
    DSR handshaking: OFF
    DSR sensitivity: OFF
    DTR circuit:     OFF
    RTS circuit:     ON

COM30

Status for device COM30:
------------------------
    Baud:            57600
    Parity:          None
    Data Bits:       8
    Stop Bits:       1
    Timeout:         OFF
    XON/XOFF:        OFF
    CTS handshaking: OFF
    DSR handshaking: OFF
    DSR sensitivity: OFF
    DTR circuit:     OFF
    RTS circuit:     ON

C:\work\avr\avrdude_test\avrdude-7.0_bin64>.\avrdude -p atmega32u4 -c avr109 -P COM30 -b 57600 -t

Connecting to programmer: .
Found programmer: Id = "CATERIN"; type = S
    Software Version = 1.0; No Hardware Version given.
Programmer supports auto addr increment.
Programmer supports buffered memory access with buffersize=128 bytes.

Programmer supports the following devices:
    Device code: 0x44

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9587 (probably m32u4)
avrdude> write flash 0x3100 "hello, world\n"
>>> write flash 0x3100 "hello, world\n"

Writing |                                                    | 0% 0.00savrdude (write): error writing 0x68 at 0x03100, rc=-1
write operation not supported on memory type "flash"
avrdude: ser_send(): write error: sorry no info avail
avrdude: ser_recv(): read error: The device does not recognize the command.

avrdude: butterfly_recv(): programmer is not responding
avrdude: error: programmer did not respond to command: set addr
avrdude: ser_send(): write error: sorry no info avail
avrdude: ser_recv(): read error: The device does not recognize the command.

avrdude: butterfly_recv(): programmer is not responding
avrdude (write): error writing 0x68 at 0x03100 cell=0x00
avrdude (write): error writing 0x65 at 0x03101, rc=-1
write operation not supported on memory type "flash"
avrdude (write): error writing 0x65 at 0x03101 cell=0x00
...

```

MCUdude, 2022-07-08:
With ATmega4808, I am not able to write to flash in terminal mode using `pickit4_updi`, `snap_updi`, or `serialupdi`.
```
$ ./avrdude -cpickit4_updi -patmega4808 -t -qq
         Vtarget                      : 5.06 V
         JTAG clock megaAVR/program   : 1000 kHz
         JTAG clock megaAVR/debug     : 100 kHz
         PDI/UPDI clock Xmega/megaAVR : 100 kHz

avrdude> write flash 0x00 "Hello World!"
>>> write flash 0x00 "Hello World!" 
avrdude (write): error writing 0x48 at 0x00000 cell=0xff
avrdude (write): error writing 0x65 at 0x00001 cell=0xff
avrdude (write): error writing 0x6c at 0x00002 cell=0xff
avrdude (write): error writing 0x6c at 0x00003 cell=0xff
avrdude (write): error writing 0x6f at 0x00004 cell=0xff
avrdude (write): error writing 0x20 at 0x00005 cell=0xff
avrdude (write): error writing 0x57 at 0x00006 cell=0xff
avrdude (write): error writing 0x6f at 0x00007 cell=0xff
avrdude (write): error writing 0x72 at 0x00008 cell=0xff
avrdude (write): error writing 0x6c at 0x00009 cell=0xff
avrdude (write): error writing 0x64 at 0x0000a cell=0xff
avrdude (write): error writing 0x21 at 0x0000b cell=0xff
avrdude (write): error writing 0x00 at 0x0000c cell=0xff
avrdude> read flash 0 0x40
>>> read flash 0 0x40 
0000  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0010  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0020  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0030  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|

avrdude> q
>>> q 

$ ./avrdude -csnap_updi -patmega4808 -t -qq
avrdude: usbhid_open(): No device found
         Vtarget                      : 5.06 V
         JTAG clock megaAVR/program   : 1000 kHz
         JTAG clock megaAVR/debug     : 100 kHz
         PDI/UPDI clock Xmega/megaAVR : 100 kHz

avrdude> write flash 0x00 "Hello World!"
>>> write flash 0x00 "Hello World!" 
avrdude (write): error writing 0x48 at 0x00000 cell=0xff
avrdude (write): error writing 0x65 at 0x00001 cell=0xff
avrdude (write): error writing 0x6c at 0x00002 cell=0xff
avrdude (write): error writing 0x6c at 0x00003 cell=0xff
avrdude (write): error writing 0x6f at 0x00004 cell=0xff
avrdude (write): error writing 0x20 at 0x00005 cell=0xff
avrdude (write): error writing 0x57 at 0x00006 cell=0xff
avrdude (write): error writing 0x6f at 0x00007 cell=0xff
avrdude (write): error writing 0x72 at 0x00008 cell=0xff
avrdude (write): error writing 0x6c at 0x00009 cell=0xff
avrdude (write): error writing 0x64 at 0x0000a cell=0xff
avrdude (write): error writing 0x21 at 0x0000b cell=0xff
avrdude (write): error writing 0x00 at 0x0000c cell=0xff
avrdude> read flash 0 0x40
>>> read flash 0 0x40 
0000  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0010  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0020  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0030  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|

avrdude> q
>>> q 

$ ./avrdude -cserialupdi -P /dev/cu.usbserial-1410 -patmega4808 -t -qq
avrdude: UPDI link initialization OK
avrdude: NVM type 0: 16-bit, page oriented write
avrdude: Entering NVM programming mode
avrdude> write flash 0x00 "Hello World!"
>>> write flash 0x00 "Hello World!" 
avrdude: Commit data command failed
avrdude (write): error writing 0x48 at 0x00000, rc=-1
write operation not supported on memory type "flash"
avrdude (write): error writing 0x48 at 0x00000 cell=0x00
^C
```

With ATmega32, `pickit4_isp` and `snap_isp` flash write works. However, I'm only able to write zeros, so a chip erase is necessary to flip the bits. It would be really cool if we could read a page, erase the page and write back the modified content.

```
$ ./avrdude -cpickit4_isp -patmega32 -t -qq
avrdude> write flash 0x00 0x10 0x55 ...
>>> write flash 0x00 0x10 0x55 ... 
avrdude> read flash 0x00 0x40
>>> read flash 0x00 0x40 
0000  55 55 55 55 55 55 55 55  55 55 55 55 55 55 55 55  |UUUUUUUUUUUUUUUU|
0010  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0020  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0030  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|

avrdude> write flash 0x00 0x10 0xAA ...
>>> write flash 0x00 0x10 0xAA ... 
avrdude: stk500v2_paged_write: write command failed
avrdude> read flash 0x00 0x40
>>> read flash 0x00 0x40 
0000  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 aa  |................|
0010  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0020  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0030  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|

avrdude> q
>>> q 

$ ./avrdude -csnap_isp -patmega32 -t -qq
avrdude: usbhid_open(): No device found
avrdude> write flash 0x20 0x10 0x55 ...
>>> write flash 0x20 0x10 0x55 ... 
avrdude> read flash 0x00 0x40
>>> read flash 0x00 0x40 
0000  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
0010  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0020  55 55 55 55 55 55 55 55  55 55 55 55 55 55 55 55  |UUUUUUUUUUUUUUUU|
0030  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|

avrdude> write flash 0x20 0x10 0xaa ...
>>> write flash 0x20 0x10 0xaa ... 
avrdude: stk500v2_paged_write: write command failed
avrdude> read flash 0x00 0x40
>>> read flash 0x00 0x40 
0000  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
0010  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0020  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 aa  |................|
0030  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|

avrdude> q
>>> q 

```

mcuee, 2022-07-09:
@MCUdude 
It is very strange that `pickit4_updi` and `snap_updi` perform differently from `pkobn_updi`. I may need to test again.

mcuee, 2022-07-09:
@stefanrueger 
So the issue may be more than classic AVRs and may affect more programmers. Not so sure if the root cause is the same or not, but the solution may be the same as you mentioned.

mcuee, 2022-07-09:
@MCUdude 
I think I have the similar issue as you after the initial good flash write with my Nano 4808.

```
PS C:\work\avr\avrdude_test\avrdude-7.0_bin64> .\avrdude.exe -qqp m4808 -c jtag2updi -P COM29 -t
avrdude> write flash 0x3100 "hello, world\n"
>>> write flash 0x3100 "hello, world\n"
avrdude> dump flash 0x3100 0x10
>>> dump flash 0x3100 0x10
3100  68 65 6c 6c 6f 2c 20 77  6f 72 6c 64 5c 6e 00 ff  |hello, world\n..|

avrdude> write flash 0x3100 "Good morning, world\n"
>>> write flash 0x3100 "Good morning, world\n"
avrdude.exe (write): error writing 0x47 at 0x03100 cell=0x40
avrdude.exe (write): error writing 0x6f at 0x03101 cell=0x65
avrdude.exe (write): error writing 0x6f at 0x03102 cell=0x6c
avrdude.exe (write): error writing 0x6d at 0x03105 cell=0x2c
avrdude.exe (write): error writing 0x6f at 0x03106 cell=0x20
avrdude.exe (write): error writing 0x69 at 0x03109 cell=0x60
avrdude.exe (write): error writing 0x6e at 0x0310a cell=0x6c
avrdude.exe (write): error writing 0x67 at 0x0310b cell=0x64
avrdude.exe (write): error writing 0x2c at 0x0310c cell=0x0c
avrdude.exe (write): error writing 0x77 at 0x0310e cell=0x00
avrdude> dump flash 0x3100 0x20
>>> dump flash 0x3100 0x20
3100  40 65 6c 64 20 2c 20 72  6e 60 6c 64 0c 20 00 6f  |@eld , rn`ld  .o|
3110  72 6c 64 5c 6e 00 ff ff  ff ff ff ff ff ff ff ff  |rld\n...........|

```

mcuee, 2022-07-09:
Same for an AVRISPMKII clone (just got it today).

```
PS C:\work\avr\avrdude_test\avrdude-7.0_bin64> .\avrdude_issue1004v3.exe -c avrispmkII  -p m328p -t

avrdude_issue1004v3.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude_issue1004v3.exe: Device signature = 0x1e950f (probably m328p)
avrdude> write flash 0x3100 "hello, world\n"
>>> write flash 0x3100 "hello, world\n"

Writing | ################################################## | 100% 0.80s

avrdude> dump flash 0x3100 0x20
>>> dump flash 0x3100 0x20

Reading | ################################################## | 100% 0.06s

3100  68 65 6c 6c 6f 2c 20 77  6f 72 6c 64 5c 6e 00 ff  |hello, world\n..|
3110  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|

avrdude> write flash 0x3100 "Good morning, world\n"
>>> write flash 0x3100 "Good morning, world\n"

Writing | ################################################## | 100% 1.23s

avrdude> dump flash 0x3100 0x20
>>> dump flash 0x3100 0x20

Reading | ################################################## | 100% 0.06s

3100  40 65 6c 64 20 2c 20 72  6e 60 6c 64 0c 20 00 6f  |@eld , rn`ld  .o|
3110  72 6c 64 5c 6e 00 ff ff  ff ff ff ff ff ff ff ff  |rld\n...........|

```

stefanrueger, 2022-07-11:
Here the worked out proposal for data representation in terminal write:
```
Usage: write <memory> <addr> <data>[,] {<data>[,]} 
       write <memory> <addr> <len> <data>[,] {<data>[,]} ...

Ellipsis ... writes <len> bytes padded by repeating the last <data> item.

<data> can be hexadecimal, octal or decimal integers, double, float,
C-style strings and chars. For numbers, an optional case-insensitive suffix
specifies the data size: HH: 8 bit, H/S: 16 bit, L: 32 bit, LL: 64 bit, F:
32-bit float. Hexadecimal floating point notation is supported. The 
ambiguous trailing F in 0x1.8F makes the number be interpreted as double;
use a zero exponent as in 0x1.8p0F to denote a hexadecimal float.

An optional U suffix makes a number unsigned. Ordinary 0x hex numbers are
always treated as unsigned. +0x or -0x hex numbers are treated as signed
unless they have a U suffix. Unsigned integers cannot be larger than 2^64-1.
If n is an unsigned integer then -n is also a valid unsigned integer as in C.
Signed integers must fall into the [-2^63, 2^63-1] range or a correspondingly
smaller range when a suffix specifies a smaller type. Out of range signed
numbers trigger a warning.

Ordinary 0x hex numbers with n hex digits (counting leading zeros) use    
the smallest size of 1, 2, 4 and 8 bytes that can accommodate any n-digit hex
number. If a suffix specifies a size explicitly the corresponding number of
least significant bytes are written. Otherwise, signed and unsigned integers
alike occupy the smallest of 1, 2, 4, or 8 bytes needed to accommodate them  
in their respective representation.
```

MCUdude, 2022-07-11:
> Analysis of the few detected edge-case problems below. I have now read the source and know what to do to fix the problems.

Thanks for taking your time to look at some of the terminal write edge cases. It looks like I haven't tested all of them myself, so there are obviously a few bugs left. If you have a solution for this, it would be great with a separate PR for this. I'll admit that I've never worked with C strings like this before, so it was quite a challenge to get it (almost) right.

> Unlike C I plan to treat hex constants without 'U' as unsigned, as users would not understand that 0xff as a signed data type would require two bytes (highest signed byte is 0x7f).

Please do!

> Generally, I'd like to treat hex numbers "special" to "do the right thing", for example, making 0x0002 write 2 bytes, so that people can cut and paste tables.

I agree! When thinking about it, I would also expect 0x0002 to write two bytes.

> Also, I'd like to introduce double as new data type for terminal write (in anticipation of Jörg's promised avr-libc extension for 64-bit double [at]dl8dtl

Great! I never got time to look at this myself. Perhaps with suffix `d` or `D`?

> ... and get strings and character constants parsed like in C

Can you elaborate? Change how strings in double-quotes and characters in single-quotes are handled?

> Here the worked out proposal for data representation in terminal write: ...

Looks great to me! Way better than a Norwegian like me could come up with. I'll assume this would be added to the TEXI documentation and perhaps the manpage?

stefanrueger, 2022-07-12:
@MCUdude PR #1025 available. I take it you volunteer to review? ;) Thanks!

> added to the TEXI documentation and perhaps the manpage

Sure, I'd like to avoid git conflicts, so prob after the PR queue has come down. I am confident @dl8dtl won't complain if we get those merged that are useful, fix problems and have been tested.

> strings and character constants parsed like in C

un-escaping backslash: `'\n'` or `"house\tyard\tgarden\n"

For now double without suffix are the default, suffix 'F' makes them 32-bit float (as before).

mcuee, 2022-07-30:
Just an update: PR #1025 has been merged.

mcuee, 2022-07-30:
avr109 (from the latest download) does not seem to work even with all the options enabled.
```
/* Uncomment the following to save code space */
//#define REMOVE_AVRPROG_SUPPORT
//#define REMOVE_FUSE_AND_LOCK_BIT_SUPPORT
//#define REMOVE_BLOCK_SUPPORT
//#define REMOVE_EEPROM_BYTE_SUPPORT
//#define REMOVE_FLASH_BYTE_SUPPORT

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_git.exe -c avr109 -P COM9
 -b 57600 -qqp m328p -D -t
Connecting to programmer: .
Found programmer: Id = "AVRBOOT"; type = S
    Software Version = 1.5; No Hardware Version given.
Programmer supports auto addr increment.
Programmer supports buffered memory access with buffersize=128 bytes.

Programmer supports the following devices:

avrdude> write flash 0x3100 "Good morning, world\n"
>>> write flash 0x3100 "Good morning, world\n"
avrdude_git.exe (write): error writing 0x47 at 0x03100, rc=-1
                         write operation not supported on memory type flash
avrdude_git.exe (write): error writing 0x47 at 0x03100 cell=0xff
avrdude_git.exe (write): error writing 0x6f at 0x03101, rc=-1
                         write operation not supported on memory type flash
avrdude_git.exe (write): error writing 0x6f at 0x03101 cell=0xff
...

```

mcuee, 2022-08-22:
FYI as well.
https://avrdudes.github.io/avrdude/7.0/avrdude_39.html#Troubleshooting

> Problem: Updating the flash ROM from terminal mode does not work with the JTAG ICEs.
>
> Solution: None at this time. Currently, the JTAG ICE code cannot write to the flash ROM one byte at a time.

Not so sure if the issue is only relevant to JTAG1 or it affects JTAG2 (JTAGICE mkii or AVR Dragon) and JTAG3 (JTAG ICE 3 and Atmel-ICE and Power Debugger) in JTAG mode. 
[Note: PICKit 4 and SNAP do not work in JTAG mode yet}.

mcuee, 2022-08-22:
@MCUdude 
Please help to check out Atmel ICE (JTAG mode) in terms of flash writing in terminal mode.

@dl8dtl 
Is this an avrdude limitation or it is a general limitation of JTAG mode?

MCUdude, 2022-08-22:
> @MCUdude
Please help to check out Atmel ICE (JTAG mode) in terms of flash writing in terminal mode.

Writing zeros works, but writing 1's doesn't.

Atmel ICE (JTAG) and ATmega128
```
$ ./avrdude -catmelice -patmega128 -t

         Vtarget                      : 5.03 V
         JTAG clock megaAVR/program   : 1500 kHz
         JTAG clock megaAVR/debug     : 200 kHz
         JTAG clock Xmega             : 1000 kHz
         PDI/UPDI clock Xmega/megaAVR : 500 kHz

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.07s

avrdude: Device signature = 0x1e9702 (probably m128)
avrdude> read flash 0x1000 0x100
>>> read flash 0x1000 0x100 

Reading | ################################################## | 100% 0.01s

1000  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
1010  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
1020  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
1030  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
1040  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
1050  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
1060  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
1070  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
1080  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
1090  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
10a0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
10b0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
10c0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
10d0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
10e0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
10f0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|

avrdude> write flash 0x1000 "Hello World! 1234567890"
>>> write flash 0x1000 "Hello World! 1234567890" 

Writing | ################################################## | 100% 1.07s

avrdude> read flash 0x1000 0x100
>>> read flash 0x1000 0x100 

Reading | ################################################## | 100% 0.00s

1000  48 65 6c 6c 6f 20 57 6f  72 6c 64 21 20 31 32 33  |Hello World! 123|
1010  34 35 36 37 38 39 30 00  ff ff ff ff ff ff ff ff  |4567890.........|
1020  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
1030  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
1040  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
1050  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
1060  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
1070  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
1080  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
1090  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
10a0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
10b0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
10c0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
10d0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
10e0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
10f0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|

avrdude> quit
[... 50 lines cut ...]

Reading | ################################################## | 100% 0.00s

1000  48 65 6c 6c 6f 20 57 6f  72 6c 64 21 20 31 32 33  |Hello World! 123|
1010  34 35 36 37 38 39 30 00  ff ff ff ff ff ff ff ff  |4567890.........|
1020  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
1030  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
1040  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
1050  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
1060  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
1070  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
1080  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
1090  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
10a0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
10b0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
10c0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
10d0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
10e0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
10f0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|

avrdude> quit
>>> quit 

avrdude done.  Thank you.
```

mcuee, 2022-08-23:
https://avrdudes.github.io/avrdude/7.0/avrdude_39.html#Troubleshooting

> Problem: Updating the flash ROM from terminal mode does not work with the JTAG ICEs.
>
> Solution: None at this time. Currently, the JTAG ICE code cannot write to the flash ROM one byte at a time.

Based on the test results, the solution may need some minor updates.

Something like the following.
> Solution: None at this time. Currently, `jtagmkI` and `jtagmkII` codes cannot write to the flash ROM one byte at a time. jtag3 codes may be able to write 0's to EEPROM but not 1's.

mcuee, 2022-08-24:
* https://github.com/avrdudes/avrdude/issues/1072#issuecomment-1225316611

https://ww1.microchip.com/downloads/en/DeviceDoc/Atmel-8151-8-bit-AVR-ATmega128A_Datasheet.pdf

For high voltage parallel programming (ST500/STK600), it is rather clear.
<I>Section 31.7.3. Chip Erase
The Chip Erase will erase the Flash and EEPROM memories plus Lock bits. The Lock bits are not reset
until the program memory has been completely erased. The Fuse bits are not changed. A Chip Erase
must be performed before the Flash and/or EEPROM are reprogrammed</I>

Same for JTAG
<I> 31.10.17. Programming the Flash
Before programming the Flash a Chip Erase must be performed. See Performing Chip Erase on page
410. </I>

mcuee, 2022-08-24:
If we based on the above datasheet, `Currently, jtagmkI and jtagmkII codes cannot write to the flash ROM one byte at a time`, that is an avrdude limitation (I will try to see if I can test using AVR Studio 4 to see if it can program Flash in two bytes a time).

mcuee, 2022-08-25:
Reference post from @MCUdude here.
* https://github.com/avrdudes/avrdude/issues/1072#issuecomment-1225243259

> I just borrowed the Atmel ICE, and I've just returned it. I may ask to do some more testing with it though.
> 
> I doubt it will be able to write 1s to EEPROM in terminal mode. It can't co it when writing a file without performing a chip and EEPROM erase first. I contacted a Microchip developer, and got the following answer:
> 
> **Q**: Is it correct that JTAG2-compatible programmers (i.e the AVR Dragon) and JTAG3/Atmel ICE only can write 0's to EEPROM and not 1's? It's like this when using Avrdude, but the EDBG documentation I've read doesn't explicitly specifies this behaviour.
> 
> **A**: Sounds like a bug to me – this rule applies of course to FLASH, not EEPROM. But EDBG is not “JTAG2-compatible”… and feature-set is _usually_ a function of the silicon, not the debugger anyway…
> 
> **Q**: I know the JTAG3/Atmel ICE can write 1's to Xmega's and AVRs with UPDI, but only when utilizing an "Atomic write" (See issue #1009 and PR #1013). However, I can't find a similar feature on AVRs with JTAG.
> 
> **A**: Correct – atomic (erase-and-write) was added for XMEGA. Not sure why this split was made…
> 
> **Q**: Does this means that it's impossible to write 1's to the EEPROM using a JTAG interface without running a chip erase first?
> 
> **A**: Not convinced, but not sure. SomeoneTM would probably have to find the RTL to know for sure – there are many subtle differences between the different entry modes (ISP/HVSP/JTAG) on those old parts – only from XMEGA onwards was there proper separation between physical and logical interfaces…

mcuee, 2022-08-26:
@stefanrueger has an interesting comment here. It is kind of the unltimate brutal force solution to work around the limitations of the programers. I think it is probably not worth doing though, in most cases.
https://github.com/avrdudes/avrdude/issues/1071#issuecomment-1222718285
> [Of course, in the end of the day, all bits can be written as they need be (that's the purpose of a programmer!) but writing a single bit to 1 may involve reading all flash and EEPROM, modifying the bit and then writing he whole shebang back; that in turn requires a fair bit of caching and cannot be done at the low-level writes]

stefanrueger, 2022-10-03:
Turns out that this issue is not a bug, after all. A litte-noted line in the documentation says:  [Terminal write] is not implemented for bank-addressed memories such as the flash memory of ATMega devices.

stefanrueger, 2022-10-03:
So PR #1106 is a proposal for a shiny new feature ;)

stefanrueger, 2022-10-05:
Closed with PR #1106

closed 2022-10-05 by stefanrueger