avrdudes/avrdude #1035: How is the systemwide avrdude.conf updated after changing src/avrdude.conf.in?

stefanrueger, 2022-07-19:
My workflow during development is like this: The first time I check out a branch I do
```
$ rm -rf build_linux
$ cmake -D CMAKE_BUILD_TYPE=RelWithDebInfo -B build_linux
$ cmake --build build_linux
```
After changes when I want to test, it's repeatedly the following, which nicely compiles and installs the to-be-tested avrdude:
```
cmake --build build_linux; sudo cmake --build build_linux --target install
```

However, I noticed that the systemwide `avrdude.conf` does not get installed/changed with the latter command even after explicitly changing `src/avrdude.conf.in`. It gets installed with the first-time creation of the build_linux dir.

What I'd like is a dependency of the systemwide conf file on the source conf.in. How do I effect that?

MCUdude, 2022-07-19:
I started contributing to the Avrdude project before Cmake was added, so I'm still using `make`

here's my workflow:
``` sh
$ ./bootstrap # Not sure if it _has_ to be done or not.
$ ./configure # creates avrdude.conf and check if dependencies are present

$ make # I run this every time I do a change to the source code

$ ./avrdude -cusbtiny -patmega2560 -Uflash:r:-:i # Success!

mariusgreuel, 2022-07-19:
I just tried this, and you are correct, the conf file dependency chain is broken. Apparently, the problem is the configure hack that I put in to implement a multi-line replace like configure does for the BEGIN/END tags:
https://github.com/avrdudes/avrdude/blob/affe4cb50a42c34a873be9bde8b5e650032cdcb5/src/avrdude.conf.in#L1413

CMakes configure_file() does not handle these multi-line sections, so I am running a multi-line regex over the entire file to keep it compatible with  automake:
https://github.com/avrdudes/avrdude/blob/affe4cb50a42c34a873be9bde8b5e650032cdcb5/src/CMakeLists.txt#L94

The regex runs only at configure time, hence the problem. As a temporary workaround, you need to run both configure and build, or run Jörgs build.sh script.

I guess we need a custom build rule to fix this. I will rename the title and assign this bug to me.

dl8dtl, 2022-07-19:
Deriving `avrdude.conf` from `avrdude.conf.in` was even a hack in the configure stuff. Basically, it's been produced there at the wrong time (during configure rather than during build).

mcuee, 2022-07-19:
> I started contributing to the Avrdude project before Cmake was added, so I'm still using `make`
> 
> here's my workflow:
> 
> ```shell
> $ ./bootstrap # Not sure if it _has_ to be done or not.
> $ ./configure # creates avrdude.conf and check if dependencies are present
> 
> $ make # I run this every time I do a change to the source code
> 
> $ ./avrdude -cusbtiny -patmega2560 -Uflash:r:-:i # Success!
> ```

Interesting. The main git tree does not have the `bootstrap` file any more.
https://github.com/avrdudes/avrdude

Now I usually just use `build.sh` under Linux and Windows (MSYS2 mingw64). Occassionally I will use VS2022/VS2019 as well under Windows.

Under macOS I use `buid.sh` as well for local testing. For system install I use the following since I use Homebrew.
1) first time: 
```
brew unlink avrdude (note: to unlink the installed avrdude-7.0 version)
brew install avrdude --HEAD
```
2) after than:
```
brew reinstall avrdude
```

mcuee, 2022-07-20:
> The regex runs only at configure time, hence the problem. As a temporary workaround, you need to run both configure and build, or run Jörgs build.sh script.

@mariusgreuel 
I do not see anything special about `build.sh`, why is it able to fix the issue?
https://github.com/avrdudes/avrdude/blob/main/build.sh

mcuee, 2022-07-20:
@dl8dtl 
BTW, even though `build.sh` only claims to support only Linux/macOS/FreeBSD, it works well under MYS2 mingw32/64. Of course the comments using `sudo` to install will not work under MSYS2 mingw32/64.

```
MINGW64 /c/work/avr/avrdude_test/avrdude_main
$ ./build.sh
-- Configuration summary:
-- ----------------------
-- DO HAVE    libelf
-- DO HAVE    libusb
-- DO HAVE    libusb_1_0
-- DO HAVE    libhidapi
-- DON'T HAVE libftdi
-- DO HAVE    libftdi1
-- DISABLED   doc
-- DISABLED   parport
-- DISABLED   linuxgpio
-- DISABLED   linuxspi
-- ----------------------
-- Configuring done
-- Generating done
-- Build files have been written to: C:/work/avr/avrdude_test/avrdude_main/build_mingw64_nt-10.0-22000
[38/56] Building C object src/CMakeFiles/libavrdude.dir/ser_win32.c.obj
C:/work/avr/avrdude_test/avrdude_main/src/ser_win32.c: In function 'net_send':
C:/work/avr/avrdude_test/avrdude_main/src/ser_win32.c:396:36: warning: pointer targets in passing argument 2 of 'send' differ in signedness [-Wpointer-sign]
  396 |                 rc = send(fd->ifd, p, (len > 1024) ? 1024 : len, 0);
      |                                    ^
      |                                    |
      |                                    const unsigned char *
In file included from C:/work/avr/avrdude_test/avrdude_main/src/ser_win32.c:30:
C:/msys64/mingw64/include/winsock2.h:1033:60: note: expected 'const char *' but argument is of type 'const unsigned char *'
 1033 |   WINSOCK_API_LINKAGE int WSAAPI send(SOCKET s,const char *buf,int len,int flags);
      |                                                ~~~~~~~~~~~~^~~
C:/work/avr/avrdude_test/avrdude_main/src/ser_win32.c: In function 'net_recv':
C:/work/avr/avrdude_test/avrdude_main/src/ser_win32.c:530:36: warning: pointer targets in passing argument 2 of 'recv' differ in signedness [-Wpointer-sign]
  530 |                 rc = recv(fd->ifd, p, (buflen - len > 1024) ? 1024 : buflen - len, 0);
      |                                    ^
      |                                    |
      |                                    unsigned char *
C:/msys64/mingw64/include/winsock2.h:1028:54: note: expected 'char *' but argument is of type 'unsigned char *'
 1028 |   WINSOCK_API_LINKAGE int WSAAPI recv(SOCKET s,char *buf,int len,int flags);
      |                                                ~~~~~~^~~
C:/work/avr/avrdude_test/avrdude_main/src/ser_win32.c: In function 'net_drain':
C:/work/avr/avrdude_test/avrdude_main/src/ser_win32.c:696:36: warning: pointer targets in passing argument 2 of 'recv' differ in signedness [-Wpointer-sign]
  696 |                 rc = recv(fd->ifd, &buf, 1, 0);
      |                                    ^~~~
      |                                    |
      |                                    unsigned char *
C:/msys64/mingw64/include/winsock2.h:1028:54: note: expected 'char *' but argument is of type 'unsigned char *'
 1028 |   WINSOCK_API_LINKAGE int WSAAPI recv(SOCKET s,char *buf,int len,int flags);
      |                                                ~~~~~~^~~
[53/56] Building C object src/CMakeFiles/avrdude.dir/term.c.obj
In function 'tokenize',
    inlined from 'terminal_mode' at C:/work/avr/avrdude_test/avrdude_main/src/term.c:1473:12:
C:/work/avr/avrdude_test/avrdude_main/src/term.c:1360:15: warning: pointer 'buf' may be used after 'realloc' [-Wuse-after-free]
 1360 |       k = buf - bufp;
      |           ~~~~^~~~~~
C:/work/avr/avrdude_test/avrdude_main/src/term.c:1344:18: note: call to 'realloc' here
 1344 |       buf_tmp  = realloc(buf, bufsize);
      |                  ^~~~~~~~~~~~~~~~~~~~~
[54/56] Building C object src/CMakeFiles/libavrdude.dir/stk500v2.c.obj
C:/work/avr/avrdude_test/avrdude_main/src/stk500v2.c: In function 'stk500v2_setparm':
C:/work/avr/avrdude_test/avrdude_main/src/stk500v2.c:3098:6: warning: 'current_value' may be used uninitialized [-Wmaybe-uninitialized]
 3098 |   if (value == current_value) {
      |      ^
C:/work/avr/avrdude_test/avrdude_main/src/stk500v2.c:3090:17: note: 'current_value' was declared here
 3090 |   unsigned char current_value;
      |                 ^~~~~~~~~~~~~
C:/work/avr/avrdude_test/avrdude_main/src/stk500v2.c: In function 'stk500v2_print_parms1':
C:/work/avr/avrdude_test/avrdude_main/src/stk500v2.c:3295:24: warning: 'osc_cmatch' may be used uninitialized [-Wmaybe-uninitialized]
[... 41 lines cut ...]
-- ----------------------
-- DO HAVE    libelf
-- DO HAVE    libusb
-- DO HAVE    libusb_1_0
-- DO HAVE    libhidapi
-- DON'T HAVE libftdi
-- DO HAVE    libftdi1
-- DISABLED   doc
-- DISABLED   parport
-- DISABLED   linuxgpio
-- DISABLED   linuxspi
-- ----------------------
-- Configuring done
-- Generating done
-- Build files have been written to: C:/work/avr/avrdude_test/avrdude_main/build_mingw64_nt-10.0-22000
ninja: no work to do.

Build succeeded.

Run

sudo cmake --build build_mingw64_nt-10.0-22000 --target install

to install.
```

mariusgreuel, 2022-07-20:
> I do not see anything special about build.sh, why is it able to fix the issue?

@mcuee The issue is simply that avrdude.conf is not rebuilt when avrdude.conf.in changes and you run `make` (or cmake --build). You need to reconfigure it by running `cmake -B`, which build.sh does both. Now, the old configure has the same problem, but one of the advantages of CMake is that these kind of problems should not occur.

mcuee, 2022-07-21:
> @mcuee The issue is simply that avrdude.conf is not rebuilt when avrdude.conf.in changes and you run `make` (or cmake --build). You need to reconfigure it by running `cmake -B`, which build.sh does [...]

Now I understand. Thanks for the explanation.

stefanrueger, 2022-08-04:
Do we really need these multi-line replacements? The multi-line replacements in avrdude.conf.in make it hard to translate the line number of something gone wrong in `avrdude.conf` to the corresponding line in `...conf.in`.  Can `avrdude.conf` with not-present programmers/host hardware descriptions  be made to behave as `avrdude.conf` without not-present stuff? For example through runtime tests?

mariusgreuel, 2022-08-07:
> Do we really need these multi-line replacements?

When I wrote the CMake project, I did not really question this, but now where you say it: I guess their only purpose is to exclude the conditional programmers from the list of programmers obtained through "-c ?". We might as well include them all (i.e. the parport and Linux GPIO/SPI programmers), which makes a long list just a tiny bit longer. A user will then find out later whether they are supported or not.

To fix the CMake build, we need to add a custom build action, which will probably add another file to it. By dropping the multi-line configure statements we can save a little complexity here.

@dl8dtl Any thoughts on this?

stefanrueger, 2022-08-07:
... and if the multi-line replacements are kept, I would suggest to change them to comment out those programmers rather than removing them, so the line numbers in `.conf` and `.conf.in` match up.

dl8dtl, 2022-08-08:
I think I started with that `avrdude.conf.in` stuff when we made the parallel port programmer optional. That removes a good number of programmers (basically all those AVRDUDE started with once) from the list for setups that don't (want to) support parallel ports.

Linux GPIO/SPI was then just the next candidate (only available on SBCs running Linux).

Commenting them out might be a good idea, indeed.

Besides printing it with `-c ?`, I think it's generally a good principle to not suggest something in the .conf file the target system/binary cannot support.

mcuee, 2022-08-08:
So we have two options.
1) to remove the multi-line configure statements -- less complexity, only the `.conf` file is a tiny bit longer. Probably a wiki or documentation can tell the user that Linux GPIO/SPI are only available on SBCs running Linux and parallel port programmers are not supported by Windows and macOS.

2) to find a replacement of the multi-line configure statements and comment out the programmers not supported.

I will vote for Option 1 actually for simplicity (as @mariusgreuel seems to be the only CMake expert here) even though Option 2 seems to be more technically correct.

stefanrueger, 2022-08-13:
I piggy-backed onto PR #1059 [code](https://github.com/avrdudes/avrdude/pull/1059/commits/297740db0e078e90387507e849cb5a7b5ded2c09) to not list programmer ids that start with dot on `-c ?`. This is now symmetric with `-p ?`.  I left `-c ?type` alone as (on reflection) it should  really list all programmer types available for constructing an entry in `avrdude.conf`.

PR #1095 (2022-09-24): CMake: Fix dependency chain between avrdude.conf.in and avrdude.conf
Fixes #1035

This PR corrects the dependency chain between avrdude.conf.in and avrdude.conf,
i.e. avrdude.conf is rebuilt when avrdude.conf.in changes.

Also, a new target 'conf' has been added for building 'avrdude.conf'.
closed 2022-09-24 by stefanrueger