avrdudes/avrdude #1037: design issue: jtag2updi is hardcoded to fail for parts with "has_pdi = yes"

ElTangas, 2022-07-20:
The jtag2updi programmer fails to recognize parts with PDI interface i.e. "has_pdi = yes".

In particular this impacts avrdude.conf files where, for compatibility with old avrdude versions, UPDI parts are defined with "has_pdi = yes" instead of "has_updi = yes".

Simple experiment: create a new programmer in avrdude.conf, named jtag2xxx, which is an exact copy of jtag2updi definition except for the name.
Now, UPDI parts with a "has_pdi = yes" definition work just fine with this programmer.

This means jtag2updi was special cased, hardcoded, inside avrdude, to fail for "has_pdi = yes". I fail to see what useful purpose this serves. In truth, I consider poor programming practice to special case user defined programmers from the avrdude.conf file, such as jtag2updi, inside avrdude itself.

In understand this was introduced here: https://github.com/avrdudes/avrdude/pull/789
I submit that the behaviour of the arduino port of avrdude, accepting both "has_pdi = yes" and "has_updi = yes" for all programmers using jtagmkii_pdi as base is preferable.

Note: using either "has_pdi = yes" or "has_updi = yes" actually generates different behaviour.
"has_updi = yes" -> performs complete chip erase before programming, uses page write for programming.
"has_pdi = yes" -> does not perform complete chip erase before programming, uses page erase/write for programming.
However both work correctly and both may be useful in specific situations, I don't advocate in any way fixing this "issue"! Therefore I think jtag2updi should not be precluded from using "has_pdi = yes".

MCUdude, 2022-07-20:
Well, I'm the one to blame here. I added Arduino's UPDI patch to `jtagmkii.c`, and a primitive check to make sure jtag2updi would print a reasonable error message when used with an ATxmega part. I didn't think of the backward compatibility error, since I've previously only used Arduino's patched version of Avrdude. Sorry for that.

This is the part causing the issue:
https://github.com/avrdudes/avrdude/blob/affe4cb50a42c34a873be9bde8b5e650032cdcb5/src/jtagmkII.c#L1308-L1314

Is there a different way we can detect ATxmegas and print a reasonable error message?

ElTangas, 2022-07-20:
Why is it necessary to detect xmegas when using jatg2updi?
They will generate an error down the line anyway, even the wiring for PDI is different from UPDI. I don't think anyone would actually connect jtag2updi to a xmega.

mcuee, 2022-07-21:
> Why is it necessary to detect xmegas when using jatg2updi? They will generate an error down the line anyway, even the wiring for PDI is different from UPDI. I don't think anyone would actually connect jtag2updi to a xmega.

To me what @MCUdude did is prudent. You do not know how people will connect the programmer to a target. Many of the users will not read much of the documentation -- and avrdude documentation is not that fantastic in this aspect.

Please refer to the discussion here.
* https://github.com/avrdudes/avrdude/discussions/1019

mcuee, 2022-07-21:
> In particular this impacts avrdude.conf files where, for compatibility with old avrdude versions, UPDI parts are defined with "has_pdi = yes" instead of "has_updi = yes".

@MCUdude and @ElTangas 
Just wondering what does `compatibility with old avrdude versions` exactly means here?

ElTangas, 2022-07-21:
> To me what @MCUdude did is prudent. You do not know how people will connect the programmer to a target. Many of the users will not read much of the documentation -- and avrdude documentation is not that fantastic in this aspect.

I agree that it is prudent. However, an error will be generated later anyway.

Further, I do not agree that a **user defined** programmer such as jtag2updi should be **hardcoded** inside avrdude. This creates an **hidden and undocumented dependence** between the avrdude executable and avrdude.conf.

If you think it's so important to detect xmegas and warn users, then IMO the correct way would be to create a **"jtagmkii_updi" base programmer** and use it as base for the jtag2updi definition.

> Just wondering what does compatibility with old avrdude versions exactly means here?

Simple. The jtagice mk2 protocol that jtag2updi uses predates the UPDI interface and does not really support it.
Therefore, when I first wrote jtag2updi in 2017-2018 avrdude.conf didn't even support "has_updi = yes" as an option.
However the PDI protocol that xmegas uses is so similar to UPDI, that I could define UPDI chips with "has_pdi = yes" and everything worked fine.
Basically, inside avrdude.conf, the UPDI chips can be defined with either "has_pdi = yes" or "has_updi = yes".
"has_pdi = yes" -> works with older avrdude versions but not with 7.0
"has_updi = yes" -> works with 7.0 but not older versions
**Both "has_updi = yes" and "has_pdi = yes" are recognised by the Arduino build of avrdude.**

MCUdude, 2022-07-21:
>> If you think it's so important to detect xmegas and warn users, then IMO the correct way would be to create a "jtagmkii_updi" base programmer and use it as base for the jtag2updi definition.

>Actually this may not be a bad idea.

> @MCUdude How do you like this idea?

I agree that this would be a good solution. I would like to hear @dl8dtl's thoughts on this though.

ElTangas, 2022-07-27:
I've tested your patch, and it now works with both old and new versions of avrdude.conf, which is a plus IMO.
This ordeal has also exposed a subtle bug in jtag2updi that I'm correcting now.

However, I'd like to propose a change to your error detection logic as follows

```
  /* Abort and print error if programmer does not support the target microcontroller */
 (original) if ((strncmp(pgm->type, "JTAGMKII_UPDI", strlen("JTAGMKII_UPDI")) == 0 && p->flags & AVRPART_HAS_PDI) ||
 (proposal) if ((strncmp(pgm->type, "JTAGMKII_UPDI", strlen("JTAGMKII_UPDI")) == 0 && !(p->flags & AVRPART_HAS_UPDI)) ||
```

Because whether a part has PDI or not is not the real issue. The issue is the absence of UPDI and that is what should be checked for.
For example, a future part might have both PDI and UPDI interfaces, and then we would be hunting bugs again - your logic would see PDI and immediately fail, even if UPDI is also supported.

MCUdude, 2022-08-02:
Fixed in #1037

closed 2022-08-02 by MCUdude