avrdudes/avrdude #1038: PICkit4 and SNAP serial numbers not read by Avrdude

MCUdude, 2022-07-20:
For some reason, the serial number string is blank with the SNAP and PICkit4.
It does however work with pkobn_updi (nEDBG) and mEDBG boards.

@janegilruud is there a different command we have to use on these gen4 programmers?

SNAP output in verbose mode:
```
$ ./avrdude -patmega4808 -csnap_updi -v

avrdude: Version 7.0-20220508
         Copyright (c) Brian Dean, http://www.bdmicro.com/
         Copyright (c) Joerg Wunsch

         System wide configuration file is "/Users/hans/Downloads/avrdude/src/avrdude.conf"
         User configuration file is "/Users/hans/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : usb
         Using Programmer              : snap_updi
avrdude: usbhid_open(): No device found
avrdude: Found CMSIS-DAP compliant device, using EDBG protocol
         AVR Part                      : ATmega4808
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
           prodsig                 0     0     0    0 no         61   61      0     0     0 0x00 0x00
           sernum                  0     0     0    0 no         10    1      0     0     0 0x00 0x00
           osccal16                0     0     0    0 no          2    1      0     0     0 0x00 0x00
           osccal20                0     0     0    0 no          2    1      0     0     0 0x00 0x00
           tempsense               0     0     0    0 no          2    1      0     0     0 0x00 0x00
           osc16err                0     0     0    0 no          2    1      0     0     0 0x00 0x00
           osc20err                0     0     0    0 no          2    1      0     0     0 0x00 0x00
           fuses                   0     0     0    0 no          9   10      0     0     0 0x00 0x00
           fuse0       wdtcfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse1       bodcfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse2       osccfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse4       tcd0cfg     0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse5       syscfg0     0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse6       syscfg1     0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse7       append      0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse8       bootend     0     0     0    0 no          1    1      0     0     0 0x00 0x00
           lock                    0     0     0    0 no          1    1      0     0     0 0x00 0x00
           data                    0     0     0    0 no          0    1      0     0     0 0x00 0x00
           userrow     usersig     0     0     0    0 no         64   64      0     0     0 0x00 0x00
           flash                   0     0     0    0 no      49152  128      0     0     0 0x00 0x00
           eeprom                  0     0     0    0 no        256   64      0     0     0 0x00 0x00

         Programmer Type : JTAGICE3_UPDI
         Description     : MPLAB(R) SNAP in UPDI mode
         ICE HW version  : 1
         ICE FW version  : 1.10 (rel. 249)
         Serial number   : 
         Vtarget         : 5.07 V
         JTAG clock megaAVR/program   : 1000 kHz
         JTAG clock megaAVR/debug     : 100 kHz
         PDI/UPDI clock Xmega/megaAVR : 100 kHz

avrdude: Partial Family_ID returned: "mega"
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.11s

avrdude: Device signature = 0x1e9650 (probably m4808)

avrdude done.  Thank you.
```

PICkit4  output in verbose mode:
```
$ ./avrdude -patmega4808 -cpickit4_updi -v

avrdude: Version 7.0-20220508
         Copyright (c) Brian Dean, http://www.bdmicro.com/
         Copyright (c) Joerg Wunsch

         System wide configuration file is "/Users/hans/Downloads/avrdude/src/avrdude.conf"
         User configuration file is "/Users/hans/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : usb
         Using Programmer              : pickit4_updi
avrdude: Found CMSIS-DAP compliant device, using EDBG protocol
         AVR Part                      : ATmega4808
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
           prodsig                 0     0     0    0 no         61   61      0     0     0 0x00 0x00
           sernum                  0     0     0    0 no         10    1      0     0     0 0x00 0x00
           osccal16                0     0     0    0 no          2    1      0     0     0 0x00 0x00
           osccal20                0     0     0    0 no          2    1      0     0     0 0x00 0x00
           tempsense               0     0     0    0 no          2    1      0     0     0 0x00 0x00
           osc16err                0     0     0    0 no          2    1      0     0     0 0x00 0x00
           osc20err                0     0     0    0 no          2    1      0     0     0 0x00 0x00
           fuses                   0     0     0    0 no          9   10      0     0     0 0x00 0x00
           fuse0       wdtcfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse1       bodcfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse2       osccfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse4       tcd0cfg     0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse5       syscfg0     0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse6       syscfg1     0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse7       append      0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse8       bootend     0     0     0    0 no          1    1      0     0     0 0x00 0x00
           lock                    0     0     0    0 no          1    1      0     0     0 0x00 0x00
           data                    0     0     0    0 no          0    1      0     0     0 0x00 0x00
           userrow     usersig     0     0     0    0 no         64   64      0     0     0 0x00 0x00
           flash                   0     0     0    0 no      49152  128      0     0     0 0x00 0x00
           eeprom                  0     0     0    0 no        256   64      0     0     0 0x00 0x00

         Programmer Type : JTAGICE3_UPDI
         Description     : MPLAB(R) PICkit 4 in UPDI mode
         ICE HW version  : 4
         ICE FW version  : 1.14 (rel. 268)
         Serial number   : 
         Vtarget         : 5.06 V
         JTAG clock megaAVR/program   : 1000 kHz
         JTAG clock megaAVR/debug     : 100 kHz
         PDI/UPDI clock Xmega/megaAVR : 100 kHz

avrdude: Partial Family_ID returned: "mega"
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.11s

avrdude: Device signature = 0x1e9650 (probably m4808)

avrdude done.  Thank you.
```

mcuee, 2022-07-28:
So looks like the following two lines do not work for PICkit 4 and SNAP.
https://github.com/avrdudes/avrdude/blob/affe4cb50a42c34a873be9bde8b5e650032cdcb5/src/jtag3.c#L2424
```
 if ((status = jtag3_command(pgm, cmd, 4, &resp, "get info (serial number)")) < 0)
    return;
```

mcuee, 2022-07-28:
From https://github.com/microchip-pic-avr-tools/pymcuprog/blob/main/pymcuprog/backend.py 
It seems to me maybe we have to rely on generic command and not JTAG ICE 3 specific comamnd.

MCUdude, 2022-07-28:
> From https://github.com/microchip-pic-avr-tools/pymcuprog/blob/main/pymcuprog/backend.py
It seems to me maybe we have to rely on generic command and not JTAG ICE 3 specific command.

That was my conclusion too. However, I couldn't find out how to send a generic message, to the programmer.

mcuee, 2022-07-28:
My SNAP was just delivered to me. I upgraded the FW using MPLAB X and then switched the mode to AVR using Microchip Studio. In ISP mode, avrdude does not even to attempt to read the serial number.

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_git.exe -c snap_isp -p m328p -v

avrdude_git.exe: Version 7.0-20220719 (affe4cb)
                 Copyright (c) Brian Dean, http://www.bdmicro.com/
                 Copyright (c) Joerg Wunsch

                 System wide configuration file is "C:/work/avr/avrdude_test/avrdude_bin/avrdude.conf"

                 Using Port                    : usb
                 Using Programmer              : snap_isp
avrdude_git.exe: usbhid_open(): No device found
avrdude_git.exe: Found CMSIS-DAP compliant device, using EDBG protocol
                 AVR Part                      : ATmega328P
                 Chip Erase delay              : 9000 us
                 PAGEL                         : PD7
                 BS2                           : PC2
                 RESET disposition             : dedicated
                 RETRY pulse                   : SCK
                 Serial program mode           : yes
                 Parallel program mode         : yes
                 Timeout                       : 200
                 StabDelay                     : 100
                 CmdexeDelay                   : 25
                 SyncLoops                     : 32
                 PollIndex                     : 3
                 PollValue                     : 0x53
                 Memory Detail                 :

                                                   Block Poll               Page                       Polled
                   Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
                   ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
                   eeprom                 65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
                   flash                  65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
                   lfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
                   hfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
                   efuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
                   lock                    0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
                   calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00
                   signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00

                 Programmer Type : JTAG3_ISP
                 Description     : MPLAB(R) SNAP in ISP mode
                 Vtarget         : 5.0 V
                 SCK period      : 500.00 us

avrdude_git.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.06s

avrdude_git.exe: Device signature = 0x1e950f (probably m328p)

avrdude_git.exe done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_git.exe -c snap_isp -p m328p -vv

avrdude_git.exe: Version 7.0-20220719 (affe4cb)
                 Copyright (c) Brian Dean, http://www.bdmicro.com/
                 Copyright (c) Joerg Wunsch

                 System wide configuration file is "C:/work/avr/avrdude_test/avrdude_bin/avrdude.conf"

                 Using Port                    : usb
                 Using Programmer              : snap_isp
avrdude_git.exe: stk500v2_jtag3_open()
avrdude_git.exe: usbhid_open(): No device found
avrdude_git.exe: Found CMSIS-DAP compliant device, using EDBG protocol
avrdude_git.exe: jtag3_edbg_prepare(): connection status 0x01
[... 33 lines cut ...]
                 Vtarget         : 5.0 V
                 SCK period      : 500.00 us

avrdude_git.exe: jtag3_setparm()
avrdude_git.exe: Sending set parameter (scope 0x12, section 0, parm 0) command: 0x80 (3 bytes msg)
avrdude_git.exe: jtag3_setparm()
avrdude_git.exe: Sending set parameter (scope 0x12, section 0, parm 1) command: 0x80 (3 bytes msg)
avrdude_git.exe: jtag3_setparm()
avrdude_git.exe: Sending set parameter (scope 0x12, section 1, parm 0) command: 0x80 (3 bytes msg)
avrdude_git.exe: AVR device initialized and ready to accept instructions

Reading |                                                    | 0% 0.00savrdude_git.exe: stk500isp_read_byte(.., signature, 0x0, ...)
avrdude_git.exe: stk500isp_read_byte(): Sending read memory command: avrdude_git.exe: stk500isp_read_byte(.., signature, 0x1, ...)
Reading | #################                                  | 33% 0.04savrdude_git.exe: stk500isp_read_byte(.., signature, 0x2, ...)
Reading | ################################################## | 100% 0.07s

avrdude_git.exe: Device signature = 0x1e950f (probably m328p)
avrdude_git.exe: stk500v2_jtag3_close()
avrdude_git.exe: jtag3_close()
avrdude_git.exe: Sending AVR sign-off command: 0x80 (3 bytes msg)
avrdude_git.exe: Sending sign-off command: 0x80 (3 bytes msg)

avrdude_git.exe done.  Thank you.

```

mcuee, 2022-07-28:
> > From https://github.com/microchip-pic-avr-tools/pymcuprog/blob/main/pymcuprog/backend.py
> > It seems to me maybe we have to rely on generic command and not JTAG ICE 3 specific command.
> 
> That was my conclusion too. However, I couldn't find out how to send a generic message, to the programmer.

Hmm, by right the following should work using hidapi.
https://github.com/avrdudes/avrdude/blob/main/src/usb_hidapi.c#L104

But then the output of the following (using -vvvv) seems to be strange.
```
avrdude_git.exe: stk500v2_jtag3_open()
avrdude_git.exe: usbhid_open(): No device found
avrdude_git.exe: usbhid_open(): Probing for max. packet size
avrdude_git.exe: usbhid_open(): Setting max_xfer from DAP_Info response to 64
avrdude_git.exe: Found CMSIS-DAP compliant device, using EDBG protocol
avrdude_git.exe: jtag3_getsync()

```

I just checked hidapi's hidtest app and it works fine.
https://github.com/libusb/hidapi/blob/master/hidtest/test.c#L85
```
Device Found
  type: 03eb 2180
  path: \\?\HID#VID_03EB&PID_2180&MI_00#8&1cdb02ac&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
  serial_number: BUR21268xxxx
  Manufacturer: Microchip Technology Incorporated
  Product:      MPLAB Snap ICD CMSIS-DAP
  Release:      100
  Interface:    0
  Usage (page): 0x1 (0xff00)

```

mcuee, 2022-07-28:
Just a simplified test from the hidapi's hidtest example application.
```
#include <stdio.h>
#include <wchar.h>
#include <string.h>
#include <stdlib.h>

#include <hidapi.h>

// Headers needed for sleeping.
#ifdef _WIN32
	#include <windows.h>
#else
	#include <unistd.h>
#endif

int main(int argc, char* argv[])
{
	(void)argc;
	(void)argv;

	int res;
	unsigned char buf[256];
	#define MAX_STR 255
	wchar_t wstr[MAX_STR];
	hid_device *handle;
	int i;

	struct hid_device_info *devs, *cur_dev;

	if (hid_init())
		return -1;

#if defined(__APPLE__) && HID_API_VERSION >= HID_API_MAKE_VERSION(0, 12, 0)
	// To work properly needs to be called before hid_open/hid_open_path after hid_init.
	// Best/recommended option - call it right after hid_init.
	hid_darwin_set_open_exclusive(0);
#endif

	devs = hid_enumerate(0x03eb, 0x2180);
	cur_dev = devs;
	while (cur_dev) {
		printf("Device Found\n  type: %04hx %04hx\n  path: %s\n  serial_number: %ls", cur_dev->vendor_id, cur_dev->product_id, cur_dev->path, cur_dev->serial_number);
		printf("\n");
		printf("  Manufacturer: %ls\n", cur_dev->manufacturer_string);
		printf("  Product:      %ls\n", cur_dev->product_string);
		printf("  Release:      %hx\n", cur_dev->release_number);
		printf("  Interface:    %d\n",  cur_dev->interface_number);
		printf("  Usage (page): 0x%hx (0x%hx)\n", cur_dev->usage, cur_dev->usage_page);
		printf("\n");
		cur_dev = cur_dev->next;
	}
	hid_free_enumeration(devs);

	/* Free static HIDAPI objects. */
	hid_exit();

#ifdef _WIN32
	system("pause");
#endif

	return 0;
}
```
```

mcuee, 2022-07-28:
@MCUdude 
Now I understand the logic of usb_hidapi.c, you need to specify the serial number and then it will match the serial number. If not it will directly open the device without matching serial number.

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_git.exe -c snap_isp -P usb:BUR21268xxxx -p m328p -v

avrdude_git.exe: Version 7.0-20220719 (affe4cb)
                 Copyright (c) Brian Dean, http://www.bdmicro.com/
                 Copyright (c) Joerg Wunsch

                 System wide configuration file is "C:/work/avr/avrdude_test/avrdude_bin/avrdude.conf"

                 Using Port                    : usb:BUR21268xxxx
                 Using Programmer              : snap_isp
avrdude_git.exe: usbhid_open(): Found MPLAB Snap ICD CMSIS-DAP, serno: BUR21268xxxx
avrdude_git.exe: Found CMSIS-DAP compliant device, using EDBG protocol
                 AVR Part                      : ATmega328P
                 Chip Erase delay              : 9000 us
                 PAGEL                         : PD7
                 BS2                           : PC2
                 RESET disposition             : dedicated
                 RETRY pulse                   : SCK
                 Serial program mode           : yes
                 Parallel program mode         : yes
                 Timeout                       : 200
                 StabDelay                     : 100
                 CmdexeDelay                   : 25
                 SyncLoops                     : 32
                 PollIndex                     : 3
                 PollValue                     : 0x53
                 Memory Detail                 :

                                                   Block Poll               Page                       Polled
                   Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
                   ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
                   eeprom                 65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
                   flash                  65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
                   lfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
                   hfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
                   efuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
                   lock                    0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
                   calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00
                   signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00

                 Programmer Type : JTAG3_ISP
                 Description     : MPLAB(R) SNAP in ISP mode
                 Vtarget         : 5.0 V
                 SCK period      : 500.00 us

avrdude_git.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.07s

avrdude_git.exe: Device signature = 0x1e950f (probably m328p)

avrdude_git.exe done.  Thank you.

```

mcuee, 2022-07-28:
@MCUdude and @mariusgreuel 

Still something is not so correct about the alternative code path. By right it should still be able to find the device, just without the serial number matching. It should not complained about `usbhid_open(): No device found`.

https://github.com/avrdudes/avrdude/blob/main/src/usb_hidapi.c#L136

```
  else
    {
      /*
       * No serial number requested, pass straight to hid_open()
       */
      dev = hid_open(pinfo.usbinfo.vid, pinfo.usbinfo.pid, NULL);
      if (dev == NULL)
      {
	avrdude_message(MSG_INFO, "%s: usbhid_open(): No device found\n",
			progname);
	return -1;
      }
    }

```

mcuee, 2022-07-28:
@MCUdude and @mariusgreuel 

With simple debugging, I know the reason of the message, because it tries to open 03eb:217f first but my SNAP is 03eb:2180.
```
PS C:\work\avr\avrdude_test\avrdude_main> git diff
diff --git a/src/usb_hidapi.c b/src/usb_hidapi.c
index 5ceca60..ac3731b 100644
--- a/src/usb_hidapi.c
+++ b/src/usb_hidapi.c
@@ -139,6 +139,7 @@ static int usbhid_open(char * port, union pinfo pinfo, union filedescriptor *fd)
        * No serial number requested, pass straight to hid_open()
        */
       dev = hid_open(pinfo.usbinfo.vid, pinfo.usbinfo.pid, NULL);
+         avrdude_message(MSG_INFO, " Trying to find HID Device: %04hx %04hx\n", pinfo.usbinfo.vid, pinfo.usbinfo.pid);
       if (dev == NULL)
       {
        avrdude_message(MSG_INFO, "%s: usbhid_open(): No device found\n",

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_issue1038.exe -c snap_isp -P usb -p m328p -v

avrdude_issue1038.exe: Version 7.0-20220719 (affe4cb)
                       Copyright (c) Brian Dean, http://www.bdmicro.com/
                       Copyright (c) Joerg Wunsch

                       System wide configuration file is "C:/work/avr/avrdude_test/avrdude_bin/avrdude.conf"

                       Using Port                    : usb
                       Using Programmer              : snap_isp
 Trying to find HID Device: 03eb 217f
avrdude_issue1038.exe: usbhid_open(): No device found
 Trying to find HID Device: 03eb 2180
avrdude_issue1038.exe: Found CMSIS-DAP compliant device, using EDBG protocol
...

```

MCUdude, 2022-07-28:
I'm able to retrieve the serial number by adding the a code snippet to `usbhid_open`:

```c

static int usbhid_open(char *port, union pinfo pinfo, union filedescriptor *fd)
{
  hid_device *dev;
  char *serno, *cp2;
  size_t x;
  unsigned char usbbuf[USBDEV_MAX_XFER_3 + 1];

 // Test code to retrieve the USB serial number
 struct hid_device_info *get_sn;
  get_sn = hid_enumerate(pinfo.usbinfo.vid, pinfo.usbinfo.pid);
  if (get_sn)
    avrdude_message(MSG_INFO, "sn2: %ls\n", get_sn->serial_number);

  ...

```

```
$ ./avrdude -csnap_updi -patmega4808
avrdude: usbhid_open(): No device found
Serial number: BUR183076609

         Vtarget                      : 5.04 V
         JTAG clock megaAVR/program   : 1000 kHz
         JTAG clock megaAVR/debug     : 100 kHz
         PDI/UPDI clock Xmega/megaAVR : 100 kHz

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.11s

avrdude: Device signature = 0x1e9650 (probably m4808)

avrdude done.  Thank you.
```

However, it doesn't work in jtag3.c, since `hidapi/hidapi.h` isn't directly included there.

MCUdude, 2022-07-28:
It looks like pymcuprog (pyedbglib) uses CMSIS-DAP messages to get the serial number. I think this _should_ be possible to replicate in jtag3.c:

https://github.com/microchip-pic-avr-tools/pyedbglib/blob/b91b24ab19a6557a67dea9082616650c20c7a01f/pyedbglib/protocols/cmsisdap.py#L74-L99

MCUdude, 2022-12-14:
@mcuee do you know if JTAG3 compatible programmers _has_ to use `hidapi`, or may they also use `libusb`?
I was thinking about looking into this issue again, and I already have a fix for `hidapi`. However, I've not tested it with libusb, and I'm not even sure how I can force JTAG3 programmers to use libusb on my mac.

mcuee, 2022-12-14:
The current avrdude codes will use hidapi first if it is available. If not it will fall back to libusb (not working for macOS.

BTW hidapi in Linux may use either hidraw backend or libusb backend. hidapi under FreeBSD uses libusb backend.

You will not be able to use libusb backend under macOS. But please go ahead to implement the hidapi way and we should be able to figure out the fallback libusb way later.

mcuee, 2022-12-14:
@MCUdude 

For libusb, you can refer to the example here to get the USB serial number.

https://github.com/libusb/libusb/blob/master/examples/xusb.c#L840-L852
```
	printf("\nReading device descriptor:\n");
	CALL_CHECK_CLOSE(libusb_get_device_descriptor(dev, &dev_desc), handle);
	printf("            length: %d\n", dev_desc.bLength);
	printf("      device class: %d\n", dev_desc.bDeviceClass);
	printf("               S/N: %d\n", dev_desc.iSerialNumber);
	printf("           VID:PID: %04X:%04X\n", dev_desc.idVendor, dev_desc.idProduct);
	printf("         bcdDevice: %04X\n", dev_desc.bcdDevice);
	printf("   iMan:iProd:iSer: %d:%d:%d\n", dev_desc.iManufacturer, dev_desc.iProduct, dev_desc.iSerialNumber);
	printf("          nb confs: %d\n", dev_desc.bNumConfigurations);
	// Copy the string descriptors for easier parsing
	string_index[0] = dev_desc.iManufacturer;
	string_index[1] = dev_desc.iProduct;
	string_index[2] = dev_desc.iSerialNumber;
```

Or here.
https://github.com/libusb/libusb/blob/master/examples/testlibusb.c#L193-L211
```
	if (handle) {
		if (desc.iManufacturer) {
			ret = libusb_get_string_descriptor_ascii(handle, desc.iManufacturer, string, sizeof(string));
			if (ret > 0)
				printf("  Manufacturer:              %s\n", (char *)string);
		}

		if (desc.iProduct) {
			ret = libusb_get_string_descriptor_ascii(handle, desc.iProduct, string, sizeof(string));
			if (ret > 0)
				printf("  Product:                   %s\n", (char *)string);
		}

		if (desc.iSerialNumber && verbose) {
			ret = libusb_get_string_descriptor_ascii(handle, desc.iSerialNumber, string, sizeof(string));
			if (ret > 0)
				printf("  Serial Number:             %s\n", (char *)string);
		}
	}
```

mcuee, 2022-12-14:
Another reference from libusb's xusb example.
https://github.com/libusb/libusb/blob/master/examples/xusb.c#L840-L852
```
	printf("\nReading device descriptor:\n");
	CALL_CHECK_CLOSE(libusb_get_device_descriptor(dev, &dev_desc), handle);
	printf("            length: %d\n", dev_desc.bLength);
	printf("      device class: %d\n", dev_desc.bDeviceClass);
	printf("               S/N: %d\n", dev_desc.iSerialNumber);
	printf("           VID:PID: %04X:%04X\n", dev_desc.idVendor, dev_desc.idProduct);
	printf("         bcdDevice: %04X\n", dev_desc.bcdDevice);
	printf("   iMan:iProd:iSer: %d:%d:%d\n", dev_desc.iManufacturer, dev_desc.iProduct, dev_desc.iSerialNumber);
	printf("          nb confs: %d\n", dev_desc.bNumConfigurations);
	// Copy the string descriptors for easier parsing
	string_index[0] = dev_desc.iManufacturer;
	string_index[1] = dev_desc.iProduct;
	string_index[2] = dev_desc.iSerialNumber;
```

MCUdude, 2022-12-15:
@mcuee I think I have a working solution for hidapi.

Would be great if you could give this a try: https://github.com/MCUdude/avrdude/tree/pickit4-snap-serialnumber.

I'll look into how it can be accomplished with libusb later.

mcuee, 2022-12-15:
It does not seem to work under Windows.
```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_issue1038_hidapi -c pickit4_isp -p m8a -v

avrdude_issue1038_hidapi: Version 7.0-20221215 (4ecb34f)
                          Copyright the AVRDUDE authors;
                          see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

                          System wide configuration file is C:\work\avr\avrdude_test\avrdude_bin\avrdude.conf

                          Using Port                    : usb
                          Using Programmer              : pickit4_isp
avrdude_issue1038_hidapi: found CMSIS-DAP compliant device, using EDBG protocol
                          AVR Part                      : ATmega8A
                          Chip Erase delay              : 10000 us
                          PAGEL                         : PD7
                          BS2                           : PC2
                          RESET disposition             : possible i/o
                          RETRY pulse                   : SCK
                          Serial program mode           : yes
                          Parallel program mode         : yes
                          Timeout                       : 200
                          StabDelay                     : 100
                          CmdexeDelay                   : 25
                          SyncLoops                     : 32
                          PollIndex                     : 3
                          PollValue                     : 0x53
                          Memory Detail                 :

                                                            Block Poll               Page                       Polled
                            Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
                            ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
                            eeprom                  4    20   128    0 no        512    4      0  9000  9000 0xff 0xff
                            flash                  33    10    64    0 yes      8192   64    128  4500  4500 0xff 0x00
                            lfuse                   0     0     0    0 no          1    1      0  2000  2000 0x00 0x00
                            signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
                            calibration             0     0     0    0 no          4    1      0     0     0 0x00 0x00

                          Programmer Type : JTAG3_ISP
                          Description     : MPLAB(R) PICkit 4 in ISP mode
                          Vtarget         : 4.87 V
                          JTAG clock megaAVR/program   : 1000 kHz
                          JTAG clock megaAVR/debug     : 100 kHz
                          PDI/UPDI clock Xmega/megaAVR : 100 kHz
                          SCK period                   : 8.00 us

avrdude_issue1038_hidapi: AVR device initialized and ready to accept instructions
avrdude_issue1038_hidapi: device signature = 0x1e9307 (probably m8a)

avrdude_issue1038_hidapi done.  Thank you.
```

mcuee, 2022-12-15:
It oes work for the UPDI mode.

```
mcuee@UbuntuSwift3:~/build/avr/avrdude_bin$ ./avrdude_issue1038 -c pickit4_updi -p m4808 -v

avrdude_issue1038: Version 7.0-20221215 (4ecb34f)
                   Copyright the AVRDUDE authors;
                   see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

                   System wide configuration file is /home/mcuee/build/avr/avrdude_bin/avrdude.conf
                   User configuration file is /home/mcuee/.avrduderc
                   User configuration file does not exist or is not a regular file, skipping

                   Using Port                    : usb
                   Using Programmer              : pickit4_updi
avrdude_issue1038: found CMSIS-DAP compliant device, using EDBG protocol
                   AVR Part                      : ATmega4808
                   RESET disposition             : dedicated
                   RETRY pulse                   : SCK
                   Serial program mode           : yes
                   Parallel program mode         : yes
                   Memory Detail                 :

                                                     Block Poll               Page                       Polled
                     Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
                     ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
                     fuse0       wdtcfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
                     fuse1       bodcfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
                     fuse2       osccfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
                     fuse4       tcd0cfg     0     0     0    0 no          1    1      0     0     0 0x00 0x00
                     fuse5       syscfg0     0     0     0    0 no          1    1      0     0     0 0x00 0x00
                     fuse6       syscfg1     0     0     0    0 no          1    1      0     0     0 0x00 0x00
                     fuse7       append      0     0     0    0 no          1    1      0     0     0 0x00 0x00
                     fuse8       bootend     0     0     0    0 no          1    1      0     0     0 0x00 0x00
                     fuses                   0     0     0    0 no          9   10      0     0     0 0x00 0x00
                     lock                    0     0     0    0 no          1    1      0     0     0 0x00 0x00
                     tempsense               0     0     0    0 no          2    1      0     0     0 0x00 0x00
                     signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
                     prodsig                 0     0     0    0 no         61   61      0     0     0 0x00 0x00
                     sernum                  0     0     0    0 no         10    1      0     0     0 0x00 0x00
                     osccal16                0     0     0    0 no          2    1      0     0     0 0x00 0x00
                     osccal20                0     0     0    0 no          2    1      0     0     0 0x00 0x00
                     osc16err                0     0     0    0 no          2    1      0     0     0 0x00 0x00
                     osc20err                0     0     0    0 no          2    1      0     0     0 0x00 0x00
                     data                    0     0     0    0 no          0    1      0     0     0 0x00 0x00
                     userrow     usersig     0     0     0    0 no         64   64      0     0     0 0x00 0x00
                     eeprom                  0     0     0    0 no        256   64      0     0     0 0x00 0x00
                     flash                   0     0     0    0 no      49152  128      0     0     0 0x00 0x00

                   Programmer Type : JTAGICE3_UPDI
                   Description     : MPLAB(R) PICkit 4 in UPDI mode
                   ICE HW version  : 4
                   ICE FW version  : 1.14 (rel. 268)
                   Serial number   : BUR2009xxxxx
                   Vtarget         : 4.72 V
                   JTAG clock megaAVR/program   : 1000 kHz
                   JTAG clock megaAVR/debug     : 100 kHz
                   PDI/UPDI clock Xmega/megaAVR : 100 kHz
avrdude_issue1038: partial Family_ID returned: "mega"
avrdude_issue1038: AVR device initialized and ready to accept instructions
avrdude_issue1038: device signature = 0x1e9650 (probably m4808)

avrdude_issue1038 done.  Thank you.

```

MCUdude, 2022-12-15:
It doesn't work in ISP mode because of #1212, but this is a separate issue.

mcuee, 2022-12-15:
> It doesn't work in ISP mode because of #1212, but this is a separate issue.

Thanks for the clarifications. UPDI mode works fine.

Test results under Windows,
```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_issue1038 -c pickit4_updi -p m4808 -v

avrdude_issue1038: Version 7.0-20221215 (4ecb34f)
                   Copyright the AVRDUDE authors;
                   see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

                   System wide configuration file is C:\work\avr\avrdude_test\avrdude_bin\avrdude.conf

                   Using Port                    : usb
                   Using Programmer              : pickit4_updi
avrdude_issue1038: found CMSIS-DAP compliant device, using EDBG protocol
                   AVR Part                      : ATmega4808
                   RESET disposition             : dedicated
                   RETRY pulse                   : SCK
                   Serial program mode           : yes
                   Parallel program mode         : yes
                   Memory Detail                 :

                                                     Block Poll               Page                       Polled
                     Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
                     ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
                     fuse0       wdtcfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
                     fuse1       bodcfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
                     fuse2       osccfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
                     fuse4       tcd0cfg     0     0     0    0 no          1    1      0     0     0 0x00 0x00
                     fuse5       syscfg0     0     0     0    0 no          1    1      0     0     0 0x00 0x00
                     fuse6       syscfg1     0     0     0    0 no          1    1      0     0     0 0x00 0x00
                     fuse7       append      0     0     0    0 no          1    1      0     0     0 0x00 0x00
                     fuse8       bootend     0     0     0    0 no          1    1      0     0     0 0x00 0x00
                     fuses                   0     0     0    0 no          9   10      0     0     0 0x00 0x00
                     lock                    0     0     0    0 no          1    1      0     0     0 0x00 0x00
                     tempsense               0     0     0    0 no          2    1      0     0     0 0x00 0x00
                     signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
                     prodsig                 0     0     0    0 no         61   61      0     0     0 0x00 0x00
                     sernum                  0     0     0    0 no         10    1      0     0     0 0x00 0x00
                     osccal16                0     0     0    0 no          2    1      0     0     0 0x00 0x00
                     osccal20                0     0     0    0 no          2    1      0     0     0 0x00 0x00
                     osc16err                0     0     0    0 no          2    1      0     0     0 0x00 0x00
                     osc20err                0     0     0    0 no          2    1      0     0     0 0x00 0x00
                     data                    0     0     0    0 no          0    1      0     0     0 0x00 0x00
                     userrow     usersig     0     0     0    0 no         64   64      0     0     0 0x00 0x00
                     eeprom                  0     0     0    0 no        256   64      0     0     0 0x00 0x00
                     flash                   0     0     0    0 no      49152  128      0     0     0 0x00 0x00

                   Programmer Type : JTAGICE3_UPDI
                   Description     : MPLAB(R) PICkit 4 in UPDI mode
                   ICE HW version  : 4
                   ICE FW version  : 1.14 (rel. 268)
                   Serial number   : BUR2009xxxxx
                   Vtarget         : 4.79 V
                   JTAG clock megaAVR/program   : 1000 kHz
                   JTAG clock megaAVR/debug     : 100 kHz
                   PDI/UPDI clock Xmega/megaAVR : 100 kHz
avrdude_issue1038: partial Family_ID returned: "mega"
avrdude_issue1038: AVR device initialized and ready to accept instructions
avrdude_issue1038: device signature = 0x1e9650 (probably m4808)

avrdude_issue1038 done.  Thank you.
```

MCUdude, 2022-12-15:
@mcuee I've just pushed now code that allows libusb to read and store the serial number as well. Can you try to download [Zadig](https://zadig.akeo.ie) and replace the driver Windows uses for PICkit4 with libusb to check if it works?

mcuee, 2022-12-15:
I am having a strange issue under macOS. But the issue is not with this PR since git main also crashed. I will need to debug further.

```
mcuee@mcuees-Mac-mini avrdude_bin % ./avrdude_issue1038 -c pickit4_updi -p m4808 -v

avrdude_issue1038: Version 7.0-20221215 (4ecb34f)
                   Copyright the AVRDUDE authors;
                   see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

                   System wide configuration file is /Users/mcuee/build/avr/avrdude_test/avrdude_bin/avrdude.conf
                   User configuration file is /Users/mcuee/.avrduderc
                   User configuration file does not exist or is not a regular file, skipping

                   Using Port                    : usb
                   Using Programmer              : pickit4_updi
zsh: segmentation fault  ./avrdude_issue1038 -c pickit4_updi -p m4808 -v
mcuee@mcuees-Mac-mini avrdude_bin % ./avrdude_git -c pickit4_updi -p m4808 -v

avrdude_git: Version 7.0-20221214 (159d126)
             Copyright the AVRDUDE authors;
             see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

             System wide configuration file is /Users/mcuee/build/avr/avrdude_test/avrdude_bin/avrdude.conf
             User configuration file is /Users/mcuee/.avrduderc
             User configuration file does not exist or is not a regular file, skipping

             Using Port                    : usb
             Using Programmer              : pickit4_updi
zsh: segmentation fault  ./avrdude_git -c pickit4_updi -p m4808 -v

mcuee@mcuees-Mac-mini avrdude_bin % lldb ./avrdude_git
(lldb) target create "./avrdude_git"
Current executable set to '/Users/mcuee/build/avr/avrdude_test/avrdude_bin/avrdude_git' (arm64).
(lldb) run -c pickit4_updi -p m4808 -v
Process 56112 launched: '/Users/mcuee/build/avr/avrdude_test/avrdude_bin/avrdude_git' (arm64)

avrdude_git: Version 7.0-20221214 (159d126)
             Copyright the AVRDUDE authors;
             see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

             System wide configuration file is /Users/mcuee/build/avr/avrdude_test/avrdude_bin/avrdude.conf
             User configuration file is /Users/mcuee/.avrduderc
             User configuration file does not exist or is not a regular file, skipping

             Using Port                    : usb
             Using Programmer              : pickit4_updi
error: libavrdude.a(usb_hidapi.c.o) failed to load objfile for /Users/mcuee/build/avr/avrdude_test/avrdude_main/build_darwin/src/libavrdude.a
error: libavrdude.a(jtag3.c.o) failed to load objfile for /Users/mcuee/build/avr/avrdude_test/avrdude_main/build_darwin/src/libavrdude.a
Process 56112 stopped
* thread #1, queue = 'com.apple.main-thread', stop reason = EXC_BAD_ACCESS (code=1, address=0x16)
    frame #0: 0x00000001001b7200 libhidapi.0.dylib`set_report + 28
libhidapi.0.dylib`set_report:
->  0x1001b7200 <+28>: ldr    w8, [x0, #0x14]
    0x1001b7204 <+32>: cbz    w8, 0x1001b7210           ; <+44>
    0x1001b7208 <+36>: mov    w8, #-0x1
    0x1001b720c <+40>: b      0x1001b7238               ; <+84>
Target 0: (avrdude_git) stopped.
(lldb) bt
error: avrdude_git debug map object file '/Users/mcuee/build/avr/avrdude_test/avrdude_main/build_darwin/src/CMakeFiles/avrdude.dir/main.c.o' has changed (actual time is 2022-12-15 21:00:44.000000000, debug map time is 2022-12-15 20:42:57.000000000) since this executable was linked, file will be ignored
* thread #1, queue = 'com.apple.main-thread', stop reason = EXC_BAD_ACCESS (code=1, address=0x16)
  * frame #0: 0x00000001001b7200 libhidapi.0.dylib`set_report + 28
    frame #1: 0x000000010006f194 avrdude_git`usbhid_open + 356
    frame #2: 0x0000000100043070 avrdude_git`jtag3_open_common + 292
    frame #3: 0x0000000100047e0c avrdude_git`jtag3_open_updi + 296
    frame #4: 0x0000000100006a7c avrdude_git`main + 4820
    frame #5: 0x0000000187e37e50 dyld`start + 2544
(lldb) quit
Quitting LLDB will kill one or more processes. Do you really want to proceed: [Y/n] Y
```

[... 43 lines cut ...]
           osccal16                0     0     0    0 no          2    1      0     0     0 0x00 0x00
           osccal20                0     0     0    0 no          2    1      0     0     0 0x00 0x00
           osc16err                0     0     0    0 no          2    1      0     0     0 0x00 0x00
           osc20err                0     0     0    0 no          2    1      0     0     0 0x00 0x00
           data                    0     0     0    0 no          0    1      0     0     0 0x00 0x00
           userrow     usersig     0     0     0    0 no         64   64      0     0     0 0x00 0x00
           eeprom                  0     0     0    0 no        256   64      0     0     0 0x00 0x00
           flash                   0     0     0    0 no      49152  128      0     0     0 0x00 0x00

         Programmer Type : JTAGICE3_UPDI
         Description     : MPLAB(R) PICkit 4 in UPDI mode
         ICE HW version  : 4
         ICE FW version  : 1.14 (rel. 268)
         Serial number   : 
         Vtarget         : 4.87 V
         JTAG clock megaAVR/program   : 1000 kHz
         JTAG clock megaAVR/debug     : 100 kHz
         PDI/UPDI clock Xmega/megaAVR : 100 kHz
avrdude: partial Family_ID returned: "mega"
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9650 (probably m4808)

avrdude done.  Thank you.

```

mcuee, 2022-12-15:
> @mcuee I've just pushed now code that allows libusb to read and store the serial number as well. Can you try to download [Zadig](https://zadig.akeo.ie) and replace the driver Windows uses for PICkit4 with libusb to check if it works?

It works well in printing out the serial number but then it does not go further. Kind of strange.

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_issue1038 -c pickit4_updi -p m4808 -v

avrdude_issue1038: Version 7.0-20221215 (4ecb34f)
                   Copyright the AVRDUDE authors;
                   see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

                   System wide configuration file is C:\work\avr\avrdude_test\avrdude_bin\avrdude.conf

                   Using Port                    : usb
                   Using Programmer              : pickit4_updi
avrdude_issue1038 usbhid_open() warning: USB device with VID: 0x03eb and PID: 0x2177 not found
avrdude_issue1038 usbhid_open() warning: USB device with VID: 0x03eb and PID: 0x2178 not found
avrdude_issue1038 usbhid_open() warning: USB device with VID: 0x03eb and PID: 0x2179 not found
avrdude_issue1038: usbdev_open(): found MPLAB PICkit 4 CMSIS-DAP, serno: BUR2009xxxxx
avrdude_issue1038: found CMSIS-DAP compliant device, using EDBG protocol

```

MCUdude, 2022-12-15:
> It works well in printing out the serial number but then it does not go further. Kind of strange

Does this happen with git main as well? I'm trying to figure out if this is caused by my commits or not.

mcuee, 2022-12-16:
> > It works well in printing out the serial number but then it does not go further. Kind of strange
> 
> Does this happen with git main as well? I'm trying to figure out if this is caused by my commits or not.

The behavior is a bit different but git main will not work either. 
I used Zadig to replace the Interface 0 from Windows HID driver to WinUSB. No change tot the PICkit 4 Virtual COM Port.

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_issue1038 -c pickit4_updi -p m4808 -v

avrdude_issue1038: Version 7.0-20221215 (4ecb34f)
                   Copyright the AVRDUDE authors;
                   see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

                   System wide configuration file is C:\work\avr\avrdude_test\avrdude_bin\avrdude.conf

                   Using Port                    : usb
                   Using Programmer              : pickit4_updi
avrdude_issue1038 usbhid_open() warning: USB device with VID: 0x03eb and PID: 0x2177 not found
avrdude_issue1038 usbhid_open() warning: USB device with VID: 0x03eb and PID: 0x2178 not found
avrdude_issue1038 usbhid_open() warning: USB device with VID: 0x03eb and PID: 0x2179 not found
avrdude_issue1038: usbdev_open(): found MPLAB PICkit 4 CMSIS-DAP, serno: BUR200973052
avrdude_issue1038: found CMSIS-DAP compliant device, using EDBG protocol

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_git -c pickit4_updi -p m4808 -v

avrdude_git: Version 7.0-20221214 (159d126)
             Copyright the AVRDUDE authors;
             see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

             System wide configuration file is C:\work\avr\avrdude_test\avrdude_bin\avrdude.conf

             Using Port                    : usb
             Using Programmer              : pickit4_updi
avrdude_git usbhid_open() warning: USB device with VID: 0x03eb and PID: 0x2177 not found
avrdude_git usbhid_open() warning: USB device with VID: 0x03eb and PID: 0x2178 not found
avrdude_git usbhid_open() warning: USB device with VID: 0x03eb and PID: 0x2179 not found
avrdude_git: usbdev_open(): found MPLAB PICkit 4 CMSIS-DAP, serno: BUR200973052
avrdude_git: found CMSIS-DAP compliant device, using EDBG protocol
avrdude_git usbdev_send() error: wrote -2 out of 912 bytes, err = No such file or directory
avrdude_git jtag3_edbg_prepare() error: unable to send command to serial port
avrdude_git main() error: unable to open programmer pickit4_updi on port usb

avrdude_git done.  Thank you.
```

mcuee, 2022-12-16:
@MCUdude 

I just tested under Linux and git main does not work either with libusb. So the issue is not with your changes.

For this test, I build avrdude without hidapi support. And I make sure the hid drver is already detached and then libusb is used. 

hidapi version works fine (with hidapi-libusb backend). So there is a bug here in git main.

```
mcuee@UbuntuSwift3:~/build/avr/avrdude_main$ cat build_nohidapi.sh 
#!/bin/sh
cmake -D CMAKE_BUILD_TYPE=RelWithDebInfo -D HAVE_LIBHIDAPI=OFF  -B build_linux
cmake --build build_linux

mcuee@UbuntuSwift3:~/build/avr/avrdude_bin$ ldd ./avrdude_git_nohidapi 
	linux-vdso.so.1 (0x00007ffc52b2f000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007fa6169f6000)
	libelf.so.1 => /lib/x86_64-linux-gnu/libelf.so.1 (0x00007fa6169da000)
	libusb-0.1.so.4 => /usr/local/lib/libusb-0.1.so.4 (0x00007fa6169d2000)
	libusb-1.0.so.0 => /usr/local/lib/libusb-1.0.so.0 (0x00007fa6169b2000)
	libftdi.so.1 => /lib/x86_64-linux-gnu/libftdi.so.1 (0x00007fa6169a8000)
	libftdi1.so.2 => /usr/local/lib/libftdi1.so.2 (0x00007fa616995000)
	libreadline.so.8 => /lib/x86_64-linux-gnu/libreadline.so.8 (0x00007fa616945000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fa616753000)
	/lib64/ld-linux-x86-64.so.2 (0x00007fa616c5f000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007fa616737000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007fa616714000)
	libudev.so.1 => /lib/x86_64-linux-gnu/libudev.so.1 (0x00007fa6166e7000)
	libtinfo.so.6 => /lib/x86_64-linux-gnu/libtinfo.so.6 (0x00007fa6166b5000)

mcuee@UbuntuSwift3:~/build/avr/avrdude_bin$ ./avrdude_git_nohidapi -c pickit4_updi -p m4808 -v

avrdude_git_nohidapi: Version 7.0-20221214 (159d126)
                      Copyright the AVRDUDE authors;
                      see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

                      System wide configuration file is /home/mcuee/build/avr/avrdude_bin/avrdude.conf
                      User configuration file is /home/mcuee/.avrduderc
                      User configuration file does not exist or is not a regular file, skipping

                      Using Port                    : usb
                      Using Programmer              : pickit4_updi
avrdude_git_nohidapi: usbdev_open(): found MPLAB PICkit 4 CMSIS-DAP, serno: BUR200973052
avrdude_git_nohidapi usbdev_open() warning: unable to set configuration 1: Device or resource busy
avrdude_git_nohidapi: found CMSIS-DAP compliant device, using EDBG protocol
avrdude_git_nohidapi usbdev_send() error: wrote -5 out of 912 bytes, err = Input/output error
avrdude_git_nohidapi jtag3_edbg_prepare() error: unable to send command to serial port
avrdude_git_nohidapi main() error: unable to open programmer pickit4_updi on port usb

avrdude_git_nohidapi done.  Thank you.

mcuee@UbuntuSwift3:~/build/avr/avrdude_bin$ ./avrdude_issue1038_nohidapi -c pickit4_updi -p m4808 -v

avrdude_issue1038_nohidapi: Version 7.0-20221215 (22ee0ec)
                            Copyright the AVRDUDE authors;
                            see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

                            System wide configuration file is /home/mcuee/build/avr/avrdude_bin/avrdude.conf
                            User configuration file is /home/mcuee/.avrduderc
                            User configuration file does not exist or is not a regular file, skipping

                            Using Port                    : usb
                            Using Programmer              : pickit4_updi
avrdude_issue1038_nohidapi: usbdev_open(): found MPLAB PICkit 4 CMSIS-DAP, serno: BUR200973052
avrdude_issue1038_nohidapi usbdev_open() warning: unable to set configuration 1: Device or resource busy
avrdude_issue1038_nohidapi: found CMSIS-DAP compliant device, using EDBG protocol
avrdude_issue1038_nohidapi usbdev_send() error: wrote -5 out of 912 bytes, err = Input/output error
avrdude_issue1038_nohidapi jtag3_edbg_prepare() error: unable to send command to serial port
avrdude_issue1038_nohidapi main() error: unable to open programmer pickit4_updi on port usb

[... 39 lines cut ...]
                     osccal16                0     0     0    0 no          2    1      0     0     0 0x00 0x00
                     osccal20                0     0     0    0 no          2    1      0     0     0 0x00 0x00
                     osc16err                0     0     0    0 no          2    1      0     0     0 0x00 0x00
                     osc20err                0     0     0    0 no          2    1      0     0     0 0x00 0x00
                     data                    0     0     0    0 no          0    1      0     0     0 0x00 0x00
                     userrow     usersig     0     0     0    0 no         64   64      0     0     0 0x00 0x00
                     eeprom                  0     0     0    0 no        256   64      0     0     0 0x00 0x00
                     flash                   0     0     0    0 no      49152  128      0     0     0 0x00 0x00

                   Programmer Type : JTAGICE3_UPDI
                   Description     : MPLAB(R) PICkit 4 in UPDI mode
                   ICE HW version  : 4
                   ICE FW version  : 1.14 (rel. 268)
                   Serial number   : BUR2009xxxxx
                   Vtarget         : 4.76 V
                   JTAG clock megaAVR/program   : 1000 kHz
                   JTAG clock megaAVR/debug     : 100 kHz
                   PDI/UPDI clock Xmega/megaAVR : 100 kHz
avrdude_issue1038: partial Family_ID returned: "mega"
avrdude_issue1038: AVR device initialized and ready to accept instructions
avrdude_issue1038: device signature = 0x1e9650 (probably m4808)

avrdude_issue1038 done.  Thank you.

```

mcuee, 2022-12-16:
@MCUdude 
I have created Issue #1221 to capture the issue with git main (issues confirmed under Linux and Windows).

mcuee, 2022-12-16:
@MCUdude 

I am thinking you can go ahead and create a PR for Issue #1028. We can deal with #1221 later. I think you can keep the libusb serial number codes as well even though it can not be tested due to #1221.

mcuee, 2022-12-16:
> I am having a strange issue under macOS. But the issue is not with this PR since git main also crashed. I will need to debug further.

The issue has been sorted out -- root cause is due to a legacy avrdude dependency libhid problem. Removing the old libhid sorted out the issue.

@MCUdude 
Your modifications work well under macOS (only HIDAPI can be used). So please go ahead with the PR. Thanks.

```

mcuee@mcuees-Mac-mini avrdude_bin % ./avrdude_issue1038v1 -c pickit4_updi -p m4808 -v   

avrdude_issue1038v1: Version 7.0-20221215 (22ee0ec)
                     Copyright the AVRDUDE authors;
                     see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

                     System wide configuration file is /Users/mcuee/build/avr/avrdude_test/avrdude_bin/avrdude.conf
                     User configuration file is /Users/mcuee/.avrduderc
                     User configuration file does not exist or is not a regular file, skipping

                     Using Port                    : usb
                     Using Programmer              : pickit4_updi
avrdude_issue1038v1: found CMSIS-DAP compliant device, using EDBG protocol
                     AVR Part                      : ATmega4808
                     RESET disposition             : dedicated
                     RETRY pulse                   : SCK
                     Serial program mode           : yes
                     Parallel program mode         : yes
                     ...
                     Programmer Type : JTAGICE3_UPDI
                     Description     : MPLAB(R) PICkit 4 in UPDI mode
                     ICE HW version  : 4
                     ICE FW version  : 1.14 (rel. 268)
                     Serial number   : BUR2009xxxxx
                     Vtarget         : 4.92 V
                     JTAG clock megaAVR/program   : 1000 kHz
                     JTAG clock megaAVR/debug     : 100 kHz
                     PDI/UPDI clock Xmega/megaAVR : 100 kHz
avrdude_issue1038v1: partial Family_ID returned: "mega"
avrdude_issue1038v1: AVR device initialized and ready to accept instructions
avrdude_issue1038v1: device signature = 0x1e9650 (probably m4808)

avrdude_issue1038v1 done.  Thank you.
```

MCUdude, 2022-12-16:
> Your modifications work well under macOS (only HIDAPI can be used). So please go ahead with the PR. Thanks.

Thanks for testing! You can actually test libusb if you use an AVRISPmkII. Now it will print its serial number, something Avrdude didn't do before. I'll create a PR tonight.

mcuee, 2022-12-16:
> > Your modifications work well under macOS (only HIDAPI can be used). So please go ahead with the PR. Thanks.
> 
> Thanks for testing! You can actually test libusb if you use an AVRISPmkII. Now it will print its serial number, something Avrdude didn't do before. I'll create a PR tonight.

Good suggestion. I have only one more proper AVRISPmkii clone and it seems to work well here.

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_issue1038v1 -c avrisp2 -p m328p -v

avrdude_issue1038v1: Version 7.0-20221215 (22ee0ec)
...
...
                     Programmer Type : STK500V2
                     Description     : Atmel AVR ISP mkII
                     Programmer Model: AVRISP mkII
                     Hardware Version: 1
                     Serial number   : xxxxxxxxxxxxx
                     Firmware Version Controller : 1.24
                     Vtarget         : 4.9 V
                     SCK period      : 8.00 us

avrdude_issue1038v1: AVR device initialized and ready to accept instructions
avrdude_issue1038v1: device signature = 0x1e950f (probably m328p)

avrdude_issue1038v1 done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_git -c avrisp2 -p m328p -v

avrdude_git: Version 7.0-20221214 (159d126)
....
             Programmer Type : STK500V2
             Description     : Atmel AVR ISP mkII
             Programmer Model: AVRISP mkII
             Hardware Version: 1
             Firmware Version Controller : 1.24
             Vtarget         : 4.9 V
             SCK period      : 8.00 us

avrdude_git: AVR device initialized and ready to accept instructions
avrdude_git: device signature = 0x1e950f (probably m328p)

avrdude_git done.  Thank you.
```

closed 2022-12-21 by stefanrueger