avrdudes/avrdude #1041: Writing fusebits to the MCU from a file does not working

masya-chel, 2022-07-22:
I am trying to write fusebits to the MCU from a file, this function does not working - an error is displayed. See the screenshot below.

[image]

`avrdude.exe: Device signature = 0x1e9307 (probably m8)
avrdude.exe: reading input file "C:\CLI\AVRdude_CLI\fw_write\high.txt"
avrdude.exe: fileio: invalid operation=0
avrdude.exe: read from file 'C:\CLI\AVRdude_CLI\fw_write\high.txt' failed`

If I try to read fusebits from the MCU to a file, then all works correctly - files with the names "high" and "low" are created in the folder, and the value of fusebits is contained inside the files. See the screenshot below.

[image]

If I try to write fusebits to the MCU directly from the console, then fusebits are written correctly. See the screenshot below.

[image]

Tested on avrdude versions: 6.3; 6.4; 7.0. The function of writing fusebits to the MCU from a file does not work.

dl8dtl, 2022-07-22:
> If I try to write fusebits to the MCU directly from the console, then fusebits are written correctly.

That suggests that something in your other (GUI, whatever it might be) execution environment must be strange.

However, what really puzzles me:

> `avrdude.exe: fileio: invalid operation=0`

`FIO_READ` *is* 0, but why does the switch/case instruction falls into the `default` clause?

I'm afraid noone outside your GUI environment might be able to reproduce that.

dl8dtl, 2022-07-22:
My only other idea: is it really sure that both invocations (GUI and commandline) are calling the exact same binary?

`FIO_READ` and `FIO_WRITE`might be a little too generic as names. So if anyone else in the toolchain defines `FIO_READ`differently than 0, and that makes it into the switch/case instruction in `fileio.c`, then passing a 0 will trigger the default case in the switch.

masya-chel, 2022-07-22:
> is it really sure that both invocations (GUI and commandline) are calling the exact same binary?

I can re-test the fusebits writing with the enabled avrdude option "verbose output" (-vvv). This option displays the full paths to the avrdude executable file in the terminal.

masya-chel, 2022-07-22:
Reading fusebits from the MCU to a file + enabled option "verbose output"

```
Текущая кодовая страница: 866

C:\CLI\AVRdude_CLI>C:\CLI\AVRdude_CLI\avrdude\avrdude.exe -P usb -c usbasp -p at
mega8 -U hfuse:r:C:\CLI\AVRdude_CLI\fw_read\high.txt:h -U lfuse:r:C:\CLI\AVRdude
_CLI\fw_read\low.txt:h -vvv
executable_abspath = C:/CLI/AVRdude_CLI/avrdude/avrdude.exe
executable_abspath_len = 38
executable_dirpath = C:/CLI/AVRdude_CLI/avrdude
executable_dirpath_len = 26
sys_config = C:/CLI/AVRdude_CLI/avrdude/avrdude.conf
sys_config_found = true

avrdude.exe: Version 7.0
             Copyright (c) Brian Dean, http://www.bdmicro.com/
             Copyright (c) Joerg Wunsch

             System wide configuration file is "C:/CLI/AVRdude_CLI/avrdude/avrdu
de.conf"

             Using Port                    : usb
             Using Programmer              : usbasp
avrdude.exe: usbasp_open("usb")
avrdude.exe: seen device from vendor ->www.fischl.de<-
avrdude.exe: seen product ->USBasp<-
             AVR Part                      : ATmega8
             Chip Erase delay              : 10000 us
             PAGEL                         : PD7
             BS2                           : PC2
             RESET disposition             : dedicated
             RETRY pulse                   : SCK
             Serial program mode           : yes
             Parallel program mode         : yes
             Timeout                       : 200
             StabDelay                     : 100
             CmdexeDelay                   : 25
             SyncLoops                     : 32
             PollIndex                     : 3
             PollValue                     : 0x53
             Memory Detail                 :

                                               Block Poll               Page
                   Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pa
ges MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ---
--- ----- ----- ---------
               eeprom                  4    20   128    0 no        512    4
  0  9000  9000 0xff 0xff
                                               Block Poll               Page
                   Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pa
ges MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ---
--- ----- ----- ---------
               flash                  33    10    64    0 yes      8192   64
128  4500  4500 0xff 0x00
                                               Block Poll               Page
                   Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pa
ges MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ---
--- ----- ----- ---------
               lfuse                   0     0     0    0 no          1    1
  0  2000  2000 0x00 0x00
                                               Block Poll               Page
                   Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pa
[... 43 lines cut ...]
exe: usbasp_spi_cmd(0x30, 0x00, 0x02, 0x00) => 0x00, 0x30, 0x00, 0x07
Reading | ################################################## | 100% 0.02s

avrdude.exe: Device signature = 0x1e9307 (probably m8)
avrdude.exe: reading hfuse memory:

Reading |                                                    | 0% 0.00savrdude.e
xe: usbasp_spi_cmd(0x58, 0x08, 0x00, 0x00) => 0x00, 0x58, 0x08, 0xc9
Reading | ################################################## | 100% 0.01s

avrdude.exe: writing output file "C:\CLI\AVRdude_CLI\fw_read\high.txt"
avrdude.exe: reading lfuse memory:

Reading |                                                    | 0% 0.00savrdude.e
xe: usbasp_spi_cmd(0x50, 0x00, 0x00, 0x00) => 0x00, 0x50, 0x00, 0x9f
Reading | ################################################## | 100% 0.00s

avrdude.exe: writing output file "C:\CLI\AVRdude_CLI\fw_read\low.txt"
avrdude.exe: usbasp_close()

avrdude.exe done.  Thank you.

C:\CLI\AVRdude_CLI>Pause
Для продолжения нажмите любую клавишу . . .
```

masya-chel, 2022-07-22:
Writing fusebits to the MCU from a file + enabled option "verbose output".

```
Текущая кодовая страница: 866

C:\CLI\AVRdude_CLI>C:\CLI\AVRdude_CLI\avrdude\avrdude.exe -P usb -c usbasp -p at
mega8 -U hfuse:w:C:\CLI\AVRdude_CLI\fw_write\high.txt:h -U lfuse:w:C:\CLI\AVRdud
e_CLI\fw_write\low.txt:h -vvv
executable_abspath = C:/CLI/AVRdude_CLI/avrdude/avrdude.exe
executable_abspath_len = 38
executable_dirpath = C:/CLI/AVRdude_CLI/avrdude
executable_dirpath_len = 26
sys_config = C:/CLI/AVRdude_CLI/avrdude/avrdude.conf
sys_config_found = true

avrdude.exe: Version 7.0
             Copyright (c) Brian Dean, http://www.bdmicro.com/
             Copyright (c) Joerg Wunsch

             System wide configuration file is "C:/CLI/AVRdude_CLI/avrdude/avrdu
de.conf"

             Using Port                    : usb
             Using Programmer              : usbasp
avrdude.exe: usbasp_open("usb")
avrdude.exe: seen device from vendor ->www.fischl.de<-
avrdude.exe: seen product ->USBasp<-
             AVR Part                      : ATmega8
             Chip Erase delay              : 10000 us
             PAGEL                         : PD7
             BS2                           : PC2
             RESET disposition             : dedicated
             RETRY pulse                   : SCK
             Serial program mode           : yes
             Parallel program mode         : yes
             Timeout                       : 200
             StabDelay                     : 100
             CmdexeDelay                   : 25
             SyncLoops                     : 32
             PollIndex                     : 3
             PollValue                     : 0x53
             Memory Detail                 :

                                               Block Poll               Page
                   Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pa
ges MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ---
--- ----- ----- ---------
               eeprom                  4    20   128    0 no        512    4
  0  9000  9000 0xff 0xff
                                               Block Poll               Page
                   Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pa
ges MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ---
--- ----- ----- ---------
               flash                  33    10    64    0 yes      8192   64
128  4500  4500 0xff 0x00
                                               Block Poll               Page
                   Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pa
ges MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ---
--- ----- ----- ---------
               lfuse                   0     0     0    0 no          1    1
  0  2000  2000 0x00 0x00
                                               Block Poll               Page
                   Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pa
ges MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ---
--- ----- ----- ---------
               hfuse                   0     0     0    0 no          1    1
  0  2000  2000 0x00 0x00
                                               Block Poll               Page
                   Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pa
ges MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ---
--- ----- ----- ---------
               lock                    0     0     0    0 no          1    1
  0  2000  2000 0x00 0x00
                                               Block Poll               Page
                   Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pa
ges MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ---
--- ----- ----- ---------
               calibration             0     0     0    0 no          4    1
  0     0     0 0x00 0x00
                                               Block Poll               Page
                   Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pa
ges MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ---
--- ----- ----- ---------
               signature               0     0     0    0 no          3    1
  0     0     0 0x00 0x00

             Programmer Type : usbasp
             Description     : USBasp, http://www.fischl.de/usbasp/

avrdude.exe: usbasp_initialize()
avrdude.exe: usbasp_spi_set_sck_period(0)
avrdude.exe: auto set sck period (because given equals null)
avrdude.exe: usbasp_program_enable()
avrdude.exe: AVR device initialized and ready to accept instructions

Reading |                                                    | 0% 0.00savrdude.e
xe: usbasp_spi_cmd(0x30, 0x00, 0x00, 0x00) => 0x00, 0x30, 0x00, 0x1e
avrdude.exe: usbasp_spi_cmd(0x30, 0x00, 0x01, 0x00) => 0x00, 0x30, 0x00, 0x93
Reading | #################                                  | 33% 0.01savrdude.
exe: usbasp_spi_cmd(0x30, 0x00, 0x02, 0x00) => 0x00, 0x30, 0x00, 0x07
Reading | ################################################## | 100% 0.01s

avrdude.exe: Device signature = 0x1e9307 (probably m8)
avrdude.exe: reading input file "C:\CLI\AVRdude_CLI\fw_write\high.txt"
avrdude.exe: fileio: invalid operation=0
avrdude.exe: read from file 'C:\CLI\AVRdude_CLI\fw_write\high.txt' failed
avrdude.exe: usbasp_close()

avrdude.exe done.  Thank you.

C:\CLI\AVRdude_CLI>Pause
Для продолжения нажмите любую клавишу . . .
```

masya-chel, 2022-07-22:
In both cases, command line calling the exact same binary, located at:
C:/CLI/AVRdude_CLI/avrdude/avrdude.exe

dl8dtl, 2022-07-22:
Who produced the binary?

stefanrueger, 2022-07-23:
This is correct behaviour. The file formats :d, :h, :o, :b are only implemented for output (ie, when reading the memory), not for input (ie, when writing to the memory). This is documented in
```
$ man avrdude
```
Look for `  d    decimal; this and the following formats are only valid on output.`

You should be able to use hex, octal, decimal or binary with the immediate :m format. Try `-U memory:w:$(cat file.txt):m` or similar in bash (have not tried).

stefanrueger, 2022-07-23:
It's a reasonable extension for AVRDUDE to implement :x, :o, :d, :b for input, too, but unclear how much effort should go into and how to priotitse

masya-chel, 2022-07-23:
> This is correct behaviour. The file formats :d, :h, :o, :b are only implemented for output (ie, when reading the memory), not for input (ie, when writing to the memory). This is documented in
> $ man avrdude

In the avrdude documentation, I did not find any restrictions on the read-write formats used.

See the [avrdude doc](https://download-mirror.savannah.gnu.org/releases/avrdude/avrdude-doc-6.3.pdf).
Chapter 2: Command Line Options
-U memtype:op:filename[:format]

dl8dtl, 2022-07-23:
> In the avrdude documentation, I did not find any restrictions on the read-write formats used.

From the man page:

" d    decimal; this and the following formats are only valid
                        on output."

dl8dtl, 2022-07-23:
I agree that a better error message is due.

masya-chel, 2022-07-23:
> Look for ` d decimal; this and the following formats are only valid on output.`

I don't use the decimal format! I use hexadecimal format to write fuses.

masya-chel, 2022-07-23:
In this case, it is necessary to explicitly write restrictions for each format flag in the documentation. The current documentation implicitly hides important information of one flag about another flag.

mcuee, 2022-07-23:
> It's a reasonable extension for AVRDUDE to implement :x, :o, :d, :b for input, too, but unclear how much effort should go into and how to prioritise. 

@stefanrueger 
With PR #1042 to address the documentation issue, I think this is probably not a priority and not that necessary either.

dl8dtl, 2022-07-23:
> I don't use the decimal format!

"this and the following formats"

dl8dtl, 2022-07-23:
I extended `avrdude.1` and `avrdude.texi` in the PR as well.

masya-chel, 2022-07-23:
Ok, thank you

stefanrueger, 2022-08-04:
Fixed by PR #1042

closed 2022-08-04 by stefanrueger