avrdudes/avrdude #1043: FLIP1 Device Signature read problem for ATmega16U2

mcuee, 2022-07-23:
Ref: The official Arduino Uno design uses ATmega16U2 (or ATmega8U2) for the USB to Serial function. The ATmega16U2 can be put into DFU mode and then you can use different host program to upload FW to the chip.
Ref: https://support.arduino.cc/hc/en-us/articles/4408887452434-Flash-USB-to-serial-firmware-in-DFU-mode

Somehow avrdude FLIP1 implementation has problem to read the device signature in this case.

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_git.exe -c usbasp -qqp m16u2 
-U flash:r:read_16u2_usbasp.hex:i && echo OK
OK

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_git.exe -c flip1 -qqp m16u2 
-U flash:v:read_16u2_usbasp.hex:i && echo OK
avrdude_git.exe: Warning: Failed to read USB device string 1: Input/output error
avrdude_git.exe: Warning: USB bDeviceClass = 0 (expected 254)
avrdude_git.exe: Warning: USB bDeviceSubClass = 0 (expected 1)
avrdude_git.exe: Expected signature for ATmega16U2 is 1E 94 89
                 Double check chip, or use -F to override this check.

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_git.exe -c flip1 -qqp m16u2 
-F -U flash:v:read_16u2_usbasp.hex:i && echo OK
avrdude_git.exe: Warning: Failed to read USB device string 1: Input/output error
avrdude_git.exe: Warning: USB bDeviceClass = 0 (expected 254)
avrdude_git.exe: Warning: USB bDeviceSubClass = 0 (expected 1)
avrdude_git.exe: Expected signature for ATmega16U2 is 1E 94 89
OK

```

mcuee, 2022-07-23:
Write has the same problem.

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_git.exe -c flip1 -qqp m16u2 -e && echo OK
avrdude_git.exe: Warning: Failed to read USB device string 1: Input/output error
avrdude_git.exe: Warning: USB bDeviceClass = 0 (expected 254)
avrdude_git.exe: Warning: USB bDeviceSubClass = 0 (expected 1)
avrdude_git.exe: Expected signature for ATmega16U2 is 1E 94 89
                 Double check chip, or use -F to override this check.

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_git.exe -c flip1 -qqp m16u2 -F -e && echo OK
avrdude_git.exe: Warning: Failed to read USB device string 1: Input/output error
avrdude_git.exe: Warning: USB bDeviceClass = 0 (expected 254)
avrdude_git.exe: Warning: USB bDeviceSubClass = 0 (expected 1)
avrdude_git.exe: Expected signature for ATmega16U2 is 1E 94 89
OK

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_git.exe -c flip1 -qqp m16u2 -F
 -U flash:w:.\Arduino-COMBINED-dfu-usbserial-atmega16u2-Uno-Rev3.hex:i && echo OK
avrdude_git.exe: Warning: Failed to read USB device string 1: Input/output error
avrdude_git.exe: Warning: USB bDeviceClass = 0 (expected 254)
avrdude_git.exe: Warning: USB bDeviceSubClass = 0 (expected 1)
avrdude_git.exe: Expected signature for ATmega16U2 is 1E 94 89
OK

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_git.exe -c usbasp -qqp m16u2 
-U flash:v:.\Arduino-COMBINED-dfu-usbserial-atmega16u2-Uno-Rev3.hex:i && echo OK
OK

```

mcuee, 2022-07-23:
This may be an issue with the bootloader FW from Arduino.
https://github.com/arduino/ArduinoCore-avr/blob/master/firmwares/atmegaxxu2/Arduino-COMBINED-dfu-usbserial-atmega16u2-Uno-Rev3.hex

I just tried another DFU firmware on the official Arduino Leonardo board and it does not seem to have such issue.

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_git.exe -c flip1 -p m32u4 -v

avrdude_git.exe: Version 7.0-20220719 (affe4cb)
                 Copyright (c) Brian Dean, http://www.bdmicro.com/
                 Copyright (c) Joerg Wunsch

                 System wide configuration file is "C:/work/avr/avrdude_test/avrdude_bin/avrdude.conf"

                 Using Port                    : usb
                 Using Programmer              : flip1
                 AVR Part                      : ATmega32U4
                 Chip Erase delay              : 9000 us
                 PAGEL                         : PD7
                 BS2                           : PA0
                 RESET disposition             : dedicated
                 RETRY pulse                   : SCK
                 Serial program mode           : yes
                 Parallel program mode         : yes
                 Timeout                       : 200
                 StabDelay                     : 100
                 CmdexeDelay                   : 25
                 SyncLoops                     : 32
                 PollIndex                     : 3
                 PollValue                     : 0x53
                 Memory Detail                 :

                                                   Block Poll               Page                       Polled
                   Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
                   ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
                   eeprom                 65    20     4    0 no       1024    4      0  9000  9000 0x00 0x00
                   flash                  65     6   128    0 yes     32768  128    256  4500  4500 0x00 0x00
                   lfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
                   hfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
                   efuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
                   lock                    0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
                   calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00
                   signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00

                 Programmer Type : flip1
                 Description     : FLIP USB DFU protocol version 1 (doc7618)

avrdude_git.exe: Found VID=0x03eb PID=0x2ff4 at 001:058
avrdude_git.exe: Warning: USB bDeviceClass = 255 (expected 254)
    USB Vendor          : ATMEL (0x03EB)
    USB Product         : ATm32U4DFU (0x2FF4)
    USB Release         : 0.0.0
    USB Serial No       : 1.0.0
    USB max packet size : 32
avrdude_git.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude_git.exe: Device signature = 0x1e9587 (probably m32u4)

avrdude_git.exe done.  Thank you.

```

mcuee, 2022-07-23:
I will close this issue as invalid after a few days. Hopefully someone can confirm that this is an Arduino DFU bootloader issue as well.

MCUdude, 2022-07-23:
I can confirm that there's an issue with the ATmega16U2 (Arduino MEGA in my case) in when using `flip1` on macOS.

ATmega16U2:
```
$ ./avrdude -cflip1 -patmega16u2

avrdude: Warning: Failed to read USB device string 1: Broken pipe
avrdude: Warning: USB bDeviceClass = 0 (expected 254)
avrdude: Warning: USB bDeviceSubClass = 0 (expected 1)
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x940000
avrdude: Expected signature for ATmega16U2 is 1E 94 89
         Double check chip, or use -F to override this check.

avrdude done.  Thank you.
```

AT90USB162:
```
$ ./avrdude -cflip1 -pat90usb162

avrdude: Warning: USB bDeviceClass = 255 (expected 254)
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9482 (probably usb162)

avrdude done.  Thank you.
```

ATmega8U2:
```
$ ./avrdude -cflip1 -patmega8u2

avrdude: Warning: USB bDeviceClass = 255 (expected 254)
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9389 (probably m8u2)

avrdude done.  Thank you.
```

MCUdude, 2022-07-23:
@mcuee do you know where I can find the "factory" DFU bootloader shipped with the ATmega16u2? The only chip I have is used on an Arduino MEGA, and it would be interesting to see if the latest version of the Atmel Flip bootloader would work better.

mcuee, 2022-07-23:
> do you know where I can find the "factory" DFU bootloader shipped with the ATmega16u2? The only chip I have is used on an Arduino MEGA, and it would be interesting to see if the latest version of the Atmel Flip bootloader would work better with Avrdude.

It is here, for both the Arduino Uno Rev 3 and Arduino Mega Rev 3. You can see they have not updated the DFU and USB Serial FW for 10 years..
https://github.com/arduino/ArduinoCore-avr/tree/master/firmwares/atmegaxxu2

I was hoping the following may be more updated, but they have the same issue.
https://github.com/NicoHood/HoodLoader2/tree/master/avr/bootloaders/HexFiles

Then I search the Arduino Core github issues list, they do not have such report. Apparently people may follow the official Arduino guide and use dfu-programmer or Microchip/Atmel FLIP, but not avrdude. BTW, dfu-util (another popular dfu related utility) does not work at all with the DFU bootloader at all.

mcuee, 2022-07-23:
Ref: https://github.com/arduino/ArduinoCore-avr/issues/431

@MCUdude 
There is another FW issue report above, on the USB serial portion but not on the DFU portion. The firmware is simply combination of the DFU FW and the USB-serial FW.

mcuee, 2022-07-24:
@MCUdude 

Latest official Microchip/Atmel DFU bootloader download is here. It has no issues with the device signature read.
https://www.microchip.com/en-us/product/ATmega32u4
https://ww1.microchip.com/downloads/aemDocuments/documents/OTH/ProductDocuments/SoftwareLibraries/Firmware/megaUSB_DFU_Bootloaders.zip

mcuee, 2022-07-24:
I will try the following later using Microchip/Atmel Studio 7.
https://tewarid.github.io/2016/08/09/arduino-usb-dfu-firmware-from-scratch.html

mcuee, 2022-07-24:
Combined hex file readback.
[atmega16u2_combined_dfufix_usbserial_remove_emptylines..hex.txt](https://github.com/avrdudes/avrdude/files/9174950/atmega16u2_combined_dfufix_usbserial_remove_emptylines.hex.txt)
[atmega16u2_combined_dfufix_usbserial.hex.txt](https://github.com/avrdudes/avrdude/files/9174951/atmega16u2_combined_dfufix_usbserial.hex.txt)

mcuee, 2022-07-24:
BTW, probably the following code is also more updated for the USB Serial codes.
https://tewarid.github.io/2016/08/10/arduino-usb-serial-firmware-from-scratch.html
https://github.com/tewarid/arduino-usb-serial

mcuee, 2022-07-24:
I have created the following issue for Arduino Core.
* https://github.com/arduino/ArduinoCore-avr/issues/490

mcuee, 2022-07-24:
I will close this issue as it is not an avrdude issue.

closed 2022-07-24 by mcuee

mcuee, 2022-07-25:
@tormodvolden

dfu-util does not seem to work well with Microchip/Atmel FLIP1 DFU bootloader. Somehow it only read one bytes from the device.
http://ww1.microchip.com/downloads/en/Appnotes/doc8457.pdf

```
$ dfu-util -l
dfu-util 0.11

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2021 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Found DFU: [03eb:2fef] ver=0000, devnum=7, cfg=1, intf=0, path="1-2.4", 
alt=0, name="UNKNOWN", serial="UNKNOWN"

$ dfu-util -U read_m16u2.bin
dfu-util 0.11

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2021 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Opening DFU capable USB device...
Device ID 03eb:2fef
Device DFU version 0110
Claiming USB DFU Interface...
Setting Alternate Interface #0 ...
Determining device status...
DFU state(2) = dfuIDLE, status(0) = No error condition is present
DFU mode device DFU version 0110
Device returned transfer size 3072
Copying data from DFU device to PC
Upload  [=========================] 100%            1 bytes
Upload done.
Received a total of 1 bytes

```

avrdude has no issues.
```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_git.exe -c flip1 -p m16u2 
-U flash:r:read_m16u2_avrdude.hex:i

avrdude_git.exe: Warning: USB bDeviceClass = 0 (expected 254)
avrdude_git.exe: Warning: USB bDeviceSubClass = 0 (expected 1)
avrdude_git.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude_git.exe: Device signature = 0x1e9489 (probably m16u2)
avrdude_git.exe: reading flash memory:

Reading | ################################################## | 100% 0.64s

avrdude_git.exe: writing output file "read_m16u2_avrdude.hex"

avrdude_git.exe done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude_bin> ls .\read_16u2_flip1.hex

    Directory: C:\work\avr\avrdude_test\avrdude_bin

Mode                 LastWriteTime         Length Name
----                 -------------         ------ ----
-a---           23/7/2022  6:00 pm          37719 read_16u2_flip1.hex
```

mcuee, 2022-07-29:
> dfu-util is for devices following the USB DFU standard. From above reference, it looks like the Microchip FLIP protocol uses some low-level elements from the DFU standard, but is not compatible. Maybe it would be possible to tweak dfu-util to support these devices as well, but if avrdude or dfu-programmer work fine with them, it shouldn't be necessary.

Thanks for the explanation. I agree that it may not be necessary.