avrdudes/avrdude #1044: Automatic verification is designed to fail for stdin

stefanrueger, 2022-07-23:
**Problem.** Currently, automatic verification is instigated during option parsing of `-U`  through
https://github.com/avrdudes/avrdude/blob/affe4cb50a42c34a873be9bde8b5e650032cdcb5/src/main.c#L528-L532
So, each `-U memory:w:file:...` will see a verification request `-U memory:v:file:...` inserted into the update action list after the write request (unless `-V` was specified *before* the `-U`). This is designed to fail if file is stdin `-` as the file is read twice.

**Proposed solution.** I would make `verify` a global variable, remove above lines and change the code for writing to memory that it checks against the already read file (so that does not need to be read twice).

(Making `verify` global also helps the micronucleus programmer print a warning that it needs to switch off `verify` because the micronucleus bootloader cannot read flash.)

stefanrueger, 2022-07-23:
The following should give a verification error
```
yes "The quick brown fox jumps over the lazy dog" | avrdude -c ... -p ... -U eeprom:w:-:r
```

MCUdude, 2022-07-23:
Thanks for reporting! Would be great if a solution to this would be something #1036 could benefit from.

> `yes "The quick brown fox jumps over the lazy dog" | avrdude -c ... -p ... -U eeprom:w:-:r`

Ooh, this could be _very_ useful!

mcuee, 2022-07-24:
Indeed this is an issue. Verfication after writing will fail even though the writing itself is good.

```
$ yes "The quick brown fox jumps over the lazy dog" | ./avrdude -c usbasp -qqp m328p 
-U eeprom:w:-:r
avrdude.exe: verification error, first mismatch at byte 0x0000
             0x54 != 0x6f
avrdude.exe: verification error; content mismatch

$ yes "The quick brown fox jumps over the lazy dog" | ./avrdude -c usbasp -qqp m328p 
-U eeprom:v:-:r && echo OK
OK
```

dl8dtl, 2022-07-24:
I'm not a big fan of global variables … but we certainly ought to do something, I agree.

mcuee, 2022-07-31:
The simplest fix is probably documentation fix to mention this limitation of the current implementation -- and then provide the workaround of using `-V` before `-U`.

mcuee, 2022-07-31:
Related issue.
* https://github.com/avrdudes/avrdude/issues/441

stefanrueger, 2022-08-04:
Fixed by PR #1053

closed 2022-08-04 by stefanrueger