avrdudes/avrdude #1047: Limited support for STK600

MCUdude, 2022-07-25:
I don't own an [STK600 board](https://www.microchip.com/en-us/development-tool/ATSTK600), so I can't confirm this. 
But from the looks of it, the STK600 can only be used in ISP, HVPP or HVSP mode with Avrdude.

The STK600 itself also supports TPI, PDI, and UPDI. I'm not sure it supports high-voltage UPDI though...

MCUdude, 2022-07-25:
It looks like PDI and TPI are implemented through the `stk600` programmer option. However, JTAG and UPDI don't seem to be implemented. And according to this site, the STK600 supports high-voltage UPDI as well!
https://microchipsupport.force.com/s/article/tinyAVR--Re-enabling-the-UPDI-interface-using-12V-programing-in-STK600

I'm not sure if the STK600 us using the jtagmkii or jtag3/EDBG protocol for its JTAG implementation.
For UPDI, It's probably jtag3/EDBG.

mcuee, 2022-07-25:
From the command line we can see avrdude only supports STK600 in three modes (ISP, HVSP and PP).

```
  stk500           = Atmel STK500
  stk500hvsp       = Atmel STK500 V2 in high-voltage serial programming mode
  stk500pp         = Atmel STK500 V2 in parallel programming mode
  stk500v1         = Atmel STK500 Version 1.x firmware
  stk500v2         = Atmel STK500 Version 2.x firmware
  stk600           = Atmel STK600
  stk600hvsp       = Atmel STK600 in high-voltage serial programming mode
  stk600pp         = Atmel STK600 in parallel programming mode
```

MCUdude, 2022-07-25:
This line can be found in the `stk600_xprog_program_enable`. Wouldn't this confirm that PDI is supported by Avrdude as well?
https://github.com/avrdudes/avrdude/blob/affe4cb50a42c34a873be9bde8b5e650032cdcb5/src/stk500v2.c#L3747

mcuee, 2022-07-25:
@MCUdude 
You are right that PDI is supported.

mcuee, 2022-07-25:
As per AVR079.

All AVR8 jtag commands start with CMD_JTAG_AVR (0x90) 
The implementation of JTAG programming in STK600 is in equal to the JTAGICE 
mkII (see T0152-0007 JTAGICE mkII Communication Protocol) but without the 
JTAGICE mkII message structure and fifo packages, i.e. only the message body is 
used in STK600.

dl8dtl, 2022-07-27:
> However, JTAG and UPDI don't seem to be implemented.

UPDI certainly not, as nobody knew about it by the time the STK600 was introduced (when I wrote that implementation).

I thought JTAG were supported, but offhand, I'm not sure. I do own an STK600 (and a number of target and routing boards) so I could look at that.

MCUdude, 2022-07-27:
> UPDI certainly not, as nobody knew about it by the time the STK600 was introduced (when I wrote that implementation).

It would be interesting to know how which protocol they used (extended/undocumented jtagmkii, edbg/jtag3) to achieve UPDI support! AVR079 has not been updated and the online docs regarding the STK600 reveals very few details.

>  I do own an STK600 (and a number of target and routing boards) so I could look at that.

I was thinking about getting one for testing purposes, but I figured it was _way_ too expensive to justify it for such a limited use. Also, the cards that sits on top also cost an arm and a leg, so even if I had an STK600 I would probably only use it as a programmer for other, external targets.

dl8dtl, 2022-07-27:
Agreed. As neat as the idea with these various routing and socket boards was, the overall price of all that prevented the STK600 from becoming a success as the STK500 turned out to be. Bottom line, I think it's a niche only anyway.

MCUdude, 2022-09-24:
Guess what, I have an STK600 now!
I can confirm that JTAG works fine. However, I'll have to disconnect the USB cable every time I use the board. Not sure what's causing this, but it's really annoying.

```
$ ./avrdude -cstk600 -p atxmega256a3bu

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9843 (probably x256a3bu)

avrdude done.  Thank you.

$ ./avrdude -cstk600 -p atxmega256a3bu

avrdude: stk500v2_command(): error in CMD_XPROG: Timeout
avrdude: stk600_xprog_program_enable(): XPRG_CMD_ENTER_PROGMODE failed
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

avrdude: stk500v2_command(): error in CMD_XPROG: Failed
avrdude: stk600_xprog_program_disable(): XPRG_CMD_LEAVE_PROGMODE failed

avrdude done.  Thank you.
```

MCUdude, 2022-09-24:
I can get JTAG to work (once) with Xmegas, but I'm not able to figure out how to make the STK600 speak JTAG with "classic" AVRs.

ATmega2560 connected to STK600 using JTAG:

```
$ ./avrdude -cstk600 -p atmega2560

avrdude: stk500v2_command(): command failed
avrdude: stk500v2_program_enable(): bad AVRISPmkII connection status: MOSI fail, RST fail, SCK fail
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

avrdude done.  Thank you.

```

MCUdude, 2022-10-22:
I can confirm that the STK600 can't be used to program over JTAG using Avrdude:

```
$ ./avrdude -patmega2560 -cstk600 -v

avrdude: Version 7.0-20220508
         Copyright (c) Brian Dean, http://www.bdmicro.com/
         Copyright (c) Joerg Wunsch

         System wide configuration file is /Users/hans/Downloads/avrdude/src/avrdude.conf
         User configuration file is /Users/hans/.avrduderc

         Using Port                    : usb
         Using Programmer              : stk600
avrdude: usbdev_open(): found STK600, serno: 0045E550833E
         AVR Part                      : ATmega2560
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : possible i/o
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom                 65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
           flash                  65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
           lfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
           hfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
           efuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
           lock                    0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
           signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
           calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00

         Programmer Type : STK600
         Description     : Atmel STK600
         Programmer Model: STK600
         Hardware Version: 3
         Firmware Version Master : 2.45
         Firmware Version Slave 1: 2.03
         Firmware Version Slave 2: 2.02
         Routing card    : Not present
         Socket card     : STK600-ATMEGA2560
         RC_ID table rev : 245
         EC_ID table rev : 1
         Vtarget         : 4.8 V
         Varef 0         : 2.99 V
         Varef 1         : 3.29 V
         SCK period      : 3.25 us
         Oscillator      : 15.992 MHz

avrdude stk500v2_command() error: command failed
avrdude stk500v2_program_enable() error: bad AVRISPmkII connection status: MOSI fail, SCK fail
avrdude main() error: initialization failed, rc=-1
        double check connections and try again or use -F to override
        this check

avrdude done.  Thank you.
```

However, it works with PDI:
[... 40 lines cut ...]
         Programmer Type : STK600
         Description     : Atmel STK600
         Programmer Model: STK600
         Hardware Version: 3
         Firmware Version Master : 2.45
         Firmware Version Slave 1: 2.03
         Firmware Version Slave 2: 2.02
         Routing card    : Not present
         Socket card     : STK600-ATMEGA2560
         RC_ID table rev : 245
         EC_ID table rev : 1
         Vtarget         : 4.8 V
         Varef 0         : 2.99 V
         Varef 1         : 3.29 V
         SCK period      : 3.25 us
         Oscillator      : 15.992 MHz

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: device signature = 0x1e9742 (probably x128a3u)

avrdude done.  Thank you.
```

MCUdude, 2023-04-07:
I decided to briefly look into this again.

It appears that Avrdude supports STK600 TPI programming, even though the documentation doesn't explicitly state that TPI programming works.

I've not succeeded in connecting an external TPI device to my STK600 (I only own an STK600 + an ATmega2560 board), and the official documentation states that it's not recommended to use it as a dedicated TPI programmer because it outputs 12V on the reset line.

@dl8dtl since you have various routing cards and adapter boards, can you check whenever the Avrdude supports TPI programming using the STK600 if you're using an on-board socket like the `STK600-ATtiny10` board?

> I thought JTAG was supported, but offhand, I'm not sure. I do own an STK600 (and a number of target and routing boards) so I could look at that.

It would be great if you could look into this. According to the docs ([here](https://ww1.microchip.com/downloads/en/Appnotes/doc8133.pdf) and [here](http://atmel-studio-doc.s3-website-us-east-1.amazonaws.com/webhelp/GUID-EE00BF20-2F37-4A8A-B685-B91173F8915D-en-US-10/index.html?GUID-8253E632-B9B6-44E8-95F5-A0ED8CA9E1A8), It's using the JTAGmkII protocol, but (let me quote):

> The implementation of JTAG programming in STK600 is in equal to the JTAGICE mkII (see T0152-0007 JTAGICE mkII Communication Protocol) but without the JTAGICE mkII message structure and fifo packages, i.e. only the message body is used in STK600.

MCUdude, 2023-06-13:
@dl8dtl do you think you have time to look at this issue before Avrdude v7.2 is ready to be released?

UPDI support isn't that important, but it would be great to get JTAG and TPI to work.

mcuee, 2024-08-23:
FYI, I saw a Chinese company started to produce an STK600 clone (only the programmer portion). So I have ordered it and hopefully it would work (to arrive in about one month's time).

MCUdude, 2024-08-23:
@mcuee do you have a link?

mcuee, 2024-08-23:
> @mcuee do you have a link?

Not so sure if you have used Taobao or not. 
https://www.taobao.com/list/item/716520275503.htm

mcuee, 2024-08-23:
The shop has some other interesting stuff.
https://world.taobao.com/dianpu/34531851.htm?spm=a21wu.12321156.shop-area.1.651e46d3yelUqu

stefanrueger, 2024-08-23:
> STK600 without High Voltage programming.

Never judge a book by the cover, but the bottom pic just looks like an AVRISP mkII does, doesn't it?

mcuee, 2024-08-23:
> > STK600 without High Voltage programming.
> 
> Never judge a book by the cover, but the bottom pic just looks like an AVRISP mkII does, doesn't it?

Plastic tooling is expensive, so they reuse this for everything: stk500v2 clone, JTAG ICE 1 clone, AVRISP mkIi clone and now STK600 clone.

stefanrueger, 2024-08-23:
> reuse the same housing

USB type B 2.0 in 2024?

mcuee, 2024-08-24:
> > reuse the same housing
> 
> USB type B 2.0 in 2024?

Haha, we are talking about clones of legacy tools in reality here.

An electronics engineer will have to have all kinds of USB cables in reality.

In fact for testing avrdude, I have different types of USB cables, including some non-stanard ones. I use a USB Type C to 4-ports USB Type A hub with my laptop for testing. 
1) USB Type A to Type A (not so standard)
2) USB to 5V DC power (not so standard)
3) USB Type A to Type C
4) USB Type A to MicroUSB
5) USB Type A to Mini USB
6) USB Type A to USB 2.0 Type B
7) USB Type A to USB Serial DIN9 cables and USB to Serial TTL cables

Then for other Open Source projects, I have more.
1) USB Type A to USB 3.0 Type B (Cypress USB FX3 development board)
2) USB Type C to Type C (USB 4.0/TB4 40Gbps cable), for use with USB 4 to 10G Network Interface card
3) USB Type C to Type C (USB 3.1/3.2 10Gbps)

MCUdude, 2025-05-27:
I'm closing this as not planned. The documentation exists, but the STK600 never got much traction, unlike its predecessor, the STK500. Of course, PRs are always welcome!

closed 2025-05-27 by MCUdude