avrdudes/avrdude #1049: Fuses sometimes fail to write

7918514, 2022-07-30:
Hello. I was advised to contact you. I posted the description of the problem here
https://github.com/ZakKemble/AVRDUDESS/issues/67

mcuee, 2022-07-31:
From the above ticket.

verbosity level =1/2/3: failed
verbosity level =4: no issues
Reducing the ISP frequency to 93750 Hz: no issues

```
>>>: avrdude -c usbasp -p m168p -P usb -v -e -B 4.0 -U flash:w:"\\192.168.1.100\other\Увлажнитель\v2\Замена синего модуля\Firmware\ATmega\168P_168PA\atmega168P_calibrated.hex":a -U lfuse:w:0xFF:m -U hfuse:w:0xDF:m -U efuse:w:0xF9:m -U lock:w:0xC0:m 

avrdude.exe: Version 7.0
             Copyright (c) Brian Dean, http://www.bdmicro.com/
             Copyright (c) Joerg Wunsch

             System wide configuration file is "C:/Other Program/AVRDUDESS/avrdude.conf"

             Using Port                    : usb
             Using Programmer              : usbasp
             Setting bit clk period        : 4.0
             AVR Part                      : ATmega168P
             Chip Erase delay              : 9000 us
             PAGEL                         : PD7
             BS2                           : PC2
             RESET disposition             : dedicated
             RETRY pulse                   : SCK
             Serial program mode           : yes
             Parallel program mode         : yes
             Timeout                       : 200
             StabDelay                     : 100
             CmdexeDelay                   : 25
             SyncLoops                     : 32
             PollIndex                     : 3
             PollValue                     : 0x53
             Memory Detail                 :

                                               Block Poll               Page                       Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               eeprom                 65    20     4    0 no        512    4      0  3600  3600 0xff 0xff
               flash                  65     6   128    0 yes     16384  128    128  4500  4500 0xff 0xff
               lfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
               hfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
               efuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
               lock                    0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
               calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00
               signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00

             Programmer Type : usbasp
             Description     : USBasp, http://www.fischl.de/usbasp/

avrdude.exe: set SCK frequency to 187500 Hz
avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude.exe: Device signature = 0x1e940b (probably m168p)
avrdude.exe: erasing chip
avrdude.exe: set SCK frequency to 187500 Hz
avrdude.exe: reading input file "\\192.168.1.100\other\Увлажнитель\v2\Замена синего модуля\Firmware\ATmega\168P_168PA\atmega168P_calibrated.hex"
avrdude.exe: input file \\192.168.1.100\other\Увлажнитель\v2\Замена синего модуля\Firmware\ATmega\168P_168PA\atmega168P_calibrated.hex auto detected as Intel Hex
avrdude.exe: writing flash (4804 bytes):

Writing | ################################################## | 100% 2.78s

avrdude.exe: 4804 bytes of flash written
avrdude.exe: verifying flash memory against \\192.168.1.100\other\Увлажнитель\v2\Замена синего модуля\Firmware\ATmega\168P_168PA\atmega168P_calibrated.hex:
avrdude.exe: input file \\192.168.1.100\other\Увлажнитель\v2\Замена синего модуля\Firmware\ATmega\168P_168PA\atmega168P_calibrated.hex auto detected as Intel Hex

Reading | ################################################## | 100% 1.41s

avrdude.exe: 4804 bytes of flash verified
avrdude.exe: reading input file "0xFF"
avrdude.exe: writing lfuse (1 bytes):

Writing | ################################################## | 100% 0.00s

avrdude.exe: 1 bytes of lfuse written
avrdude.exe: verifying lfuse memory against 0xFF:

Reading | ################################################## | 100% 0.00s

avrdude.exe: 1 bytes of lfuse verified
avrdude.exe: reading input file "0xDF"
avrdude.exe: writing hfuse (1 bytes):

Writing |  ***failed;  
################################################## | 100% 0.03s

avrdude.exe: 1 bytes of hfuse written
avrdude.exe: verifying hfuse memory against 0xDF:

Reading | ################################################## | 100% 0.00s

avrdude.exe: verification error, first mismatch at byte 0x0000
             0xff != 0xdf
avrdude.exe: verification error; content mismatch

avrdude.exe done.  Thank you.

```

mcuee, 2022-07-31:
I can not reproduce the issue here.

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_git.exe -c usbasp -p m168p -P usb -v -U lfuse:w:0xFF:m -U hfuse:w:0xDF:m -U efuse:w:0xF9:m -U lock:w:0xC0:m

avrdude_git.exe: Version 7.0-20220719 (affe4cb)
                 Copyright (c) Brian Dean, http://www.bdmicro.com/
                 Copyright (c) Joerg Wunsch

                 System wide configuration file is "C:/work/avr/avrdude_test/avrdude_bin/avrdude.conf"

                 Using Port                    : usb
                 Using Programmer              : usbasp
                 AVR Part                      : ATmega168P
                 Chip Erase delay              : 9000 us
                 PAGEL                         : PD7
                 BS2                           : PC2
                 RESET disposition             : dedicated
                 RETRY pulse                   : SCK
                 Serial program mode           : yes
                 Parallel program mode         : yes
                 Timeout                       : 200
                 StabDelay                     : 100
                 CmdexeDelay                   : 25
                 SyncLoops                     : 32
                 PollIndex                     : 3
                 PollValue                     : 0x53
                 Memory Detail                 :

                                                   Block Poll               Page                       Polled
                   Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
                   ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
                   eeprom                 65    20     4    0 no        512    4      0  3600  3600 0xff 0xff
                   flash                  65     6   128    0 yes     16384  128    128  4500  4500 0xff 0xff
                   lfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
                   hfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
                   efuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
                   lock                    0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
                   calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00
                   signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00

                 Programmer Type : usbasp
                 Description     : USBasp, http://www.fischl.de/usbasp/

avrdude_git.exe: auto set sck period (because given equals null)
avrdude_git.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude_git.exe: Device signature = 0x1e940b (probably m168p)
avrdude_git.exe: reading input file "0xFF"
avrdude_git.exe: writing lfuse (1 bytes):

Writing | ################################################## | 100% 0.00s

avrdude_git.exe: 1 bytes of lfuse written
avrdude_git.exe: verifying lfuse memory against 0xFF:

Reading | ################################################## | 100% 0.00s

avrdude_git.exe: 1 bytes of lfuse verified
avrdude_git.exe: reading input file "0xDF"
avrdude_git.exe: writing hfuse (1 bytes):

Writing | ################################################## | 100% 0.01s

avrdude_git.exe: 1 bytes of hfuse written
avrdude_git.exe: verifying hfuse memory against 0xDF:

Reading | ################################################## | 100% 0.00s

avrdude_git.exe: 1 bytes of hfuse verified
avrdude_git.exe: reading input file "0xF9"
avrdude_git.exe: writing efuse (1 bytes):

Writing | ################################################## | 100% 0.01s

avrdude_git.exe: 1 bytes of efuse written
avrdude_git.exe: verifying efuse memory against 0xF9:

Reading | ################################################## | 100% 0.00s

avrdude_git.exe: 1 bytes of efuse verified
avrdude_git.exe: reading input file "0xC0"
avrdude_git.exe: writing lock (1 bytes):

Writing | ################################################## | 100% 0.01s

avrdude_git.exe: 1 bytes of lock written
avrdude_git.exe: verifying lock memory against 0xC0:

Reading | ################################################## | 100% 0.00s

avrdude_git.exe: 1 bytes of lock verified

avrdude_git.exe done.  Thank you.

```

mcuee, 2022-07-31:
@7918514 
As for ATtiny45/85, not so sure if it is related to Pull Request #983. You may want to try to increase the chip erase delay to see if that helps. 
* https://github.com/avrdudes/avrdude/pull/983

Another thing, you can try to split the commands into three steps to see if that helps: erase first, wrting fuse and lastly writing flash.

Or you can split into two steps as what Arduino ususally does (writing fuse first, then writing flash (which will erase the flash if `-D` is not specified).

mcuee, 2022-08-14:
@7918514 

Issue #983 is probably not related unless you are also using the gemma bootloader. Have you tried my suggestion of using two steps?

mcuee, 2022-08-22:
@7918514 
Any updates from your side? Thanks.

dl8dtl, 2022-08-24:
> @7918514
> 
> Issue #983 is probably not related unless you are also using the gemma bootloader. Have you tried my suggestion of using two steps?

Bootloaders cannot write fuses or lock bits anyway.

mcuee, 2022-08-25:
> Bootloaders cannot write fuses or lock bits anyway.

Good point. Thanks for pointing that out.

mcuee, 2022-09-05:
Closing this for now as not heard back from the OP. Please reopen when you have more info. Thanks.

closed 2022-09-05 by mcuee