avrdudes/avrdude #1052: JTAG programming doesn't work for SNAP and PICkit4

MCUdude, 2022-08-02:
I added the following programmer to avrdude.conf to see if I could get JTAG programming to work. JTAG works just with both the PK4 and the SNAP in MPLAB X. I've enabled both the OCDEN and the JTAGEN fuse for good measure.

```
programmer
  id    = "pickit4";
  desc  = "MPLAB(R) PICkit 4 in JTAG mode";
  type  = "jtagice3";
  connection_type = usb;
  usbpid = 0x2177, 0x2178, 0x2179;
;

programmer
  id    = "snap";
  desc  = "MPLAB(R) SNAP in JTAG mode";
  type  = "jtagice3";
  connection_type = usb;
  usbpid = 0x217F, 0x2180, 0x2181;
  hvupdi_support = 1;
;
```

PICkit4
```
$ ./avrdude -cpickit4 -p atmega1284p -B8

         Vtarget                      : 5.07 V
         JTAG clock megaAVR/program   : 125 kHz
         JTAG clock megaAVR/debug     : 100 kHz
         PDI/UPDI clock Xmega/megaAVR : 100 kHz

avrdude: JTAGEN fuse disabled?
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

avrdude done.  Thank you.
```

SNAP:
```
$ ./avrdude -csnap -p atmega1284p -B8
avrdude: usbhid_open(): No device found

         Vtarget                      : 5.06 V
         JTAG clock megaAVR/program   : 125 kHz
         JTAG clock megaAVR/debug     : 100 kHz
         PDI/UPDI clock Xmega/megaAVR : 100 kHz

avrdude: JTAGEN fuse disabled?
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

```

(Note that the debugWire support for these programmers is also lacking, but I haven't checked to see if dW works with these or not)

bonidydy, 2022-10-12:
Programming with SNAP over JTAG works with gdb and "Bloom" debug server [https://github.com/bloombloombloom/Bloom](url).

MCUdude, 2022-10-23:
I've tested a little more and can confirm that the SNAP and PICkit4 work with Xmega targets using JTAG. However, I'm still having issues communicating with "standard" 8-bit AVRs over JTAG. Really weird... Especially since other JTAG3/EDBG compatible programmers have no issues with JTAG.

```
$ ./avrdude -patxmega256a3bu -csnap -B10

         Vtarget                      : 3.30 V
         JTAG clock megaAVR/program   : 1000 kHz
         JTAG clock megaAVR/debug     : 100 kHz
         JTAG clock Xmega             : 100 kHz
         PDI/UPDI clock Xmega/megaAVR : 100 kHz
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude: device signature = 0x1e9843 (probably x256a3bu)

avrdude done.  Thank you.

$ ./avrdude -patxmega256a3bu -cpickit4 -B1

         Vtarget                      : 3.31 V
         JTAG clock megaAVR/program   : 1000 kHz
         JTAG clock megaAVR/debug     : 100 kHz
         JTAG clock Xmega             : 1000 kHz
         PDI/UPDI clock Xmega/megaAVR : 100 kHz
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude: device signature = 0x1e9843 (probably x256a3bu)

avrdude done.  Thank you.
```

mcuee, 2022-10-28:
@bonidydy 

Are you using the SNAP / PICKit 4 with class AVR in JTAG mode or with ATxmega parts? What is the issue you encounterd?

mcuee, 2022-11-05:
I can reproduce the issue.
Connection as per Page65 from PICKit 4 user guide. target is ATmega32A. OCDEN and JTAGEN are programmed (as 0).
http://ww1.microchip.com/downloads/en/DeviceDoc/PICkit4%20In-Circuit%20DebuggerUG-DS50002751B.pdf

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c pickit4 -p m32a -B8

         Vtarget                      : 4.93 V
         JTAG clock megaAVR/program   : 125 kHz
         JTAG clock megaAVR/debug     : 100 kHz
         PDI/UPDI clock Xmega/megaAVR : 100 kHz
avrdude error: JTAGEN fuse disabled?
avrdude error: initialization failed, rc=-1
        - double check the connections and try again
        - use -B to set lower ISP clock frequency, e.g. -B 200kHz
        - use -F to override this check

avrdude done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c avrispmkii -p m32a -qq -t
avrdude> dump lfuse
>>> dump lfuse
0000  bf                                                |.               |

avrdude> dump hfuse
>>> dump hfuse
0000  06                                                |.               |

avrdude> dump lock
>>> dump lock
0000  ff                                                |.               |

avrdude> quit
>>> quit
avrdude>

```

mcuee, 2022-11-05:
Need to learn how to use debugWire. It does not seem to work. But I need to check if it works with Microchip Studio or not first. Somehow once I enable DW bit, I can not communicate with the ATmega328P with AVRISP mkii (reset pin is now DW pin).

```

#------------------------------------------------------------
# pickit4 in DW mode
#------------------------------------------------------------

programmer
  id    = "pickit4_dw";
  desc  = "MPLAB(R) PICkit 4 in debugWire mode";
  type  = "jtagice3_dw";
  prog_modes = PM_debugWIRE;
  connection_type = usb;
  usbpid = 0x2177, 0x2178, 0x2179;
;

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c pickit4_dw -p m328p

         Vtarget                      : 4.94 V
         JTAG clock megaAVR/program   : 1000 kHz
         JTAG clock megaAVR/debug     : 100 kHz
         PDI/UPDI clock Xmega/megaAVR : 100 kHz
avrdude error: initialization failed, rc=-1
        - double check the connections and try again
        - use -B to set lower ISP clock frequency, e.g. -B 200kHz
        - use -F to override this check

avrdude done.  Thank you.

```

mcuee, 2022-11-05:
@MCUdude 
I check under Microchip Studio, PICKit 4 will not support DW mode of ATmega328P.

bonidydy, 2022-11-05:
Again I would like to point out the project "Bloom" [https://github.com/bloombloombloom/Bloom](url) which is a debug server for gdb. It can not only debug with PICKit4 and SNAP over DebugWire but also program targets with the gdb's "load" command. The same applies to JTAG, PDI and UPDI.

mcuee, 2022-11-05:
I will check out Bloom, as of now it seems to only support Linux, so it is probably not a solution for many avrdude users on Windows and macOS.

MCUdude, 2022-12-19:
@xedbg you're probably busy right before Christmas, but do you know if something as trivial as a flag or state needs to be set on these gen 4 programmers in order to support megaAVR JTAG programming?

MCUdude, 2022-12-19:
For reference, here are two verbose output logs. One for Atmel ICE (that works), and one for PICkit4 that fails. The PICkit4 fails at the sign-on command, and responds with a 0x33 error, even though the Atmel ICE and PICkit4 run the exact same initialization routine.

<details>
<summary>Atmel ICE JTAG init success (-vvv)</summary>

```
$ ./avrdude -catmelice -patmega16 -vvv
executable_abspath = /Users/hans/Downloads/avrdude/src/avrdude
executable_abspath_len = 41
executable_dirpath = /Users/hans/Downloads/avrdude/src
executable_dirpath_len = 33
sys_config = /Users/hans/Downloads/avrdude/src/avrdude.conf
sys_config_found = true

avrdude: Version 7.0-20220508
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is /Users/hans/Downloads/avrdude/src/avrdude.conf
         User configuration file is /Users/hans/.avrduderc

         Using Port                    : usb
         Using Programmer              : atmelice
avrdude: jtag3_open()
avrdude: usbhid_open(): probing for max packet size
avrdude: usbhid_open(): setting max_xfer from DAP_Info response to 512
avrdude: found CMSIS-DAP compliant device, using EDBG protocol
avrdude: jtag3_getsync()

avrdude: jtag3_edbg_prepare()
avrdude: jtag3_edbg_prepare(): connection status 0x01
avrdude: sending sign-on command: 
avrdude: jtag3_edbg_send(): sending 3 bytes
avrdude: jtag3_recv(): Got message seqno 0 (command_sequence == 0)

[general] OK
         AVR Part                      : ATmega16
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : possible i/o
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom                  4    10   128    0 no        512    4      0  9000  9000 0xff 0xff
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           flash                  33     6   128    0 yes     16384  128    128  4500  4500 0xff 0xff
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           lfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           hfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
[... 336 lines cut ...]
avrdude: retrying with external reset applied
avrdude jtag3_initialize() [jtag3.c:1338] error: JTAGEN fuse disabled?
avrdude main() [main.c:1244] error: initialization failed, rc=-1
        - double check the connections and try again
        - use -B to set lower ISP clock frequency, e.g. -B 200kHz
        - use -F to override this check

avrdude: jtag3_close()
avrdude: sending AVR sign-off command: 
avrdude: jtag3_edbg_send(): sending 3 bytes
avrdude: jtag3_recv(): Got message seqno 15 (command_sequence == 15)

[AVR] OK
avrdude: sending sign-off command: 
avrdude: jtag3_edbg_send(): sending 4 bytes
avrdude: jtag3_recv(): Got message seqno 16 (command_sequence == 16)

[general] OK

avrdude: jtag3_edbg_signoff()

avrdude done.  Thank you.
```

</details>

MCUdude, 2022-12-20:
My setup is also working, because I can ping the device over JTAG using pymcuprog:

```
$ pymcuprog ping -d atmega324pb
Connecting to anything possible
Connected to MPLAB PICkit 4 CMSIS-DAP from Microchip Technology Incorporated (serial number BUR2040xxxxx)
Debugger firmware version 1.14.268
Debugger hardware revision 4
pymcuprog.nvm - WARNING - 
pymcuprog.nvm - WARNING - megaAVR-JTAG stack is in Alpha state
pymcuprog.nvm - WARNING - Expect some features to be missing
pymcuprog.nvm - WARNING - 
Pinging device...
Ping response: 1E9403
Done.
```

xedbg, 2022-12-20:
Looks to me like there is some config missing before activating physical: reason 0x33
there are a few "must haves" for debugging (which unfortunately means programming too :/ )
- flash page size
- flash size
- smallest boot start
- sram start
- OCD reg address

For programming, the first two are needed - the rest can really be dummies

MCUdude, 2022-12-20:
It looks like the flash page size and flash size are already present. and I can confirm that it works with Atmel ICE and the Power Debugger, so my initial thought was that GEN4 programmers need a slightly different initialization.

Pymcuprog/pyedbglib is a nice rabbit hole to go down, but I'm not able to get a proper verbose debug output, so I have no clue which initialization packets are sent. If there where I could compare them to the ones Avrdude sends to see if I could figure out what Avrdude is doing wrong.

https://github.com/avrdudes/avrdude/blob/159d12693e8787fab5bf68d87f331bedf85164a1/src/jtag3.c#L1243-L1299

xedbg, 2022-12-20:
@MCUdude give us a minute then... Christmas lunch is only at 12... ;)

janegilruud, 2022-12-20:
PR submitted ;-)