avrdudes/avrdude #1054: AVR JTAG ICE can not program EEPROM properly

mcuee, 2022-08-03:
PR #1031 fixed Issue #443 (AVR JTAG ICE is unable to write fuse and lock bits). It helps EEPROM write a bit but not yet fixed the issue of EEPROM write.

* https://github.com/avrdudes/avrdude/pull/1031#issuecomment-1189562329

Comment from @dl8dtl 
> Yes, I think EEPROM should be added to these dummy reads. Without, the JTAG1 completely wedges, and the remaining AVRDUDE session is unusable (needs a reconnect). With that, it only fails to actually write to EEPROM.

mcuee, 2022-08-04:
@dl8dtl 
Can you share your failed test case?

My JTAGICE clone (square unit) seems to be dead but luckily I just got another cheaper clone (small size like the ATmega88 based USBISP clone). I did a simple test with an JTAG enabled ATmega16A and EEPROM writing seems to be okay now, verfied using usbasp as well.

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude.exe -c jtag1 -P COM8 -p m16a 
-U eeprom:w:.\hex\entest.eep:i -v

avrdude.exe: Version 7.0-20220802 (e91f733)
             Copyright (c) Brian Dean, http://www.bdmicro.com/
             Copyright (c) Joerg Wunsch

             System wide configuration file is "C:/work/avr/avrdude_test/avrdude_bin/avrdude.conf"

             Using Port                    : COM8
             Using Programmer              : jtag1
             AVR Part                      : ATmega16A
             Chip Erase delay              : 9000 us
             PAGEL                         : PD7
             BS2                           : PA0
             RESET disposition             : dedicated
             RETRY pulse                   : SCK
             Serial program mode           : yes
             Parallel program mode         : yes
             Timeout                       : 200
             StabDelay                     : 100
             CmdexeDelay                   : 25
             SyncLoops                     : 32
             PollIndex                     : 3
             PollValue                     : 0x53
             Memory Detail                 :

                                               Block Poll               Page                       Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               eeprom                  4    10   128    0 no        512    4      0  9000  9000 0xff 0xff
               flash                  33     6   128    0 yes     16384  128    128  4500  4500 0xff 0xff
               lock                    0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               lfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               hfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
               calibration             0     0     0    0 no          4    1      0     0     0 0x00 0x00

             Programmer Type : JTAGMKI
             Description     : Atmel JTAG ICE (mkI)
             ICE HW version: 0xc3
             ICE FW version: 0x80
             Vtarget       : 6.2 V
             JTAG clock    : 1 MHz (1.0 us)

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.02s

avrdude.exe: Device signature = 0x1e9403 (probably m16a)
avrdude.exe: reading input file ".\hex\entest.eep"
avrdude.exe: writing eeprom (512 bytes):

Writing | ################################################## | 100% 1.84s

avrdude.exe: 512 bytes of eeprom written
avrdude.exe: verifying eeprom memory against .\hex\entest.eep:

Reading | ################################################## | 100% 32.54s

avrdude.exe: 512 bytes of eeprom verified

avrdude.exe done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude.exe -c usbasp -qqp m16a 
-U eeprom:v:.\hex\entest.eep:i && echo OK
OK

```

MCUdude, 2022-08-05:
I happened to have a few spare ATmega16's laying around, so I might as well build a DIY JTAG ICE mkI programmer with it to help out with further testing.

@mcuee you don't happen to have the latest ICE mkI firmware, do you?

mcuee, 2022-08-05:
> @mcuee you don't happen to have the latest ICE mkI firmware, do you?

I do not have myself. But maybe @dl8dtl will have.

mcuee, 2022-08-05:
@MCUdude 
I think the following website should get you there.
https://scienceprog.com/build-your-own-avr-jtagice-clone/

dl8dtl, 2022-08-05:
The usual way to get the firmware is that you flash the ATmega16 with an AVR109 bootloader, and then let AVR Studio 4.x updgrade the firmware. ;-) (That's the reasons why these clones used to pop up everywhere by their time. Atmel charged USD 300 for a complete device.)

@MCUdude If you're interested, I could send you a bare PCB for a clone. I had to order three of them but only need one. It includes a footprint for an FT232RL so you can directly attach it to USB. (Alas, obtaining FTDIs might be an issue these days.)

MCUdude, 2022-08-05:
@dl8dtl Thanks for the offer! At the moment I only have ATmega16's in DIP packages. However, I might know a guy that has a spare, ATmega16 in a TQFP package. Would you like to share a picture of your board?

dl8dtl, 2022-08-05:
> Would you like to share a picture of your board?

Sure.

One is the Kicad rendering, the other one my real device.

[image]
[image]

dl8dtl, 2022-08-05:
> Can you share your failed test case?

IIRC I tried to write some bytes into EEPROM in terminal mode.

mcuee, 2022-08-06:
> > Can you share your failed test case?
> 
> IIRC I tried to write some bytes into EEPROM in terminal mode.

Indeed this does not work.

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c jtag1 -P COM8 -qqp m16a -t
avrdude> write eeprom 0x0 "hello world!"
>>> write eeprom 0x0 "hello world!"
avrdude.exe (write): error writing 0x68 at 0x00000 cell=0x54
avrdude.exe (write): error writing 0x65 at 0x00001 cell=0x68
avrdude.exe (write): error writing 0x6c at 0x00002 cell=0x65
avrdude.exe (write): error writing 0x6c at 0x00003 cell=0x20
avrdude.exe (write): error writing 0x6f at 0x00004 cell=0x71
avrdude.exe (write): error writing 0x20 at 0x00005 cell=0x75
avrdude.exe (write): error writing 0x77 at 0x00006 cell=0x69
avrdude.exe (write): error writing 0x6f at 0x00007 cell=0x63
avrdude.exe (write): error writing 0x72 at 0x00008 cell=0x6b
avrdude.exe (write): error writing 0x6c at 0x00009 cell=0x20
avrdude.exe (write): error writing 0x64 at 0x0000a cell=0x62
avrdude.exe (write): error writing 0x21 at 0x0000b cell=0x72
avrdude.exe (write): error writing 0x00 at 0x0000c cell=0x6f
avrdude> d ee
>>> d ee
0000  54 68 65 20 71 75 69 63  6b 20 62 72 6f 77 6e 20  |The quick brown |
0010  66 6f 78 20 6a 75 6d 70  73 20 6f 76 65 72 20 74  |fox jumps over t|
0020  68 65 20 6c 61 7a 79 20  64 6f 67 0a 54 68 65 20  |he lazy dog The |
0030  71 75 69 63 6b 20 62 72  6f 77 6e 20 66 6f 78 20  |quick brown fox |
0040  6a 75 6d 70 73 20 6f 76  65 72 20 74 68 65 20 6c  |jumps over the l|
0050  61 7a 79 20 64 6f 67 0a  54 68 65 20 71 75 69 63  |azy dog The quic|
0060  6b 20 62 72 6f 77 6e 20  66 6f 78 20 6a 75 6d 70  |k brown fox jump|
0070  73 20 6f 76 65 72 20 74  68 65 20 6c 61 7a 79 20  |s over the lazy |
0080  64 6f 67 0a 54 68 65 20  71 75 69 63 6b 20 62 72  |dog The quick br|
0090  6f 77 6e 20 66 6f 78 20  6a 75 6d 70 73 20 6f 76  |own fox jumps ov|
00a0  65 72 20 74 68 65 20 6c  61 7a 79 20 64 6f 67 0a  |er the lazy dog |
00b0  54 68 65 20 71 75 69 63  6b 20 62 72 6f 77 6e 20  |The quick brown |
00c0  66 6f 78 20 6a 75 6d 70  73 20 6f 76 65 72 20 74  |fox jumps over t|
00d0  68 65 20 6c 61 7a 79 20  64 6f 67 0a 54 68 65 20  |he lazy dog The |
00e0  71 75 69 63 6b 20 62 72  6f 77 6e 20 66 6f 78 20  |quick brown fox |
00f0  6a 75 6d 70 73 20 6f 76  65 72 20 74 68 65 20 6c  |jumps over the l|

```

mcuee, 2022-08-09:
@dl8dtl 
Your design is quite nice. I actually like your design better than my clones as it makes FW upgrade much easier with the ISP connector for the ATmega16 (I believe J4 is for ATmega16).

I am trying to revive my dead clone using `AVRprog.exe` from AVR Studion 3.x/4.x, but somehow it does not work (verficiation error).

My dead JTAG ICE clone (square box type) is using ATmega16A, CH340T and 74HC244. I have another working one (small USBAS{/USBISP shapre with metal housing) is using ATmega16, CH340T and 74HC245.
[image]

dl8dtl, 2022-08-09:
@mcuee Can you send me your email address?

mcuee, 2022-08-09:
@dl8dtl 
It is listed here.
https://github.com/libusb/libusb (Xiaofan Chen)

MCUdude, 2022-08-09:
Sorry for being so late to reply. If you still have a PCB left, I'll happily take one! I got my hands on an ATmega16 in a TQFP package, so I'll be able to get it running after a bit of soldering.

mcuee, 2022-08-10:
I guess my bad unit has the corrupted bootloader. AVRProg can program the flash (using the official ebn file) but verification failed. avrdude does not seem to work with it.

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_git.exe -c avr109 -P COM3 -p m16a -v

avrdude_git.exe: Version 7.0-20220804 (5f5002e)
                 Copyright (c) Brian Dean, http://www.bdmicro.com/
                 Copyright (c) Joerg Wunsch

                 System wide configuration file is "C:/work/avr/avrdude_test/avrdude_bin/avrdude.conf"

                 Using Port                    : COM3
                 Using Programmer              : avr109
                 AVR Part                      : ATmega16A
                 Chip Erase delay              : 9000 us
                 PAGEL                         : PD7
                 BS2                           : PA0
                 RESET disposition             : dedicated
                 RETRY pulse                   : SCK
                 Serial program mode           : yes
                 Parallel program mode         : yes
                 Timeout                       : 200
                 StabDelay                     : 100
                 CmdexeDelay                   : 25
                 SyncLoops                     : 32
                 PollIndex                     : 3
                 PollValue                     : 0x53
                 Memory Detail                 :

                                                   Block Poll               Page                       Polled
                   Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
                   ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
                   eeprom                  4    10   128    0 no        512    4      0  9000  9000 0xff 0xff
                   flash                  33     6   128    0 yes     16384  128    128  4500  4500 0xff 0xff
                   lock                    0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
                   lfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
                   hfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
                   signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
                   calibration             0     0     0    0 no          4    1      0     0     0 0x00 0x00

                 Programmer Type : butterfly
                 Description     : Atmel AppNote AVR109 Boot Loader

Connecting to programmer: .
Found programmer: Id = "AVRBOOT"; type = S
    Software Version = 1.0; No Hardware Version given.
Programmer supports auto addr increment.
avrdude_git.exe: error: buffered memory access not supported. Maybe it isn't
a butterfly/AVR109 but a AVR910 device?
avrdude_git.exe: initialization failed, rc=-1
                 Double check connections and try again, or use -F to override
                 this check.

avrdude_git.exe done.  Thank you.
```

mcuee, 2022-08-10:
Interestinly it seems to work better with `-c avr910`.

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_git.exe -c avr910 -P COM3 -p m16 -v

avrdude_git.exe: Version 7.0-20220804 (5f5002e)
                 Copyright (c) Brian Dean, http://www.bdmicro.com/
                 Copyright (c) Joerg Wunsch

                 System wide configuration file is "C:/work/avr/avrdude_test/avrdude_bin/avrdude.conf"

                 Using Port                    : COM3
                 Using Programmer              : avr910
                 avr910_devcode (avrdude.conf) : 0x74
                 AVR Part                      : ATmega16
                 Chip Erase delay              : 9000 us
                 PAGEL                         : PD7
                 BS2                           : PA0
                 RESET disposition             : dedicated
                 RETRY pulse                   : SCK
                 Serial program mode           : yes
                 Parallel program mode         : yes
                 Timeout                       : 200
                 StabDelay                     : 100
                 CmdexeDelay                   : 25
                 SyncLoops                     : 32
                 PollIndex                     : 3
                 PollValue                     : 0x53
                 Memory Detail                 :

                                                   Block Poll               Page                       Polled
                   Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
                   ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
                   eeprom                  4    10   128    0 no        512    4      0  9000  9000 0xff 0xff
                   flash                  33     6   128    0 yes     16384  128    128  4500  4500 0xff 0xff
                   lock                    0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
                   lfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
                   hfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
                   signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
                   calibration             0     0     0    0 no          4    1      0     0     0 0x00 0x00

                 Programmer Type : avr910
                 Description     : Atmel Low Cost Serial Programmer

Found programmer: Id = "AVRBOOT"; type = S
    Software Version = 1.0; Hardware Version = ?.
Programmer supports auto addr increment.

Programmer supports the following devices:
    Device code: 0x74 = ATmega16

avrdude_git.exe: avr910_devcode selected: 0x74
avrdude_git.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude_git.exe: Device signature = 0x1e9402 (probably m163)
avrdude_git.exe: Expected signature for ATmega16 is 1E 94 03
                 Double check chip, or use -F to override this check.

avrdude_git.exe done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_git.exe -c avr910 -P COM3 -p m16 -F -D -U .\jtagice-7f.hex

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_git.exe -c avr910 -P COM3 -p m16 -F -D -U .\jtagice-7f.hex -v

avrdude_git.exe: Version 7.0-20220804 (5f5002e)
                 Copyright (c) Brian Dean, http://www.bdmicro.com/
                 Copyright (c) Joerg Wunsch

[... 44 lines cut ...]

Reading | ################################################## | 100% 0.01s

avrdude_git.exe: Device signature = 0x1e9402 (probably m163)
avrdude_git.exe: Expected signature for ATmega16 is 1E 94 03
avrdude_git.exe: input file .\jtagice-7f.hex auto detected as Intel Hex
avrdude_git.exe: reading input file .\jtagice-7f.hex for flash
                 with 14814 bytes in 1 section within [0, 0x39dd]
                 using 116 pages and 34 pad bytes
avrdude_git.exe: writing 14814 bytes flash ...

Writing | ################################################## | 100% 56.70s

avrdude_git.exe: 14814 bytes of flash written
avrdude_git.exe: verifying flash memory against .\jtagice-7f.hex

Reading | ################################################## | 100% 27.53s

avrdude_git.exe: verification error, first mismatch at byte 0x1800
                 0x00 != 0xc4
avrdude_git.exe: verification error; content mismatch

avrdude_git.exe done.  Thank you.

```

mcuee, 2022-08-22:
Actually the current documentation already mentioned the above limitation. So I change the label to `enhancement`.

https://avrdudes.github.io/avrdude/7.0/avrdude_39.html#Troubleshooting
> Problem: Updating the flash ROM from terminal mode does not work with the JTAG ICEs.
> Solution: None at this time. Currently, the JTAG ICE code cannot write to the flash ROM one byte at a time.

mcuee, 2022-10-04:
I believe this is fixed by #1106.
* https://github.com/avrdudes/avrdude/pull/1106#issuecomment-1266858952

stefanrueger, 2022-10-05:
Fixed by PR #1106

closed 2022-10-05 by stefanrueger