avrdudes/avrdude #1062: Buffer overrun error in lexer.l

stefanrueger, 2022-08-10:
While reading `lexer.l` it occurred to me that the parsing of strings takes place in a 1024 byte buffer without checking  boundaries. A long string in `avrdude.conf` can crash avrdude:

```
$ cat test.conf
#------------------------------------------------------------
# wiring
#------------------------------------------------------------

programmer
    id                  = "mywiring";
    desc                = "Wiringhhhhhhhhhhhhhhhhhhhh\
012345678901234567890123456789012345678901234567890123456789abc\
012345678901234567890123456789012345678901234567890123456789abc\
012345678901234567890123456789012345678901234567890123456789abc\
\
... and many more lines so string is much larger than 1024 chars \
\
012345678901234567890123456789012345678901234567890123456789abc";

    type                = "wiring";
    connection_type     = serial;
;

$ avrdude -C test.conf
Segmentation fault (core dumped)
```

I'll correct that as part of the ongoing PR #1059.

mcuee, 2022-08-12:
Indeed this is an issue and PR #1059 fixed it.

```
MINGW64 /c/work/avr/avrdude_test/avrdude_bin
$ ./avrdude -C test.conf
Segmentation fault

$ ./avrdude_pr1059v4 -C test.conf

avrdude_pr1059v4.exe: no programmer has been specified on the command line or the config file
                      Specify a programmer using the -c option and try again

```

stefanrueger, 2022-08-15:
Fixed with PR #1059

closed 2022-08-15 by stefanrueger