avrdudes/avrdude #1064: Cmake installs avrdude.1 in the wrong place on Linux

stefanrueger, 2022-08-12:
When I use cmake to install `avrdude.1` gets copies into the wrong place
``` sh
$ ls -l /usr/local/share/man/
total 36K
drwxr-xr-x 2 root 4.0K Aug 12 16:03 man1
drwxrwxr-x 2 root  28K May 10 19:40 man3
drwxr-xr-x 2 root 4.0K Mar 17 22:37 man5
$ sudo cmake --build build_linux --target install
$ ls -l /usr/local/share/man/
total 84K
-rw-r--r-- 1 root  46K Aug  7 08:35 avrdude.1
drwxr-xr-x 2 root 4.0K Aug 12 16:03 man1
drwxrwxr-x 2 root  28K May 10 19:40 man3
drwxr-xr-x 2 root 4.0K Mar 17 22:37 man5
```

Using `sudo make install` in `src` copies `avrdude.1` into the right place `/usr/local/share/man/man1/avrdude.1` (on this system).

sbngross, 2022-08-12:
Might look into it if you like, since I brought this on to you and I'm quite familiar with `cmake`.

mcuee, 2022-08-12:
> Might look into it if you like, since I brought this on to you and I'm quite familiar with `cmake`.

That will be great. PR welcome. Thanks.

There are a few CMake related issues as well.
* https://github.com/avrdudes/avrdude/issues/1035
* https://github.com/avrdudes/avrdude/issues/955
* https://github.com/avrdudes/avrdude/issues/952
* https://github.com/avrdudes/avrdude/issues/947

sbngross, 2022-08-12:
Well this one is easy enough. @stefanrueger should I put the fix on top of #1063 ?

@mcuee I can certainly try. But I'm note sure if I can provide for other OS's than linux.

stefanrueger, 2022-08-12:
> put the fix on top of PR #1063

Either way is fine. Thanks for looking into this!

sbngross, 2022-08-12:
Run a short test with the `cmake --build` and the bare `make` in order to see if this breaks anything. As you noticed different behaviour of those two. Looks fine to me.

```
brutzzl3r@diamond-tiara ~/pkg % cmake --build tmp --target install          
[ 91%] Built target libavrdude
[100%] Built target avrdude
Install the project...
-- Install configuration: ""
...
-- Installing: /home/brutzzl3r/sandbox/opt/share/man/man1/avrdude.1

brutzzl3r@diamond-tiara ~/pkg % rm -rf ~/sandbox/opt              

brutzzl3r@diamond-tiara ~/pkg % make --no-print-directory -C tmp/src install
[ 91%] Built target libavrdude  
[100%] Built target avrdude
Install the project...
-- Install configuration: ""
...
-- Installing: /home/brutzzl3r/sandbox/opt/share/man/man1/avrdude.1

brutzzl3r@diamond-tiara ~/pkg % man -M /home/brutzzl3r/sandbox/opt/share/man -w 1 avrdude
/home/brutzzl3r/sandbox/opt/share/man/man1/avrdude.1
```

mcuee, 2022-08-13:
> @mcuee I can certainly try. But I'm note sure if I can provide for other OS's than linux.

I see that you are looking at #947 which is probably the most important of the issues related to CMake. Thanks.
* https://github.com/avrdudes/avrdude/issues/947

#955 is half done if you can fix for Linux (macOS and FreeBSD will be similar).
* https://github.com/avrdudes/avrdude/issues/955

#1035 seems to have a simple solution (option 1) even though Option 2 may be better.
* https://github.com/avrdudes/avrdude/issues/1035

#952 is diffcult and not only a CMake issue, rather it depends on other changes.
* https://github.com/avrdudes/avrdude/issues/952

stefanrueger, 2022-08-14:
Fixed by PR #1063

closed 2022-08-14 by stefanrueger