avrdudes/avrdude #1065: Warning message for teensy bootloader

mcuee, 2022-08-13:
There is a warning message for teensy bootlaoder.
`avrdude.exe: WARNING: Failed to write page: hid_write/GetOverlappedResult: (0x0000001F) A device attached to the system is not functioning.`

Run log with Teensy 2.0.
```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c teensy -p m32u4 -x wait -V -U flash:w:.\Blink.ino.TEENSY2.hex:i

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude.exe: Device signature = 0x1e9587 (probably m32u4)
avrdude.exe: NOTE: "flash" memory has been specified, an erase cycle will be performed
             To disable this feature, specify the -D option.
avrdude.exe: erasing chip
avrdude.exe: reading input file .\Blink.ino.TEENSY2.hex for flash
avrdude.exe: writing 2742 bytes flash ...

Writing | ################################################## | 100% 0.26s

avrdude.exe: 2742 bytes of flash written
avrdude.exe: WARNING: Failed to write page: hid_write/GetOverlappedResult: (0x0000001F) 
A device attached to the system is not functioning.

avrdude.exe done.  Thank you.

```

mcuee, 2022-08-13:
`-vvvv` run log shows more details.
```
avrdude.exe: teensy_powerdown()
avrdude.exe: teensy_reboot()
avrdude.exe: teensy_write_page(address=0xFFFFFFFF, size=0)
avrdude.exe: WARNING: Failed to write page: hid_write/GetOverlappedResult: (0x0000001F) A device attached to the system is not functioning.
```

<details>
<summary> full avrdude run log with -vvvv </summary>

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c teensy -p m32u4 -x wait -V -U flash:w:.\Blink.ino.TEENSY2.hex:i -vvvv
executable_abspath = C:/work/avr/avrdude_test/avrdude_bin/avrdude.exe
executable_abspath_len = 48
executable_dirpath = C:/work/avr/avrdude_test/avrdude_bin
executable_dirpath_len = 36
sys_config = C:/work/avr/avrdude_test/avrdude_bin/avrdude.conf
sys_config_found = true

avrdude.exe: Version 7.0-20220804 (5f5002e)
             Copyright (c) Brian Dean, http://www.bdmicro.com/
             Copyright (c) Joerg Wunsch

             System wide configuration file is "C:/work/avr/avrdude_test/avrdude_bin/avrdude.conf"

avrdude.exe: teensy_setup()
avrdude.exe: teensy_parseextparams()
             Using Port                    : usb
             Using Programmer              : teensy
avrdude.exe: teensy_open("usb")
             AVR Part                      : ATmega32U4
             Chip Erase delay              : 9000 us
             PAGEL                         : PD7
             BS2                           : PA0
             RESET disposition             : dedicated
             RETRY pulse                   : SCK
             Serial program mode           : yes
             Parallel program mode         : yes
             Timeout                       : 200
             StabDelay                     : 100
             CmdexeDelay                   : 25
             SyncLoops                     : 32
             PollIndex                     : 3
             PollValue                     : 0x53
             Memory Detail                 :

                                               Block Poll               Page                       Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               eeprom                 65    20     4    0 no       1024    4      0  9000  9000 0x00 0x00
                                               Block Poll               Page                       Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               flash                  65     6   128    0 yes     32768  128    256  4500  4500 0x00 0x00
                                               Block Poll               Page                       Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               lfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
                                               Block Poll               Page                       Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               hfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
                                               Block Poll               Page                       Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               efuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
                                               Block Poll               Page                       Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
[... 72 lines cut ...]
Writing | #########################################          | 81% 0.27savrdude.exe: teensy_paged_write(page_size=0x80, addr=0x900, n_bytes=0x80)
avrdude.exe: teensy_write_page(address=0x000900, size=128)
Writing | ###########################################        | 86% 0.29savrdude.exe: teensy_paged_write(page_size=0x80, addr=0x980, n_bytes=0x80)
avrdude.exe: teensy_write_page(address=0x000980, size=128)
Writing | #############################################      | 90% 0.30savrdude.exe: teensy_paged_write(page_size=0x80, addr=0xA00, n_bytes=0x80)
avrdude.exe: teensy_write_page(address=0x000A00, size=128)
Writing | ################################################   | 95% 0.32savrdude.exe: teensy_paged_write(page_size=0x80, addr=0xA80, n_bytes=0x80)
avrdude.exe: teensy_write_page(address=0x000A80, size=128)
Writing | ################################################## | 100% 0.33s

avrdude.exe: 2742 bytes of flash written
avrdude.exe: teensy_powerdown()
avrdude.exe: teensy_reboot()
avrdude.exe: teensy_write_page(address=0xFFFFFFFF, size=0)
avrdude.exe: WARNING: Failed to write page: hid_write/GetOverlappedResult: (0x0000001F) A device attached to the system is not functioning.
avrdude.exe: teensy_disable()
avrdude.exe: teensy_close()

avrdude.exe done.  Thank you.

avrdude.exe: teensy_teardown()

```

</details>

mcuee, 2022-08-13:
The same happens to Teensy++ 2.0 as well.
```

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c teensy -p usb1286 -x wait -V 
-U flash:w:.\Blink.ino.TEENSY2PP.hex:i -vvvv
...
avrdude.exe: 2950 bytes of flash written
avrdude.exe: teensy_powerdown()
avrdude.exe: teensy_reboot()
avrdude.exe: teensy_write_page(address=0xFFFFFFFF, size=0)
avrdude.exe: WARNING: Failed to write page: hid_write/GetOverlappedResult: (0x0000001F) A device attached to the system is not functioning.
avrdude.exe: teensy_disable()
avrdude.exe: teensy_close()

avrdude.exe done.  Thank you.

avrdude.exe: teensy_teardown()
```

MCUdude, 2022-08-14:
@mcuee can you check if you get the same warning when using @mariusgreuel's [Avrdude fork](https://github.com/mariusgreuel/avrdude)?

mcuee, 2022-08-14:
> @mcuee can you check if you get the same warning when using @mariusgreuel's [Avrdude fork](https://github.com/mariusgreuel/avrdude)?

Same issue.

```
PS C:\work\avr\avrdude_test\avrdude_win\out\build\x64-Debug\src\Debug> .\avrdude -c teensy -p m32u4 -x wait -V -U flash:w:.\Blink.ino.TEENSY2.hex:i
avrdude.exe: No device found, waiting for device to be plugged in...
avrdude.exe: Press CTRL-C to terminate.

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.08s

avrdude.exe: Device signature = 0x1e9587 (probably m32u4)
avrdude.exe: NOTE: "flash" memory has been specified, an erase cycle will be performed
             To disable this feature, specify the -D option.
avrdude.exe: erasing chip
avrdude.exe: reading input file ".\Blink.ino.TEENSY2.hex"
avrdude.exe: writing flash (2742 bytes):

Writing | ################################################## | 100% 0.44s

avrdude.exe: 2742 bytes of flash written
avrdude.exe: WARNING: Failed to write page: A device attached to the system is not functioning.

avrdude.exe done.  Thank you.
```

Detail log shows the same as well.
```
avrdude.exe: 2742 bytes of flash written
avrdude.exe: teensy_powerdown()
avrdude.exe: teensy_reboot()
avrdude.exe: teensy_write_page(address=0xFFFFFFFF, size=0)
avrdude.exe: WARNING: Failed to write page: A device attached to the system is not functioning.
avrdude.exe: teensy_disable()
avrdude.exe: teensy_close()

avrdude.exe done.  Thank you.

avrdude.exe: teensy_teardown()
```

I also tried the previous release of the fork and the results are the same.
```
PS C:\work\avr\avrdude_test\avrdude_binary_archive\avrdude-v6.3.1.1-windows> .\avrdude -c teensy -p m32u4 -x wait -V -U flash:w:.\Blink.ino.TEENSY2.hex:i
avrdude.exe: No device found, waiting for device...
avrdude.exe: Press CTRL-C to terminate.

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude.exe: Device signature = 0x1e9587 (probably m32u4)
avrdude.exe: NOTE: "flash" memory has been specified, an erase cycle will be performed
             To disable this feature, specify the -D option.
avrdude.exe: erasing chip
avrdude.exe: reading input file ".\Blink.ino.TEENSY2.hex"
avrdude.exe: writing flash (2742 bytes):

Writing | ################################################## | 100% 0.30s

avrdude.exe: 2742 bytes of flash written

avrdude.exe: safemode: Fuses OK (E:FF, H:FF, L:FF)
avrdude.exe: WARNING: Failed to write page: A device attached to the system is not functioning.

avrdude.exe done.  Thank you.
```

Detail log shows the same issue.
```
avrdude.exe: 2742 bytes of flash written

avrdude.exe: teensy_read_byte(desc=lfuse, addr=0x0)
avrdude.exe: safemode read 1, lfuse value: ff
avrdude.exe: teensy_read_byte(desc=lfuse, addr=0x0)
avrdude.exe: safemode read 2, lfuse value: ff
avrdude.exe: teensy_read_byte(desc=lfuse, addr=0x0)
avrdude.exe: safemode read 3, lfuse value: ff
avrdude.exe: safemode: lfuse reads as FF
avrdude.exe: teensy_read_byte(desc=hfuse, addr=0x0)
avrdude.exe: safemode read 1, hfuse value: ff
avrdude.exe: teensy_read_byte(desc=hfuse, addr=0x0)
avrdude.exe: safemode read 2, hfuse value: ff
avrdude.exe: teensy_read_byte(desc=hfuse, addr=0x0)
avrdude.exe: safemode read 3, hfuse value: ff
avrdude.exe: safemode: hfuse reads as FF
avrdude.exe: teensy_read_byte(desc=efuse, addr=0x0)
avrdude.exe: safemode read 1, efuse value: ff
avrdude.exe: teensy_read_byte(desc=efuse, addr=0x0)
avrdude.exe: safemode read 2, efuse value: ff
avrdude.exe: teensy_read_byte(desc=efuse, addr=0x0)
avrdude.exe: safemode read 3, efuse value: ff
avrdude.exe: safemode: efuse reads as FF
avrdude.exe: safemode: Fuses OK (E:FF, H:FF, L:FF)
avrdude.exe: teensy_powerdown()
avrdude.exe: teensy_reboot()
avrdude.exe: teensy_write_page(address=0xFFFFFFFF, size=0)
avrdude.exe: WARNING: Failed to write page: A device attached to the system is not functioning.
avrdude.exe: teensy_disable()
avrdude.exe: teensy_close()

avrdude.exe done.  Thank you.

avrdude.exe: teensy_teardown()
```

mcuee, 2022-08-14:
Somehow it does not work under Linux.
```
mcuee@UbuntuSwift3:~/build/avr/avrdude_bin$ ./avrdude -C ./avrdude.conf -c teensy -p m32u4 -x wait -vvvv
sys_config = ./avrdude.conf
sys_config_found = false

avrdude: Version 7.0-20220804 (5f5002e)
         Copyright (c) Brian Dean, http://www.bdmicro.com/
         Copyright (c) Joerg Wunsch

         System wide configuration file is "./avrdude.conf"
         User configuration file is "/home/mcuee/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

avrdude: teensy_setup()
avrdude: teensy_parseextparams()
         Using Port                    : usb
         Using Programmer              : teensy
avrdude: teensy_open("usb")
avrdude: No device found, waiting for device to be plugged in...
avrdude: Press CTRL-C to terminate.
         AVR Part                      : ATmega32U4
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom                 65    20     4    0 no       1024    4      0  9000  9000 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           flash                  65     6   128    0 yes     32768  128    256  4500  4500 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           lfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           hfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           efuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           lock                    0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00

         Programmer Type : teensy
         Description     : Teensy Bootloader
avrdude: teensy_display()

avrdude: teensy_enable()
avrdude: teensy_initialize()
avrdude: WARNING: Cannot detect board type (HID usage is 0)
avrdude: HID usage: 0x00
avrdude: Board: Unknown Board
avrdude: Available flash size: 32256
avrdude: Page size: 128
avrdude: Signature: 0x1E0000
avrdude: AVR device initialized and ready to accept instructions

Reading |                                                    | 0% 0.00savrdude: teensy_read_sig_bytes()
Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e0000
avrdude: Expected signature for ATmega32U4 is 1E 95 87
         Double check chip, or use -F to override this check.
avrdude: teensy_powerdown()
avrdude: teensy_disable()
avrdude: teensy_close()

avrdude done.  Thank you.

avrdude: teensy_teardown()

```

mcuee, 2022-08-14:
@mariusgreuel 
hidapi issue reported to hidapi project.
* https://github.com/libusb/hidapi/issues/443

mcuee, 2022-08-14:
@mariusgreuel 
A simple fix is to prefer hidapi-hidraw to hidapi-libusb.

```
mcuee@UbuntuSwift3:~/build/avr/avrdude$ git diff
diff --git a/CMakeLists.txt b/CMakeLists.txt
index 5efc8d4..2250065 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -38,8 +38,8 @@ option(BUILD_SHARED_LIBS "Build shared libraries" OFF)
 if(WIN32)
     # Prefer static libraries over DLLs on Windows
     option(USE_STATIC_LIBS "Use static libraries" ON)
-else()
-    option(USE_STATIC_LIBS "Use static libraries" OFF)
+#else()
+#    option(USE_STATIC_LIBS "Use static libraries" OFF)
 endif()
 
 include(CheckIncludeFile)
```

In that case, we do not need to use `-F` anymore. I have not set up the proper udev rule so that I need to use sudo for the hidapi hidraw backend.
```
mcuee@UbuntuSwift3:~/build/avr/avrdude$ sudo ./build_linux/src/avrdude -c teensy 
-p m32u4 -x wait -V -U ../avrdude_bin/Blink.ino.TEENSY2.hex 
avrdude: No device found, waiting for device to be plugged in...
avrdude: Press CTRL-C to terminate.

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9587 (probably m32u4)
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: input file ../avrdude_bin/Blink.ino.TEENSY2.hex auto detected as Intel Hex
avrdude: reading input file ../avrdude_bin/Blink.ino.TEENSY2.hex for flash
avrdude: writing 2742 bytes flash ...

Writing | ################################################## | 100% 0.20s

avrdude: 2742 bytes of flash written
avrdude: WARNING: Failed to write page: Protocol error

avrdude done.  Thank you.
```

mcuee, 2022-08-15:
Answer from hidapi developer.
https://github.com/libusb/hidapi/issues/443#issuecomment-1214433164
> This is a known and documented issue:
> 
> https://github.com/libusb/hidapi/blob/c3ae3d0b582ba5e6b26af603aea37d988e7eda7b/libusb/hid.c#L729
> 
> As I mentioned [here](https://github.com/libusb/hidapi/pull/432#issuecomment-1214167047) - now there is a workaround (ask for USAGE after the device is open).

mariusgreuel, 2022-09-24:
> Answer from hidapi developer

Thanks for contacting them. I will give their suggestions a shot.

mariusgreuel, 2022-09-24:
> There is a warning message for teensy bootlaoder.
> A device attached to the system is not functioning.

It seems that your PC is a bit more picky then mine about USB communication. I suspect when the AVR reboots it does not properly reply to the control packet.

It does not happen on my PC. Could you share some details about the make of your PC? Are you using a VM? Just wondering what CPU/USB chipset it has.

Also, does the Teensy successfully reboot? A simply workaround would be to just suppress the warning for the reboot.

mcuee, 2022-09-25:
Interesting that you do not have issues with your PC.

My Acer laptop: dual boot Windows 11 or Ubuntu 20.04, not using VM.
Processor	11th Gen Intel(R) Core(TM) i5-1135G7 @ 2.40GHz   2.42 GHz
Installed RAM	16.0 GB

Standard Windows xHCI driver for Intel chipset.

mcuee, 2022-10-10:
> A simply workaround would be to just suppress the warning for the reboot.

@mariusgreuel That is a good workaround. I agree with this.

MCUdude, 2022-10-10:
> A simply workaround would be to just suppress the warning for the reboot.

A PR would be nice! 😀

mariusgreuel, 2022-10-11:
@mcuee I was unable to reproduce the problem, however, this PR should fix the warning you see. Could you please give this a try?

mcuee, 2022-10-12:
@mariusgreuel 
Yes, PR #1122 is good. Thanks.

PR #1122 (2022-10-17): Suppress Teensy USB communication error message on reboot
Fixes #1065
closed 2022-10-17 by stefanrueger