avrdudes/avrdude #1067: cmake: Used syntax mismatches minimum_required version

sbngross, 2022-08-14:
`install` commands in `src/CMakeLists.txt` have the `TYPE` attribute. This syntax was introduced in `cmake` `3.14`

minimum_required_version is `3.12`.

Came across it when I tested my PR with a stone-age debian 10. There is a cmake `3.13`.
It just runs in qemu so I can't provide logs, but this error is obvious.

The question remains how to deal with it. Raise min_req_ver or get rid of new syntax?

PS: We had to adapt the `man` install anyway to the `DESTINATION` syntax for #1063

mcuee, 2022-08-15:
I tend to think it is easier to raise the minimum_required_version to 3.14. But let's hear from @mariusgreuel or @dl8dtl.

mcuee, 2022-09-05:
@mariusgreuel 
Please help to check this one. Thanks.

PR #1243 (2022-12-29): CMake updates
Some minor updates:

- Fixes #1067
- Rename option from USE_EXTERNAL to USE_EXTERNAL_LIBS to make it more descriptive
- Remove some superfluous MSVC compiler options
- Add CMake options file for Visual Studio
closed 2022-12-29 by stefanrueger