avrdudes/avrdude #1069: AVRDUDE 7 no longer works with RS232 ports

avrfreak, 2022-08-15:
I use the same  (REAL) serial port programmer (not usbtoTTL, or USB2RS232) on com1.
After avrdude6.3, none of the avrdude versions work with any serial port programmer
USB only is working.
Has the serial port support been disabled in newer versions of avrdude?

MCUdude, 2022-08-15:
> Has the serial port support been disabled in newer versions of avrdude?

Not that I'm aware of. If we're going to help you, we need some more information.

* What operating system are you using?
* Which RS232 based programmer are you using?
* Can you post the Avrdude output from a failed session?

dl8dtl, 2022-08-15:
I'm pretty sure I've been testing my STK500 on a real serial port before the 7.0 release.

avrfreak, 2022-08-16:
> > Has the serial port support been disabled in newer versions of avrdude?
> 
> Not that I'm aware of. If we're going to help you, we need some more information.
> 
> * What operating system are you using?
> * Which RS232 based programmer are you using?
> * Can you post the Avrdude output from a failed session?

jtag2updi with a MAX232, using an arduino micro board (M328P)

OR
an arduino UNO with a CH340G

w7 x64

avrfreak, 2022-08-16:
Here is a log from the newest version avrdude V7 downloaded from here.
`
avrdude -Cavrdude.conf -c jtag2updi -p avr128db28 -P COM1 -e -b 115200 -U flash:w:test.hex:a -U eeprom:w:test.eep:a

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.07s

avrdude: Device signature = 0x1e970e (probably avr128db28)
avrdude: erasing chip
avrdude: reading input file "test.hex"
avrdude: input file test.hex auto detected as Intel Hex
avrdude: writing flash (11976 bytes):

Writing | ################################################## | 100% 2.40s

avrdude: 11976 bytes of flash written
avrdude: verifying flash memory against test.hex:
avrdude: input file test.hex auto detected as Intel Hex

Reading | ################################################## | 100% 2.27s

avrdude: 11976 bytes of flash verified
avrdude: reading input file "test.eep"
avrdude: input file test.eep auto detected as Intel Hex
avrdude: writing eeprom (16 bytes):

Writing |                                                    | 0% 0.00savrdude:
jtagmkII_write_byte(): bad response to write memory command: RSP_ILLEGAL_MCU_STATE
 ***failed;
Writing | ###                                                | 6% 0.05savrdude:
jtagmkII_write_byte(): bad response to write memory command: RSP_ILLEGAL_MCU_STATE
 ***failed;
Writing | ######                                             | 12% 0.09savrdude:
 jtagmkII_write_byte(): bad response to write memory command: RSP_ILLEGAL_MCU_STATE
 ***failed;
Writing | #########                                          | 18% 0.13savrdude:
 jtagmkII_write_byte(): bad response to write memory command: RSP_ILLEGAL_MCU_STATE
 ***failed;
Writing | #############                                      | 25% 0.17savrdude:
 jtagmkII_write_byte(): bad response to write memory command: RSP_ILLEGAL_MCU_STATE
 ***failed;
Writing | ################                                   | 31% 0.21savrdude:
 jtagmkII_write_byte(): bad response to write memory command: RSP_ILLEGAL_MCU_STATE
 ***failed;
Writing | ###################                                | 37% 0.25savrdude:
 jtagmkII_write_byte(): bad response to write memory command: RSP_ILLEGAL_MCU_STATE
 ***failed;
Writing | ######################                             | 43% 0.29savrdude:
 jtagmkII_write_byte(): bad response to write memory command: RSP_ILLEGAL_MCU_STATE
 ***failed;
Writing | #########################                          | 50% 0.33savrdude:
 jtagmkII_write_byte(): bad response to write memory command: RSP_ILLEGAL_MCU_STATE
 ***failed;
Writing | ############################                       | 56% 0.37savrdude:
 jtagmkII_write_byte(): bad response to write memory command: RSP_ILLEGAL_MCU_STATE
 ***failed;
Writing | ###############################                    | 62% 0.41savrdude:
 jtagmkII_write_byte(): bad response to write memory command: RSP_ILLEGAL_MCU_STATE
 ***failed;
Writing | ##################################                 | 68% 0.45savrdude:
 jtagmkII_write_byte(): bad response to write memory command: RSP_ILLEGAL_MCU_STATE
 ***failed;
Writing | ######################################             | 75% 0.49savrdude:
 jtagmkII_write_byte(): bad response to write memory command: RSP_ILLEGAL_MCU_STATE
 ***failed;
Writing | #########################################          | 81% 0.53savrdude:
 jtagmkII_write_byte(): bad response to write memory command: RSP_ILLEGAL_MCU_STATE
 ***failed;
Writing | ############################################       | 87% 0.57savrdude:
 jtagmkII_write_byte(): bad response to write memory command: RSP_ILLEGAL_MCU_STATE
 ***failed;
Writing | ###############################################    | 93% 0.62savrdude:
 jtagmkII_write_byte(): bad response to write memory command: RSP_ILLEGAL_MCU_STATE
 ***failed;
Writing | ################################################## | 100% 0.66s

avrdude: 16 bytes of eeprom written
avrdude: verifying eeprom memory against test.eep:
avrdude: input file test.eep auto detected as Intel Hex

Reading | ################################################## | 100% 0.78s

avrdude: verification error, first mismatch at byte 0x0000
         0xff != 0x04
avrdude: verification error; content mismatch
avrdude: jtagmkII_recv(): checksum error
avrdude: jtagmkII_program_disable(): timeout/error communicating with programmer
 (status -4)
avrdude: jtagmkII_recv(): checksum error
avrdude: jtagmkII_close(): timeout/error communicating with programmer (status -4)

avrdude done.  Thank you.
`

mcuee, 2022-08-16:
@avrfreak 
No need to create a new issue, just change the title to reflect that this is related to EEPROM using jtag2updi. For example, you can change the title to "avrdude 7.0 jtag2updi not working with EEPROM". 

For Windows, the only official release is in this github repo. 
https://github.com/avrdudes/avrdude/releases/tag/v7.0

When you say version 6.3 works, you mean the version from Arduino IDE and not the official version, right?

mcuee, 2022-08-16:
For my Arduini Nano Every, the error message is different but indeed it does not work with EEPROM writing, using latets git.

```
PS C:\work\avr\avrdude_test\avrdude_bin> mode COM7 baud=12 dtr=on > nul
PS C:\work\avr\avrdude_test\avrdude_bin> mode COM7 baud=12 dtr=off > nul

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude.exe -C .\avrdude.conf -patmega4809 -cjtag2updi
 -PCOM7 -b115200 -Ueeprom:w:entest_256B.eep:i

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.29s

avrdude.exe: Device signature = 0x1e9651 (probably m4809)
avrdude.exe: reading input file entest_256B.eep for eeprom
avrdude.exe: writing 256 bytes eeprom ...

Writing |                                                    | 0% 0.00savrdude.exe: jtagmkII_paged_write(): bad response to write memory command: RSP_ILLEGAL_MEMORY_TYPE
Writing | #############                                      | 25% 0.01savrdude.exe: jtagmkII_write_byte(): fatal timeout/error communicating with programmer (status -1)
 ***failed;
avrdude.exe: jtagmkII_write_byte(): fatal timeout/error communicating with programmer (status -1)
 ***failed;
avrdude.exe: jtagmkII_write_byte(): fatal timeout/error communicating with programmer (status -1)
 ***failed;

```

avrfreak, 2022-08-16:
> Where do you download the non-working version?
I saw a link somewhere on AVRFreaks (IIRC one of the jtag2updi threads) but i can't remember where.
There are other versions, 6.99, all do the same thing. No serial port access and IIRC they did work with the jtag2updi code in my arduino UNO USB (with ch340).

> When you say version 6.3 works, you mean the version from Arduino IDE and not the official version, right? Does that version work with EEPROM?

I am assuming I have the official version. I remember getting it from savannah a long time ago.
No I am not using the one from the Arduino IDE. 

How do you know if you have the official version or not?

avrfreak, 2022-08-16:
this appears to be a jtag2updi issue only. I tested avrdude V7 with an atmega8515 writing flash and eeprom with an old avr910 serial port programmer and it works.

The eeprom programming does work with avrdude 6.3 using the same jtag2updi programmer.

mcuee, 2022-08-16:
Then your 6.3 version can not be from the official channel as the avrdude  6.3 version does not support jtag2updi at all, not even 6.4 version.

You may want to post the -v verbose run log of your good 6.3 version.

avrfreak, 2022-08-16:
if I use -vvvv then avrdude6.3 does not work for the jtag2updi with the avr128db28 I am attempting to program.
Any more than two "-vv" and I receive a timeout message from serial port and checksum invalid message.
here is the working log with  "-vv" Programs and verifies successfully.
`
C:\test>avrdude -Cavrdude.conf -c jtag2updi -p avr128db28 -P COM1 -v -e -b 11520
0 -U flash:w:test.hex:a -U eeprom:w:test.eep:a

avrdude: Version 6.3, compiled on Feb 17 2016 at 09:25:53
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "avrdude.conf"

         Using Port                    : COM1
         Using Programmer              : jtag2updi
         Overriding Baud Rate          : 115200
JTAG ICE mkII sign-on message:
Communications protocol version: 1
M_MCU:
  boot-loader FW version:        1
  firmware version:              6.00
  hardware version:              1
S_MCU:
  boot-loader FW version:        1
  firmware version:              6.00
  hardware version:              1
Serial number:                   00:00:00:00:00:00
Device ID:                       JTAGICE mkII
         AVR Part                      : AVR128DB28
         Chip Erase delay              : 0 us
         PAGEL                         : P00
         BS2                           : P00
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 0
         StabDelay                     : 0
         CmdexeDelay                   : 0
         SyncLoops                     : 0
         ByteDelay                     : 0
         PollIndex                     : 0
         PollValue                     : 0x00
         Memory Detail                 :

                                  Block Poll               Page
      Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  Max
W   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ---
-- ---------
           signature      0     0     0    0 no          3    0      0     0 0 0x00 0x00
           prodsig        0     0     0    0 no        128  128      0     0 0 0x00 0x00
           fuses          0     0     0    0 no          9    0      0     0 0 0x00 0x00
           fuse0          0     0     0    0 no          1    0      0     0 0 0x00 0x00
           fuse1          0     0     0    0 no          1    0      0     0 0 0x00 0x00
           fuse2          0     0     0    0 no          1    0      0     0 0 0x00 0x00
           fuse5          0     0     0    0 no          1    0      0     0 0 0x00 0x00
           fuse6          0     0     0    0 no          1    0      0     0 0 0x00 0x00
           fuse7          0     0     0    0 no          1    0      0     0 0 0x00 0x00
           fuse8          0     0     0    0 no          1    0      0     0 0 0x00 0x00
           fuserow        0     0     0    0 no         16    0      0     0 0 0x00 0x00
           lock           0     0     0    0 no          4    0      0     0 0 0x00 0x00
           data           0     0     0    0 no          0    0      0     0 0 0x00 0x00
           usersig        0     0     0    0 no         32   32      0     0 0 0x00 0x00
           mmflash        0     0     0    0 no      32768  512      0     0 0 0x00 0x00
           flash          0     0     0    0 no     131072  512      0     0 0 0x00 0x00
           eeprom         0     0     0    0 no        512    1      0     0 0 0x00 0x00

         Programmer Type : JTAGMKII_PDI
         Description     : JTAGv2 to UPDI bridge
         M_MCU hardware version: 1
         M_MCU firmware version: 6.00
         S_MCU hardware version: 1
         S_MCU firmware version: 6.00
         Serial number:          00:00:00:00:00:00
         Vtarget         : 5.0 V

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.07s

avrdude: Device signature = 0x1e970e (probably avr128db28)
avrdude: erasing chip
avrdude: reading input file "test.hex"
avrdude: input file test.hex auto detected as Intel Hex
avrdude: writing flash (1024 bytes):

Writing | ################################################## | 100% 0.19s

avrdude: 1024 bytes of flash written
avrdude: verifying flash memory against test.hex:
avrdude: load data flash data from input file test.hex:
avrdude: input file test.hex auto detected as Intel Hex
avrdude: input file test.hex contains 1024 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 0.18s

avrdude: verifying ...
avrdude: 1024 bytes of flash verified
avrdude: reading input file "test.eep"
avrdude: input file test.eep auto detected as Intel Hex
avrdude: writing eeprom (16 bytes):

Writing | ################################################## | 100% 0.19s

avrdude: 16 bytes of eeprom written
avrdude: verifying eeprom memory against test.eep:
avrdude: load data eeprom data from input file test.eep:
avrdude: input file test.eep auto detected as Intel Hex
avrdude: input file test.eep contains 16 bytes
avrdude: reading on-chip eeprom data:

Reading | ################################################## | 100% 0.14s

avrdude: verifying ...
avrdude: 16 bytes of eeprom verified

avrdude done.  Thank you.`

avrfreak, 2022-08-16:
i think the version of avrdude 6.3 I am using is official but the avrdude.conf has been modified to work with jtag2updi and it works always, 100%.

MCUdude, 2022-08-16:
@avrfreak can you find the jtag2updi programmer in your avrdude.conf file and post it here?

MCUdude, 2022-08-16:
I've tested with the Avrdude 6.3 version that Arduino IDE ships with, and the latest Avrdude 7.0 release, and both can read and write to EEPROM using a jtag2updi programmer with fresh firmware. I can't read or write to EEPROM in terminal mode, but that's probably a jtag2updi related issue.

For reference, this was tested on macOS using a CH340N USB to serial adapter.

avrfreak, 2022-08-16:
> @avrfreak can you find the jtag2updi programmer in your avrdude.conf file and post it here?

`#------------------------------------------------------------
# jtag2updi programmer definitions
#------------------------------------------------------------

programmer
  id    = "jtag2updi";
  desc  = "JTAGv2 to UPDI bridge";
  type  = "jtagmkii_pdi";
  connection_type = serial;
  baudrate = 115200;
;`

MCUdude, 2022-08-16:
@avrfreak it looks like you're using a "vanilla" Avrdude 6.3 build with @eltangas' patched avrdude.conf to make it work with the jtag2updi programmer.

I don't have a machine with a dedicated RS232 port, so I can't help you there. But my jtag2updi programmer with when connected to a USB to serial adapter at least.

Here's a test that might tell us something. Can you try to build the latest Avrdude from source? If you don't know how to or don't want to, you can download builds here: https://github.com/avrdudes/avrdude/actions/runs/2857628485#artifacts

Then add the following to your avrdude.conf file:

```
#------------------------------------------------------------
# jtag2updi programmer definitions
#------------------------------------------------------------

programmer
  id    = "jtag2updi_test";
  desc  = "JTAGv2 to UPDI bridge";
  type  = "jtagmkii_pdi";
  connection_type = serial;
  baudrate = 115200;
;

#------------------------------------------------------------
# AVR-Dx class common values
#------------------------------------------------------------

part
    id		= ".avrdx_test";
    desc	= "AVR-Dx class common values";
#    has_updi	= yes;
    has_pdi	= yes;
    nvm_base	= 0x1000;
#    ocd_base	= 0x0F80;

    memory "signature"	# these are the first 3 bytes of the SIGROW memory area
        size		= 3;
        offset		= 0x1100;
    ;

    memory "prodsig"	# this is the name avrdude recognises for the SIGROW memory area
        size		= 0x300;
        offset		= 0x1100;
        page_size	= 0x100;
        readsize	= 0x100;
    ;

    memory "fuses"	# first 9 bytes of fuse memory area, writing to undocumented fuses may be dangerous!
        size		= 9;
        offset		= 0x1050;
    ;

    memory "fuse0"
        size		= 1;
        offset		= 0x1050;
    ;

    memory "fuse1"
        size		= 1;
        offset		= 0x1051;
    ;

    memory "fuse2"
        size		= 1;
        offset		= 0x1052;
    ;

    memory "fuse5"
        size		= 1;
        offset		= 0x1055;
    ;

    memory "fuse6"
        size		= 1;
        offset		= 0x1056;
    ;

    memory "fuse7"
        size		= 1;
        offset		= 0x1057;
    ;

    memory "fuse8"
        size		= 1;
        offset		= 0x1058;
    ;

    memory "fuserow"	# this is the entire fuse memory area, writing to undocumented fuses may be dangerous!
        size		= 0x10;
        offset		= 0x1050;
    ;

    memory "lock"
        size		= 4;
        offset		= 0x1040;
    ;

    memory "data"
        # SRAM, only used to supply the offset
        offset		= 0x1000000;
    ;

    memory "usersig"
        size		= 0x20;
        offset		= 0x1080;
        page_size	= 0x20;
        readsize	= 0x20;
    ;

    memory "mmflash"	# memory mapped flash window (aka MAPPED_PROGMEM).
        size      = 0x8000;
        offset    = 0x8000;
        page_size = 0x200;
        readsize  = 0x100;
    ;
;

#------------------------------------------------------------
# AVR128DA sub-family common values
#------------------------------------------------------------

part parent    ".avrdx_test"
    id			= ".avr128da_test";
    desc		= "AVR-DA with 128KB flash common values";
 
    memory "flash"
        size      = 0x20000;
        offset    = 0x800000;
        page_size = 0x200;
        readsize  = 0x100;
    ;

    memory "eeprom"
        size      = 0x200;
        offset    = 0x1400;
        page_size = 0x1;
        readsize  = 0x100;
    ;
;

#------------------------------------------------------------
# AVR128DB sub-family common values
#------------------------------------------------------------

part parent    ".avr128da_test"	# The AVR-DB have identical memory layout to AVR-DA
    id			= ".avr128db_test";
    desc		= "AVR-DB with 128KB flash common values";
;

#------------------------------------------------------------
# AVR128DB28
#------------------------------------------------------------

part parent    ".avr128db_test"
    id        = "avr128db28_test";
    desc      = "AVR128DB28";
    signature = 0x1E 0x97 0x0E;
;
```

Now here's what you do:

* Execute the downloaded Avrdude binary, and use programmer `jtag2updi_test` and part `avr128db28_test`. Please paste the output here.
* Run the downloaded Avrdude binary again, but this time use programmer `jtag2updi` and part `avr128db28` instead. Please paste the output for this one as well.

The reason why I'm asking you to do this is because by using `*_test`, we're testing the "pre-7.0" jtag2updi patch. If `*_test` works, it means that there's something wrong with the "proper" jtag2updi implementation that was introduced in 7.0

avrfreak, 2022-08-16:
Here is the result from this command:
avrdude -Cavrdude.conf -c jtag2updi_test -p avr128db28_test -P COM1 -vvv -e -b 115200 -U flash:w:test.hex:a -U eeprom:w:test.eep:a
`avrdude: Version 7.0-20220815 (a4bfa82)
         Copyright (c) Brian Dean, http://www.bdmicro.com/
         Copyright (c) Joerg Wunsch

         System wide configuration file is "avrdude.conf"

avrdude: input file test.hex auto detected as Intel Hex
avrdude: input file test.eep auto detected as Intel Hex
         Using Port                    : COM1
         Using Programmer              : jtag2updi_test
         Overriding Baud Rate          : 115200
avrdude: jtagmkII_open_pdi()
avrdude: jtagmkII_getsync()
avrdude: jtagmkII_getsync() attempt 1 of 10: Sending sign-on command:
avrdude: jtagmkII_send(): sending 1 bytes

avrdude: jtagmkII_recv(): Got message seqno 0 (command_sequence == 0)

Sign-on succeeded

JTAG ICE mkII sign-on message:
Communications protocol version: 1
M_MCU:
  boot-loader FW version:        1
  firmware version:              6.00
  hardware version:              1
S_MCU:
  boot-loader FW version:        1
  firmware version:              6.00
  hardware version:              1
Serial number:                   00:00:00:00:00:00
Device ID:                       JTAGICE mkII
avrdude: jtagmkII_getsync(): Using a 298-byte device descriptor
avrdude: jtagmkII_setparm()
avrdude: jtagmkII_setparm(): Sending set parameter command (parm 0x03, 1 bytes):

avrdude: jtagmkII_send(): sending 3 bytes

avrdude: jtagmkII_recv(): Got message seqno 1 (command_sequence == 1)

OK

avrdude: jtagmkII_getsync(): Sending get sync command:
avrdude: jtagmkII_send(): sending 1 bytes

avrdude: jtagmkII_recv(): Got message seqno 2 (command_sequence == 2)

OK

         AVR Part                      : AVR128DB28_test
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Memory Detail                 :

                                           Block Poll               Page
               Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages
MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------
----- ----- ---------
           signature               0     0     0    0 no          3    1      0
    0     0 0x00 0x00
                                           Block Poll               Page
               Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages
MinW  MaxW   ReadBack
[... 199 lines cut ...]
avrdude: error reading signature data, rc=-2
avrdude: jtagmkII_program_disable(): Sending leave progmode command:
avrdude: jtagmkII_send(): sending 1 bytes

avrdude: jtagmkII_recv(): Got message seqno 8 (command_sequence == 8)

No target power

avrdude: jtagmkII_program_disable(): bad response to leave progmode command: RSP
_NO_TARGET_POWER
avrdude: jtagmkII_close()
avrdude: jtagmkII_close(): Sending GO command:
avrdude: jtagmkII_send(): sending 1 bytes

avrdude: jtagmkII_recv(): Got message seqno 9 (command_sequence == 9)

OK

avrdude: jtagmkII_close(): Sending sign-off command:
avrdude: jtagmkII_send(): sending 1 bytes

avrdude: jtagmkII_recv(): Got message seqno 10 (command_sequence == 10)

OK
avrdude done.  Thank you.`

avrfreak, 2022-08-16:
Here is the result from this command:
avrdude -Cavrdude.conf -c jtag2updi -p avr128db28 -P COM1 -vvv -e -b 115200 -U flash:w:test.hex:a -U eeprom:w:test.eep:a

`avrdude: input file test.hex auto detected as Intel Hex
avrdude: input file test.eep auto detected as Intel Hex
         Using Port                    : COM1
         Using Programmer              : jtag2updi
         Overriding Baud Rate          : 115200
avrdude: jtagmkII_open_pdi()
avrdude: jtagmkII_getsync()
avrdude: jtagmkII_getsync() attempt 1 of 10: Sending sign-on command:
avrdude: jtagmkII_send(): sending 1 bytes

avrdude: jtagmkII_recv(): Got message seqno 0 (command_sequence == 0)

Sign-on succeeded

JTAG ICE mkII sign-on message:
Communications protocol version: 1
M_MCU:
  boot-loader FW version:        1
  firmware version:              6.00
  hardware version:              1
S_MCU:
  boot-loader FW version:        1
  firmware version:              6.00
  hardware version:              1
Serial number:                   00:00:00:00:00:00
Device ID:                       JTAGICE mkII
avrdude: jtagmkII_getsync(): Using a 298-byte device descriptor
avrdude: jtagmkII_setparm()
avrdude: jtagmkII_setparm(): Sending set parameter command (parm 0x03, 1 bytes):

avrdude: jtagmkII_send(): sending 3 bytes

avrdude: jtagmkII_recv(): Got message seqno 1 (command_sequence == 1)

OK

avrdude: jtagmkII_getsync(): Sending get sync command:
avrdude: jtagmkII_send(): sending 1 bytes

avrdude: jtagmkII_recv(): Got message seqno 2 (command_sequence == 2)

OK

         AVR Part                      : AVR128DB28
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Memory Detail                 :

                                           Block Poll               Page
               Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages
MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------
----- ----- ---------
           signature               0     0     0    0 no          3    1      0
    0     0 0x00 0x00
                                           Block Poll               Page
               Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages
MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------
----- ----- ---------
           prodsig                 0     0     0    0 no        125  125      0
    0     0 0x00 0x00
                                           Block Poll               Page
[... 203 lines cut ...]
avrdude: jtagmkII_program_disable(): Sending leave progmode command:
avrdude: jtagmkII_send(): sending 1 bytes

avrdude: jtagmkII_recv(): Got message seqno 8 (command_sequence == 8)

No target power

avrdude: jtagmkII_program_disable(): bad response to leave progmode command: RSP
_NO_TARGET_POWER
avrdude: jtagmkII_close()
avrdude: jtagmkII_close(): Sending GO command:
avrdude: jtagmkII_send(): sending 1 bytes

avrdude: jtagmkII_recv(): Got message seqno 9 (command_sequence == 9)

OK

avrdude: jtagmkII_close(): Sending sign-off command:
avrdude: jtagmkII_send(): sending 1 bytes

avrdude: jtagmkII_recv(): Got message seqno 10 (command_sequence == 10)

OK

avrdude done.  Thank you.`

MCUdude, 2022-08-16:
Thanks, @avrfreak! It didn't change anything though, the issue is present anyways. And this _only_ happens when using the physical RS232 port on your PC, not with a USB to serial adapter?

avrfreak, 2022-08-16:
The non-official versions of avrdude I have don't work at all with the real serial port but avrdude6.3 does.
The newest version you asked me to download does not program the eeprom (flash only) in both the RS232 port and the USB port.
I am sure I have the newest firmware of jtag2updi loaded.

MCUdude, 2022-08-16:
> The non-official versions of avrdude I have don't work at all with the real serial port but avrdude6.3 does.

I'll assume it's the Avrdude version bundled with Arduino IDE? Its 6.3 version doesn't have any RS232 related patches IIRC, so I have no idea why it doesn't work for you.

> The non-official versions of avrdude I have don't work at all with the real serial port but avrdude6.3 does.

This is really weird. I don't have an AVR128DB28 chip, but I can read and write to EEPROM just fine using a jtag2updi programmer and an ATmega4808 as the target. @eltangas might have to check in on this one...

avrfreak, 2022-08-16:
> This is really weird. I don't have an AVR128DB28 chip, but I can read and write to EEPROM just fine using a jtag2updi programmer and an ATmega4808 as the target. @ElTangas might have to check in on [...]

I just tried an Atmega1608 and avrdude 7 works using the RS232 jtag2updi programmer.

mcuee, 2022-08-16:
So this seems to be a timing issue with the AVR128DB28 for the jtag2updi, and it may be worse if using real serial port under Windows.

BTW, my issue is probably related to the SAM D11 UPDI FW, it is supposed to support EEPROM but seems there is a FW issue here.
https://github.com/arduino/ArduinoCore-megaavr/tree/master/firmwares/MuxTO

I have another Nano 4808 which still shows the warning message but at least EEPROM writing is okay in the end.
```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c jtag2updi -P COM13 
-b 115200 -p m4808 -U eeprom:w:entest_256B.eep:i

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.29s

avrdude.exe: Device signature = 0x1e9650 (probably m4808)
avrdude.exe: reading input file entest_256B.eep for eeprom
avrdude.exe: writing 256 bytes eeprom ...

Writing |                                                    
| 0% 0.00savrdude.exe: jtagmkII_paged_write(): 
bad response to write memory command: RSP_ILLEGAL_MEMORY_TYPE
Writing | ################################################## | 100% 2.04s

avrdude.exe: 256 bytes of eeprom written
avrdude.exe: verifying eeprom memory against entest_256B.eep

Reading | ################################################## | 100% 0.33s

avrdude.exe: 256 bytes of eeprom verified

avrdude.exe done.  Thank you.
```

avrfreak, 2022-08-17:
> So this seems to be a timing issue with the AVR128DB28 for the jtag2updi, and it may be worse if using real serial port under Windows.
A timing issue for the jtag2updi programmer or avrdude?
I do know that the AVR128DB28  eeprom is programmed byte by byte, (no page programming). I briefly glanced at the avrdude.conf it shows page size 1.  This as I said before works well with avrdude6.3. It is very slow but it does work.

mcuee, 2022-08-17:
@MCUdude 
> I've tested with the Avrdude 6.3 version that Arduino IDE ships with, and the latest Avrdude 7.0 release, and both can read and write to EEPROM using a jtag2updi programmer with fresh firmware. I [...]

It seems to me that there are different implementations of the jtag2updi and they may behave a bit differently. Both will probably not behave the same as the Microchip/Atmel official JTAGICE mkii (JTAGMKII_UPDI)
1: https://github.com/ElTangas/jtag2updi
2: https://github.com/arduino/ArduinoCore-megaavr/tree/master/firmwares/MuxTO (the Nano 4808 clones may use the same)

Interestingly you say you can not read/write EEPROM in terminal mode, but my Nano 4808 clone seems to be okay.

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude.exe -C .\avrdude.conf -patmega4808 -cjtag2updi -PCOM13 -b115200 -t

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.29s

avrdude.exe: Device signature = 0x1e9650 (probably m4808)
avrdude> d ee
>>> d ee

Reading | ################################################## | 100% 1.29s

0000  54 68 65 20 71 75 69 63  6b 20 62 72 6f 77 6e 20  |The quick brown |
0010  66 6f 78 20 6a 75 6d 70  73 20 6f 76 65 72 20 74  |fox jumps over t|
0020  68 65 20 6c 61 7a 79 20  64 6f 67 0a 54 68 65 20  |he lazy dog The |
0030  71 75 69 63 6b 20 62 72  6f 77 6e 20 66 6f 78 20  |quick brown fox |
0040  6a 75 6d 70 73 20 6f 76  65 72 20 74 68 65 20 6c  |jumps over the l|
0050  61 7a 79 20 64 6f 67 0a  54 68 65 20 71 75 69 63  |azy dog The quic|
0060  6b 20 62 72 6f 77 6e 20  66 6f 78 20 6a 75 6d 70  |k brown fox jump|
0070  73 20 6f 76 65 72 20 74  68 65 20 6c 61 7a 79 20  |s over the lazy |
0080  64 6f 67 0a 54 68 65 20  71 75 69 63 6b 20 62 72  |dog The quick br|
0090  6f 77 6e 20 66 6f 78 20  6a 75 6d 70 73 20 6f 76  |own fox jumps ov|
00a0  65 72 20 74 68 65 20 6c  61 7a 79 20 64 6f 67 0a  |er the lazy dog |
00b0  54 68 65 20 71 75 69 63  6b 20 62 72 6f 77 6e 20  |The quick brown |
00c0  66 6f 78 20 6a 75 6d 70  73 20 6f 76 65 72 20 74  |fox jumps over t|
00d0  68 65 20 6c 61 7a 79 20  64 6f 67 0a 54 68 65 20  |he lazy dog The |
00e0  71 75 69 63 6b 20 62 72  6f 77 6e 20 66 6f 78 20  |quick brown fox |
00f0  6a 75 6d 70 73 20 6f 76  65 72 20 74 68 65 20 6c  |jumps over the l|

avrdude> w ee 0 0xff
>>> w ee 0 0xff

Writing | ################################################## | 100% 0.47s

avrdude> d ee
>>> d ee

Reading | ################################################## | 100% 1.28s

0000  ff 68 65 20 71 75 69 63  6b 20 62 72 6f 77 6e 20  |.he quick brown |
0010  66 6f 78 20 6a 75 6d 70  73 20 6f 76 65 72 20 74  |fox jumps over t|
0020  68 65 20 6c 61 7a 79 20  64 6f 67 0a 54 68 65 20  |he lazy dog The |
0030  71 75 69 63 6b 20 62 72  6f 77 6e 20 66 6f 78 20  |quick brown fox |
0040  6a 75 6d 70 73 20 6f 76  65 72 20 74 68 65 20 6c  |jumps over the l|
0050  61 7a 79 20 64 6f 67 0a  54 68 65 20 71 75 69 63  |azy dog The quic|
0060  6b 20 62 72 6f 77 6e 20  66 6f 78 20 6a 75 6d 70  |k brown fox jump|
0070  73 20 6f 76 65 72 20 74  68 65 20 6c 61 7a 79 20  |s over the lazy |
0080  64 6f 67 0a 54 68 65 20  71 75 69 63 6b 20 62 72  |dog The quick br|
0090  6f 77 6e 20 66 6f 78 20  6a 75 6d 70 73 20 6f 76  |own fox jumps ov|
00a0  65 72 20 74 68 65 20 6c  61 7a 79 20 64 6f 67 0a  |er the lazy dog |
00b0  54 68 65 20 71 75 69 63  6b 20 62 72 6f 77 6e 20  |The quick brown |
00c0  66 6f 78 20 6a 75 6d 70  73 20 6f 76 65 72 20 74  |fox jumps over t|
00d0  68 65 20 6c 61 7a 79 20  64 6f 67 0a 54 68 65 20  |he lazy dog The |
00e0  71 75 69 63 6b 20 62 72  6f 77 6e 20 66 6f 78 20  |quick brown fox |
00f0  6a 75 6d 70 73 20 6f 76  65 72 20 74 68 65 20 6c  |jumps over the l|

```

mcuee, 2022-08-17:
> So this seems to be a timing issue with the AVR128DB28 for the jtag2updi, and it may be worse if using real serial port under Windows.
>> A timing issue for the jtag2updi programmer or avrdude?

To me it will be jtag2updi FW issue. The avrdude timing for the chip should be okay -- please refer to my EEPROM test results using AVR128DB48 (similar to your chip) with the Curiosity Nano on-board pkobn_updi programmer (JTAGICE3_UPDI based).
* https://github.com/avrdudes/avrdude/issues/1009#issuecomment-1166276923

> > I do know that the AVR128DB28  eeprom is programmed byte by byte, (no page programming). I briefly glanced at the avrdude.conf it shows page size 1.  This as I said before works well with avrdude6.3. It is very slow but it does work.

Please refer to my test results above, it works for my Nano 4808 clone and indeed it supports byte programming as well as page programming. Again the issue for my Arduino Nano Every and the Nano 4808 Clone are more related to the FW.

mcuee, 2022-08-17:
> It seems to me that there are different implementations of the jtag2updi and they may behave a bit differently. Both will probably not behave the same as the Microchip/Atmel official JTAGICE mkii [...]

@MCUdude and @avrfreak 
Even I tend to think the issues with EEPROM are more related to the jtag2updi implementation, probably there are workarounds which can be done in avrdude, since we can differentiate the official AVR JTAGICE mkii with jtag2updi.

Maybe @ElTangas can help here, at least for his implementation (which is your use case). 

For Arduino megaavr SAMD11 FW implementation, I have created an issue with Arduino.
* https://github.com/arduino/ArduinoCore-megaavr/issues/120

avrfreak, 2022-08-17:
thank you for all your help

mcuee, 2022-08-17:
@MCUdude and @avrfreak 

Just for your info, my Nano 4808 clone is like the following. So the FW should be the same as the jtag2updi with ATmega328P (similar to yours). So based on my testing, it seems to work okay (CH340 + ATmega328P based UPDI programmer)
https://github.com/Thinary/ThinaryArduino/tree/master/Thinary-megaavr

mcuee, 2022-10-16:
Indeed there is an issue with AVR Dx parts, from @MCUdude's test results here.
* https://github.com/avrdudes/avrdude/pull/1121#issuecomment-1279799851

mcuee, 2022-10-16:
@avrfreak 
Since this is now the same issue as #1128, let's continue there (issue with jtag2updi for AVR Dx parts). I will close this one.

closed 2022-10-16 by mcuee