avrdudes/avrdude #1070: macOS CMake Doc build

mcuee, 2022-08-16:
Pull request #1066 will mostly address #947.
* #1066 
* #947 

macOS will need more work -- therefore this issue is raised to carry out work on macOS.

@sbngross wrote in the #1066 comments:
> macos now somehow builds. At least that awk related stuff. Errors from last build are related to some `makeinfo` and `avrdude.texi` issues AFAIK. From my part the job is done except for commit clean-up after review. 
> 
> can you put that macos stuff in a new ticket, so we are good here?

mcuee, 2022-08-16:
From https://github.com/avrdudes/avrdude/pull/1066#issuecomment-1216539274

I can build the docs under macOS with the following commands.

```
brew install texinfo
export PATH="/opt/homebrew/opt/texinfo/bin:$PATH"
brew install gnu-tar
PATH="/opt/homebrew/opt/gnu-tar/libexec/gnubin:$PATH"

brew install texlive
1038  git clone https://github.com/avrdudes/avrdude.git avrdude_pr1066
 1039  cd avrdude_pr1066
 1040  git fetch origin pull/1066/head && git checkout FETCH_HEAD
 1041  echo $PATH
 1042  cmake -B build_darwin -DCMAKE_INSTALL_PREFIX=/home/mcuee/bin -D BUILD_DOC=ON -D CMAKE_BUILD_TYPE=Debug -D CMAKE_C_FLAGS=-I/opt/homebrew/include -D CMAKE_EXE_LINKER_FLAGS=-L/opt/homebrew/lib
 1043  cmake --build build_darwin
```

sbngross, 2022-08-27:
Doc build for mac now also works in CI.

[Run 2938907870](avrdudes/avrdude/actions/runs/2938907870)

It is necessary to isntall dependencies and set either path or makeinfo binary correct.

This issue might be closed by #1066

mcuee, 2022-09-05:
Closing this issue as both #1066 and #1096 can fix this issue.

closed 2022-09-05 by mcuee