avrdudes/avrdude #1072: JTAG is not able to write 1s to EEPROM

mcuee, 2022-08-20:
https://github.com/avrdudes/avrdude/issues/1071#issuecomment-1220886097

From @dl8dtl 
> I think it was always the case that JTAG could not write 1s to EEPROM.

I think avrdude documentation needs to reflect this one.

MCUdude, 2022-08-21:
We should perhaps confirm this by testing the JTAG1's EEPROM write capabilities in AVR studio 4 before stating that writing 1's to the EEPROM isn't possible. If it is correct, it's a major bug!

mcuee, 2022-08-22:
@MCUdude 

@dl8dtl mentioned this as a generic AVR JTAG programmer limitation, not limited to JTAG ICE 1. My test results show that JTAG1 and AVR Dragon (JTAG2) have the issue under avrdude.

We should test JTAG ICE 1 under AVR Studio 4 (not supported by Microchip Studio), and other JTAG programmers under Microchip Studio (eg: JTAGICE mkii, AVR Dragon, JTAGICE 3, Atmel-ICE, Power Debugger, etc),

I see that you have access to AVR Dragon and Atmel-ICE, please help to test this out under Microchip Studio. It will be great that you can check out the two test cases mentioned here.
* https://github.com/avrdudes/avrdude/issues/1071#issuecomment-1219543781

1) Uses jtag3.c or jtagmkII.c with any of the nine parts that have `eecr = 0x3c` which also have a JTAG interface: 
ATmega128 ATmega128A ATmega16 ATmega162 ATmega16A ATmega32 ATmega32A ATmega64 ATmega64A

2) Uses jtagmkII.c with any "classic" part other than the five parts that avrdude -p*/st | grep eecr outputs?
```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -p*/st | grep eecr
.pt     AT90CAN128      eecr    0x3f
.pt     AT90CAN64       eecr    0x3f
.pt     AT90CAN32       eecr    0x3f
.pt     ATmega165       eecr    0x3f
.pt     ATmega406       eecr    0x3f
```

mcuee, 2022-08-22:
I just tried to use AVR JTAGICE 1 under Atmel Studio 4.18 and indeed it has some issues.

```
Setting device parameters for jtag programming ..OK
Entering programming mode.. OK
Programming EEPROM ..       OK
Reading EEPROM ..       OK
WARNING: EEPROM address 0x0000 is 0x50 (should be 0x54).. FAILED!
Leaving programming mode.. OK
``
![jtag1_atmel_studio4_eeprom_error](https://user-images.githubusercontent.com/1294740/185822814-8a9b9a2d-5721-4d40-84a5-a733687ab361.JPG)
`

mcuee, 2022-08-22:
BTW, JTAG1 has also a separated issue with the terminal mode as mentioned in #1054.
* #1054

```
C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c jtag1 -P COM4 -qqp m128 -t
avrdude> d ee 0x0 0x10
>>> d ee 0x0 0x10
0000  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|

avrdude>  write eeprom 0x0 "hello world!"
>>> write eeprom 0x0 "hello world!"
avrdude.exe (write): error writing 0x68 at 0x00000 cell=0xff
avrdude.exe (write): error writing 0x65 at 0x00001 cell=0xff
avrdude.exe (write): error writing 0x6c at 0x00002 cell=0xff
avrdude.exe (write): error writing 0x6c at 0x00003 cell=0xff
avrdude.exe (write): error writing 0x6f at 0x00004 cell=0xff
avrdude.exe (write): error writing 0x20 at 0x00005 cell=0xff
avrdude.exe (write): error writing 0x77 at 0x00006 cell=0xff
avrdude.exe (write): error writing 0x6f at 0x00007 cell=0xff
avrdude.exe (write): error writing 0x72 at 0x00008 cell=0xff
avrdude.exe (write): error writing 0x6c at 0x00009 cell=0xff
avrdude.exe (write): error writing 0x64 at 0x0000a cell=0xff
avrdude.exe (write): error writing 0x21 at 0x0000b cell=0xff
avrdude.exe (write): error writing 0x00 at 0x0000c cell=0xff
avrdude> quit
>>> quit
```

mcuee, 2022-08-22:
Hmm, I need to read the documentation in more detail. It has already mentioned the above limitation.

https://avrdudes.github.io/avrdude/7.0/avrdude_39.html#Troubleshooting
> Problem: Page-mode programming the EEPROM (using the -U option) does not erase EEPROM cells before writing, and thus cannot overwrite any previous value != 0xff.
> Solution: None. This is an inherent feature of the way JTAG EEPROM programming works, and is documented that way in the Atmel AVR datasheets. In order to successfully program the EEPROM that way, a prior chip erase (with the EESAVE fuse unprogrammed) is required. This also applies to the STK500 and STK600 in high-voltage programming mode.

mcuee, 2022-08-22:
@MCUdude and @dl8dtl 

Shall I close this issue?  Or shall we improve the documentation to clearly mention this limitation about JTAG programmers  and not just in the troubleshooting section?

mcuee, 2022-08-24:
@MCUdude 
I can see that Atmel-ICE can perform a bit differently compared to JTAG ICE 1 and JTAGICE mkii based on your test results below regarding Flash Write in terminal mode. Please help to test for EEPROM writing in terminal mode as well when you got time. Thanks.
* https://github.com/avrdudes/avrdude/issues/1020#issuecomment-1222498695

MCUdude, 2022-08-24:
I just borrowed the Atmel ICE, and I've just returned it. I may ask to do some more testing with it though.

I doubt it will be able to write 1s to EEPROM in terminal mode. It can't co it when writing a file without performing a chip and EEPROM erase first. I contacted a Microchip developer, and got the following answer:

**Q**: Is it correct that JTAG2-compatible programmers (i.e the AVR Dragon) and JTAG3/Atmel ICE only can write 0's to EEPROM and not 1's? It's like this when using Avrdude, but the EDBG documentation I've read doesn't explicitly specifies this behaviour.

**A**: Sounds like a bug to me – this rule applies of course to FLASH, not EEPROM.  But EDBG is not “JTAG2-compatible”… and feature-set is _usually_ a function of the silicon, not the debugger anyway…

**Q**: I know the JTAG3/Atmel ICE can write 1's to Xmega's and AVRs with UPDI, but only when utilizing an "Atomic write" (See issue #1009 and PR #1013). However, I can't find a similar feature on AVRs with JTAG.

**A**: Correct – atomic (erase-and-write) was added for XMEGA.  Not sure why this split was made…

**Q**: Does this means that it's impossible to write 1's to the EEPROM using a JTAG interface without running a chip erase first?

**A**: Not convinced, but not sure.  SomeoneTM would probably have to find the RTL to know for sure – there are many subtle differences between the different entry modes (ISP/HVSP/JTAG) on those old parts – only from XMEGA onwards was there proper separation between physical and logical interfaces…

mcuee, 2022-08-24:
@MCUdude 
Great info. Thanks. Hopefully you can ask the Microchip developer to carry out the simple test to confirm.

mcuee, 2022-08-24:
BTW, I was trying to find the ATmega128A datasheet on the JTAG side. I do not see limitation mentioned in the datasheet with regard to EEPROM.
https://ww1.microchip.com/downloads/en/DeviceDoc/Atmel-8151-8-bit-AVR-ATmega128A_Datasheet.pdf

For high voltage parallel programming (ST500/STK600), it is rather clear.
>Section 31.7.3. Chip Erase
> The Chip Erase will erase the Flash and EEPROM memories plus Lock bits. The Lock bits are not reset
until the program memory has been completely erased. The Fuse bits are not changed. A Chip Erase
must be performed before the Flash and/or EEPROM are reprogrammed

mcuee, 2022-08-24:
@MCUdude 
Maybe you can point out the above info from the datasheet to the Microchip developer and see if he or she has further comments. Thanks.

mcuee, 2022-08-24:
In the end, I think avrdude documentation already mentions the limitation (classic AVR chip limitation in JTAG mode). Still we may need to improve the documentation to mention the chips and JTAG programmers affected.

MCUdude, 2022-08-25:
@mcuee I don't want to bother the Microchip developers more than necessary. I think we have the answers we need, and we can test using AVR studio 4 just to be sure when it comes to JTAG + EEPROM.

mcuee, 2022-08-25:
> I don't want to bother the Microchip developers more than necessary. I think we have the answers we need, and we can test using AVR studio 4 just to be sure when it comes to JTAG + EEPROM.

I agree. In terms of JTAG and EEPROM, I think it is rather clear. And the current AVR documentation is not wrong. The only thing is that we may want to improvement the documentation a bit further to mention the JTAG programmers and Chips (Classic AVRs with JTAG).

mcuee, 2022-10-04:
This may get fixed by #1106.
* #1106

stefanrueger, 2022-10-05:
closed with PR #1106

closed 2022-10-05 by stefanrueger