avrdudes/avrdude #1075: avrdude vs Microchip Studio for an xmega hex file

mcuee, 2022-08-20:
This is probably not a bug but rather I feel strange that Microchip Studio seems to be able to program a hex file considered as `invalid`  by avrdude.

The hex file is provided by a vendor of ATxmega32U4 core board as the factory DFU bootloader (no source codes provided). It does not seem to work under my system but neverthess Microchip Studio is able to program the chip using an AVRISP mkii with PDI. avrdude pops out an error.

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c avrispmkii -p x32a4u -U .\USB_bootloader_xmega32a4u_mini_EO.hex

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude.exe: Device signature = 0x1e9541 (probably x32a4u)
avrdude.exe: NOTE: Programmer supports page erase for Xmega devices.
             Each page will be erased before programming it, but no chip erase is performed.
             To disable page erases, specify the -D option; for a chip-erase, use the -e option.
avrdude.exe: ERROR: address 0x8010 out of range at line 2 of .\USB_bootloader_xmega32a4u_mini_EO.hex
avrdude.exe: read from file .\USB_bootloader_xmega32a4u_mini_EO.hex failed

avrdude.exe done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude_bin> head .\USB_bootloader_xmega32a4u_mini_EO.hex
:020000020000FC
:1080000000C00091780005FD6AC0F092800608E18A
:10801000009390060FEF0A950023E9F70091880678
:1080200000FFECC0E0E0F0E0079116910F3F19F47B
:108030001F3F09F4E3C053C03091CF0137FDFCCF9F
:108040003BB72BBFF8012091CA0113E21093CA017C
:108050000A01E8952093CA013BBF08953BB72BBFA7
:10806000F8012091CA014093CA013DE930933400E0
:10807000E8952093CA013BBF089520E006D0E9F7B8
:108080000895259102D0E9F708952D9301501040ED

```
[USB_bootloader_xmega32a4u_mini_EO.hex.txt](https://github.com/avrdudes/avrdude/files/9387216/USB_bootloader_xmega32a4u_mini_EO.hex.txt)

mcuee, 2022-08-20:
The initial intention is to test out FLIP2 (xmega DFU bootloader) and see if the patch in #733 works or not.

* #733 

By right there is also a factory DFU bootloader but I have not figured out how to enter that one yet.

stefanrueger, 2022-08-20:
The hex file is OK. The trouble is that the default memory for ATxmega parts for `-U` is `application` not `flash` or `boot`. Try `-U flash:w:file:i`

mcuee, 2022-08-20:
> The hex file is OK. The trouble is that the default memory for ATxmega parts for `-U` is `application` not `flash` or `boot`. Try `-U flash:w:file:i`

Thanks. Now it works and the USB DFU device also works. 

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c avrispmkii -p x32a4u -U flash:w:USB_bootloader_xmega32a4u_mini_EO.hex:i

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude.exe: Device signature = 0x1e9541 (probably x32a4u)
avrdude.exe: NOTE: Programmer supports page erase for Xmega devices.
             Each page will be erased before programming it, but no chip erase is performed.
             To disable page erases, specify the -D option; for a chip-erase, use the -e option.
avrdude.exe: reading input file USB_bootloader_xmega32a4u_mini_EO.hex for flash
             with 3940 bytes in 2 sections within [0x8000, 0x8fb1]
             using 16 pages and 156 pad bytes
avrdude.exe: writing 3940 bytes flash ...

Writing | ################################################## | 100% 0.03s

avrdude.exe: 3940 bytes of flash written
avrdude.exe: verifying flash memory against USB_bootloader_xmega32a4u_mini_EO.hex

Reading | ################################################## | 100% 0.03s

avrdude.exe: 3940 bytes of flash verified

avrdude.exe done.  Thank you.

```

closed 2022-08-20 by mcuee

mcuee, 2022-08-20:
But avrdude `-c flip2` does not seem to work. Need to study futher.

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c flip2 -p x32a4u -D -U flash:w:flip2_demo_app.hex:i -v

avrdude.exe: Version 7.0-20220815 (a4bfa82)
             Copyright (c) Brian Dean, http://www.bdmicro.com/
             Copyright (c) Joerg Wunsch

             System wide configuration file is "C:/work/avr/avrdude_test/avrdude_bin/avrdude.conf"

             Using Port                    : usb
             Using Programmer              : flip2
             AVR Part                      : ATxmega32A4U
             RESET disposition             : dedicated
             RETRY pulse                   : SCK
             Serial program mode           : yes
             Parallel program mode         : yes
             Memory Detail                 :

                                               Block Poll               Page                       Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               fuse1                   0     0     0    0 no          1    1      0     0     0 0x00 0x00
               fuse2                   0     0     0    0 no          1    1      0     0     0 0x00 0x00
               fuse4                   0     0     0    0 no          1    1      0     0     0 0x00 0x00
               fuse5                   0     0     0    0 no          1    1      0     0     0 0x00 0x00
               lock                    0     0     0    0 no          1    1      0     0     0 0x00 0x00
               signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
               prodsig                 0     0     0    0 no         50   50      0     0     0 0x00 0x00
               data                    0     0     0    0 no          0    1      0     0     0 0x00 0x00
               eeprom                  0     0     0    0 no       1024   32      0     0     0 0x00 0x00
               flash                   0     0     0    0 no      36864  256      0     0     0 0x00 0x00
               application             0     0     0    0 no      32768  256      0     0     0 0x00 0x00
               apptable                0     0     0    0 no       4096  256      0     0     0 0x00 0x00
               boot                    0     0     0    0 no       4096  256      0     0     0 0x00 0x00
               usersig                 0     0     0    0 no        256  256      0     0     0 0x00 0x00

             Programmer Type : flip2
             Description     : FLIP USB DFU protocol version 2 (AVR4023)

avrdude.exe: Found VID=0x03eb PID=0x2fe4 at 001:044
avrdude.exe: Error: DFU_DNLOAD failed: Input/output error
avrdude.exe: Error: Failed to get DFU status: Input/output error
avrdude.exe: Error: Failed to set memory unit 0x05 (signature)
avrdude.exe: AVR device initialized and ready to accept instructions

Reading |                                                    
| 0% 0.00savrdude.exe: error reading signature data for part "ATxmega32A4U", rc=-1
avrdude.exe: error reading signature data, rc=-1

avrdude.exe done.  Thank you.

```

mcuee, 2022-08-21:
Looks like I need to use `application` for the upload. But it still failed.

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c flip2 -p x32a4u -D -U application:w:flip2_demo_app.hex:i -v

avrdude.exe: Version 7.0-20220819 (38aa131)
             Copyright (c) Brian Dean, http://www.bdmicro.com/
             Copyright (c) Joerg Wunsch

             System wide configuration file is "C:/work/avr/avrdude_test/avrdude_bin/avrdude.conf"

             Using Port                    : usb
             Using Programmer              : flip2
             AVR Part                      : ATxmega32A4U
             RESET disposition             : dedicated
             RETRY pulse                   : SCK
             Serial program mode           : yes
             Parallel program mode         : yes
             Memory Detail                 :

                                               Block Poll               Page                       Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               fuse1                   0     0     0    0 no          1    1      0     0     0 0x00 0x00
               fuse2                   0     0     0    0 no          1    1      0     0     0 0x00 0x00
               fuse4                   0     0     0    0 no          1    1      0     0     0 0x00 0x00
               fuse5                   0     0     0    0 no          1    1      0     0     0 0x00 0x00
               lock                    0     0     0    0 no          1    1      0     0     0 0x00 0x00
               signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
               prodsig                 0     0     0    0 no         50   50      0     0     0 0x00 0x00
               data                    0     0     0    0 no          0    1      0     0     0 0x00 0x00
               eeprom                  0     0     0    0 no       1024   32      0     0     0 0x00 0x00
               flash                   0     0     0    0 no      36864  256      0     0     0 0x00 0x00
               application             0     0     0    0 no      32768  256      0     0     0 0x00 0x00
               apptable                0     0     0    0 no       4096  256      0     0     0 0x00 0x00
               boot                    0     0     0    0 no       4096  256      0     0     0 0x00 0x00
               usersig                 0     0     0    0 no        256  256      0     0     0 0x00 0x00

             Programmer Type : flip2
             Description     : FLIP USB DFU protocol version 2 (AVR4023)

avrdude.exe: Found VID=0x03eb PID=0x2fe4 at 001:036
    USB Vendor          : 0x03EB
    USB Product         : 0x2FE4
    USB Release         : 0.0.4
    Part signature      : 0x1E9541
    Part revision       : E
    Bootloader version  : 2.0.4
    USB max packet size : 64
avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude.exe: Device signature = 0x1e9541 (probably x32a4u)
avrdude.exe: reading input file flip2_demo_app.hex for application
             with 29166 bytes in 1 section within [0, 0x71ed]
             using 114 pages and 18 pad bytes
avrdude.exe: writing 29166 bytes application ...

Writing | ######                                             | 12% 0.07savrdude.exe: Error: DFU_DNLOAD failed: short write
avrdude.exe: Error: Failed to set memory unit 0x00 (Flash)
Writing | #######                                            | 13% 0.07savrdude.exe: Error: Failed to get DFU status: short read
avrdude.exe: Error: Failed to set memory page 0x0000
 ***failed;
avrdude.exe: Error: Failed to get DFU status: short read
avrdude.exe: Error: Failed to write 0x0001 bytes at 0x027E
 ***failed;
avrdude.exe: Error: DFU_DNLOAD failed: short write
avrdude.exe: Error: Failed to set memory page 0x0000
 ***failed;
avrdude.exe: Error: DFU_DNLOAD failed: short write
avrdude.exe: Error: Failed to set memory unit 0x00 (Flash)
 ***failed;
avrdude.exe: Error: Failed to get DFU status: short read
avrdude.exe: Error: Failed to write 0x0001 bytes at 0x02AE
 ***failed;
avrdude.exe: Error: DFU_DNLOAD failed: short write
avrdude.exe: Error: Failed to write 0x0001 bytes at 0x0348
 ***failed;
avrdude.exe: Error: DFU_DNLOAD failed: short write
avrdude.exe: Error: Failed to set memory unit 0x00 (Flash)
 ***failed;
avrdude.exe: Error: DFU_DNLOAD failed: short write
avrdude.exe: Error: Failed to set memory page 0x0000
 ***failed;
avrdude.exe: Error: DFU_DNLOAD failed: short write
avrdude.exe: Error: Failed to write 0x0001 bytes at 0x03E8
 ***failed;
avrdude.exe: Error: DFU_DNLOAD failed: short write
avrdude.exe: Error: Failed to write 0x0001 bytes at 0x03F5
 ***failed;
avrdude.exe: Error: DFU_DNLOAD failed: short write
avrdude.exe: Error: Failed to set memory page 0x0000
 ***failed;
avrdude.exe: Error: DFU_DNLOAD failed: short write
avrdude.exe: Error: Failed to write 0x0001 bytes at 0x05AE
 ***failed;
avrdude.exe: Error: DFU_DNLOAD failed: short write
avrdude.exe: Error: Failed to write 0x0001 bytes at 0x06EF
 ***failed;
avrdude.exe: Error: DFU_DNLOAD failed: short write
avrdude.exe: Error: Failed to write 0x0001 bytes at 0x082E
 ***failed;
CTRL-C

```