avrdudes/avrdude #1082: flip2: ATxmega32A4U DFU bootloader does not work even though Atmel FLIP 3.47 works fine.

mcuee, 2022-08-21:
I have got the DFU bootloader to work on a simple breakout board of ATxmega32A4U (from MCUzone). Atmel FLIP 3.47 works fine. However avrdude does not seem to work.

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c flip2 -p x32a4u -U application:w:flip2_demo_app.hex:i

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude.exe: Device signature = 0x1e9541 (probably x32a4u)
avrdude.exe: NOTE: "application" memory has been specified, an erase cycle will be performed
             To disable this feature, specify the -D option.
avrdude.exe: erasing chip
avrdude.exe: reading input file flip2_demo_app.hex for application
avrdude.exe: writing 29166 bytes application ...

Writing | ############################                       
| 56% 0.31savrdude.exe: Error: DFU_DNLOAD failed: short write
avrdude.exe: Error: Failed to set memory page 0x0000
Writing | #############################                      
| 57% 0.32savrdude.exe: Error: DFU_DNLOAD failed: short write
avrdude.exe: Error: Failed to set memory unit 0x00 (Flash)
 ***failed;
avrdude.exe: Error: DFU_DNLOAD failed: short write
avrdude.exe: Error: Failed to write 0x0001 bytes at 0x0006
 ***failed;
avrdude.exe: Error: Failed to get DFU status: short read
avrdude.exe: Error: Failed to write 0x0001 bytes at 0x0078
 ***failed;
avrdude.exe: Error: DFU_DNLOAD failed: short write
avrdude.exe: Error: Failed to set memory page 0x0000
 ***failed;
avrdude.exe: Error: DFU_DNLOAD failed: short write
avrdude.exe: Error: Failed to set memory page 0x0000
 ***failed;
avrdude.exe: Error: DFU_DNLOAD failed: short write
avrdude.exe: Error: Failed to write 0x0001 bytes at 0x00A2
 ***failed;
avrdude.exe: Error: DFU_DNLOAD failed: short write
avrdude.exe: Error: Failed to set memory page 0x0000
 ***failed;
avrdude.exe: Error: DFU_DNLOAD failed: short write
avrdude.exe: Error: Failed to write 0x0001 bytes at 0x00A5
 ***failed;
avrdude.exe: Error: DFU_DNLOAD failed: short write
avrdude.exe: Error: Failed to write 0x0001 bytes at 0x0101
 ***failed;
avrdude.exe: Error: DFU_DNLOAD failed: short write
avrdude.exe: Error: Failed to set memory page 0x0000
 ***failed;
<CTRL-C>
```

```

mcuee, 2022-08-22:
Microchip xmega DFU bootloader AN (FLIP2)
https://ww1.microchip.com/downloads/en/Appnotes/doc8429.pdf

Alternative DFU bootloader implementations.
https://github.com/kuro68k/xmega_dfu_bootloader (GPL v3)
https://github.com/ganzziani/Xproto-Watch-Bootloader

Official PC host tool: Atmel FLIP (which supports flip1 and flip2)
https://www.microchip.com/en-us/development-tool/flip

Alternative host program other than avrdude
https://sourceforge.net/projects/dfu-programmer/

MCUdude, 2022-10-03:
I wonder if this has anything to do with the ATxmega32A4U being a device with less than 128kiB flash.
I have a hybrid Arduino UNO board with an ATxmega128A3U I can modify to piggyback a USB connector on, so I can check if `flip2` works or not on this chip.

I've had a look at Xmega-based Xplained boards, and very few of them have a "native" USB connector intended for the flip2 bootloader. I have an STK600 that could be used for this, but the socket adapters are so freaking expensive that I can't justify buying one just to help out and potentially fix a single issue.

mcuee, 2023-03-26:
> I wonder if this has anything to do with the ATxmega32A4U being a device with less than 128kiB flash.
> I have a hybrid Arduino UNO board with an ATxmega128A3U I can modify to piggyback a USB connector on, so I can check if flip2 works or not on this chip.

@MCUdude 

I am going through the existing bug reports and this is one of them. Just wondering if you have got a chance to test on your side. Thanks.

mcuee, 2023-05-07:
Unfortunately I can not get the USB bootloader working now for this device. Strange. The device can not be recognized by Windows 11 or Ubuntu Linux 20.04.

mcuee, 2023-05-07:
The bootloader works fine with dfu-programmer (https://github.com/dfu-programmer/dfu-programmer).
```
mcuee@UbuntuSwift3:~/build/avr/dfu-programmer$ dfu-programmer --version
dfu-programmer 1.0.0

mcuee@UbuntuSwift3:~/build/avr/dfu-programmer$ dfu-programmer atxmega32a4u erase
Checking memory from 0x0 to 0x7FFF...  Not blank at 0x1.
Erasing flash...  Success
Checking memory from 0x0 to 0x7FFF...  Empty.

mcuee@UbuntuSwift3:~/build/avr/avrdude_bin$ dfu-programmer atxmega32a4u flash flip2_demo_app.hex 
Checking memory from 0x0 to 0x71FF...  Empty.
0%                            100%  Programming 0x7200 bytes...
[>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>]  Success
0%                            100%  Reading 0x8000 bytes...
[>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>]  Success
Validating...  Success
0x7200 bytes written into 0x8000 bytes memory (89.06%).
```

mcuee, 2023-05-07:
Hmm, now it works under Linux.

```
mcuee@UbuntuSwift3:~/build/avr/avrdude_bin$ ./avrdude -c flip2 -p x32a4u -e
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9541 (probably x32a4u)
avrdude: erasing chip

avrdude done.  Thank you.

mcuee@UbuntuSwift3:~/build/avr/avrdude_bin$ ./avrdude -c flip2 -p x32a4u -U application:w:flip2_demo_app.hex:i
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9541 (probably x32a4u)
avrdude: Note: application memory has been specified, an erase cycle will be performed.
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file flip2_demo_app.hex for application
         with 29166 bytes in 1 section within [0, 0x71ed]
         using 114 pages and 18 pad bytes
avrdude: writing 29166 bytes application ...
Writing | ################################################## | 100% 0.55 s 
avrdude: 29166 bytes of application written
avrdude: verifying application memory against flip2_demo_app.hex
Reading | ################################################## | 100% 0.21 s 
avrdude: 29166 bytes of application verified

avrdude done.  Thank you.
```

mcuee, 2023-05-07:
It works under Windows as well.

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c flip2 -p x32a4u -U application:w:flip2_demo_app.hex:i
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9541 (probably x32a4u)
avrdude: Note: application memory has been specified, an erase cycle will be performed.
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file flip2_demo_app.hex for application
         with 29166 bytes in 1 section within [0, 0x71ed]
         using 114 pages and 18 pad bytes
avrdude: writing 29166 bytes application ...
Writing | ################################################## | 100% 0.61 s
avrdude: 29166 bytes of application written
avrdude: verifying application memory against flip2_demo_app.hex
Reading | ################################################## | 100% 0.49 s
avrdude: 29166 bytes of application verified

avrdude done.  Thank you.
```

closed 2023-05-07 by mcuee