avrdudes/avrdude #1085: 4 parts in avrdude.conf claim interfaces they don't have

stefanrueger, 2022-08-23:
According to the corresponding ATDF files
 - ATtiny11 and ATtiny28 do not have an ISP interface, so `avrdude.conf.in` should state `serial = no;`
 - ATxmega32A4 and ATxmega16A4 do not have a JTAG interface, so `avrdude.conf.in` should state `has_jtag = no;`

MCUdude, 2022-08-23:
Another error in avrdude.conf I'm aware of is `reset`. It can either be `dedicated` or `io`, and not all devices has this. I'm not sure what this is even used for, and I'm not sure all occurrences are correct.

stefanrueger, 2022-08-23:
`reset = [dedicated|io];` see https://github.com/avrdudes/avrdude/blob/731d5813985ff88608c2199fc42f0c30decc16e2/src/libavrdude.h#L161-L164

I am guessing(!) that when grepping for the reset vector caption in ATDF
```
$ grep '"0" *name=.reset' -i atdf/ATmega* | head
atdf/ATmega1280.atdf:        <interrupt index="0" name="RESET" caption="External Pin,Power-on Reset,Brown-out Reset,Watchdog Reset,and JTAG AVR Reset. See Datasheet.     "/>
atdf/ATmega1281.atdf:        <interrupt index="0" name="RESET" caption="External Pin,Power-on Reset,Brown-out Reset,Watchdog Reset,and JTAG AVR Reset. See Datasheet.     "/>
atdf/ATmega1284.atdf:        <interrupt index="0" name="RESET" caption="External Pin,Power-on Reset,Brown-out Reset,Watchdog Reset,and JTAG AVR Reset. See Datasheet.     "/>
atdf/ATmega1284P.atdf:        <interrupt index="0" name="RESET" caption="External Pin,Power-on Reset,Brown-out Reset,Watchdog Reset,and JTAG AVR Reset. See Datasheet.     "/>
atdf/ATmega1284RFR2.atdf:        <interrupt index="0" name="RESET" caption="External Pin,Power-on Reset,Brown-out Reset,Watchdog Reset,and JTAG AVR Reset. See Datasheet."/>
atdf/ATmega128A.atdf:        <interrupt index="0" name="RESET" caption="External Pin, Power-on Reset, Brown-out Reset, Watchdog Reset and JTAG AVR Reset"/>
atdf/ATmega128.atdf:        <interrupt index="0" name="RESET" caption="External Pin, Power-on Reset, Brown-out Reset, Watchdog Reset and JTAG AVR Reset"/>
atdf/ATmega128RFA1.atdf:        <interrupt index="0" name="RESET" caption="External Pin,Power-on Reset,Brown-out Reset,Watchdog Reset,and JTAG AVR Reset. See Datasheet.     "/>
atdf/ATmega128RFR2.atdf:        <interrupt index="0" name="RESET" caption="External Pin,Power-on Reset,Brown-out Reset,Watchdog Reset,and JTAG AVR Reset. See Datasheet."/>
atdf/ATmega162.atdf:        <interrupt index="0" name="RESET" caption="External Pin,Power-on Reset,Brown-out Reset,Watchdog Reset,and JTAG AVR Reset. See Datasheet.     "/>
```
if the capions says External Pin, it will be `io`.

Used to inititialise an STK500 device in v1 protocol, so ... not really pressing anymore, and I actually cannot imagine what the STK500 programmer would do with that info? Change the fuses? With HVPP? Well, ...

stefanrueger, 2022-08-23:
It's initialised to `dedicated`, so would expect that all `io` devices need to spell this out. Above grep is a crude tool and only catches 195 parts, 162 of which mention External Pin, and 33 don't.

dl8dtl, 2022-08-23:
It's probably useless information, that has only been added once since the STK500v1 protocol used to ask for it.

mcuee, 2022-08-24:
> It's probably useless information, that has only been added once since the STK500v1 protocol used to ask for it.

The codes are still there.
https://github.com/avrdudes/avrdude/blob/731d5813985ff88608c2199fc42f0c30decc16e2/src/stk500.c#L578-583

dl8dtl, 2022-08-24:
> The codes are still there.

Yes, sure, since the protocol requires it.

But I agree with you that the information is likely to be useless, as I cannot imagine as well what the STK500 might do with it. Someone could reverse engineer their code to see if it's being ignored completely or not. ;-)

mcuee, 2022-08-24:
> > The codes are still there.
> 
> Yes, sure, since the protocol requires it.
> 
> But I agree with you that the information is likely to be useless, as I cannot imagine as well what the STK500 might do with it. Someone could reverse engineer their code to see if it's being ignored completely or not. ;-)

I see. If I read the commit message of https://github.com/avrdudes/avrdude/commit/8aa98920ef523de648ffa901f3785f44eeea5709 correctly, it is only useful for high voltage programming of STK500 v1. So in the end it does not matter now that we deprecate the old STK500 v1 FW (for the original Atmel STK500 HW) in favor of the current partial implementation like optiboot or `Arduino as ISP`, the issue becomes an non issue. How it works with the HVPP, I have no idea. 

But I might give it try when I got time with my STK500 HV programmer which is compatible with the original Atmel STK500 using the V1 FW, to see if it will fail if I modify avrdude.conf. I guess I have to use old avrdude version as the current avrdude seems to support only HVSP/HVPP with V2 FW.
```
  stk500           = Atmel STK500
  stk500hvsp       = Atmel STK500 V2 in high-voltage serial programming mode
  stk500pp         = Atmel STK500 V2 in parallel programming mode
  stk500v1         = Atmel STK500 Version 1.x firmware
  stk500v2         = Atmel STK500 Version 2.x firmware
```

dl8dtl, 2022-08-24:
I could also give that a try with the original STK500.

I wonder which parts the information might really be important for, HVSP maybe?

dl8dtl, 2022-08-24:
> very old avrdude version as the current avrdude seems to support only HVSP/HVPP with V2 FW

Thinking more about it: AVRDUDE very likely never supported HV programming with V1 firmware. The entire STK500v1 support was started by using the "SPI MULTI" command to just wrap all the ISP logic that was already present in AVRDUDE. That seemed to be the fastest path to support the STK500 at all by its time. HV support would have required a major overhaul of the entire code which was then only done for V2 (and even then, "SPI MULTI" was still used for many details).

mcuee, 2022-08-25:
> Thinking more about it: AVRDUDE very likely never supported HV programming with V1 firmware. The entire STK500v1 support was started by using the "SPI MULTI" command to just wrap all the ISP logic [...]

@dl8dtl 
Indeed that is the case. Only STK500v2 supports high voltage programming. The initial related commit for HVPP was on Jul 17, 2006 and in the commit message you clear mentioned that it was for V2 FW.
https://github.com/avrdudes/avrdude/commit/8bdbf1774ef397adc5ab75ab351a8665a3bb636f

HVSP was added a bit later on  on Jul 22, 2006.
https://github.com/avrdudes/avrdude/commit/dbf083fb4d46e75be6175008080dd53e169c0763

stefanrueger, 2022-08-30:
Fixed with PR #1086 
Also repaired the `reset = dedicated | io` components, principally because it's documented and might as well be correct.

closed 2022-08-30 by stefanrueger