avrdudes/avrdude #1087: avrdude.conf warnings _may_ be a bit too much...

MCUdude, 2022-08-25:
Just for fun, I tested the avrdude.conf file Arduino IDE ships with, and it appears that it has a few errors in it. None of them are crucial when using official Arduino hardware. Wouldn't it be better to output the avrdude.conf warnings in the developer options instead? IMO this is a bit noisy when using a legacy avrdude.conf file.

```
$ ./avrdude -C /Applications/Arduino.app/Contents/Java/hardware/tools/avr/etc/avrdude.conf -cusbasp -patmega2560
avrdude: warning at /Applications/Arduino.app/Contents/Java/hardware/tools/avr/etc/avrdude.conf:5684: ATmega329's flash writepage misses a necessary address bit a13
avrdude: warning at /Applications/Arduino.app/Contents/Java/hardware/tools/avr/etc/avrdude.conf:5910: ATmega649's flash writepage misses a necessary address bit a14
avrdude: warning at /Applications/Arduino.app/Contents/Java/hardware/tools/avr/etc/avrdude.conf:9107: AT90PWM2's eeprom writepage misses a necessary address bit a8
avrdude: warning at /Applications/Arduino.app/Contents/Java/hardware/tools/avr/etc/avrdude.conf:10965: ATtiny44's eeprom writepage misses a necessary address bit a7
avrdude: warning at /Applications/Arduino.app/Contents/Java/hardware/tools/avr/etc/avrdude.conf:11149: ATtiny84's eeprom writepage misses a necessary address bit a8
avrdude: warning at /Applications/Arduino.app/Contents/Java/hardware/tools/avr/etc/avrdude.conf:11434: a4 would normally be expected to be a5
avrdude: warning at /Applications/Arduino.app/Contents/Java/hardware/tools/avr/etc/avrdude.conf:11434: a3 would normally be expected to be a4
avrdude: warning at /Applications/Arduino.app/Contents/Java/hardware/tools/avr/etc/avrdude.conf:11434: a2 would normally be expected to be a3
avrdude: warning at /Applications/Arduino.app/Contents/Java/hardware/tools/avr/etc/avrdude.conf:11434: a1 would normally be expected to be a2
avrdude: warning at /Applications/Arduino.app/Contents/Java/hardware/tools/avr/etc/avrdude.conf:11434: a0 would normally be expected to be a1
avrdude: warning at /Applications/Arduino.app/Contents/Java/hardware/tools/avr/etc/avrdude.conf:11434: too few opcode bits in instruction
avrdude: warning at /Applications/Arduino.app/Contents/Java/hardware/tools/avr/etc/avrdude.conf:11453: ATtiny43u's eeprom read misses a necessary address bit a5
avrdude: warning at /Applications/Arduino.app/Contents/Java/hardware/tools/avr/etc/avrdude.conf:11639: ATmega32U4's eeprom writepage misses a necessary address bit a2
avrdude: warning at /Applications/Arduino.app/Contents/Java/hardware/tools/avr/etc/avrdude.conf:12034: AT90USB1286's eeprom writepage misses a necessary address bit a11
avrdude: warning at /Applications/Arduino.app/Contents/Java/hardware/tools/avr/etc/avrdude.conf:14790: ATtiny1634's flash writepage misses a necessary address bit a5

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9801 (probably m2560)

avrdude done.  Thank you.
```

mcuee, 2022-08-25:
I agree. Similar output here.
* https://github.com/avrdudes/avrdude/issues/1060#issuecomment-1226807809

On the other hand, it may help push people to upgrade to new avrdude version. Haha.

stefanrueger, 2022-08-25:
The legacy `avrdude.conf` is actually broken for these parts. These are serious errors. Should AVRDUDE really ignore these errors in the *hope* the user won't need those parts? What is the use case for running a new avrdude with a legacy .conf file? 

Without these warnings the project wouldn't know that part descriptions that users contribute are broken (which has happened in the past to the extent that 25% of the classic parts had broken SPI programming commands).

mcuee, 2022-08-25:
@stefanrueger 
I tend to agree that there are no important use case to use old avrdude.conf with new release of avrdude.

In the following case, it is purely for testing purpose -- @avrfreak provides an avrdude.conf mod based on old version. and then I will tend to test using latest git avrdude.
* https://github.com/avrdudes/avrdude/issues/1060#issuecomment-1226807809

MCUdude, 2022-08-25:
> I tend to agree that there are no important use case to use old avrdude.conf with new release of avrdude.

Well, I do agree. If Arduino decides to bundle Avrdude 7.1 with their IDE, they should fix or replace their avrdude.conf.

MCUdude, 2022-08-28:
It would be really cool if one could actually utilize the Avrdude to fix a defective avrdude.conf file. Perhaps through the development options?

mcuee, 2022-08-29:
> It would be really cool if one could actually utilize the Avrdude to fix a defective avrdude.conf file. Perhaps through the development options?

That may be a bit dangerous if the user has a heavily customized avrdude.conf file.

dl8dtl, 2022-08-29:
> That may be a bit dangerous if the user has a heavily customized avrdude.conf file.

`avrdude.conf` is normally not meant to be customized. Its placement under `etc/` is basically wrong, it ought to be in `share/` instead. The user config file is meant to be the home for customizations.

MCUdude, 2022-08-29:
> That may be a bit dangerous if the user has a heavily customized avrdude.conf file.

Perhaps it would be better to create a new file, based on the old one, where the fixes have been applied

MCUdude, 2022-09-10:
Missing MCU IDs also seem to cause a bunch of warnings:
This is when using a slightly older avrdude.conf file.
I do agree that real errors like pointed out previously is a real problem, but missing MCU IDs aren't really a problem unless the user is using urclock/urboot. How about only showing the MCU ID error when urclock/urboot is being used?

```
$ ./avrdude -patmega4808 -cserialupdi -P /dev/cu.usbserial-1410 -Uflash:w:/var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_84595/Blink.ino.hex 
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:1612: assigned mcuid = 8 to part ATtiny11
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:1745: assigned mcuid = 9 to part ATtiny12
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:1918: assigned mcuid = 10 to part ATtiny13
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:1927: assigned mcuid = 11 to part ATtiny13A
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:2060: assigned mcuid = 12 to part ATtiny15
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:2176: assigned mcuid = 183 to part AT90S1200
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:2289: assigned mcuid = 190 to part AT90S4414
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:2402: assigned mcuid = 186 to part AT90S2313
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:2529: assigned mcuid = 188 to part AT90S2333
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:2653: assigned mcuid = 189 to part AT90S2343
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:2778: assigned mcuid = 191 to part AT90S4433
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:2861: assigned mcuid = 192 to part AT90S4434
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:2975: assigned mcuid = 193 to part AT90S8515
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:3096: assigned mcuid = 195 to part AT90S8535
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:3235: assigned mcuid = 84 to part ATmega103
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:3416: assigned mcuid = 72 to part ATmega64
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:3425: assigned mcuid = 73 to part ATmega64A
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:3605: assigned mcuid = 85 to part ATmega128
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:3614: assigned mcuid = 86 to part ATmega128A
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:3806: assigned mcuid = 176 to part AT90CAN128
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:3998: assigned mcuid = 172 to part AT90CAN64
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:4190: assigned mcuid = 171 to part AT90CAN32
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:4368: assigned mcuid = 49 to part ATmega16
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:4377: assigned mcuid = 50 to part ATmega16A
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:4567: assigned mcuid = 111 to part ATmega324P
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:4652: assigned mcuid = 93 to part ATmega164P
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:4661: assigned mcuid = 94 to part ATmega164PA
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:4671: assigned mcuid = 92 to part ATmega164A
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:4681: assigned mcuid = 113 to part ATmega324PB
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:4691: assigned mcuid = 112 to part ATmega324PA
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:4701: assigned mcuid = 110 to part ATmega324A
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:4891: assigned mcuid = 127 to part ATmega644
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:4900: assigned mcuid = 128 to part ATmega644A
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:4910: assigned mcuid = 129 to part ATmega644P
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:4920: assigned mcuid = 130 to part ATmega644PA
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:5110: assigned mcuid = 140 to part ATmega1284
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:5120: assigned mcuid = 141 to part ATmega1284P
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:5316: assigned mcuid = 90 to part ATmega162
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:5476: assigned mcuid = 91 to part ATmega163
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:5660: assigned mcuid = 104 to part ATmega169
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:5670: assigned mcuid = 105 to part ATmega169A
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:5680: assigned mcuid = 106 to part ATmega169P
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:5690: assigned mcuid = 107 to part ATmega169PA
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:5875: assigned mcuid = 121 to part ATmega329
[... cut ...]
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:19335: assigned mcuid = 227 to part LGT8F88P
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:19343: assigned mcuid = 228 to part LGT8F168P
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:19351: assigned mcuid = 229 to part LGT8F328P
avrdude: ser_open(): can't open device "/dev/cu.usbserial-1410": Resource busy
avrdude: opening programmer "serialupdi" on port "/dev/cu.usbserial-1410" failed

avrdude done.  Thank you.
```

stefanrueger, 2022-09-11:
Fair point. Commit [168a8b9](https://github.com/avrdudes/avrdude/commit/168a8b92e966ac2e42d455fb28faae10ccd392d2) solves this.

closed 2022-09-11 by stefanrueger

MCUdude, 2022-09-11:
@stefanrueger sorry, but the only thing that changed was the warning message, nothing else really...

```
$ ./avrdude -C /users/hans/Downloads/avrdude/src/avrdude.conf -patmega4808 -cserialupdi -P /dev/cu.usbserial-1410 -Uflash:w:/var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_84595/Blink.ino.hex 
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:1927: overwriting mcuid of part ATtiny13A to be 11
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:3425: overwriting mcuid of part ATmega64A to be 73
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:3614: overwriting mcuid of part ATmega128A to be 86
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:4377: overwriting mcuid of part ATmega16A to be 50
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:4652: overwriting mcuid of part ATmega164P to be 93
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:4661: overwriting mcuid of part ATmega164PA to be 94
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:4671: overwriting mcuid of part ATmega164A to be 92
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:4681: overwriting mcuid of part ATmega324PB to be 113
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:4691: overwriting mcuid of part ATmega324PA to be 112
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:4701: overwriting mcuid of part ATmega324A to be 110
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:4900: overwriting mcuid of part ATmega644A to be 128
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:4910: overwriting mcuid of part ATmega644P to be 129
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:4920: overwriting mcuid of part ATmega644PA to be 130
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:5120: overwriting mcuid of part ATmega1284P to be 141
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:5670: overwriting mcuid of part ATmega169A to be 105
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:5680: overwriting mcuid of part ATmega169P to be 106
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:5690: overwriting mcuid of part ATmega169PA to be 107
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:5884: overwriting mcuid of part ATmega329A to be 122
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:5894: overwriting mcuid of part ATmega329P to be 123
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:5904: overwriting mcuid of part ATmega329PA to be 124
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:5914: overwriting mcuid of part ATmega3290 to be 150
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:5924: overwriting mcuid of part ATmega3290A to be 151
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:5934: overwriting mcuid of part ATmega3290P to be 152
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:5944: overwriting mcuid of part ATmega3290PA to be 153
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:6138: overwriting mcuid of part ATmega649A to be 136
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:6148: overwriting mcuid of part ATmega649P to be 137
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:6158: overwriting mcuid of part ATmega6490 to be 157
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:6168: overwriting mcuid of part ATmega6490A to be 158
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:6178: overwriting mcuid of part ATmega6490P to be 159
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:6503: overwriting mcuid of part ATmega32A to be 59
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:6673: overwriting mcuid of part ATmega8A to be 46
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:7340: overwriting mcuid of part ATtiny261A to be 31
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:7537: overwriting mcuid of part ATtiny461A to be 34
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:7734: overwriting mcuid of part ATtiny861A to be 39
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:7991: overwriting mcuid of part ATmega48A to be 68
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:8001: overwriting mcuid of part ATmega48P to be 69
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:8011: overwriting mcuid of part ATmega48PA to be 70
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:8022: overwriting mcuid of part ATmega48PB to be 71
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:8219: overwriting mcuid of part ATmega88A to be 80
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:8229: overwriting mcuid of part ATmega88P to be 81
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:8239: overwriting mcuid of part ATmega88PA to be 82
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:8250: overwriting mcuid of part ATmega88PB to be 83
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:8449: overwriting mcuid of part ATmega168A to be 100
[... cut ...]
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:19335: overwriting mcuid of part LGT8F88P to be 227
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:19343: overwriting mcuid of part LGT8F168P to be 228
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:19351: overwriting mcuid of part LGT8F328P to be 229
avrdude: ser_open(): can't open device "/dev/cu.usbserial-1410": No such file or directory
avrdude: opening programmer "serialupdi" on port "/dev/cu.usbserial-1410" failed

avrdude done.  Thank you.
```

stefanrueger, 2022-09-11:
Hmm, I don't know how this can be: avr_new_part(void) initialises mcuid to -1; unless mcuid is set in your old avrdude.conf the lines https://github.com/avrdudes/avrdude/blob/168a8b92e966ac2e42d455fb28faae10ccd392d2/src/config.c#L857-L858 should protect you from the warning. I need to give it a closer look.

Setting known mcuid as default is a feature that does not need a warning. The warning is only warranted if the avrdude.conf file sets mcuid to an incompatible value in which case it is overwritten.

MCUdude, 2022-09-11:
Here's something interesting. I edited the source code to output the existing MCU ID. The avrdude.conf file I used does not have _any_ `mcuid` parameters at all.

```
$ ./avrdude -C /users/hans/Downloads/avrdude/src/avrdude.conf -patmega4808 -cserialupdi -P /dev/cu.usbserial-1410 -Uflash:w:/var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_84595/Blink.ino.hex 
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:1927: overwriting mcuid of part ATtiny13A to be 11, was 10
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:3425: overwriting mcuid of part ATmega64A to be 73, was 72
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:3614: overwriting mcuid of part ATmega128A to be 86, was 85
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:4377: overwriting mcuid of part ATmega16A to be 50, was 49
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:4652: overwriting mcuid of part ATmega164P to be 93, was 111
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:4661: overwriting mcuid of part ATmega164PA to be 94, was 93
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:4671: overwriting mcuid of part ATmega164A to be 92, was 93
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:4681: overwriting mcuid of part ATmega324PB to be 113, was 111
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:4691: overwriting mcuid of part ATmega324PA to be 112, was 111
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:4701: overwriting mcuid of part ATmega324A to be 110, was 111
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:4900: overwriting mcuid of part ATmega644A to be 128, was 127
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:4910: overwriting mcuid of part ATmega644P to be 129, was 127
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:4920: overwriting mcuid of part ATmega644PA to be 130, was 127
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:5120: overwriting mcuid of part ATmega1284P to be 141, was 140
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:5670: overwriting mcuid of part ATmega169A to be 105, was 104
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:5680: overwriting mcuid of part ATmega169P to be 106, was 104
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:5690: overwriting mcuid of part ATmega169PA to be 107, was 104
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:5884: overwriting mcuid of part ATmega329A to be 122, was 121
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:5894: overwriting mcuid of part ATmega329P to be 123, was 121
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:5904: overwriting mcuid of part ATmega329PA to be 124, was 121
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:5914: overwriting mcuid of part ATmega3290 to be 150, was 121
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:5924: overwriting mcuid of part ATmega3290A to be 151, was 121
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:5934: overwriting mcuid of part ATmega3290P to be 152, was 121
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:5944: overwriting mcuid of part ATmega3290PA to be 153, was 121
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:6138: overwriting mcuid of part ATmega649A to be 136, was 135
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:6148: overwriting mcuid of part ATmega649P to be 137, was 135
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:6158: overwriting mcuid of part ATmega6490 to be 157, was 135
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:6168: overwriting mcuid of part ATmega6490A to be 158, was 135
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:6178: overwriting mcuid of part ATmega6490P to be 159, was 135
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:6503: overwriting mcuid of part ATmega32A to be 59, was 58
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:6673: overwriting mcuid of part ATmega8A to be 46, was 45
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:7340: overwriting mcuid of part ATtiny261A to be 31, was 30
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:7537: overwriting mcuid of part ATtiny461A to be 34, was 33
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:7734: overwriting mcuid of part ATtiny861A to be 39, was 38
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:7991: overwriting mcuid of part ATmega48A to be 68, was 67
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:8001: overwriting mcuid of part ATmega48P to be 69, was 67
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:8011: overwriting mcuid of part ATmega48PA to be 70, was 67
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:8022: overwriting mcuid of part ATmega48PB to be 71, was 67
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:8219: overwriting mcuid of part ATmega88A to be 80, was 79
[... cut ...]
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:19335: overwriting mcuid of part LGT8F88P to be 227, was 79
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:19343: overwriting mcuid of part LGT8F168P to be 228, was 99
avrdude: warning at /Users/hans/Downloads/avrdude/src/avrdude.conf:19351: overwriting mcuid of part LGT8F328P to be 229, was 118
avrdude: ser_open(): can't open device "/dev/cu.usbserial-1410": No such file or directory
avrdude: opening programmer "serialupdi" on port "/dev/cu.usbserial-1410" failed

avrdude done.  Thank you.
```

stefanrueger, 2022-09-11:
Ahh, I get it. The remaining warnings are from parts that are instantiated through a parent: the parent gets mcuid automagically set as the part was known at compile time and mcuid is still in uninitialised state; now the child has a mcuid, which is no longer uninitialised but still wrong.

I have raised the warning level to DEBUG so you normally won't see these remaining warnings.

stefanrueger, 2022-09-11:
> missing MCU IDs aren't really a problem unless the user is using urclock/urboot. How about only showing the MCU ID error when urclock/urboot is being used?

BTW, I don't quite agree with the suggestion that errors in avrdude.conf should only be pointed out when they would actually be of consequence in a specific session. I prefer avrdude.conf to be correct no matter what.

In the case of wrong mcuids they can fortuitously be corrected from independent compile-time knowledge. Therefore the warning is less relevant in this particular case. Generally, I feel we actually should warn more often when parameters do not pass sanity checks, of which there are in my opinion far too few. For example, avrdude.conf has had wrong ISP commands for many parts over a long period, because AVRDUDE missed to implement sanity checks and therefore warned too little.

On the other hand, some of the current sanity checks stop AVRDUDE from progressing with a hard error rather than a warning. Here one might argue that exiting with an error is too harsh, as it could really be that the session would have progressed correctly, as the error pertained to a programmer or part or method not used in that particular session. I tend to be in the warn-rather-than-exit camp here (though there might be cases where an error prevented a user from damaging a part).

MCUdude, 2022-09-12:
> I have raised the warning level to DEBUG so you normally won't see these remaining warnings.

I'm fine with this solution, thanks!

> BTW, I don't quite agree with the suggestion that errors in avrdude.conf should only be pointed out when they would actually be of consequence in a specific session. I prefer avrdude.conf to be correct no matter what.

To me, it's important that Avrdude "allows" a user to use an older avrdude.conf file without throwing a bunch of warnings, so many that it's annoying. I agree that _actual_ errors that may affect the programming of AVRs should throw a warning. But in this case, 150+ lines of warnings for a newly added ID number that isn't relevant for all current programmers was IMO a bit too much.