avrdudes/avrdude #1090: Error reading signature data for part "ATtiny1616"

cuiqingwei, 2022-08-28:
xxx\avrdude-v7.0-windows-x64>avrdude.exe -cstk500v1 -PCOM3 -b9600 -pATtiny1616 -e -v

avrdude.exe: Version 7.0
             Copyright (c) Brian Dean, http://www.bdmicro.com/
             Copyright (c) Joerg Wunsch

             System wide configuration file is "C:/Users/Gary/Desktop/MR8910/avrdude-v7.0-windows-x64/avrdude.conf"

             Using Port                    : COM3
             Using Programmer              : stk500v1
             Overriding Baud Rate          : 9600
             AVR Part                      : ATtiny1616
             RESET disposition             : dedicated
             RETRY pulse                   : SCK
             Serial program mode           : yes
             Parallel program mode         : yes
             Memory Detail                 :

                                               Block Poll               Page                       Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
               prodsig                 0     0     0    0 no         61   61      0     0     0 0x00 0x00
               sernum                  0     0     0    0 no         10    1      0     0     0 0x00 0x00
               osccal16                0     0     0    0 no          2    1      0     0     0 0x00 0x00
               osccal20                0     0     0    0 no          2    1      0     0     0 0x00 0x00
               tempsense               0     0     0    0 no          2    1      0     0     0 0x00 0x00
               osc16err                0     0     0    0 no          2    1      0     0     0 0x00 0x00
               osc20err                0     0     0    0 no          2    1      0     0     0 0x00 0x00
               fuses                   0     0     0    0 no          9   10      0     0     0 0x00 0x00
               fuse0       wdtcfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
               fuse1       bodcfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
               fuse2       osccfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
               fuse4       tcd0cfg     0     0     0    0 no          1    1      0     0     0 0x00 0x00
               fuse5       syscfg0     0     0     0    0 no          1    1      0     0     0 0x00 0x00
               fuse6       syscfg1     0     0     0    0 no          1    1      0     0     0 0x00 0x00
               fuse7       append      0     0     0    0 no          1    1      0     0     0 0x00 0x00
               fuse8       bootend     0     0     0    0 no          1    1      0     0     0 0x00 0x00
               lock                    0     0     0    0 no          1    1      0     0     0 0x00 0x00
               data                    0     0     0    0 no          0    1      0     0     0 0x00 0x00
               userrow     usersig     0     0     0    0 no         32   32      0     0     0 0x00 0x00
               flash                   0     0     0    0 no      16384   64      0     0     0 0x00 0x00
               eeprom                  0     0     0    0 no        256   32      0     0     0 0x00 0x00

             Programmer Type : STK500
             Description     : Atmel STK500 Version 1.x firmware
             Hardware Version: 0
             Firmware Version: 16.0
             Topcard         : Unknown
             Vtarget         : 0.0 V
             Varef           : 0.0 V
             Oscillator      : Off
             SCK period      : 0.1 us

avrdude.exe: AVR device initialized and ready to accept instructions

Reading |                                                    | 0% 0.00savr_read(): error reading address 0x0000
    read operation not supported for memory "signature"
avrdude.exe: error reading signature data for part "ATtiny1616", rc=-2
avrdude.exe: error reading signature data, rc=-2
avrdude.exe: stk500_recv(): programmer is not responding

avrdude.exe done.  Thank you.

mcuee, 2022-08-28:
@cuiqingwei 
I assume you are using a bootloader here. Please state which bootloader you are using? Are you using the optiboot_x bootloader here? 
https://github.com/SpenceKonde/megaTinyCore/tree/master/megaavr/bootloaders

You may want to use `-c arduino` and try again to see if that helps.

cuiqingwei, 2022-08-28:
I use the bootloader generated by the ATMEL（[https://start.atmel.com/](https://start.atmel.com/)） wizard。

mcuee, 2022-08-29:
@cuiqingwei 
It seems to me this is not an avrdude problem but your bootloader is not working. So you may want to ask in some other forum (eg: avrfreaks) to see if you can get the bootloader working or not.

Or you can try the bootloader here.
https://github.com/SpenceKonde/megaTinyCore/tree/master/megaavr/bootloaders

mcuee, 2022-08-31:
Close this for now. Please reopen if you have found potential issues with avrdude.

mcuee, 2022-08-31:
Reference here: apparently the STK500 bootloader middleware is badly written and it is not a good example to follow.
https://www.avrfreaks.net/forum/atmel-start-stk500-serial-bootloader-tiny817

cuiqingwei, 2022-09-01:
I have fixed the problem of stk500 bootloader middleware. 
Avrdude's support for ATtiny1616 is still problematic.
1. default，The phenomenon is as follows:
```
>>>: avrdude -c stk500v1 -p t1616 -P COM3 -b 115200 -F -V -D -U flash:w:"C:\Users\Gary\Desktop\MR8910\tools\avrdude-v7.0-windows-x64\ATtiny1616_APP.hex":i 

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | avr_read(): error reading address 0x0000
    read operation not supported for memory "signature"
avrdude.exe: error reading signature data for part "ATtiny1616", rc=-2
avrdude.exe: error reading signature data, rc=-2
avrdude.exe: stk500_disable(): protocol error, expect=0x14, resp=0xff

avrdude.exe done.  Thank you.
```

2. After modifying the **avrdude.conf** configuration file, 

```
#------------------------------------------------------------
# ATtiny1616
#------------------------------------------------------------

part parent    ".avr8x_tiny"
    id        = "t1616";
    desc      = "ATtiny1616";
    signature = 0x1E 0x94 0x21;

    memory "flash"
        size      = 0x4000;
        offset    = 0x8000;
        page_size = 0x40;
        readsize  = 0x100;
    ;

    memory "eeprom"
        size      = 0x100;
        offset    = 0x1400;
        page_size = 0x20;
        readsize  = 0x100;
    ;
	
	#   ATtiny1616 has Signature Bytes: 0x1E 0x94 0x21
     memory "signature"
         size            = 3;
         read            = "0  0  1  1   0  0  0  0   0  0  0  x   x  x  x  x",
                           "x  x  x  x   x  x a1 a0   o  o  o  o   o  o  o  o";
       ;
;
```

the phenomenon is as follows:
```
>>>: avrdude -c stk500v1 -p t1616 -P COM3 -b 115200 -F -V -D -U flash:w:"C:\Users\Gary\Desktop\MR8910\tools\avrdude-v7.0-windows-x64\ATtiny1616_APP.hex":i 

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude.exe: Device signature = 0x000000 (retrying)

Reading | ################################################## | 100% 0.00s

avrdude.exe: Device signature = 0x000000 (retrying)

Reading | ################################################## | 100% 0.00s

avrdude.exe: Device signature = 0x000000
avrdude.exe: Yikes!  Invalid device signature.
avrdude.exe: Expected signature for ATtiny1616 is 1E 94 21
avrdude.exe: reading input file "C:\Users\Gary\Desktop\MR8910\tools\avrdude-v7.0-windows-x64\ATtiny1616_APP.hex"
avrdude.exe: writing flash (3404 bytes):

Writing | ################################################## | 100% 0.00s

avrdude.exe: 3404 bytes of flash written
avrdude.exe: stk500_disable(): protocol error, expect=0x14, resp=0xff

avrdude.exe done.  Thank you.
```

mcuee, 2022-09-01:
@cuiqingwei 
Please upload the bootloader Atmel Start project so that the people here can check whether they can reproduce the issue or not. And they may be able to check to see if it is a problem with the bootloader or with avrdude. 

I am not so sure why adding the `signature` memory region helps your situation. Apparently the bootloader is still not working properly since it can not even read the device signature.

cuiqingwei, 2022-09-01:
```
void BlockLoad(uint8_t buffer[], uint8_t mem, uint16_t size, native_pointer_t *address)
{

	native_pointer_t tempaddress;
	uint8_t          i;

	// EEPROM memory type.
	if ((mem == 'E') && BOOTLOADER_ENABLE_EEPROM_SUPPORT) {
		/* Fill buffer first, as EEPROM is too slow to copy with UART speed */
		for (tempaddress = 0; tempaddress < size; tempaddress++)
			buffer[tempaddress] = USART_0_read();

		/* Then program the EEPROM */
		FLASH_0_write_eeprom_block(*address, buffer, size);
		*address = *address + size;
		return;
	}

	// Flash memory type.
	else if (mem == 'F') {        // NOTE: For flash programming, 'address' is given in words.
		//(*address) <<= 1;         // Convert address to bytes temporarily.
		tempaddress = (*address); // Store address in page.

		for (i = 0; i < size; i++) {
			/* Programmer always writes complete pages, so finalize parameter is set to false. */
			// FLASH_0_write_flash_stream(BOOTSIZE + tempaddress + i, USART_0_read(), false);
			/* 地址偏移还是需要再build固件的之后制定，.text=BOOTSIZE */
			FLASH_0_write_flash_stream(tempaddress + i, USART_0_read(), false);
		}

		*address = *address + size;
		//(*address) >>= 1; // Convert address back to Flash words again.
		return;
	}
}
```
modify void **BlockLoad(uint8_t buffer[], uint8_t mem, uint16_t size, native_pointer_t *address)** in bootloader.c

cuiqingwei, 2022-09-01:
My main problem is how to make **avrdude** not to verify the signature with attiny1616？

mcuee, 2022-09-01:
> My main problem is how to make **avrdude** not to verify the signature of ATtiny1616？

Which is done by `-F`. 

But you have to fix your bootloader in the end. That is the right solution. You need to implement the bootloader function to be able to read the device signature. Your bootloader is broken without that function.

cuiqingwei, 2022-09-01:
```
SUDT ACCESSPORT LOG FILE - Monitor mode

监控: COM3
创建时间: 2022-09-01, 09:19:23
计算机名: GARY‘S-PC
系统版本:  (Build 9200)

#	Time		Duration (s)	Process		Request                             	Port	Result	Data ( Hex )	

1489	9:06:52.233	0.00181540	avrdude.exe	IRP_MJ_CREATE                       	COM3	SUCCESS	Port Opened	
1490	9:06:52.235	0.00000290	avrdude.exe	IOCTL_SERIAL_SET_QUEUE_SIZE         	COM3	SUCCESS	InSize: 1024, OutSize: 1024	
1491	9:06:52.235	0.00030350	avrdude.exe	IOCTL_SERIAL_SET_BAUD_RATE          	COM3	SUCCESS	Baud Rate: 115200	
1492	9:06:52.236	0.00008720	avrdude.exe	IOCTL_SERIAL_CLR_RTS                	COM3	SUCCESS		
1493	9:06:52.236	0.00052320	avrdude.exe	IOCTL_SERIAL_CLR_DTR                	COM3	SUCCESS		
1494	9:06:52.236	0.00045190	avrdude.exe	IOCTL_SERIAL_SET_LINE_CONTROL       	COM3	SUCCESS	StopBits: 1, Parity: No, DataBits: 8	
1495	9:06:52.236	0.00014340	avrdude.exe	IOCTL_SERIAL_SET_CHARS              	COM3	SUCCESS	EofChar: 0x0, ErrorChar: 0x0, BreakChar: 0x0, EventChar: 0x0, XonChar: 0x0, XoffChar: 0x0	
1496	9:06:52.236	0.00025730	avrdude.exe	IOCTL_SERIAL_SET_HANDFLOW           	COM3	SUCCESS	ControlHandShake: 0x0, FlowReplace: 0x0, XonLimit: 0, XoffLimit: 0	
1497	9:06:52.237	0.00000210	avrdude.exe	IOCTL_SERIAL_SET_TIMEOUTS           	COM3	SUCCESS	ReadIntervalTimeout: 0, ReadTotalTimeoutMultiplier: 0, ReadTotalTimeoutConstant: 0, WriteTotalTimeoutMultiplier: 0, WriteTotalTimeoutConstant: 0	
1498	9:06:52.237	0.00013690	avrdude.exe	IOCTL_SERIAL_SET_TIMEOUTS           	COM3	SUCCESS	ReadIntervalTimeout: 250, ReadTotalTimeoutMultiplier: 250, ReadTotalTimeoutConstant: 250, WriteTotalTimeoutMultiplier: 0, WriteTotalTimeoutConstant: 0	
1499	9:06:52.237	0.50011580	avrdude.exe	IRP_MJ_READ                         	COM3	TIMEOUT	Length: 0, Data: 	
1500	9:06:52.737	0.00001500	avrdude.exe	IOCTL_SERIAL_SET_TIMEOUTS           	COM3	SUCCESS	ReadIntervalTimeout: 500, ReadTotalTimeoutMultiplier: 500, ReadTotalTimeoutConstant: 500, WriteTotalTimeoutMultiplier: 0, WriteTotalTimeoutConstant: 0	
1501	9:06:52.737	0.00038460	avrdude.exe	IRP_MJ_WRITE                        	COM3	SUCCESS	Length: 2, Data: 30 20 	
1502	9:06:52.737	0.00001150	avrdude.exe	IOCTL_SERIAL_SET_TIMEOUTS           	COM3	SUCCESS	ReadIntervalTimeout: 250, ReadTotalTimeoutMultiplier: 250, ReadTotalTimeoutConstant: 250, WriteTotalTimeoutMultiplier: 0, WriteTotalTimeoutConstant: 0	
1503	9:06:52.737	0.00041170	avrdude.exe	IRP_MJ_READ                         	COM3	SUCCESS	Length: 1, Data: 14 	
1504	9:06:52.739	0.00000650	avrdude.exe	IRP_MJ_READ                         	COM3	SUCCESS	Length: 1, Data: 10 	
1505	9:06:52.739	0.50993630	avrdude.exe	IRP_MJ_READ                         	COM3	TIMEOUT	Length: 0, Data: 	
1506	9:06:53.249	0.00001320	avrdude.exe	IOCTL_SERIAL_SET_TIMEOUTS           	COM3	SUCCESS	ReadIntervalTimeout: 500, ReadTotalTimeoutMultiplier: 500, ReadTotalTimeoutConstant: 500, WriteTotalTimeoutMultiplier: 0, WriteTotalTimeoutConstant: 0	
1507	9:06:53.249	0.00030750	avrdude.exe	IRP_MJ_WRITE                        	COM3	SUCCESS	Length: 2, Data: 30 20 	
1508	9:06:53.249	0.00001130	avrdude.exe	IOCTL_SERIAL_SET_TIMEOUTS           	COM3	SUCCESS	ReadIntervalTimeout: 250, ReadTotalTimeoutMultiplier: 250, ReadTotalTimeoutConstant: 250, WriteTotalTimeoutMultiplier: 0, WriteTotalTimeoutConstant: 0	
1509	9:06:53.249	0.00053110	avrdude.exe	IRP_MJ_READ                         	COM3	SUCCESS	Length: 1, Data: 10 	
1510	9:06:53.250	0.00000630	avrdude.exe	IRP_MJ_READ                         	COM3	SUCCESS	Length: 1, Data: 10 	
1511	9:06:53.250	0.50959560	avrdude.exe	IRP_MJ_READ                         	COM3	TIMEOUT	Length: 0, Data: 	
1512	9:06:53.759	0.00001580	avrdude.exe	IOCTL_SERIAL_SET_TIMEOUTS           	COM3	SUCCESS	ReadIntervalTimeout: 500, ReadTotalTimeoutMultiplier: 500, ReadTotalTimeoutConstant: 500, WriteTotalTimeoutMultiplier: 0, WriteTotalTimeoutConstant: 0	
1513	9:06:53.759	0.00021310	avrdude.exe	IRP_MJ_WRITE                        	COM3	SUCCESS	Length: 2, Data: 30 20 	
1514	9:06:53.759	0.00001180	avrdude.exe	IOCTL_SERIAL_SET_TIMEOUTS           	COM3	SUCCESS	ReadIntervalTimeout: 5000, ReadTotalTimeoutMultiplier: 5000, ReadTotalTimeoutConstant: 5000, WriteTotalTimeoutMultiplier: 0, WriteTotalTimeoutConstant: 0	
1515	9:06:53.759	0.00048120	avrdude.exe	IRP_MJ_READ                         	COM3	SUCCESS	Length: 1, Data: 14 	
1516	9:06:53.760	0.00001210	avrdude.exe	IOCTL_SERIAL_SET_TIMEOUTS           	COM3	SUCCESS	ReadIntervalTimeout: 5000, ReadTotalTimeoutMultiplier: 5000, ReadTotalTimeoutConstant: 5000, WriteTotalTimeoutMultiplier: 0, WriteTotalTimeoutConstant: 0	
1517	9:06:53.760	0.00000380	avrdude.exe	IRP_MJ_READ                         	COM3	SUCCESS	Length: 1, Data: 10 	
1518	9:06:53.760	0.00000740	avrdude.exe	IOCTL_SERIAL_SET_TIMEOUTS           	COM3	SUCCESS	ReadIntervalTimeout: 500, ReadTotalTimeoutMultiplier: 500, ReadTotalTimeoutConstant: 500, WriteTotalTimeoutMultiplier: 0, WriteTotalTimeoutConstant: 0	
1519	9:06:53.760	0.00017410	avrdude.exe	IRP_MJ_WRITE                        	COM3	SUCCESS	Length: 3, Data: 41 80 20 	
1520	9:06:53.761	0.00001150	avrdude.exe	IOCTL_SERIAL_SET_TIMEOUTS           	COM3	SUCCESS	ReadIntervalTimeout: 5000, ReadTotalTimeoutMultiplier: 5000, ReadTotalTimeoutConstant: 5000, WriteTotalTimeoutMultiplier: 0, WriteTotalTimeoutConstant: 0	
1521	9:06:53.761	0.00061420	avrdude.exe	IRP_MJ_READ                         	COM3	SUCCESS	Length: 1, Data: 14 	
[... cut ...]
1625	9:06:53.789	0.00014000	avrdude.exe	IRP_MJ_WRITE                        	COM3	SUCCESS	Length: 2, Data: 51 20 	
1626	9:06:53.789	0.00005360	avrdude.exe	IOCTL_SERIAL_SET_TIMEOUTS           	COM3	SUCCESS	ReadIntervalTimeout: 5000, ReadTotalTimeoutMultiplier: 5000, ReadTotalTimeoutConstant: 5000, WriteTotalTimeoutMultiplier: 0, WriteTotalTimeoutConstant: 0	
1627	9:06:53.789	0.00035360	avrdude.exe	IRP_MJ_READ                         	COM3	SUCCESS	Length: 1, Data: FF 	

```
I have captured the data of the whole communication process. What step is the signature reading performed?

mcuee, 2022-09-05:
Maybe @dl8dtl or @stefanrueger can help answer your question anout `What step is the signature reading performed?`.

cuiqingwei, 2022-09-05:
Thank you. I hope you can help me find the real cause of the problem as soon as possible.

mcuee, 2022-09-05:
> Thank you. I hope you can help me find the real cause of the problem as soon as possible.

In the mean time, why not give the other bootloader a try. I am pretty sure your bootloader is broken.
https://github.com/SpenceKonde/megaTinyCore/tree/master/megaavr/bootloaders

cuiqingwei, 2022-09-06:
With the code you provided, the same phenomenon occurs.
1. modify and build optiboot_x  [optiboot_x.zip](https://github.com/avrdudes/avrdude/files/9494188/optiboot_x.zip)
[image]
2. Test communication and obtain version number.
[image]
3. Test avrdude，Same, no~
[image]

mcuee, 2022-09-06:
@cuiqingwei 
Thanks for the update.

Can you try the avrdude version inside Arduino megaTinyCore (https://github.com/SpenceKonde/megaTinyCore)? Thanks.

cuiqingwei, 2022-09-06:
In order to support ATtiny1616, I made two modifications, passed the compilation, and verified that the serial communication works normally.
[image]
[image]

mcuee, 2022-09-06:
Sorry I am not able to help you much further as I do not have the necessary HW to carryout the test and I am not familiar with the bootloader either.

When I say avrdude version which comes with the megaTinyCore, basically that means the avrdude version which comes with Arduino (megaTinyCore is installed on top of Arduino). It is supposed to work together with the bootloader. If that does not wotk either, you can always create an issue for the megaTinyCore project if you think your bootloader build is good yet it does not work with avrdude (either official version or the Arduino version).
https://github.com/SpenceKonde/megaTinyCore

cuiqingwei, 2022-09-07:
Is it normal for ATtiny817 xplaned Mini evaluation board to use avrdude?

cuiqingwei, 2022-09-07:
```
avrdude.exe: AVR device initialized and ready to accept instructions

Reading | avr_read(): error reading address 0x0000
    read operation not supported for memory "signature"
avrdude.exe: error reading signature data for part "ATtiny1616", rc=-2
avrdude.exe: error reading signature data, rc=-2
```
The whole process was not sent Cmnd_STK_READ_SIGN 0x75 command，So this should be the problem of avrdude.exe itself with ATtiny1616.

cuiqingwei, 2022-09-07:
[image]

What platform is it verified on?

cuiqingwei, 2022-09-07:
I guess it's avrdude.exe there is a problem with the inherited configuration file. Otherwise, how can I explain that I manually modified the configuration file earlier?
[image]
[image]
[image]
[image]

cuiqingwei, 2022-09-07:
@mcuee Can anyone answer the above question?

mcuee, 2022-09-07:
> @mcuee Can anyone answer the above question?

Sorry I am not able to help here. Hopefully someone else can help you.

cuiqingwei, 2022-09-07:
> 

@mcuee Thank you for your reply.

mcuee, 2022-09-12:
> @mcuee Thank you for your reply.
> 
> So, do you agree that this is a bug ？

No I do not agree. To me the bootloader is not working properly.

mcuee, 2022-09-12:
> Is it normal for ATtiny817 xplaned Mini evaluation board to use avrdude?

Yes it is normal. But I do not use the bootloader.

mcuee, 2022-09-12:
@MCUdude and @SpenceKonde

Just wondering if you have ever tested the optiboot_x with ATtiny1616 (or other chips) using avrdude (either official version or Arduino version). Thanks.

It seems to be a bit tricky to set up.
https://github.com/SpenceKonde/megaTinyCore#bootloader-optiboot-support
https://github.com/SpenceKonde/megaTinyCore/blob/master/megaavr/extras/Ref_Reset.md

MCUdude, 2022-09-12:
@mcuee I don't think I have any ATtinyxx16 hardware, so I'm not able to test Optiboot on these devices. However, I highly doubt there is an Avrdude-related issue since @spencekonde would have caught this a long time ago.

> I guess it's avrdude.exe there is a problem with the inherited configuration file. Otherwise, how can I explain that I manually modified the configuration file earlier?

If you compare the ATtiny1616 section in the"official" avrdude.conf file with the one shipped with megaTinyCore, you can see that they are practically identical except for some formatting. Try a different bootloader or use a dedicated UPDI programmer such as SerialUPDI or similar.

mcuee, 2022-09-12:
> @mcuee I don't think I have any ATtinyxx16 hardware, so I'm not able to test Optiboot on these devices. However, I highly doubt there is an Avrdude-related issue since @SpenceKonde would have caught this a long time ago.

If you can test the megaTinyCore optiboot_x bootloader using ATtiny817, that will be good. In that case, we are sure that avrdude is not the issue and the optiboot_x bootloader itself is also not the issue. Then I can close this issue.

mcuee, 2022-09-12:
> I don't think I have any ATtinyxx16 hardware, so I'm not able to test Optiboot on these devices. However, I highly doubt there is an Avrdude-related issue since @SpenceKonde would have caught this a long time ago.

@MCUdude 
On the other hand, I just looked at the closed issues from megaTinyCore object, the bootloader should work properly. So I will agree with you.
https://github.com/SpenceKonde/megaTinyCore/issues?q=is%3Aissue+is%3Aclosed+bootloader

@cuiqingwei 
As mentioned above, we believe this is not an issue with avrdude. And you can try to get support from megaTinyCore project to see if @SpenceKonde can help you there.

I will close the issue now and I will reopen if the issue is confirmed to be on avrdude side later.

closed 2022-09-12 by mcuee

SpenceKonde, 2022-10-08:
Optiboot works beautifully with avrdude 6.3, which is what should be getting used if you do a board manager install. Do not use 7.0 until it has been tested and qualified with megaTinyCore which will happen after we have a stable 2.6.x mTC and 1.5.x DxCore. 

As right now we have neither, so a high risk upgrade of a core tool which I cannot bypass the development team just isn't in the cards. 

The only people who should be running into this is perople who "updated" avrdude to an unsupported version of the tool that is apparently not backwards compatible. What did you *expect* to happen? 

[image]
^Let's see if this gif about what your expectation should be using an unsupported version of a third party tool which the author of the core dislikes and only tolerates because he doesn't have time to write replacements in a civilized language like python. should be displays :-)

MCUdude, 2022-10-08:
It turned out that the hardware I used when working on the JTAG3 HVUPDI implementation was using an ATtiny1616. I can confirm that there's an issue with the Attiny1616 bootloader (bundled with megaTinyCore), and the upstream Avrdude version. It works great with Avrdude 6.3 and 7.0 (tested with Arduino's fork), but not the upstream version:

Arduino 6.3:

```
$ /Users/hans/Library/Arduino15/packages/DxCore/tools/avrdude/6.3.0-arduino17or18/bin/avrdude -C/Users/hans/Library/Arduino15/packages/megaTinyCore/hardware/megaavr/2.6.1/avrdude.conf -pattiny1616 -carduino -D -P/dev/cu.usbserial-1410 -b115200 -P/dev/cu.usbserial-1410 -Uflash:w:/var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_148565/Blink.ino.hex:i

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9421 (probably t1616)
avrdude: reading input file "/var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_148565/Blink.ino.hex"
avrdude: writing flash (1278 bytes):

Writing | ################################################## | 100% 0.06s

avrdude: 1278 bytes of flash written
avrdude: verifying flash memory against /var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_148565/Blink.ino.hex:
avrdude: load data flash data from input file /var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_148565/Blink.ino.hex:
avrdude: input file /var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_148565/Blink.ino.hex contains 1278 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 0.05s

avrdude: verifying ...
avrdude: 1278 bytes of flash verified

avrdude: safemode: Fuses OK (E:FF, H:FF, L:FF)

avrdude done.  Thank you.
```

Avrdude 7.0:
```
$ /Users/hans/Library/Arduino15/packages/MegaCoreX/tools/avrdude/7.0.0-arduino2/bin/avrdude -C/Users/hans/Library/Arduino15/packages/megaTinyCore/hardware/megaavr/2.6.1/avrdude.conf -pattiny1616 -carduino -D -P/dev/cu.usbserial-1410 -b115200 -P/dev/cu.usbserial-1410 -Uflash:w:/var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_148565/Blink.ino.hex:i 

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9421 (probably t1616)
avrdude: reading input file "/var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_148565/Blink.ino.hex"
avrdude: writing flash (1278 bytes):

Writing | ################################################## | 100% 0.06s

avrdude: 1278 bytes of flash written
avrdude: verifying flash memory against /var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_148565/Blink.ino.hex:

Reading | ################################################## | 100% 0.05s

avrdude: 1278 bytes of flash verified

avrdude done.  Thank you.

```

Upstream version:
```
$ ./avrdude -C/Users/hans/Library/Arduino15/packages/megaTinyCore/hardware/megaavr/2.6.1/avrdude.conf -pattiny1616 -carduino -D -P/dev/cu.usbserial-1410 -b115200 -P/dev/cu.usbserial-1410 -Uflash:w:/var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_148565/Blink.ino.hex:i 

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9421 (probably t1616)
avrdude: reading input file /var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_148565/Blink.ino.hex for flash
         with 766 bytes in 1 section within [0x200, 0x4fd]
         using 12 pages and 2 pad bytes
avrdude: writing 766 bytes flash ...

Writing | ################################################## | 100% 0.05s

avrdude: 766 bytes of flash written
avrdude: verifying flash memory against /var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_148565/Blink.ino.hex

Reading | ################################################## | 100% 0.05s

avrdude: verification error, first mismatch at byte 0x0200
         0x04 != 0x3d
avrdude: verification error; content mismatch

avrdude done.  Thank you.
```

MCUdude, 2022-10-08:
I can confirm that commit 3d06457 is breaking Optiboot on (certain?) ATtinys.

If I run `git checkout 29c6645`(previous commit, relative to 3d06457), it works just fine.

mcuee, 2022-10-09:
@MCUdude 
Good to know that optiboot_x is working fine with the megaTinyCore bundled version of avrdude 6.3. In that case, I will keep this issue closed and we can continue the discussion in #1120.

mcuee, 2022-10-09:
@cuiqingwei 
As per the above, please use the megaTinyCore bundled version of avrdude 6.3 as of now until #1120 is fixed. If you still have issues with avrdude 6.3, you can try to get the support from megaTinyCore since that will point to your bootloader issue.

mcuee, 2022-10-09:
@SpenceKonde 

Fair enough for you to keep using the avrdude 6.3 version in current megaTinyCore as avrdude 7.0 is not ready and it apparently has issues with optiboot_x based on the testing results from @MCUdude.

MCUdude, 2022-10-09:
> Fair enough for you to keep using the avrdude 6.3 version in current megaTinyCore as avrdude 7.0 is not ready and it apparently has issues with optiboot_x based on the testing results from @MCUdude.

7.0 works just fine. It's the upstream version that's the problem.

mcuee, 2022-10-09:
> > Fair enough for you to keep using the avrdude 6.3 version in current megaTinyCore as avrdude 7.0 is not ready and it apparently has issues with optiboot_x based on the testing results from @MCUdude.
> 
> 7.0 works just fine. It's the upstream version that's the problem.

My bad, 7.0 works fine. git main does not work as of now.

SpenceKonde, 2022-10-10:
Not surprised this broke my forks of Optiboot when someone fiddled with it (was that change made to address a complaint?). Arguably the hardest part of porting optiboot to Dx-series parts was figuring out to get AVRdude to use the correct addressing modes. Everything else was relatively straightforward. But DA and DB can have 128k parts. AVRdude uses byte addresses talking to optiboot_x for the tinyAVR parts, while I used a weird trick to fool it into using word addressing for the AVR128Dx parts  (see my avrdude.conf from DxCore). If there's a better way that will work with both 6.3 and future 7.x versions please let me know (based on experience, it *must* be backwards compatible - I can change what's in the avrdude.conf to make it work with whatever new version - or ideally both versions, but I can't say "Yeah from here on out you need a new bootloader binary",  people make an unholy stink*) but *shrug* that was the trick I found to make it work on 6.3 (haven't tested 7.0 - I'm not allowed to look at that until both megaTinyCore and DxCore have stable releases)

`*`  especially the ones who are using tinyAVRs with the updi pin fused to act as reset or I/O and who don't own an HVUPDI programmer (virtually nobody does HV updi. most of microchip's programmers don't even support it as implemented on tinyAVR (and small wonder: you need to have a line at 12v, then go back to sending normal data over it.  Even microchip retreated from the shared HV pulse + data line concept on the DD's! So there are two HV UPDI's now :-( The DD's put the pulse on reset, which can be fused as input, but never output (reset sucks when used as output anyway - sub-milliamp current? screw that) and if UPDI is fused as GPIO, then a 12V pulse on reset followed rapidly by a KEY on the (now once more) UPDI line is what you need to do to reprogram it. On a tinyAVR, instead, you power-cycle it, almost immediately put the 12v pulse on the UPDI/Reset/GPIO line... there may or may not be a timeout that you have to send that KEY within on some parts; certainly some parts don't, because multiple people have told me they do it with 12v doorbell batteries and nerves of steel),

mcuee, 2022-10-10:
> Not surprised this broke my forks of Optiboot when someone fiddled with it (was that change made to address a complaint?).

Indeed that change https://github.com/avrdudes/avrdude/commit/3d06457a1614a2247efc846b8e73788cb8585978 is because of a long standing bug that typical optiboot implemnetation does not follow the stk500v1 exactly.
* https://github.com/avrdudes/avrdude/issues/967