avrdudes/avrdude #1104: Programmer features and terminal mode

MCUdude, 2022-09-25:
I recently got my hands on a [Power debugger](https://www.microchip.com/en-us/development-tool/ATPOWERDEBUGGER) and an [STK600](https://www.microchip.com/en-us/development-tool/ATSTK600) and was curious to see if their "special features" like programmable target voltage and the programmable clock was supported in Avrdude terminal mode.

There are the commands terminal mode supports, and is printed regardless of which programmer is in use:

```
avrdude> ?
>>> ? 
Valid commands:
  dump    : dump <memory> [<addr> <len> | <addr> ... | <addr> | ...]
  read    : alias for dump
  write   : write <memory> <addr> [<data>[,] {<data>[,]} | <len> <data>[,] {<data>[,]} ...]
  erase   : perform a chip erase
  sig     : display device signature bytes
  part    : display the current part information
  send    : send a raw command: send <b1> <b2> <b3> <b4>
  parms   : display adjustable parameters (STK500 and Curiosity Nano only)
  vtarg   : set <V[target]> (STK500 and Curiosity Nano only)
  varef   : set <V[aref]> (STK500 only)
  fosc    : set <oscillator frequency> (STK500 only)
  sck     : set <SCK period> (STK500 only)
  spi     : enter direct SPI mode
  pgm     : return to programming mode
  verbose : change verbosity
  quell   : set quell level for progress bars
  help    : help
  ?       : help
  quit    : quit
```

The list of commands is a bit misleading. For instance, the STK600 supports `parms`, `vtarg`, `varef`, `fosc `and `sck`.
Actually, several programmers supports `sck`, even though it clearly says "STK500 only".

It would be great if the list somehow could be built based on what the programmer supports.
All terminal functions check if the function pointer is NULL, so maybe this could be utilized to build/print the list? Then we could remove all the "STK500 only" info, since lots of it isn't even true:

(For reference, NULL check in the `cmd_sck` function)
```c
  if (pgm->set_sck_period == NULL) {
    terminal_message(MSG_INFO, "%s (sck): the %s programmer cannot set SCK period\n",
      progname, pgm->type);
    return -2;
  }
```

MCUdude, 2022-10-06:
I took a closer look at the source code, and it's not 100% how to solve this... Maybe @stefanrueger can look into it (only print commands a programmer actually supports)?

stefanrueger, 2022-10-06:
@MCUdude `offsetof()` is your friend. See PR #1113

stefanrueger, 2022-10-17:
Closed as PR #1113 addresses this issue

closed 2022-10-17 by stefanrueger